pgaskin/kobopatch-patches #2: kobopatch.yaml seems not to be used

norbusan, 2018-07-11:
Hi
thanks for the new patcher. I wanted to switch over from the old patcher to this one, and found that the changes in kobopatch.yaml are *not* used.

Here the relevant part of `kobopatch_4.9.11311/kobopatch.yaml` (all patch name lines are correctly intended by 4 spaces):

```
overrides:
  src/nickel.patch:
    Custom left & right margins: yes
    Fix three KePub fullScreenReading bugs: yes
    Change dicthtml strings to micthtml: yes
  src/libadobe.so.patch:
  src/libnickel.so.1.0.0.patch:
    Show all games: yes
    Custom reading footer style: yes
    Dictionary pop-up frame size increase: yes
    New home screen increasing cover size: yes
    Fix reading stats/author name cut off when series is showing: yes
    Increase the cover size in library: yes
    Increase the view details container size: yes
    Custom font to collection and author titles: yes
    New home screen subtitle custom font: yes
    Remove footer (row3) and increase cover size on new home screen: yes
  src/librmsdk.so.1.0.0.patch:
    Default ePub monospace font (Courier): yes
```

after that I run the .sh or the respective binary and get:
```
kobopatch v0.7.6
https://github.com/geek1011/kobopatch

Reading config file (kobopatch.yaml)
Opening input file
Patching ./usr/local/Kobo/librmsdk.so.1.0.0
...
```
but all patches are marked as disabled and skipped.

Thanks

Norbert

geek1011, 2018-07-11:
Thanks for testing kobopatch! I fixed it just now in v2. That was a remaining part of the old format backwards-compatibility.

For you, you can just change all occurences of `.patch` to `.yaml`.

closed 2018-07-11 by geek1011

norbusan, 2018-07-11:
Thanks, confirmed that after fixing .patch to .yaml (and my mix up of locations of patches between nickel and nickel.so.0.0.0) it all worked smoothly. Next I will see how it works on my device.

Thanks a lot

geek1011, 2018-07-11:
No problem. Another thing which might be of interest to you is the new (I just added it) `files` section in kobopatch.yaml. It allows you to add additional files to KoboRoot.tgz.