pgaskin/kobopatch-patches #5: "Fix three KePub fullScreenReading bugs" Better Patch

sekkit, 2018-10-05:
- FindReplaceString:
      Find:    "img { display: block; margin-left: auto !important; margin-right: auto !important; width: auto !important; height: auto !important; }"
      Replace: "body{margin:0!important;padding:0!important;}body>div{margin-right:0.2em !important; margin-bottom:0.25em !important;               }"
  
The default 3 patches still have text cut off in Japanese(Vertical), English under difference circumstances.
This one is bruteforce but enables Japanese, Eng/Chinese aligned properly without being cut off.
Any better adjustment will be welcomed.
Plz merge this.

sekkit, 2018-10-05:
I think change 0.Xem to XXpx could be better in compatibility.
Because fonts differs in weight & sizes, I tested many works, few(very few) like MI LANTING still cut off Japanese with this patch. BUT default fonts all works fine.

geek1011, 2018-11-04:
This patch should not be necessary since 4.11.11911, as Kobo now officially supports disabling headers and footers.

closed 2018-11-04 by geek1011