pgaskin/kobopatch-patches #9: Make it easier to use for macOS users

geek1011, 2019-02-20:
https://www.mobileread.com/forums/showthread.php?t=315383

NiLuJe, 2019-02-20:
Provided the exec bit is set, a shell script with a .script extension will be double-clickable, run in Terminal, won't close by itself.

The only quirk you have to account for is that PWD is enforced to HOME ;).

NiLuJe, 2019-02-20:
That still leaves GateKeeper to deal with, so, not actually double click able in practice... Right click > open > gatekeeper pop-up

geek1011, 2019-03-02:
Hmm. I'll try this in the next version and see if I can do anything about GateKeeper.

geek1011, 2019-09-23:
@NiLuJe on a *slightly* unrelated note, what do you think of merging the kobopatch binaries into a fat hybrid sh-batch script so there's only one binary to run (this might be a really, really bad idea...)?

NiLuJe, 2019-09-23:
In practice, most people will only ever run the batch/shell script, so I'm not sure what the interest would be, unless I'm missing something ;).

Ironically enough, on macOS, I don't *think* (I'll check tomorrow) GateKeeper shouts at you for running an untrusted binary from a shell script once you've authorized the shell-script itself, so that wouldn't even help with stuff like that.

geek1011, 2019-09-23:
@NiLuJe It was just a curiosity, but it'd be nice to have less files for the sake of minimalism. I think the cost of it would be too high in this case, though. It would also allow me to simplify the instructions, though.

I just checked the GateKeeper thing in my Mojave VM, and it seems like this wouldn't change anything. I did notice that running the binary directly needs an admin password, unlike running the shell script by double-clicking a *.command file. For both methods, you still have to appease GateKeeper in the settings panel.

NiLuJe, 2019-09-25:
Fun fact: making the .command a symlink doesn't work ;). Clicking that would open the default app for shell-scripts (f.g., XCode).

NiLuJe, 2019-09-25:
Also, on a slightly related note about the threads themselves:

* There's no mention of macOS at all in the bit that points the user to the wrapper script ;).
* The link to the matching kobopatch zip for the version the thread targets could be automatically put in there, the GH Release page might be a bit scary/confusing for some...

geek1011, 2019-09-25:
> There's no mention of macOS at all in the bit that points the user to the wrapper script ;).

That was a bit of an oversight on my part while I was writing the new instructions. I assumed everyone knew macOS was unix-like ;).

> The link to the matching kobopatch zip for the version the thread targets could be automatically put in there, as the GH Release page might be a bit scary/confusing for some...

I could do that. The main reasons why I link to the releases page is so people can see the release notes without me having to convert it to bbcode (and hopefully the savvy users can also catch on to the fact they can pull in-progress stuff from GitHub), and so I don't need to edit every post on each release (MR started limiting the number of edits you can do per day, and how long after a post you can edit it). It also makes the post quote-friendly. I guess one way I could solve this is with a simple proxy to redirect to a specified file in the latest GitHub release for a repo.

NiLuJe, 2019-09-25:
Yeah, I was just thinking that getting people to see the release notes was the reasoning behind it as I wrote that ;p.

geek1011, 2019-10-18:
@NiLuJe, oh no... https://developer.apple.com/news/?id=09032019a.

NiLuJe, 2019-10-18:
@geek1011: That only applies to signed binaries.

For the foreseeable future, unsigned binaries are "okay" (i.e., GateKeeper will shout at you, but you'll be able to run 'em).

NiLuJe, 2019-10-18:
c.f., this for a neat recap: https://eclecticlight.co/2019/10/04/will-gatekeeper-let-me-run-that-app-in-catalina/

And this for a confirmation that things are as they always were for unsigned stuff: https://eclecticlight.co/2019/10/05/what-to-do-when-a-newly-installed-app-cant-be-opened/

geek1011, 2019-10-18:
What about the last part here: https://eclecticlight.co/2019/06/07/notarization-in-mojave-and-catalina/

NiLuJe, 2019-10-18:
Yeah, stay tuned for more fun crap next year, I guess ;p.

NiLuJe, 2019-10-29:
Whee, realized there was a nasty thinko in my original comment. I meant *.command*. .script is for AppleScript -_-".

NiLuJe, 2019-12-13:
@geek1011: Re: the thread linking to the Release without a direct link to the zip: someone was just bit the asset list not being sorted naturally ;). (This was one of the many things I'd expected people to get wrong with the current approach, kinda surprised it even took this long ^^).

geek1011, 2019-12-13:
I'm surprised it took that long too, TBH, although you'd expect people with e-readers to be literate :P

I could link to the new kobopatch webpage, which has it sorted properly, but it doesn't work on older browsers (it uses newer JS APIs like fetch).

I'll probably put together another download page (either using more compatible JS or rendered server-side) which pulls the data through the GH API and puts it in a more user friendly form.

Do you have any other suggestions for that page when I put it together?

NiLuJe, 2019-12-13:
None that I think of right now ;).

On a slihgtly related subject, I should get around to actually do/test the macOS .command thingy because it's extremely trivial ;p.

geek1011, 2020-02-16:
@NiLuJe, does kobopatch still work for you after https://developer.apple.com/news/?id=12232019a?

NiLuJe, 2020-02-16:
That's for App Store submissions only ;).

geek1011, 2020-02-16:
What's this about, then:

> In June, we announced that all Mac software distributed outside the Mac App Store must be notarized by Apple in order to run by default on macOS Catalina. In September, we temporarily adjusted the notarization prerequisites to make this transition easier and to protect users on macOS Catalina who continue to use older versions of software. Starting February 3, 2020, all submitted software must meet the original notarization prerequisites

NiLuJe, 2020-02-16:
I take that "by default" to mean without a Gate Keeper warning.

NiLuJe, 2020-02-16:
Yup, [business as usual](https://eclecticlight.co/2020/02/03/hardening-and-notarization-finally-arrive-in-catalina/) ;).

geek1011, 2020-05-19:
Since nothing's completely broken with the recent macOS updates AFAIK, I'm going to close this issue for now.

closed 2020-05-19 by geek1011