pgaskin/kobopatch-patches #13: Enable rotation on all devices:

ddutchie, 2019-04-09:
Thanks for all the work!

There's a note under this option:

>   # This patch is not necessary anymore, as there is a fully working config setting
>   # now, but is being kept for backwards compatibility with old kobopatch configs.

How do I access this config setting?

geek1011, 2019-04-10:
ForceAllowLandscape in the DeveloperSettings section.

closed 2019-04-12 by ddutchie

Lucasxp32, 2019-04-24:
It "works" on the Kobo Glo HD, however, the "touch position" doesn't seem to rotate well. Touch is offset & some areas don't work at all.

geek1011, 2019-04-25:
Which firmware version are you running?

Lucasxp32, 2019-04-25:
> Which firmware version are you running?

- 4.13.12638
- KSM 9 (update 2) -- although, I've tried booting nickel directly (runsettings.txt / runmenu=never); but same issue.

It behaves the same whether I activate it through the Config or the latest kobopatch (v44).
I didn't test it out unmodded & in a older firmware & get logs. But I could do that if necessary.

Thanks for your attention!

geek1011, 2019-04-29:
Hmm. I'd need to see logs to say anything more on this. Maybe @NiLuJe might know?

NiLuJe, 2019-04-29:
Did it ever work properly?

Nickel might not be translating coordinates entirely properly in some circumstances. It is an experimental option, after all, and handling Kobo touch input shenanigans properly is... unfun.

NiLuJe, 2019-04-29:
For people unfamiliar with how this works: rotating the framebuffer rotates the "display" coordinates, but the touch coordinates are *fixed*, so you need to rotate 'em to match the display orientation. Nickel *always* do this, to some extent, as the touch coordinates ahve *never* matched the default Portrait display orientation.

I'm using "rotate" in a generic manner, it's sometimes a wee bit more involved on some devices.