pgaskin/kobopatch-patches #16: Changing the info panel in full size screensaver

benchen71, 2019-06-05:
Everything on MobileRead suggests that "Changing the info panel in full size screensaver" should work on all versions. But it isn't working on my Glo HD.

I would love to be able to customise the info panel, for example,
- moving it to the upper left (or upper right)
- adding the battery level
- removing "Sleeping"

geek1011, 2019-06-05:
This may be due to the recent CSS changes in nickel. I'll look into it sometime in the next few days.

geek1011, 2019-06-10:
@jcn363?

geek1011, 2019-06-21:
@benchen71, have a look at [this](https://www.mobileread.com/forums/showthread.php?p=3858938) post by jackie_w on MobileRead:

> I don't use jcn363's full screensaver patch myself but I do have my own patch for that particular CSS stream so I can make a few general comments...
>
> It looks to me as if all jcn363's patch is doing is changing the font-size of some of the text (battery charging status only ???) in the semi-transparent rectangular (rather large & ugly ) info panel which appears two-thirds the way down the left-hand edge of the full screensaver, obscuring some of the cover.
>
> The ability to change the position of this panel (to one of the corners) disappeared many firmwares ago, or at least it's no longer obvious how to do it.
>
> You can still change the size of the panel by changing the #infoWidget min/max-width value.
>
> I'm not sure what benchen71 actually means by 'removing Sleeping'. I think it's probably possible to change the 'Sleeping' and 'Powered off' text to something you like better as they are both strings in libnickel.so.1.0.0 (D1E774). However, as they are both quite short strings I don't know how much flexibility there would be in renaming them. Overtyping them with normal spaces sounds plausible in theory.
>
> For my own purposes I wanted to use the full, rather than small, screensaver and also be able to see at a glance whether my device is sleeping or powered off. So my own custom patch:
- significantly reduces info panel size when Powered off
- reduces info panel size to zero when Sleeping, so that the cover is fully visible.
>
> As for 'adding the battery level'. I thought the info panel already displayed the battery level whilst sleeping if the device is currently charging. It certainly used to, but I can't say for sure because of my custom patch.

benchen71, 2019-06-22:
Thanks for sharing that.  I've now made my own custom patch (which works well on my Glo HD):

Resize info panel:
  - Enabled: yes
  - Description: |
      Custom patch to change the width of the info panel.
  - FindZlib: "#infoWidget"
  - ReplaceZlibGroup:
      Replacements:
        - {Find: "max-width: 300px;", Replace: "max-width: 150px;"} # Trilogy
        - {Find: "min-width: 300px;", Replace: "min-width: 150px;"} # Trilogy
        - {Find: "qproperty-topMargin: 12px;", Replace: "qproperty-topMargin: 6px;"} # Trilogy
        - {Find: "qproperty-bottomMargin: 12px;", Replace: "qproperty-bottomMargin: 6px;"} # Trilogy
        - {Find: "qproperty-rightMargin: 12px;", Replace: "qproperty-rightMargin: 6px;"} # Trilogy
        - {Find: "qproperty-leftMargin: 24px;", Replace: "qproperty-leftMargin: 12px;"} # Trilogy
        - {Find: "max-width: 380px;", Replace: "max-width: 190px;"} # Phoenix
        - {Find: "min-width: 380px;", Replace: "min-width: 190px;"} # Phoenix
        - {Find: "qproperty-topMargin: 16px;", Replace: "qproperty-topMargin: 8px;"} # Phoenix
        - {Find: "qproperty-bottomMargin: 16px;", Replace: "qproperty-bottomMargin: 8px;"} # Phoenix
        - {Find: "qproperty-rightMargin: 16px;", Replace: "qproperty-rightMargin: 8px;"} # Phoenix
        - {Find: "qproperty-leftMargin: 32px;", Replace: "qproperty-leftMargin: 16px;"} # Phoenix
        - {Find: "max-width: 470px;", Replace: "max-width: 235px;"} # Dragon
        - {Find: "min-width: 470px;", Replace: "min-width: 235px;"} # Dragon
        - {Find: "qproperty-topMargin: 22px;", Replace: "qproperty-topMargin: 11px;"} # Dragon
        - {Find: "qproperty-bottomMargin: 22px;", Replace: "qproperty-bottomMargin: 11px;"} # Dragon
        - {Find: "qproperty-rightMargin: 22px;", Replace: "qproperty-rightMargin: 11px;"} # Dragon
        - {Find: "qproperty-leftMargin: 44px;", Replace: "qproperty-leftMargin: 22px;"} # Dragon
        - {Find: "max-width: 611px;", Replace: "max-width: 305px;"} # Daylight
        - {Find: "min-width: 611px;", Replace: "min-width: 305px;"} # Daylight
        - {Find: "qproperty-topMargin: 28px;", Replace: "qproperty-topMargin: 14px;"} # Daylight
        - {Find: "qproperty-bottomMargin: 28px;", Replace: "qproperty-bottomMargin: 14px;"} # Daylight
        - {Find: "qproperty-rightMargin: 28px;", Replace: "qproperty-rightMargin: 14px;"} # Daylight
        - {Find: "qproperty-leftMargin: 56px;", Replace: "qproperty-leftMargin: 28px;"} # Daylight

closed 2019-09-19 by geek1011