pgaskin/kobopatch-patches #21: Footer/header/navbar FAQ

geek1011, 2019-09-21:
There are at least 5 different patches about this, and it's not clear (even to me) what they all do. Let's collect some notes here which can be turned into a FAQ.

cc @jackiew1

jackiew1, 2019-09-21:
Re: height of header/footer I think there now needs to be only 2 patches (nickel.yaml)
1. GeoffR: Reduce top/bottom page spacer
2. jackie_w: Reduce new header/footer height
Both are customisable to single pixel level, so when it comes to perceived "wasted vertical whitespace" everyone should be able to find an option to suit if they make the effort.

You can enable either, neither or both of patch 1 & 2.
1. controls the height of the header/footer when the user sets to OFF Chapter progress and Book progress respectively.
2. controls the height of the header/footer when the user sets to not-OFF Chapter progress and Book progress respectively.

a) An only-fullpage-reading-will-do user, who always sets both to OFF only needs to enable patch 1.
b) A user who likes to have both header & footer ON, but thinks they're a bit too tall, only needs to enable patch 2.
c) A user who wants header or footer ON, but not both, and who wants to minimise space taken up by the one that's OFF should enable both.
d) A user who's not sure yet but thinks the default header & footer are too tall - same as c)

geek1011, 2019-09-21:
I think I've figured out how the different spacing patches and header/footer options interact. I'm working on an interactive demo to show what it looks like.

geek1011, 2019-09-22:
@jackiew1, can you test https://pgaskin.net/kobopatch-patches/hdrftrptch.html?

closed 2019-09-22 by geek1011

NiLuJe, 2019-09-22:
I don't recall if the spacer patch actually affects anything when the header/fotter is shown, but, otherwise, yeah, that's the idea.

I do have a bit of a cropping issue with the patched full-screen setup, though:

[image]

(Granted, that could theoretically happen in Nickel at one time, but I think those days are mostly gone ;p).

geek1011, 2019-09-22:
> I don't recall if the spacer patch actually affects anything when the header/fotter is shown, but, otherwise, yeah, that's the idea.
I did it by eye (with an ordinary book, not the margin test one), and it seemed to have a small difference, but looking again, I think you're right. I'll fix that.

> I do have a bit of a cropping issue with the patched full-screen setup, though
Yes, that's an issue, as I did the page body the lazy way by clipping, instead of using proper CSS columns and overflow tricks. I'll see if there's a simple way to fix it without losing compatibility or consistency too much.

NiLuJe, 2019-09-22:
The debug option that displays the widget regions and names was mildy useful (and/or at least entertaining to watch) when we had to go digging for the spacer the first time around ;).

(I *think* that's `ShowLayoutRectangles` in `FeatureSettings`. Or possibly `ShowWidget` in `DeveloperSettings`. Or actually each does what it says on the tin and I was thinking of both enabled ;p).

jackiew1, 2019-09-22:
When I was testing I noticed that after the initial reboot after patching if I opened my previous book there were sometimes weird issues with cutoff but once I paged across a chapter boundary (or opened a new book) everything sorted itself out. I've paged through both an epub and a kepub and am not seeing any cutoffs - no matter whether header or footer are ON/OFF.

P.S. I don't have time for a long discussion now, it's 01:30 and I'm old.

geek1011, 2019-09-22:
@NiLuJe I've fixed everything (and made some other improvements). The new URL is: https://pgaskin.net/kobopatch-patches/hdrftrptch-13651.html.

NiLuJe, 2019-09-22:
@jackiew1: Yep, the first thing shown when opening a book being a cached render vaguely rings a bell ;).

@geek1011: :+1:

Sidebar re: the progress bar: From the very quick glance I took at it, it shouldn't affect page layout at all (it's effectively an overlay, you can check on a cover page, you'll be able to see a few lines of the images below the bar).

geek1011, 2019-09-22:
@NiLuJe That's why I [used](https://github.com/geek1011/kobopatch-patches/commit/abc7ed2e2ff6be00feb5c6f2c6f969169aadba84#diff-5d44db4fe56f65c2e30dbf7fa72e98b4R123) absolute positioning instead of having it part of the flexbox layout.

commit c483927c (2019-09-22): Added header/footer patch interactive demo (fixes #21)