pgaskin/kobopatch-patches #22: Add FeatureSettings Screenshots

tERyceNzAchE, 2019-09-22:
Add support for `FeatureSettings Screenshots=true/false`.

geek1011, 2019-09-22:
I just remembered the reason why I originally chose not to include this setting: it hijacks the power button, so it's not very helpful for long-term use, and these patches were intended for options which are useful being on for an extended period of time.

tERyceNzAchE, 2019-09-22:
I disagree, it's very helpful.  I've had it enabled on my Kobo for years.  Yes it does hijack the power button, but if you have a sleepcover then you don't need the sleep button.  I would recommend adding it with a warning.

geek1011, 2019-09-22:
@jackiew1, what do you think?

jackiew1, 2019-09-23:
My first reaction was the same as yours. Problems if user sets-then-forgets. If we knew that only switched-on users would enable it then it would be OK with a warning.

The theory about sleepcover is fine but, based on personal experience, when I enable mine manually in the .conf file I always disable it again ASAP, because otherwise I end up with loads of accidental screenshots over the next few days. Then next time I connect/disconnect from PC they're all imported as books and it's a PITA to delete them.

OTOH, if you were able to create a patch to both enable/disable Screenshots from the device, (like InvertScreen) then I'd be all for it :)

geek1011, 2019-09-23:
> OTOH, if you were able to create a patch to both enable/disable Screenshots from the device, (like InvertScreen) then I'd be all for it :)

That would be possible, but you'd lose the invert screen one from both developer settings and the patch (whatever you had set would remain, though), as I'd have to reuse it's set function, as there isn't any unused ones anymore I can modify.

jackiew1, 2019-09-23:
InvertScreen beats ScreenShot from my POV.

c02y, 2019-09-23:
This feature is requested a long time ago, check https://www.mobileread.com/forums/showthread.php?p=3777339#post3777339

but there is no update about it in that thread.

NiLuJe, 2019-09-23:
Stupid question ahead: I fail to actually see the advantage of using a patch to set these settings.

You can just append a duplicate section+key to the config file, and Nickel will pick it up on the next restart/unplug.

The only case where this makes sense is if you intend to trash your .kobo folder on a regular basis (i.e., account switching).

And even then, for such a weird, quirky setting, that doesn't seem like a common use-case.

geek1011, 2019-09-23:
> but I didn't see any update about implementation in that thread.

Yes, what @jackiew1 said pretty much sums up why.

---

> You can just append a duplicate section+key to the end of the config file, and Nickel will pick it up on the next restart or unplug, and QSettings will sanitize everything properly.

I did notice that. That's a double-sided thing though, as it's what caused the bugs in the invert screen patch when also setting it in the config file and/or the developer settings.

> The only advantage I can see is if you intend to trash your .kobo folder on a regular basis (i.e., account switching), in which case you can't rely on always having a config to edit ;).

That sounds a lot like me :). I've been doing some crazy experiments with the database and root filesystem, so I'm always resetting and signing out (that's the main reason I added these patches). Also, I like to know when something is removed, and having it as part of the patches does this implicitly (the build breaks).

> And even then, for such a weird, quirky setting, that doesn't seem like a common use-case.

That's what I was thinking.

NiLuJe, 2019-09-23:
Plus, if you need on-demand screenshots, might I introduce you to our lord and savior [FBGrab](https://www.mobileread.com/forums/showpost.php?p=2854326&postcount=20)? ;).

jackiew1, 2019-09-23:
@NiLuJe, That's the same 'silly question' I asked last week via PM :)

@geek1011, I fully understand that you might like to have all kinds of arcane patches. But you're not planning to distribute them as part of the standard pack, are you?

jackiew1, 2019-09-23:
@NiLuJe, re: FBGrab. Was that aimed at me? I'm OK with FBInk as part of frostschutz's handily packaged MiniClock hack but outside of that I wouldn't know where to start.

geek1011, 2019-09-23:
@jackiew1, it's not any trouble to maintain, as it's identical to quite a few patches except for the symbol name. If one breaks, so do all the others (and vice versa for fixing).

@NiLuJe, useful. Didn't know that existed before. Knowing about that code before would have saved some time implementing my VNC server (it is a lot easier to digest than FBInk, which has multiple layers of abstraction, but it still has the Kobo quirks fixed).

@jackiew1, you would just need to copy the FBGrab binary over (usb, ssh, telnet, whatever), then run it over ssh or telnet.

NiLuJe, 2019-09-23:
@jackiew1: Nope, that was just a random infodump on the topic of screenshots, because I know it's not necessarily well advertised, as @geek1011 just confirmed ;p.

NiLuJe, 2019-09-23:
@geek1011: It has *most* of the purely framebuffer related quirks handled, yeah, as long as the device operate more or less under "normal" parameters.

That's never an issue on Kindle, but on Kobo, expect to have to sometimes rotate the image yourself, for instance ;p.

geek1011, 2019-09-23:
@NiLuJe Yes, that makes sense. And it's good I'll need to rotate the image myself, as I've been trying to re-rotate the touch to match the fixed and unquirked dump from `fbink_dump` (and I haven't managed to get it consistent enough between devices and rotations...). And about the "normal" parameter requirement, that's fine for me, as the VNC server is just for my own use (it should make it easier to test and update the patches and other mods if I don't have to do it physically and over a USB connection, and I'll also be able to remotely get my reading position if my sync script fails for whatever reason :D).

NiLuJe, 2019-09-23:
@geek1011: an fbink_dump should always match the current fb rotation (in fact, restore will shout at you if you try to restore a dump done in another rotation ;p).

That's potentially clearer for full-screen dumps, because the temporary insanity that made me implement cropping made the region shenanigans nigh unreadable in restore :D.

geek1011, 2019-09-23:
@NiLuJe I'm not sure I understand what you're trying to say there ... what I meant was that the touch orientation and position wouldn't match FBInk's dump output, and it's hard to get it to match. And I'm watching your comment change as I type this...

NiLuJe, 2019-09-23:
@geek1011: Sorry, I should be done editing ;p.

And, yeah, that's to be expected on Kobo, that's more or less what I was trying to say ^^. The dump will honor the fb rotation, which, obviously, can be changed, but the touch coordinates are fixed, and they generally never match the standard Portrait orientation on Kobo.

NiLuJe, 2019-09-23:
@geek1011: What @baskerville did in Plato is probably one of the most concise & logical ways to deal with it (I had a hard time wrapping my brain around it, but after having played with multiple different device,s i'm starting to see the appeal ;p).

c.f.,  https://github.com/baskerville/plato/blob/bfc1b2734ebf6192a97a2541c4b67fa6d3212e86/src/device.rs#L119 & https://github.com/baskerville/plato/blob/bfc1b2734ebf6192a97a2541c4b67fa6d3212e86/src/device.rs#L119

NiLuJe, 2019-09-23:
@geek1011: Define "weird things". Possibly as simple as a BGRA vs. RGBA issue? (the fb, and as such the dump, should be BGRA on Kobo).

You should be able to pull the bitdepth (i.e., 32), viewport dimensions, screen dimensions, as well as line_length from [fbink_get_state](https://github.com/NiLuJe/FBInk/blob/f0606bc8cb03047ea2c4c1d9a957fde623d1e666/fbink.h#L196).

jackiew1, 2019-09-23:
> @jackiew1, it's not any trouble to maintain, as it's identical to quite a few patches except for the symbol name. If one breaks, so do all the others (and vice versa for fixing).

@geek1011  I'm 100% sure about your ability to maintain them. My concerns are with the practicality of who's going to field any forum questions. As long as that's also 100% you ... :)

> @jackiew1, you would just need to copy the FBGrab binary over (usb, ssh, telnet, whatever), then run it over ssh or telnet.

@geek1011 LOL! You're very funny! He-ll-o-o, typical dumb Windows user here ...

geek1011, 2019-09-23:
@NiLuJe That part about the colors makes sense. It's so hard to find documentation on eInk and framebuffers.

And that rotation business is driving me crazy. I wish rotation was done in a single place, and that it matched the touchscreen.

I did have a look at plato, and @baskerville has done it in quite a concise way. At the moment, my goal for the VNC server is to have it able to send the unrotated screen and receive touch events properly with no more than 100ms delay (I'll need to implement a proper thread-safe event loop) and a minimal amount of code.

@jackiew1, as all those patches do is enable a config option, I'd send them to @davidfor for answers if I had questions :).

You're certainly not a typical dumb Windows user. If you're dumb, I wouldn't want to imagine what all the usual users would be! To expand on what I said earlier, you'd copy the FBGrab binary from the Kobo directory of NiLuJe's tarball to somewhere on the Kobo, then you'd connect to a terminal over SSH or telnet (built-in) and run /path/to/fbgrab.

NiLuJe, 2019-09-23:
@geek1011: I had high hopes for the Libra when I saw that it was booting UpRight, the boot progress (pickel) was keeping it UpRight, and nickel kept it UpRight, the device actually was UpRight, the panel dimension was finally reported in Portrait and not Landscape... But, no, apparently, it's still fun to have the touch layer report events at a whatever-it-actually-is rotation (probably+/- 90°, but, blergh).

geek1011, 2019-09-23:
> But, no, apparently, it's still fun to have the touch layer report events at a whatever-it-actually-is rotation (probably +/- 90° from that initial nickel rota, but, blergh).

Haha. I'm guessing there's no way to automatically determine this reliably?

> (Worse, looking at the kernel, I'm starting to think the driver is purposefully mangling them this way).

You've got to be kidding me. Is this a driver Kobo modified, or a badly written original driver?

jackiew1, 2019-09-23:
> @jackiew1, as all those patches do is enable a config option, I'd send them to @davidfor for answers if I had questions :).

Ah, to @davidfor who (I think) isn't a kobopatch user. He's a major participant in the Kobo beta programme. I would think he needs to keep his Kobos "pure".

> You're certainly not a typical dumb Windows user. If you're dumb, I wouldn't want to imagine what all the usual users would be! To expand on what I said earlier, you'd copy the FBGrab binary from [...]

No, I'm not dumb about many things - but SSH, telnet - never used, nor considered using. Anyway, long past time for me to leave the rest of this techno-conversation to you guys.

geek1011, 2019-09-23:
> Ah, to @davidfor who (I think) isn't a kobopatch user. He's a major participant in the Kobo beta programme. I would think he needs to keep his Kobos "pure".

The config options are relatively pure, and they don't require patches (those are just for convenience).

> No, I'm not dumb about many things - but SSH, telnet - never used, nor considered using. Anyway, long past time for me to leave the rest of this techno-conversation to you guys.

Sounds good.

---

@NiLuJe I've just been looking at the device trees, and they seem to give quite a bit of useful info (especially about things like hardware params and sensor-key mappings).

NiLuJe, 2019-09-23:
@geek1011: Yeah, they're not a particularly fun read, but they can come in handy ;p.

Short of keeping a device->misc_rota_crap mapping, no, I unfortunately don't have a silver bullet :/.

That's the whole *reason d'être* for [this enum](https://github.com/NiLuJe/FBInk/blob/f0606bc8cb03047ea2c4c1d9a957fde623d1e666/fbink.h#L186) in FBInk, which covers part of it, which got me [that](https://github.com/NiLuJe/FBInk/blob/f0606bc8cb03047ea2c4c1d9a957fde623d1e666/fbink.c#L260) which possibly behaves at least on the Forma. Although I did recently add a "actually double-check that for real", because I'm drawing a blank on doing an actual practical test of this given how maddening this is ^^.

davidfor, 2019-09-24:
> Ah, to @davidfor who (I think) isn't a kobopatch user. He's a major participant in the Kobo beta programme. I would think he needs to keep his Kobos "pure".

There's that, but, it's more that I don't have any personal need to use many of the patches. I occasionally suggest to Kobo that they should have an option for the screenshots on either the beta features of the developer options, so this come close to something I would use. But, my screenshot use tends to be at times that I can do the connection. And with covers, it just means a few extra images in the library until I remember to disable it.

Some day I'll get around to creating an options setting as part of the Kobo Utilities plugin. But, usually when I think of it, I decide I'd prefer to read a book.

closed 2022-08-06 by pgaskin