pgaskin/kobopatch-patches #23: User-supplied update for Cyrillic keyboard

jackiew1, 2019-09-25:
@geek1011, I don't know whether you saw this but an MR user has posted an update if you wanted to include it in the standard pack for next time. 
https://www.mobileread.com/forums/showpost.php?p=3894655&postcount=84

I think we need more users like this :)

geek1011, 2019-09-25:
> @geek1011, I don't know whether you saw this but an MR user has posted an update if you wanted to include it in the standard pack for next time.

Yes, I noticed that. I'm probably going to include it, as it has been requested a few times, and is well written.

> I think we need more users like this :)

I agree that was a pleasant surprise. Especially that the offsets and string lengths were correct, and that they had the thoughtfulness to put everything in order by offset (that was one of the main reasons I didn't want to rewrite all of the original ones from scratch). Also, they used a good find string instead of just putting something random from around the area which may not have stayed unique.

jackiew1, 2019-09-27:
@geek1011 
Updated Greek keyboard posted by Bald Eagle here https://www.mobileread.com/forums/showpost.php?p=3895481&postcount=125

geek1011, 2019-09-27:
Yep. I'll add them for the next release. I'm probably going to put them in `geoffr.yaml` with a note about who updated it. I've also sent Bald Eagle a PM about this.

geek1011, 2019-10-05:
@jackie_w, I've sent a PM to Semwize (it's also been forwarded to you too) about becoming a collaborator on this repository, as he's correctly updated multiple patches now (and without assistance, too).

commit e52743bf (2019-11-10): Added updated Greek and Cyrillic keyboard patches by Bald Eagle for 4.18.13737 (closes #23)
closed 2019-11-10 by geek1011