pgaskin/kobopatch-patches #25: GeoffR's multi-version "My 10 line spacing values" patch for libnickel

jackiew1, 2019-09-27:
@geek1011 
This is what I've been using. I don't know if it answers your question but I think it's working OK. 
I believe the only bits I changed myself are the last 10 lines of ReplaceFloat custom line-height values. You're on your own with the FindBaseAddressHex, ReplaceBytes bits :)
```
My 10 line spacing values - MV:
  - Enabled: no
  - PatchGroup: Line spacing values alternatives
  - Description: |
      This patch changes the values on the line spacing adjustment slider,
      reducing the number of spacing options from 15 to 10, but allowing
      much narrower spacing values to be set.
      Created by GeoffR for fw 4.11.11980 
      https://www.mobileread.com/forums/showpost.php?p=3772937&postcount=248
    # Bypass code-constructed values 1.0, 1.2, 1.5, 2.0, 3.0
  - FindBaseAddressHex: 4F F0 00 09 07 F1 80 01 4F F0 00 08 C3 F6 F0 79 3C 1D F3 58 20 46 61 E9 1E 89 7B 60
  - ReplaceBytes: {Offset: 20, FindH: 20 46 61 E9 1E 89, ReplaceH: 7B 60 03 E0 00 BF}
  - ReplaceBytes: {Offset: 108, FindH: 20 46, ReplaceH: 03 E0}
  - ReplaceBytes: {Offset: 150, FindH: 20 46, ReplaceH: 03 E0}
  - ReplaceBytes: {Offset: 210, FindH: 20 46, ReplaceH: 03 E0}
  - ReplaceBytes: {Offset: 312, FindH: 20 46, ReplaceH: 03 E0}
    # 10 values for replacement, change these replace_float values to suit:
  - FindBaseAddressHex: CD CC CC CC CC CC F0 3F 1F 85 EB 51 B8 1E F1 3F
  - ReplaceFloat: {Offset: 0,  Find: 1.05,  Replace: 0.8}
  - ReplaceFloat: {Offset: 8,  Find: 1.07,  Replace: 0.85}
  - ReplaceFloat: {Offset: 16, Find: 1.1,   Replace: 0.9}
  - ReplaceFloat: {Offset: 24, Find: 1.35,  Replace: 0.92}
  - ReplaceFloat: {Offset: 32, Find: 1.7,   Replace: 0.94}
  - ReplaceFloat: {Offset: 40, Find: 1.8,   Replace: 0.96}
  - ReplaceFloat: {Offset: 48, Find: 2.2,   Replace: 0.98}
  - ReplaceFloat: {Offset: 56, Find: 2.4,   Replace: 1.0}
  - ReplaceFloat: {Offset: 64, Find: 2.6,   Replace: 1.02}
  - ReplaceFloat: {Offset: 72, Find: 2.8,   Replace: 1.04}
```

geek1011, 2019-09-28:
Oh, I see how this works. It seems that instead of replacing the append instructions, the common instructions before the append call are patched to jump over it instead of replacing it directly. And, since this means the stack size needs to be fixed, the first few instructions are patched to change it.

The main reason why I didn't include this at that time is because that stack-size logic was a bit brittle for my liking (plus, it's a pain to re-calculate if needed).

I guess I'll probably see if I can get the PLT offsets resolved in kobopatch, in which case I can make it multi-version without messing with the stack (by referencing the append call by its name instead of its address). If not, I'll consider adding this along with a few notes on the reasoning behind the stack fixing for future reference.

geek1011, 2019-11-11:
I've decided to leave this patch as-is right now for a few reasons:

- Although the state it's in right now needs updating, it's quite quick to do so (under a minute).
- The multi-version one doesn't need updating, but when it does, it'll be a lot more work to do so, and it's a bit harder to understand.
- When I get around to the PLT stuff, I'll be able to make this patch multi-version without either of these disadvantages. If it turns out not to be possible, I still think it'll be better to keep it as-is.

closed 2022-08-06 by pgaskin