pgaskin/kobopatch-patches #26: Create new patch to customise style of new header/footer

jackiew1, 2019-09-27:
GeoffR's old patch "Custom reading footer style" included customisation of both height and style of the old footer.

The new patch "Reduce new header/footer height" handles the height part for both header/footer. There is not yet a patch to handle the style aspects. I am working on adding it.

It will be of most use to users where the default Avenir font serves their native language quite badly because it is not a Kobo-supported language, e.g. Russian, Vietnamese. However it will be usable by anyone for purely cosmetic purposes or just-because-we-like-to-tinker.

NiLuJe, 2019-09-27:
Random remark: choosing a font with poor Unicode coverage as a font used to display chapter/book titles seem like a weird choice, given the fact that even English language books might contain non-Latin titles...

jackiew1, 2019-09-27:
Can't argue with that. The old method of using the user-selected font for both book content and header/footer was better.

Personally I prefer sans-serif for all GUI stuff (serif for book content), but at least provide a half-decent sans-serif. Perhaps it's too expensive. The CJK fallback for Avenir is horrendous for Cyrillic - every char double-spaced.

geek1011, 2019-09-29:
And, here's a patch to prevent uppercasing the header/footer:

```yaml
Make book header/footer lowercase:
  - Enabled: no
  - PatchGroup: Header/footer page number text
  # In ReadingView::getChapterTitle (header), NOP QString::toUpper call:
  - ReplaceBytesNOP: {Offset: 0xA366EC, FindBLX: 0x440474}
  # In ReadingView::updateFooter, NOP QString::toUpper call:
  - ReplaceBytesNOP: {Offset: 0xA39C36, FindBLX: 0x440474}
  # Page number text
  - FindReplaceString: {Find: "%1 OF %2", Replace: "%1 of %2"}
```

jackiew1, 2019-09-29:
@geek1011
Following up on your post https://www.mobileread.com/forums/showpost.php?p=3896072&postcount=149 ...

I've spent a lot of time over the last few days looking at nickel's GUI CSS. I think know more about it now but whether I know _enough_ is still open to debate ... :)

I'm documenting here some observations about forced UPPERCASE in the GUI.

There are many instances of **text-transform:uppercase** in nickel's CSS. I wrote quick temporary patches to convert all of them to **text-transform:none** just to see what's going on. That'll do the trick, I thought ... wrongly. It worked for some bits of the GUI but not others.

The main areas where it doesn't work are:
1. new Header & Footer (nickel ReadingFooter CSS stream, #caption)
1. Series metadata in the various book lists and in  book details (MetaDataElidedLabel)
1. several (but not all) parts of the GUI which use the MetaDataLabel generic widget

Setting **text-transform:none or capitalize** don't work for items 1/2. Setting **text-transform:lowercase** _does_ have an effect, but no-one would want to use it.

My conclusion (I could be wrong) is that the fw must be delivering pre-UPPERCASE-ed text to some of these widgets.  force-lowercase-followed-by-force-capitalize might have been a klutzy workaround but I don't see how CSS could be used to do that.

Back to your potential 'I could write a patch'. How localised could we make it? I think items 2/3 should probably be left as-is. Too many other things on the go at the moment to be dealing with 'You wrecked my GUI' complaints.

I could do testing on the GUI I can see, but my main concern is that I've never enabled wi-fi (nor do I intend to) so there are many screen layouts I've _never even seen_ even though I got my first Kobo in 2012.

If we could localize to new Header/Footer, I think that should be safe. Famous last words ...

BTW you coerced me, kicking & screaming, into using GitHub. Are you sorry yet? :D

jackiew1, 2019-09-29:
... and now I see you've already done the work whilst I was wring the above. 

I really wish you hadn't posted it in open forum, though. For things that affect my patches I'd like to get everything organised in my own head before people start playing around with them. 'Home screen wars 2' beckons if you're not careful ...

geek1011, 2019-09-29:
> I've spent a lot of time over the last few days looking at nickel's GUI CSS. I think know more about it now but whether I know enough is still open to debate ... :)

Yes, it's pretty long. Although, @NiLuJe and I are working (slowly) on figuring out the plugin system (see the private repo I added you to), and we might be able to put something together to live-edit the nickel CSS.

> My conclusion (I could be wrong) is that the fw must be delivering pre-UPPERCASE-ed text to some of these widgets. force-lowercase-followed-by-force-capitalize might have been a klutzy workaround but I don't see how CSS could be used to do that.

That's correct.

> BTW you coerced me, kicking & screaming, into using GitHub. Are you sorry yet? :D

Haha, it looks like you're enjoying the features it gives you more than PMs. I don't think you regret your decision :smiley:.

> I really wish you hadn't posted it in open forum, though.

Sorry about that. Although, it doesn't affect any of your patches directly, and there's not much choice about how to do it.

P.S. I've finally made a simple [homepage](https://pgaskin.net/kobopatch-patches/) for the patches so there are links to everything in one place.

jackiew1, 2019-09-29:
I hope this is an environment where I can say what I think. I don't mean any offence but I need to say it for the sake of a harmonious future ...

> Sorry about that. Although, it doesn't affect any of your patches directly, and there's not much choice about how to do it.

Maybe so, but it would have been much better for me if you'd liaised with me so it could be released at the same time as its partner, i.e. this new ReadingFooter patch. I'm trying to handle too many Kobo-related MR queries ATM and you just created potential for more.

What I really want to concentrate on, for a little while, is improving the GUI for users of Kobo non-supported languages. Anything we come up with will also have cosmetic benefits for those who enjoy tinkering. I'm just about to create a new thread for it, to avoid de-railing the main kobopatch/firmware threads. It may be a total damp squib but I feel the need to at least try. What we discover there may have a knock-on effect on how to do the 2 new patches (issues 26/27) and hopefully will lead to other new long-term helpful patches. I don't want to rush things out and have to change/fix them almost immediately - like you just had to. I _hate_ false starts, non-techie users get confused enough as it is.

Honestly, I do know you're only trying to make everyone happy as fast as you can :) but the hi-speed bull-at-a-gate approach doesn't work at all for me. As the old saying goes _"no-one will remember how fast you did it but they will remember how well you did it"_. We're all a product of our own past work experiences and user-support was a major part of mine.

geek1011, 2019-09-29:
> Maybe so, but it would have been much better for me if you'd liaised with me so it could be released at the same time as its partner, i.e. this new ReadingFooter patch. I'm trying to handle too many Kobo-related MR queries ATM and you just created potential for more.

Point taken.

> I'm just about to create a new thread for it, to avoid de-railing the main kobopatch/firmware threads.

OK, I was about to do that myself (and ask the mods to move those posts from the firmware thread).

> I don't want to rush things out and have to change/fix them almost immediately - like you just had to. I hate false starts, non-techie users get confused enough as it is.

Do you want to include #26/#27 with the next release (v53), which I'm planning to do sometime this week (I'm giving the release notes some time to settle first, then I want to release it for 4847f39deeac85beca376fb1a1498f69a26699ea, 7992a174d0efaba01db4b0ef71c5165a879e62dd, a4a9723be795e2b46aaa12b1b2e511e2491ef893, 46203146b109ea85b1bd757e85ca077c7c0a5cb6, cd78216dc2ce6e816d4d398daea4ea9c584c6432, and 5ca1c6361b6efa822b7a5be0e0eb054a0be6e8ab)?

> The hi-speed bull-at-a-gate approach doesn't work at all for me. As the old saying goes "no-one will remember how fast you did it but they will remember how well you did it". We're all a product of our own past work experiences and user-support was a major part of mine.

:+1: I'm also planning to take a break for a week or two after v53.

jackiew1, 2019-09-29:
> I'm just about to create a new thread for it, to avoid de-railing the main kobopatch/firmware threads.

This is my new thread. https://www.mobileread.com/forums/showthread.php?t=323350

> Do you want to include #26/#27 with the next release (v53), which I'm planning to do sometime this week

It depends whether there's much (or even any?) reaction to my new thread above. If there isn't I'll update the jackie_w nickel.yaml file with what I've already done for #26/#27. I'll also need to prepare a special post for each, with screenshots, So a few hours warning before planned release would be helpful - this time, please!, NOT in the middle of the night in my UK timezone which is GMT +/- 1hr this week (_I can never remember which way round it is_).

> 👍 I'm also planning to take a break for a week or two after v53.

I was a bit snappy at a couple of aggravating users this week, so you might be right. Time for a break :)

jackiew1, 2019-09-29:
> And, here's a patch to prevent uppercasing the header/footer:
> 
> ```yaml
> Make book header/footer lowercase:
>   - Enabled: no
>   - PatchGroup: Header/footer page number text
>   # In ReadingView::getChapterTitle (header), replace QString::toUpper call with
>   # QString::trimmed (to copy the string, and not need to add a mov instruction
>   # and NOP the destructor):
>   - ReplaceBLX: {Offset: 0xA366EC, FindBLX: 0x440474, ReplaceBLX: 0x451410}
>   # In ReadingView::updateFooter, do the same:
>   - ReplaceBLX: {Offset: 0xA39C36, FindBLX: 0x440474, ReplaceBLX: 0x451410}
>   # Page number text
>   - FindReplaceString: {Find: "%1 OF %2", Replace: "%1 of %2"}
> ```

@geek1011 
Call me a PITA but this is the conversation we needed to have before you launched it. :) 

How do you propose we handle the case when users want to un-force UPPERCASE for header/footer and also change '%1 OF %2' to '%1 / %2'? 

IMHO creating a new PatchGroup is the wrong solution. 'Same-but-slightly-different' patch proliferation is to be avoided at all costs.  I keep banging on about it, but no more Home screen fiasco, please! I may be misunderstanding what you are doing but if I'm confused what chance everyone else?

If un-force UPPERCASE and simple '%1 OF %2' text rename can't be kept completely independent of each other then we need to come up with a way of replacing my earlier trivial "Custom header/footer page number text" with your new one, 

Keep the old name but put it in your libnickel.yaml file, not mine, because you'll need to do the hex-location stuff every fw update.

Yours needs to be split into sections with more Help Notes so users can easily enable/disable the combo of options they want. 

Comment out your new option so that the old simple rename option is the default so as not to confuse those who have already set up their kobopatch.yaml for the old simple option and will never want anything more than that.

geek1011, 2019-09-29:
> Call me a PITA but this is the conversation we needed to have before you launched it. :)

I didn't make it official, yet. Remember that patches I post directly on MR don't count as final, tested, working, or even supported. And, the one in the repo isn't released yet.

> How do you propose we handle the case when users want to un-force UPPERCASE for header/footer and also change '%1 OF %2' to '%1 / %2'?

One option would be to have an always-on patch which replaces `OF` with `of`, and adds CSS to transform it to uppercase. Then, my patch can remove that CSS, while yours changes `of` to `/`. That would solve all the issues you mentioned, with the disadvantage of having an always-on patch.

> IMHO creating a new PatchGroup is the wrong solution. 'Same-but-slightly-different' patch proliferation is to be avoided at all costs. I keep banging on about it, but no more Home screen fiasco, please! I may be misunderstanding what you are doing but if I'm confused what chance everyone else?

Remember that since then, I added support for more than one PatchGroup. Now, it means something more like "Conflicts" or "Touches".

NiLuJe, 2019-09-29:
> I'll also need to prepare a special post for each, with screenshots, So a few hours warning before planned release would be helpful - this time, please!, NOT in the middle of the night in my UK timezone which is GMT +/- 1hr this week (_I can never remember which way round it is_).

+1 for DST ;). (I'm on GMT+1 +  DST,  1 hour difference, but easier to remember ;p).

jackiew1, 2019-09-29:
> > Call me a PITA but this is the conversation we needed to have before you launched it. :)
> 
> I didn't make it official, yet. Remember that patches I post directly on MR don't count as final, tested, working, or even supported. And, the one in the repo isn't released yet.

But you have created the expectation that it'll be in the next release real soon now.

> One option would be to have an always-on patch which replaces `OF` with `of`, and adds CSS to transform it to uppercase. Then, my patch can remove that CSS, while yours changes `of` to `/`. That [...]

I don't understand any of the above. Until further notice I'm considering it something you're going to handle entirely yourself.

> Remember that since then, I added support for more than one PatchGroup. Now, it means something more like "Conflicts" or "Touches".

Ability to handle Conflicts - Good. 
Ability to avoid creating Conflicts in the first place - Better :)

Just because you created a technical solution to detect conflicts doesn't mean you created a practical solution to avoid user confusion.

jackiew1, 2019-09-30:
@geek1011,
Added new patch to nickel.yaml/jackie_w.yaml ready for next release.

geek1011, 2019-09-30:
It's too long: https://cloud.drone.io/geek1011/kobopatch-patches/150/2/18 .

And, do you want to remove the need for a patch group for the page text patches by having an always-on normalization patch, like stated above?

jackiew1, 2019-09-30:
I swear I've spent more time writing today than a real author would have ...

> It's too long: https://cloud.drone.io/geek1011/kobopatch-patches/150/2/18 .

Ah, I see. I never thought about your auto-tester. The last 4 lines are commented out and provide the ability, with help notes, for the user to un-comment as necessary to shorten enough to pass kobopatch, by removing Japanese/Chinese-specific stuff. 

I guess what you're telling me is that I'll need to un-comment those 4 F/R by default. I'll also need to change the help to reflect that.

To be consistent I'll need to do that on the 2nd new patch as well. I'll update both later tonight.

> And, do you want to remove the need for a patch group for the page text patches by having an always-on normalization patch, like stated above?

We don't appear to be communicating very well, do we? As I said at the time I didn't understand any of that and there's no need to explain. Let me spell out what I want.

I want you to completely remove me from the equation by fully adopting the old "Custom header/footer page number text". Move it from libnickel.so.1.0.0.yaml/jackie_w.yaml to libnickel.so.1.0.0.yaml/geek1011.yaml. Or I can just delete mine.

Once you've done that you can decide how best to allow users any A/B combo they like, _as long as it's still feasible_, from:
A:
- header/footer is UPPERCASE-ed (current default)
- header/footer is not UPPERCASE-ed

B:
- page text is %1 OF %2 (current default, but would anyone _really_ miss it?
- page text is %1 of %2
- page text is %1 / %2

As you and your 1 (or 2) patches will now be completely responsible for this, including writing helpful patch comments or handling any user confusion because there aren't any helpful comments, I won't mind how you do it.

I don't think it affects my remaining patches at all, does it? All #26 now does is allow custom font-family/font-size for header/footer. Actual text content of header/footer shouldn't matter, should it?

I think you can leave me with "Custom navigation menu page number text" or is that also going to be affected by your new patch? If 'yes' then you need to adopt that one, too.

BTW, don't you think the new name "Make book header/footer lowercase" is misleading? Presumably the result will be MixedCase not lowercase. And aren't you 'stopping forced uppercase' rather than 'converting to lowercase'? Maybe rename to "Book header/footer un-force uppercase". That still looks like nerd-speak but at least it's factual. There's several other patches with 'un-force' in the name and no-one's complained :)

geek1011, 2019-09-30:
> I guess what you're telling me is that I'll need to un-comment those 4 F/R by default. I'll also need to change the help to reflect that.

I did some testing, and it seems like uncommenting the Japanese one is enough. One thing I tried (it was almost enough) was condensing the Chinese stuff like `{Find: "#caption[localeName=\"zh\"] {\n  font-family: Sans-SerifZH-Traditional, sans-serif;\n  font-style: normal;\n}\n#caption[localeName=\"zh-HK\"] {\n  font-family: Sans-SerifZH-Traditional, sans-serif;\n  font-style: normal;\n}\n#caption[localeName=\"zh-TW\"] {\n  font-family: Sans-SerifZH-Traditional, sans-serif;\n  font-style: normal;\n}", Replace: "#caption[localeName=\"zh\"],#caption[localeName=\"zh-HK\"],#caption[localeName=\"zh-TW\"]{font-family:Sans-SerifZH-Traditional,sans-serif;font-style:normal}"}`.

> To be consistent I'll need to do that on the 2nd new patch as well. I'll update both later tonight.

When you update it, also add it back to 4.18.13737 (replace the line like `# Waiting (jackie_w) (zlib doesn't fit with default config): Custom new header/footer font`).

> I want you to completely remove me from the equation by fully adopting the old "Custom header/footer page number text". Move it from libnickel.so.1.0.0.yaml/jackie_w.yaml to libnickel.so.1.0.0.yaml/geek1011.yaml. Or I can just delete mine.

OK, that's easy enough to do. I'll just do as I said earlier, and add a comment about the always-on patch. That way, all 3 options will be usable without the need to have manual customizations.

> BTW, don't you think the new name "Make book header/footer lowercase" is misleading? Presumably the result will be MixedCase not lowercase. And aren't you 'stopping forced uppercase' rather than [...]

Yes, it's a bit misleading. I'm a bit on the fence about this, as I don't like the wordings I can think of. @NiLuJe, any ideas?

---

I'm almost done with my part for #28, so tell me once you're ready for a release. It'll either be sometime in the next 3 hours, or it'll be tomorrow around the time I created #28 (or 3 hours from now tomorrow).

NiLuJe, 2019-09-30:
The potentially simpler and more pleasing option would be "Ensure header/footer is TitleCased" or something to the effect, but it is, of course, technically wrong, as I'm guessing it's just preventing it from being uppercased, so, it'll only be Title Cased if that how it was to begin with ;p.

TL;DR: "Don't uppercase header/footer"? It's still vague enough to maybe warrant a technical description somewhere, but hopefully gets the gist of it across at first glance?

geek1011, 2019-09-30:
> TL;DR: "Don't uppercase header/footer"? (+/- "titles", in case that applies to the full string, or only book/chapter titles).

I think I like that. I'll update the names (it's a good thing it isn't released yet, as we did enough breaking changes which I usually avoid last version).

> It's still vague enough to maybe warrant a technical description somewhere, unless it's already part of one of the preview webpages, but hopefully gets the gist of it across at first glance?

I've already added it to the preview.

jackiew1, 2019-09-30:
Is there any compelling reason to rush this out tonight? Surely a 24 hour turnaround will still be pretty acceptable.

It's nearly 01:00 here and I still need to make those corrections you need to the new patches. I got derailed by the 4.18 announcement and no, I don't want you to do anything to the 2 new patches on my behalf.

"Don't uppercase header/footer" sounds OK. Minor nit, not sure about apostrophes in names, though.

geek1011, 2019-10-01:
> Is there any compelling reason to rush this out tonight? Surely a 24 hour turnaround will still be pretty acceptable.

Not really. I'll leave it for tomorrow night.

> It's nearly 01:00 here and I still need to make those corrections you need to the new patches. I got derailed by the 4.18 announcement and no, I don't want you to do anything to the 2 new patches on my behalf.

OK, sounds good.

> "Don't uppercase header/footer" sounds OK. Minor nit, not sure about apostrophes in names, though.

Apostrophes are OK when they aren't at the beginning or end.

geek1011, 2019-10-01:
> One option would be to have an always-on patch which replaces OF with of, and adds CSS to transform it to uppercase. Then, my patch can remove that CSS, while yours changes of to /. That would solve all the issues you mentioned, with the disadvantage of having an always-on patch.

I just realized I can't do this, as I don't have a way of specifying dependencies (and I don't think it's worth adding it just for this), and the patch order is not guaranteed to be deterministic. I'll probably just have 3 patches for now, and combine them if I find a better way. I could also merge them into 1, but then customizations would be needed for every update.

jackiew1, 2019-10-01:
Well, good luck explaining all that in words of one syllable your average user will understand.!

geek1011, 2019-10-01:
> Well, good luck explaining all that in words of one syllable your average user will understand.!

Haha, I'm not planning to explain the reasoning behind the decision in the patch, I'm just going to split it into 3 and put it in a PatchGroup. 

And, before anyone asks, I'm putting it into 3 patches instead of merging them, as it gives me more flexibility in the future, and also allows people to use the overrides option with the individual parts of the patch. Also, the fact that they depend on each other is an internal firmware detail which shouldn't be exposed to users.

closed 2019-10-01 by geek1011