pgaskin/kobopatch-patches #28: Patches for 4.18.13737

geek1011, 2019-09-30:
And here we go again...

geek1011, 2019-09-30:
- The advanced font controls patch needs to be partly rewritten; some logic has moved from loadView to setupFontFaceDropdown.

NiLuJe, 2019-09-30:
@geek1011: If that lives in .kobo, we're safe, as both the mods and my config leave .kobo/.adobe alone ;).

geek1011, 2019-09-30:
I'm going to take a short break, then I'll see if I can fit @jackiew1's patch into the zlib stream without uncommenting the language removal stuff, I'll test some Dropbox stuff, I'll probably implement #29, and I'll test the patches and update my Kobos.

jackiew1, 2019-09-30:
@geek1011,
Re: Geoff, I think we'll have to assume he won't be updating anymore. 

I was interested to see that Semwize made an attempt at updating Geoff's "Custom font sizes". If he knows coding that well have you considered asking him to contribute more formally? Maybe he could figure out the "My 24 line spacing values" one, too :)

Re: update to "Custom collection/author header title font". 
I'd already figured out the new FindZlib and I was expecting to change it myself. Any reason why you did it?

Also, how did you know that the change to the 2nd new conflicting stream also needed adding to this patch? As the new stream is Dropbox related have you checked what it actually does? Won't the patch name and description need updating to reflect the extra feature?

geek1011, 2019-09-30:
> I was interested to see that Semwize made an attempt at updating Geoff's "Custom font sizes". If he knows coding that well have you considered asking him to contribute more formally? Maybe he could figure out the "My 24 line spacing values" one, too :)

I'd have to ask if he knows how to use a disassembler and at least have a basic understanding of assembly. There's two possible ways he did what he did: find the hex nearby in the old version, and did a search for it (in which case, Semwize is more resourceful than the average user, but not enough to do new development on it [such as the Libra values]), or he actually understood the assembly and updated it using that (in which case, he'd be a good fit for taking over that patch).

> I'd already figured out the new FindZlib and I was expecting to change it myself. Any reason why you did it?

I came across it as I was doing my usual inspections, so I did it to make the automatic tests pass. If you have a better FindZlib string, you can replace it.

> Also, how did you know that the change to the 2nd new conflicting stream also needed adding to this patch? As the new stream is Dropbox related have you checked what it actually does? Won't the patch name and description need updating to reflect the extra feature?

Based on a quick look at it, it's pretty much the same situation as what happened with the kepub progress chart one two versions ago. It's just a modification to the original stream which doesn't apply to everything yet, and the old one will probably be gone for the next version.

geek1011, 2019-09-30:
And ....... something's changed about one of the TODOs with the libnickel stuff. It now segfaults when opening a book.

geek1011, 2019-10-01:
TODO: https://www.mobileread.com/forums/showpost.php?p=3896805&postcount=32

geek1011, 2019-10-01:
So here's what is left:
- [ ] @geek1011: Fix the larger sleep/poweroff timeouts patch.
- [ ] @geek1011: #29 (maybe)
- [ ] @jackiew1: https://github.com/geek1011/kobopatch-patches/issues/26#issuecomment-536800107
- [ ] @geek1011: https://github.com/geek1011/kobopatch-patches/issues/26#issuecomment-536332966

Did I miss anything?

geek1011, 2019-10-01:
FYI @jackiew1, I'm planning to do the release in either 2 hours or 8 hours from now (I'm busy in between).

geek1011, 2019-10-01:
@NiLuJe or @jackiew1, do you mind confirming the issue JSWolf reported with the sleep timeouts patch (I don't have a reader with me right now)?

jackiew1, 2019-10-01:
> FYI @jackiew1, I'm planning to do the release in either 2 hours or 8 hours from now (I'm busy in between).

2 hours is good for me.

NiLuJe, 2019-10-01:
I'm not using the patch in question, but there may be some specific HW quirks that make >24H wakeups tricky.

Not enough users (and tricky to test) to have more info right now ;).

jackiew1, 2019-10-01:
> @NiLuJe or @jackiew1, do you mind confirming the issue JSWolf reported with the sleep timeouts patch (I don't have a reader with me right now)?

@geek1011 
I will but I can't test a 24 hour timeout before planned release of 4.18 :) So you'll need to decide whether to exclude that patch until it can be tested properly.

I will mention here that in 4.17 I had problems with my Forma not auto-powering down after 12 hours when it was put to sleep by its sleepcover. It was OK with shorter times if I just left the cover open - but no-one is going to do that in real life.
My ClaraHD in its sleepcover seemed to be working OK with 12 hours in 4.17.

I had a feeling that my Forma's problem might be nothing to do with kobopatch and everything to do with its MiniClock hack installation which is looking for button presses as well as touch. I thought I'd wait to see what others reported.  I haven't really had the time to pursue it but I will do when the madness finally subsides.

NiLuJe, 2019-10-01:
@jackiew1: The SleepCover is essentially interpreted as a button, with press/release.... and *repeat* events, so it's quite likely to be annoyingly screwy to deal with.

jackiew1, 2019-10-01:
For my own practical purposes I'm not too bothered because the Forma is the device least likely to go 12 hours without me opening it. But if there are problems other users will certainly keep reporting it ad infinitum.
1. No release
1. Release with copious warnings
1. Release and keep handling complaints.

IMO, _please_, not 3.

NiLuJe, 2019-10-01:
Pessimistic view here, coming from a few intense experiences with dealing with suspend-related power management stuff on Kobo, but I can see approximately a zillion ways this could potentially silently go wrong, so, eh, at the very least I'd slap a permanent "beta" sticker on it, with copious warnings (i.e., might not work, worse, might leave the device *completely awake* instead of just in suspend).

jackiew1, 2019-10-01:
> ... (i.e., might not work, worse, might leave the device _completely awake_ instead of just in suspend).

Speaking as a user, I'm fairly tolerant of hackers' "best efforts", but any kind of battery drain experience would make me disable it permanently.

jackiew1, 2019-10-01:
P.S. I've thought some more. Surely the only sane decision is to exclude it, at least until some better testing can be done.

geek1011, 2019-10-01:
> Pessimistic view here, coming from a few intense experiences with dealing with suspend-related power management stuff on Kobo, but I can see approximately a zillion ways this could potentially [...]

I'm certain it won't cause more battery drain than the original options, as all it does is change the multiplier used for setting the internal timer, for comparing if sleep or power-off comes first, and for setting the RTC timer.

I'm not even totally convinced JSWolf's issue is an issue with this patch, and not a firmware or other issue. That's why I asked you about testing it.

In addition, I don't see any major changes with power-handling since the last version, where it was confirmed to work by multiple users. I also tested it a bit on the last firmware version for up to 9 hours, and it seemed to work fine. I didn't test the power-off as much, though.

Assuming it is an issue caused by the interaction between this patch and the firmware, I'd say it's most likely to be an issue with the power-off timeout being interrupted by the sync timeout. This is because the sync timeout is set separately, and causes a wakeup on it's own IIRC.

Because of this, I have decided to leave it in, but add a warning about it not powering off for long times (and saying that turning off auto-sync may fix it).

jackiew1, 2019-10-01:
But the only reason people have asked for it is for the longer time power offs, i.e. 24hours+

FTR my auto-sync is OFF, wi-fi is OFF, sleepcover is ON, and always have been, on every device I own.

geek1011, 2019-10-01:
> But the only reason people have asked for it is for the longer time power offs, i.e. 24hours+

And >4 hours (I'm one of those people), due to the other patch only going up to 4.

geek1011, 2019-10-01:
Unless anyone has objections/last minute fixes, I'll make the release in 15 minutes (*:10 PM/AM).

geek1011, 2019-10-01:
And, for v54, I might take over @geoffreynolds patches if Kobo doesn't release another firmware version within the next two weeks.

jackiew1, 2019-10-01:
> And >4 hours (I'm one of those people), due to the other patch only going up to 4.

I'm not buying it. Being able to set say 5-10 hours is not going to satisfy many. Jon will probably keep on at you until multiple days works. Are you prepared for that?

With my typical user hat on. My 7 values were 
- approx 10min, 15min, 25min (only used for sleep)
- approx 12hrs, 24hr, 36hrs (only used for power off)
- Never

I'm OK with 15mins release

NiLuJe, 2019-10-01:
(Especially given the fact that if Pocket is truly wonky, we might see a new release soon-ish...).

geek1011, 2019-10-01:
> I'm not buying it. Being able to set say 5-10 hours is not going to satisfy many. Jon will probably keep on at you until multiple days works. Are you prepared for that?

Yep. I'll be able to keep up with that. I'll probably make him the lead tester ;).

> (Especially given the fact that if Pocket is truly wonky, we might see a new release soon-ish...).

Oh no ... we all need a break.

jackiew1, 2019-10-01:
> > I'm not buying it. Being able to set say 5-10 hours is not going to satisfy many. Jon will probably keep on at you until multiple days works. Are you prepared for that?
> 
> Yep. I'll be able to keep up with that. I'll probably make him the lead tester ;).

To be fair, he did some sterling work as beta tester for "Custom header/footer font" over the last few days. :)

geek1011, 2019-10-01:
Done! Posted: https://www.mobileread.com/forums/showthread.php?t=323419. Building: https://cloud.drone.io/geek1011/kobopatch-patches/167.

closed 2019-10-01 by geek1011

jackiew1, 2019-10-01:
Hang on ... why is my name still against this
> - jackie_w: Custom header/footer page number text

Apart from that I managed to get in at the top this time. :)

geek1011, 2019-10-01:
Release notes are done: https://github.com/geek1011/kobopatch-patches/releases/tag/v53.

> Hang on ... why is my name still against this

You were the original creator. I'll add a note about the transfer.

> Apart from that I managed to get in at the top this time. :)

I noticed!

jackiew1, 2019-10-04:
> But the only reason people have asked for it is for the longer time power offs, i.e. 24hours+
> 
> FTR my auto-sync is OFF, wi-fi is OFF, sleepcover is ON, and always have been, on every device I own.

I know this thread is closed but I thought you might like an extra datapoint. I applied the Larger poweroff patch on 4.18 on my ClaraHD, setting same as quoted above. I set poweroff to slightly less than 24hrs and put it to sleep with my sleepcover. I opened it  27hrs+ later and it was powered off as it should be. 

Not able to re-test on the Forma yet because I keep needing to open the sleepcover.

geek1011, 2019-10-04:
Thanks @jackiew1! That seems to confirm my theory about the cause of the intermittent cases of the patch not working.

P.S. Issues are closed when the main point has been resolved, but it doesn't mean it isn't open for discussion (unless it's locked too, which I don't do).

jackiew1, 2019-10-04:
Perhaps the poweroff stuff should be moved to an open issue of its own. Having said that, you haven't had any complaints yet so maybe it's not an issue :)

jackiew1, 2020-01-15:
> I know this thread is closed but I thought you might like an extra datapoint. I applied the Larger poweroff patch on 4.18 on my ClaraHD, setting same as quoted above. I set poweroff to slightly less [...]

@geek1011 @NiLuJe A further update ...

There was a time when the Forma (in a sleepcover) wouldn't auto-poweroff after >12hours. My current patch setting is for it to auto-poweroff after approx. 36hours. This now seems to be working OK  ... although I don't know the exact time when it did poweroff, nor do I know exactly when it started to work OK.

Is this maybe something to do with changes in MiniClock rather than anything to do with the patch?

Whatever the reason, it's an improvement. 😃

NiLuJe, 2020-01-15:
Hmm, I don't *think* MiniClock should have ever had any interaction with the SleepCover events, but, who knows ;).

As another data point, mine is set to poweroff after 15 days in KOReader, and that works ;). No sleepcover, though.

geek1011, 2020-01-15:
To properly debug the patch, we need to track power states and events. The Linux power trace points, if enabled on the Kobo, could be useful for this.

skontar, 2020-02-17:
Hello. I am not exactly sure if this is the best place to put my feedback...

> If you are using this patch with timeouts above 12 hours, and can confirm
>       if it works/doesn't work, please PM me or comment on GitHub with your
>       setting values, sleepcover or not, auto-sync interval, wifi on/off, firmware
>       version, and Kobo model.

Setting values: sleep 8m, power off 24h1m
Sleep cover: yes
Auto-sync: Was on (overnight), but Wifi was off
Wifi: Off
Firmware: 4.19.14123
Model: Kobo Aura H2O

Device powers-off much faster, I think less then 8 hours (over night).
I have now tried to disable auto-sync and see if it makes any difference.

geek1011, 2020-02-17:
Thanks for the report. What was the shift value?

NiLuJe, 2020-02-17:
FWIW (because this was mentioned on MR), the actual device *is* capable of >24H rtc alarms (it works fine in KOReader ;)).

geek1011, 2020-02-17:
Yes, but the issue was that nickel sets its own alarms for autosync and possibly other stuff.

skontar, 2020-02-18:
@geek1011 it was the patch default, so I think it is 19? Anyway, even with disabled auto-sync it again powered down during night, so in about 8 hours instead of 24.

geek1011, 2020-02-18:
Hmm. I'll take a closer look at this when I update the patches for the next version.

skontar, 2020-02-19:
@geek1011 Disabling sleep cover in settings did not help either.

Also "8m" setting for sleep is working as expected. I kept it in sleep during work so I could keep an eye on it and it powered off in ~7 hours with "24h1m" setting.

geek1011, 2020-02-19:
Does setting a 24h sleep (rather than power off) time work?

gonzule, 2020-03-27:
Hi, i´m also having issues with the large sleep/power-off timeouts. I have a Kobo aura on 4.20.14622 with a sleepcover. I have power off time configured as 24h1m, but the reader will always be powered off when I pick it up to continue reading. I read at nights, so it´s like 23 hours between when I close the sleepcover and when I pick it up again to continue reading.

Autosyn is off (i had it on until 1 week ago, but i had the same issue)
Sleepcover works properly
Its a normal Kobo Aura, not HD
Wifi is ALWAYS off
Sleep time is 8 minutes
Power off time is 24h1m
Automatic sleep and wakeup is enabled (because of the cover)

Large power off time used to work well a while ago when modifying  koboereader.conf (see here: https://www.mobileread.com/forums/showpost.php?p=3738078&postcount=3)  but that stopped working a few months ago so i started using kobopatch

skontar, 2020-03-27:
@geek1011 24h sleep (rather than power-off) did not work either. It did not go to sleep early as with the poweroff but did not go to sleep in 24 hours either. However this may be irrelevant as I have noticed that you mentioned different offsets in the new version. Maybe I will try it later with new version.