pgaskin/kobopatch-patches #30: GeoffR's "Custom font sizes"

jackiew1, 2019-10-02:
Semwize posted a new version for 4.18. https://www.mobileread.com/forums/showpost.php?p=3897542&postcount=31

I asked whether he knew how to add the LibraH2O stuff. Answer is no.

ezel3, 2019-10-26:
Hmm, seems I reinvented the wheel. I also made a new version before I saw this one. But if someone is interested, I also documented the hex patterns (see 'Find hex string' in the zip) so it is much easier to locate the places that must be altered. Hope this is of some use for future patches.
[custom_font_sizes.zip](https://github.com/geek1011/kobopatch-patches/files/3775065/custom_font_sizes.zip)

geek1011, 2019-11-10:
So, it seems the Libra H2O uses the same values as the Clara. This can be determined based on my [research](https://gist.github.com/geek1011/613b34c23f026f7c39c50ee32f5e167e) (those multi-layer device codenames are a bit annoying) and tracing the complicated (because of GCC optimization) set of branches on Device::is*. I've updated Semwize's updated patch with this info.

commit 5aea5475 (2019-11-10): Updated "Custom font sizes" patch for 4.18.13737 based on Semwize's updated version (closes #30)

The Libra uses the same values as the Clara (this is based on density). This
can be confirmed with information from https://gist.github.com/geek1011/613b34c23f026f7c39c50ee32f5e167e
and the Device::is* calls in FontTypeUtil::fontSizes.
closed 2019-11-10 by geek1011