pgaskin/kobopatch-patches #31: "Custom new header/footer font" - right-half progress part

jackiew1, 2019-10-03:
I want to try to track down why using "Custom new header/footer font" to change font-family from Avenir to _whatever_ only works on the Chapter name, Book title parts of header/footer, not the progress part on the right.

@geek1011 @NiLuJe 
I'm not asking you to do this for me (_at least not yet_ :)) but before I spend a lot of effort with trial & error in nickel CSS I'm asking whether you already know the answer from when you were researching "Don't uppercase header/footer text"?

NiLuJe, 2019-10-03:
Do you happen to have the relevant CSS string on hand, just for reference?

jackiew1, 2019-10-03:
Do you mean the CSS for "Custom new header/footer font" - that's the stream which contains **ReadingFooter**

jackiew1, 2019-10-04:
I spent quite a lot of time yesterday trying to track this down - no joy. It's tedious and time consuming. @geek1011 , @NiLuJe , if you do have any insights to point me in the right direction, I'm all ears.

It shouldn't be considered a priority because the only 2 users who have complained now have an alternative solution. All that's left is me failing to understand why applying CSS to hdr/ftr works for all of the text it when it's font-size/color etc but not the book progress right-half when it's font-family.

jackiew1, 2019-10-05:
Update:
I remembered that it was faster to hack global changes to nickel CSS using the old pipcat method, so I used it to remove every single trace of Avenir. The right-half of both header/footer is still displayed in Avenir.

Conclusion either
- the relevant CSS is not in nickel, it's elsewhere (libnickel.so.1.0.0 ???)
or
- there is CSS in nickel that neither @geek1011's cssextract-windows nor pipcat's old extract is currently finding

geek1011, 2019-11-10:
I'm going to close this issue for now.

jackiew1, 2020-01-15:
@geek1011 as per your post here https://www.mobileread.com/forums/showpost.php?p=3940243&postcount=432 the best way I can think of reminding you is by re-opening this issue for when you have some time/inclination for further investigation.

geek1011, 2020-05-19:
I looked into this, and there isn't any way simple enough to be worth doing for this change.

closed 2020-05-19 by geek1011