pgaskin/kobopatch-patches #32: Book title in header/ Chapter name in footer

jackiew1, 2019-10-04:
Several users are complaining about Kobo's decision to reverse which data was in header/footer.

I don't even know if this is possible and I wouldn't expect any early action from anyone but at least the thought won't get lost.

geek1011, 2019-10-04:
OK, so I decided to take a short break from my break :wink: to have a quick look at a hunch I had about a simpler way to do this, and it turned out to be correct.

When I originally said it wasn't easy enough to do without adding too much time for updates, I was looking at swapping the widgets themselves. Earlier today, I had the idea of swapping the updates instead of the widgets, as both are children of ReadingView and are ReadingFooter widgets (thus the update methods are compatible). From there, I just had to make sure they were not referenced anywhere else for anything else, and it wasn't, so I only needed to swap the offsets for the LDR instruction (0x8 vs 0x14).

closed 2019-10-04 by geek1011

jackiew1, 2019-10-04:
@geek1011 , LOL, you just can't resist, can you? It was supposed to be a back-burner...

Have you got part 2 ready, when someone says they want the Book title as header but the full book progress kept in the footer? :D 

Did you see anything that might help me with #31 when you were poking around?

geek1011, 2019-10-05:
> LOL, you just can't resist, can you? It was supposed to be a back-burner...

Yes ... but ... I had an idea about it so I decided to try it :laughing: .

> Have you got part 2 ready, when someone says they want the Book title as header but the full book progress kept in the footer? :D

That will actually go on the back burner. I'm not planning on making a patch for that if someone requests it.

> Did you see anything that might help me with #31 when you were poking around?

Not really, although I can say the page number is a separate QLabel, as you can see in the assembly file I attached (just note the value of r0 and where it came from before the QLabel::setText calls; MOV a, b is like a = b, and LDR a, [b, c] is like setting a to the value at the address b + c).

[libnickel-13737_procc11f14.asm.txt](https://github.com/geek1011/kobopatch-patches/files/3692843/libnickel-13737_procc11f14.asm.txt)

geek1011, 2019-10-05:
Note to self: fix https://github.com/geek1011/kobopatch-patches/commit/71eb7447dcc877ad5fc1aa594919c58338406a31#r35371887 before the next release.

geek1011, 2019-10-06:
Another note: check extra footer options for references to the labels to fix part of header/footer missing.

jackiew1, 2019-10-06:
> Another note: check extra footer options for references to the labels to fix part of header/footer missing.

@geek1011 is this note referring to this problem reported by Zodwallop https://www.mobileread.com/forums/showpost.php?p=3899147&postcount=59

If so, I'm not convinced it's anything to do with your new patch. I've seen similar on a completely unpatched AuraHD.

geek1011, 2019-10-06:
Yes, as I believe I may be able to partially fix it.

jackiew1, 2019-10-06:
OK, don't let me stop you 😄 The MR post read as if the patch had caused the problem.

commit 71eb7447 (2019-10-04): Added new "Swap reading header/footer" patch (closes #32)