pgaskin/kobopatch-patches #33: Kobo Forma's orientation is momentarily upside down after every wake or power-on.

hub2git, 2019-10-14:
I like holding my Kobo Forma in my right hand.
On my Kobo Forma, I have inverted "Next" and "Previous" page-control hardware buttons. 

These two things combined seem to cause my Forma, after waking up or powering on, to momentarily have the text or the enter-your-PIN page to be upside down, before correcting itself to be right-side up. I don't want to waste time every time I pick up my Forma waiting for it to have the right orientation.  That wastes 5 seconds every time.

I noticed that when my Next and Previous buttons are not inverted, I don't have this temporary hiccup. But, I will not choose "non-inverted" just to avoid this hiccup. I would like to invert my page-control buttons AND not have this glitch.

geek1011, can you add a patch that fixes this?
Maybe we can we can force the Kobo Forma into one of five orientations (regardless of Kobo's pitch axis):
1. handle on the right (for me, since I like to hold the Kobo with my right hand)
2. handle on the left
3. handle at the bottom (landscape)
4. handle at the top (bats, circus performers, and handstand-walkers would appreciate this orientation)
5. Automatic (based on pitch-axis sensor)

geek1011, 2019-11-11:
The orientation stuff won't be worth the effort to patch, but I can easily make a workaround which inverts the page buttons without having that option selected. Would that be sufficient for your needs?

geek1011, 2019-11-11:
You can use this libnickel patch on 4.18.13737 to invert the page buttons without affecting anything else:

```yaml
Invert page turn buttons:
  - Enabled: yes
  # In ReadingView::keyPressEvent(QKeyEvent*), invert the branch to next page:
  - ReplaceBytes: {Offset: 0xAD2824, FindH: 28 B9, ReplaceH: 28 B1} # CBNZ R0, #0xE -> CBZ R0, #0xE
```

If this is not sufficient for working around your issue, you'll need to wait until Kobo fixes it or I come across a simple way to fix it.

hub2git, 2019-11-11:
Thanks Patrick, I'll try your patch. Which file do I add your code into?
And should I disable any other patch?

hub2git, 2019-11-11:
@geek1011, I've just installed/applied your patch. Happy to announce that it removes the temporary upside-down hiccup. Thanks for your help!

Note to self and others with the same predicament: Before applying the patch, I put the buttons back from "Inverted" to "Default".

poczynek, 2019-11-13:
@geek1011 Will this solve my current issue with orientation?  So every time the Forma even slightly flips while holding it, it flips the screen upside down!  Very frustrating.

geek1011, 2019-11-14:
No, only the issue mentioned in the first post.

hub2git, 2019-11-14:
@geek1011, it seems that your patch still allows the glitch through on the
rare occasion, like once every 15 times. I'm grateful that it's not all the
time, as it was before your patch. Just thought I should let you know.

Also, I noticed that when the PIN pad is _not_ upside down to begin with,
it seems that the right-side up PIN pad's text ("Please enter your current
4-idigt PIN.") shifts a few pixels to the left. This isn't a big deal in
itself, but during this " shift-to-the-left phasing", I can't enter my
4-digit PIN. I don't recall if this "phasing" is due to your patch, but
thought I'd let you know.

Curiously, when I hold my Forma with my left hand and open the SleepCover,
I noticed the phasing is upward. (I don't tend to use my left hand to hold
the Forma, so this doesn't concern me.)

Also, I noted that I must keep the "Button Controls" under the Reading
Settings set to "Inverted" in order to retain my preferred inverted
controls when I'm in "My Books", "Authors", "My Collections", "My
Articles". Just thought I should let you know about this. (I had thought
reverting the Button Controls to "Default", in conjunction with your patch,
would fix this thread's issue.)

poczynek, 2019-11-14:
> No, only the issue mentioned in the first post.

No problem - thanks anyway.

hub2git, 2019-11-17:
I should add that if I [enabled the patch](https://github.com/geek1011/kobopatch-patches/issues/33#issuecomment-552296649) and set the Page Control buttons to Inverted, then -- something like when you multiply a negative number by another negative number -- the Page Control buttons become "Default" (regular). 
What this means is that I can't have inverted controls everywhere with the patch enabled. With the patch enabled, I can have inverted controls either:
--only while reading a book
or
--when I'm My Books/Collections/Authors/etc.

That's too bad.

geek1011, 2019-12-28:
I'm not planning on implementing this as a patch because there are too many places which would need to be patched, and there isn't enough interest to justify the time required.

Something like [setkeycodes](https://git.busybox.net/busybox/tree/console-tools/setkeycodes.c?id=2e068e725ce693afe5e89e10feaa9f24bef5f789) running at boot will probably work, though.

closed 2019-12-28 by geek1011