pgaskin/kobopatch-patches #35: [Request] Change system font

babygau, 2019-11-08:
Hi,

I'd like to have the system font changed. It would be great that we could
use side-loaded fonts for system font.

Many thanks.

jackiew1, 2019-11-08:
To some extent you can. You haven't told us what your native language is but there should be some useful information for you in these 3 recent threads on MobileRead.

kobopatch users & **non-Kobo-supported languages** https://www.mobileread.com/forums/showthread.php?t=323350

New patch - **"Custom page navigation scrubber"** https://www.mobileread.com/forums/showpost.php?p=3897174&postcount=3

New patch - **"Custom new header/footer font"**  https://www.mobileread.com/forums/showpost.php?p=3897175&postcount=4

babygau, 2019-11-08:
@jackiew1 Tks for your quick response. FYI, I'm Vietnamese. I already successfully applied my font to the header, footer and navigation scrubber thanks to your new patch. I will try the solution in the first post first thing tmr as it's late already. I'll let you know asap. Many thanks!

babygau, 2019-11-08:
@geek1011, I could confirm that the patch work like a charm, put a Vietnamese
font to `fonts` directory and put its name into your patch, reboot, the open a
`kepub` book and all Vietnamese characters displayed beatifully.

closed 2019-11-08 by babygau