pgaskin/kobopatch-patches #36: Earlier start time for Comfortlight

jackiew1, 2019-11-09:
@geek1011  For your consideration once you feel like looking at patches again.

There was a request in the MR 'Kobo future firmware feature request thread' https://www.mobileread.com/forums/showpost.php?p=3913806&postcount=1710 regarding having the ability to set Comfortlight start time earlier than 21:00.

Naturally I'm bringing it to your attention because it's one I'd be interested in myself.😁 It can be dark by 15:00 in the UK in winter. Absolutely no urgency.

geek1011, 2019-11-09:
I'll take a look at it later today, and do the patch if it is straightforward (i.e. under 10 minutes to do). I'd also be interested in this myself once I get a Clara (I'm waiting for it to go on sale).

geek1011, 2019-11-10:
@jackiew1, do the [hours](https://doc.qt.io/qt-5/qtime.html#QTime-2) 7, 18, and 5 mean anything to you (possibly shifted, so also consider the intervals in between)? If so, this will be a really simple patch, if not, I'll have to look further later.

```asm
00a499de 3046                   mov        r0, r6 ; CODE XREF=_ZN14BedtimeManager17adjustTemperatureEv+72
00a499e0 7AF6E8E2               blx        _ZN14BedtimeManager18daytimeTemperatureEv@PLT ; BedtimeManager::daytimeTemperature()
00a499e4 3861                   str        r0, [r7, #0x10]
00a499e6 3046                   mov        r0, r6
00a499e8 64F68CE0               blx        _ZN14BedtimeManager18bedtimeTemperatureEv@PLT ; BedtimeManager::bedtimeTemperature()
00a499ec 4346                   mov        r3, r8
00a499ee 7860                   str        r0, [r7, #0x4]
00a499f0 4246                   mov        r2, r8
00a499f2 0721                   movs       r1, #7
00a499f4 2046                   mov        r0, r4
00a499f6 CDF80080               str.w      r8, [sp, #0x70 + var_70]
00a499fa 61F68AE3               blx        _ZN5QTimeC1Eiiii@PLT ; QTime::QTime(int, int, int, int)
00a499fe D7F82C90               ldr.w      sb, [r7, #0x2c]
00a49a02 4346                   mov        r3, r8
00a49a04 4246                   mov        r2, r8
00a49a06 1221                   movs       r1, #18
00a49a08 CDF80080               str.w      r8, [sp, #0x70 + var_70]
00a49a0c 2046                   mov        r0, r4
00a49a0e B9F1FF3F               cmp.w      sb, #0xffffffff
00a49a12 08BF                   it         eq
00a49a14 4FF00009               mov.weq    sb, #0x0
00a49a18 61F67AE3               blx        _ZN5QTimeC1Eiiii@PLT ; QTime::QTime(int, int, int, int)
00a49a1c D7F82CB0               ldr.w      fp, [r7, #0x2c]
00a49a20 4346                   mov        r3, r8
00a49a22 CDF80080               str.w      r8, [sp, #0x70 + var_70]
00a49a26 4246                   mov        r2, r8
00a49a28 0521                   movs       r1, #5
00a49a2a 2046                   mov        r0, r4
00a49a2c BBF1FF3F               cmp.w      fp, #0xffffffff
00a49a30 08BF                   it         eq
00a49a32 4FF0000B               mov.weq    fp, #0x0
00a49a36 61F66CE3               blx        _ZN5QTimeC1Eiiii@PLT ; QTime::QTime(int, int, int, int)
00a49a3a FB6A                   ldr        r3, [r7, #0x2c]
00a49a3c 7B61                   str        r3, [r7, #0x14]
00a49a3e 0133                   adds       r3, #0x1
00a49a40 00F05881               beq.w      loc_a49cf4
```

jackiew1, 2019-11-10:
That's a bit cryptic for me 😄 Here's the only actual info I can see, none of which answers your question.

The config setting for "Bedtime" is a drop-down menu with 13 entries:
21:00 to 03:00 inclusive, at 30min intervals. That's a range of 6 hours.

I have my Clara and Forma set at the minimum of 21:00 which is stored in the Kobo .conf file's [PowerOptions] section with a value 
BedTime=@Variant(\0\0\0\xf\x4\x81\x90\x80)

geek1011, 2019-11-10:
OK, thanks, that helps.

BTW, I'm going to release v54 in the next few minutes. All the missing patches are now updated.

jackiew1, 2019-11-10:
... I typed that @Variant with a capital 'V' but it's displaying as small 'v'

geek1011, 2019-11-10:
That's because GitHub's interpreting that as a mention to the user `variant`.

jackiew1, 2019-11-10:
Well then, lets hope user variant can answer your question if he reads it 😄

jackiew1, 2019-11-10:
A bit more info I missed ... 
There is also another .conf setting in [PowerOptions] for the colour temperature slider value.
ColorSetting=6400 (for full blue)
ColorSetting=1500 (for full orange)

geek1011, 2019-11-10:
I found the relevant code in a function which had it's name optimized out by going back from references to QTime::addSecs (since the values aren't hard-coded, I guessed they would be generated this way). I just have to figure out where to patch it.

geek1011, 2019-11-10:
@jackiew1, can you tell me the earliest and latest time in the list?

geek1011, 2019-11-10:
@jackiew1, can you try this:

```yaml
Change ComfortLight bedtime range:
  - Enabled: no
  - Description: Change the times available in the ComfortLight bedtime dropdown.
    # In an unnamed subroutine two layers into FrontLightPopupController::loadView
    # (find it by going back from QTime::addSecs), the times for the dropdown are
    # generated into a QVector<QPair<QString, QTime>> with a simple loop.
    #
    # The abbreviated structure of the loop:
    #
    #    r0 = new QTime(r1 = 21, 0, 0)
    #    QVector<QPair<QString, QTime>>::reallocData(int, r2 = 13, QFlags<QArrayData::AllocationOption>)   ; https://code.woboq.org/qt5/qtbase/src/corelib/tools/qvector.h.html#_ZN7QVector11reallocDataEii6QFlagsIN10QArrayData16AllocationOptionEE
    #    r5 = 13
    #    do {
    #      QLocale::toString(QTime const&, QLocale::FormatType)
    #      QVector<QPair<QString, QTime>>::append(QPair<QString, QTime> const&)
    #      QTime::addSecs(r1 = 1800)
    #      r5--
    #    } while (r5 > 0)
    #
  # Change the starting hour passed to the QTime constructor (mov r1, #21):
  - ReplaceInt: {Offset: 0xA50E3E, Find: 21, Replace: 17}      # 21:00 (9PM) -> 17:00 (5PM)
  # Optional: Change the second increment passed to QTime::addSecs:
  - ReplaceInt: {Offset: 0xA50EA2, Find: 1800, Replace: 1800}  # 30m
  # Change the number of increments (movs r5, #13):
  - ReplaceInt: {Offset: 0xA50E76, Find: 13, Replace: 21}      # 13 [9PM..3AM]+30m -> 21 [5PM..3AM]+30m
  # Change the number of elements pre-allocated (I don't think this is actually
  # required, as Qt will realloc as necessary, but we might as well do this just
  # in case):
  - ReplaceInt: {Offset: 0xA510DE, Find: 13, Replace: 21}      # same as prev replacement
```

And, I suggest you make a backup first, as this may need a reset if I did it wrong and it segfaults (although if I messed this up, it'd most likely only trigger when pressing the frontlight menu). I think I did this correctly, though.

jackiew1, 2019-11-10:
Testing now. Sorry for delay, it's been a busy morning.

jackiew1, 2019-11-10:
I installed the patch as-is on a KA1. It gave me the expected range of Bedtimes 17:00-03:00, 21 entries at 30min intervals. It's only 14:50 here so I need to customise the patch to force an earlier colour-changing.

I re-applied the patch with a customised minimum Bedtime of 15:00 and commented out all other find/replace. After installation, as expected, the drop-down has 13 entries at 30min intervals 15:00-21:00. I've selected 15:00 from the drop-down and made sure the Natural light Auto switch is on. 

I would expect the colour to start changing immediately so it would gradually get to "full orange" by 15:00 . So far it's remained on "full blue". 😕  

When you were digging around did you notice the algorithm for how the colour-change is meant to happen?

geek1011, 2019-11-10:
I did take note of where the color change was handled (BedTimeManager::adjustTemperature), but I didn't bother to try to understand it. I was just focusing on the menu and config file yesterday. Does the color change happen if you set the time even further back?

One suspicion I have is that those times I saw earlier also need to be adjusted (my patch doesn't touch anything to do with a wake-up time). Do you know when the auto color change ends (i.e. goes back to normal)?

jackiew1, 2019-11-10:
> Does the color change happen if you set the time even further back?

I haven't changed the patch on the KA1 but it's 15:12 now and the KA1 is still full blue. So passing the Bedtime value didn't kickstart anything.

> One suspicion I have is that those times I saw earlier also need to be adjusted (my patch doesn't touch anything to do with a wake-up time). Do you know when the auto color change ends (i.e. goes back to normal)?

I don't know for sure when orange starts to move back to blue. When I've finished reading for the night (early morning) the device will be full orange. If I open the book next morning it will be full blue, although I have a feeling that the immediate colour will be wherever it was when I closed the cover and re-opening it immediately starts to auto-decrease the orange. It's gradual but fast-gradual.

jackiew1, 2019-11-10:
Follow-up on the KA1 ...

- I disabled the auto-Natural light button at 15:30
- manually set the orange slider to max
- re-enabled the auto-Natural light button. The orange slider immediately starts to inch its way back from max to min.

So there's definitely more to it than we have so far.

jackiew1, 2019-11-10:
Follow-up 2 ...

I've now installed the patch on my Clara. Bedtime 21 entries 15:00-01:00 set at 16:00. I plan to do some normal reading now and note at what time the orange starts to kick in (if it ever does).

I've left this page open on screen so if you have more ideas I should see them quite quickly.

jackiew1, 2019-11-10:
Both KA1 and Clara started to move at 18:00. Both auto-stepped from min to max in under a minute. I have to admit that I've never paid attention to how long it normally takes to move from min to max.

My Forma remains unpatched and has Bedtime set at the default min of 21:00. It's at full blue at the moment, so the patch has obviously made some kind of difference. 

I'll now continue reading on the unpatched Forma and pay more attention to when the colour-change kicks in and how fast it moves.

geek1011, 2019-11-10:
Can you enable all logging categories in developer options?

jackiew1, 2019-11-10:
I've turned on dev mode and ticked every box in Logging Category.

jackiew1, 2019-11-10:
I've just checked the colour-slider. It's started to move. It's about 1cm from min. I wouldn't have noticed the change if I hadn't gone looking.

geek1011, 2019-11-10:
These are the calculations "before day time start":

```asm
00a49d1e 7B69                   ldr        r3, [r7, #0x14] ; CODE XREF=dword_a4a6c4+1492, dword_a4a6c4+1538, dword_a4a6c4+2376, sub_a4b086+112
00a49d20 7A68                   ldr        r2, [r7, #0x4]
00a49d22 3969                   ldr        r1, [r7, #0x10]
00a49d24 ED1A                   subs       r5, r5, r3
00a49d26 C3EB0909               rsb        sb, r3, sb
00a49d2a 07EE902A               vmov       s15, r2
00a49d2e 8B1A                   subs       r3, r1, r2
00a49d30 F8EEE71B               vcvt.f64.s32 d17, s15
00a49d34 07EE905A               vmov       s15, r5
00a49d38 F8EEE70B               vcvt.f64.s32 d16, s15
00a49d3c 07EE909A               vmov       s15, sb
00a49d40 F8EEE73B               vcvt.f64.s32 d19, s15
00a49d44 07EE903A               vmov       s15, r3
00a49d48 F8EEE72B               vcvt.f64.s32 d18, s15
00a49d4c C0EEA30B               vdiv.f64   d16, d16, d19
00a49d50 42EEA01B               vmla.f64   d17, d18, d16
00a49d54 FDEEE17B               vcvt.s32.f64 s15, d17
00a49d58 C7ED047A               vstr       s15, [r7, #0x10]
00a49d5c C9E6                   b          loc_a49af2
```

These are "after sunset but not bed time yet":

```asm
00a4a54e 3A69                   ldr        r2, [r7, #0x10] ; CODE XREF=dword_a4a6c4+1894, dword_a4a6c4+1944, dword_a4a6c4+2406
00a4a550 CBEB0505               rsb        r5, fp, r5
00a4a554 CBEB0808               rsb        r8, fp, r8
00a4a558 7968                   ldr        r1, [r7, #0x4]
00a4a55a 07EE902A               vmov       s15, r2
00a4a55e 531A                   subs       r3, r2, r1
00a4a560 F8EEE72B               vcvt.f64.s32 d18, s15
00a4a564 07EE905A               vmov       s15, r5
00a4a568 F8EEE71B               vcvt.f64.s32 d17, s15
00a4a56c 07EE908A               vmov       s15, r8
00a4a570 F8EEE70B               vcvt.f64.s32 d16, s15
00a4a574 07EE903A               vmov       s15, r3
00a4a578 F8EEE73B               vcvt.f64.s32 d19, s15
00a4a57c C1EEA00B               vdiv.f64   d16, d17, d16
00a4a580 43EEE02B               vmls.f64   d18, d19, d16
00a4a584 FDEEE27B               vcvt.s32.f64 s15, d18
00a4a588 C7ED047A               vstr       s15, [r7, #0x10]
00a4a58c FFF7B1BA               b.w        loc_a49af2
```

And this calculation is used for something else:

```asm
00a49b5e 801A                   subs       r0, r0, r2
00a49b60 9FEDB86A               vldr       s12, =0x458ca000 ; 0xa49e44
00a49b64 0028                   cmp        r0, #0x0
00a49b66 DFEDB87A               vldr       s15, =0x4270 ; 0xa49e4a
00a49b6a B8BF                   it         lt
00a49b6c 4042                   rsbslt     r0, r0, #0x0
00a49b6e DFEDB76A               vldr       s13, =0x9b8e447a ; 0xa49e4e
00a49b72 1146                   mov        r1, r2
00a49b74 0122                   movs       r2, #0x1
00a49b76 07EE100A               vmov       s14, r0
00a49b7a 2846                   mov        r0, r5
00a49b7c B8EEC77A               vcvt.f32.s32 s14, s14
00a49b80 87EE067A               vdiv.f32   s14, s14, s12
00a49b84 67EE277A               vmul.f32   s15, s14, s15
00a49b88 67EEA67A               vmul.f32   s15, s15, s13
00a49b8c FDEEE77A               vcvt.s32.f32 s15, s15
00a49b90 17EE903A               vmov       r3, s15
00a49b94 61F6F8E4               blx        _ZN10FrontLight14setTemperatureEibi@PLT ; FrontLight::setTemperature(int, bool, int)
00a49b98 BAF1000F               cmp.w      sl, #0x0
00a49b9c BED1                   bne        loc_a49b1c
```

(these are mainly for my own reference, I'll need to work out the math when I get around to it)

jackiew1, 2019-11-10:
I hope all that makes more sense to you than it does to me.

I won't be particularly surprised if I see a slider jump every 30mins rather than a constant gradual creep.

geek1011, 2019-11-10:
> I hope all that makes more sense to you than it does to me.

It does. It's easier than it seems at first, as quite a few of those instructions are just variants of a few simple ones (blx to call functions, v... for floating point numbers, etc, ...lt for executing only if less-than, {mul,div} for math operations, ldr to load from a memory address, etc). For example, the last part (after the vldr) of the last one basically says `FrontLight::setTemperature(r2, true, s14/s12*s15*s13)`. The main thing I have to figure out is what's currently stored in s14, s12, and s15, for which I need to follow the address in vldr back to their source.

I won't be doing much with this today, though. Once you have the logs, can you post them or PM them to me (specifically the lines about adjustTemperature)?

jackiew1, 2019-11-10:
How long do you want me to let it run before connecting to access the logs? Is it one big log or lots of little ones?

geek1011, 2019-11-10:
It saves whenever you connect USB. And it'd be nice to have the logs at the beginning and end of the change (remember to keep track of what bedtime you used).

jackiew1, 2019-11-10:
It just jumped again at 19:00. My best guess at the moment is that there will be 6 jumps in all (1st at 18:30), one every 30mins arriving at max at 21:00 which is the Bedtime I have set (and never changed).

jackiew1, 2019-11-10:
Is it just the syslog-xxx file you want? Or the dmesg-xxx too?

geek1011, 2019-11-10:
Just the syslog. Thanks!

jackiew1, 2019-11-10:
If there's such a thing as a spoiler tag perhaps you can let me know how to do it for next time 😄 

```
Nov 10 17:59:59 nickel: (   364.730 @ 0x19af8d0 / ui.warning) static bool EntitlementManager::subscriptionExpired(const Entitlement&, const UsageData&) Check isSubscription first "" 
Nov 10 18:00:07 nickel: (   370.696 @ 0x19af8d0) QObject::connect: No such slot KepubBookReader::calculatePageForBookmark(const Bookmark&, int*)
Nov 10 18:00:07 nickel: (   370.696 @ 0x19af8d0) QObject::connect:  (sender name:   'ReadingView')
Nov 10 18:00:07 nickel: (   370.741 @ 0x19ea010 / packetdump.warning) dropping stale GA  QUrl( "https://ssl.google-analytics.com/collect?v=1&tid=UA-6177406-38&cid=3ed87734-ce1c-4613-8255-7933733eef93&uid=3b68b458-02e4-4e18-8b8f-e51e493f0952&av=4.18.13737&an=nickel&sr=1440x1920&ul=en-gb&t=event&ec=ReadingExperience&ea=OpenReadingSettingsMenu&el=Scrubber" )  false 249 
Nov 10 18:00:07 nickel: (   371.414 @ 0x19ea010 / packetdump.warning) dropping stale GA  QUrl( "https://ssl.google-analytics.com/collect?v=1&tid=UA-6177406-38&cid=3ed87734-ce1c-4613-8255-7933733eef93&uid=3b68b458-02e4-4e18-8b8f-e51e493f0952&av=4.18.13737&an=nickel&sr=1440x1920&ul=en-gb&t=event&ec=ReadingExperience&ea=OpenReadingSettingsMenu" )  false 249 
Nov 10 18:00:09 nickel: (   373.566 @ 0x19ea010 / packetdump.warning) dropping stale GA  QUrl( "https://ssl.google-analytics.com/collect?v=1&tid=UA-6177406-38&cid=3ed87734-ce1c-4613-8255-7933733eef93&uid=3b68b458-02e4-4e18-8b8f-e51e493f0952&av=4.18.13737&an=nickel&sr=1440x1920&ul=en-gb&t=event&ec=ReadingExperience&ea=LeaveContent&el=application/x-kobo-epub+zip&ev=29" )  false 249 
Nov 10 18:00:10 nickel: (   373.692 @ 0x19ea010 / packetdump.warning) dropping stale GA  QUrl( "https://ssl.google-analytics.com/collect?v=1&tid=UA-6177406-38&cid=3ed87734-ce1c-4613-8255-7933733eef93&uid=3b68b458-02e4-4e18-8b8f-e51e493f0952&av=4.18.13737&an=nickel&sr=1440x1920&ul=en-gb&t=event&ec=Home&ea=HomeWidgetClicked&el=CurrentRead" )  false 249 
Nov 10 18:09:31 nickel: (   395.936 @ 0x19ea010 / packetdump.warning) dropping stale GA  QUrl( "https://ssl.google-analytics.com/collect?v=1&tid=UA-6177406-38&cid=3ed87734-ce1c-4613-8255-7933733eef93&uid=3b68b458-02e4-4e18-8b8f-e51e493f0952&av=4.18.13737&an=nickel&sr=1440x1920&ul=en-gb&cd60=Home&t=event&ec=ReadingExperience&ea=OpenContent&el=application/x-kobo-epub+zip" )  false 249 
Nov 10 18:09:31 nickel: (   396.116 @ 0x19ea010 / packetdump.warning) dropping stale GA  QUrl( "https://ssl.google-analytics.com/collect?v=1&tid=UA-6177406-38&cid=3ed87734-ce1c-4613-8255-7933733eef93&uid=3b68b458-02e4-4e18-8b8f-e51e493f0952&av=4.18.13737&an=nickel&sr=1440x1920&ul=en-gb&t=event&ec=Battery&ea=PluggedIn&el=TimeSinceLastPlugIn&ev=142502" )  false 249 
Nov 10 18:09:33 nickel: (   397.721 @ 0x19ea010 / packetdump.warning) dropping stale GA  QUrl( "https://ssl.google-analytics.com/collect?v=1&tid=UA-6177406-38&cid=3ed87734-ce1c-4613-8255-7933733eef93&uid=3b68b458-02e4-4e18-8b8f-e51e493f0952&av=4.18.13737&an=nickel&sr=1440x1920&ul=en-gb&t=event&ec=Battery&ea=PluggedIn&el=BatteryLevel&ev=74" )  false 249 
Nov 10 18:09:35 nickel: (   399.726 @ 0x19ea010 / packetdump.warning) dropping stale GA  QUrl( "https://ssl.google-analytics.com/collect?v=1&tid=UA-6177406-38&cid=3ed87734-ce1c-4613-8255-7933733eef93&uid=3b68b458-02e4-4e18-8b8f-e51e493f0952&av=4.18.13737&an=nickel&sr=1440x1920&ul=en-gb&t=event&ec=ReadingExperience&ea=LeaveContent&el=application/x-kobo-epub+zip&ev=25" )  false 249 
Nov 10 18:09:37 nickel: (   401.425 @ 0x19ea010 / packetdump.warning) dropping stale GA  QUrl( "https://ssl.google-analytics.com/collect?v=1&tid=UA-6177406-38&cid=3ed87734-ce1c-4613-8255-7933733eef93&uid=3b68b458-02e4-4e18-8b8f-e51e493f0952&av=4.18.13737&an=nickel&sr=1440x1920&ul=en-gb&sc=start&t=event&ec=Home&ea=HomeWidgetClicked&el=CurrentRead" )  false 249 
Nov 10 18:09:38 nickel: (   403.345 @ 0x19ea010 / packetdump.warning) dropping stale GA  QUrl( "https://ssl.google-analytics.com/collect?v=1&tid=UA-6177406-38&cid=3ed87734-ce1c-4613-8255-7933733eef93&uid=3b68b458-02e4-4e18-8b8f-e51e493f0952&av=4.18.13737&an=nickel&sr=1440x1920&ul=en-gb&cd60=Home&t=event&ec=ReadingExperience&ea=OpenContent&el=application/x-kobo-epub+zip" )  false 249 
Nov 10 18:09:43 nickel: (   407.686 @ 0x19ea010 / packetdump.warning) dropping stale GA  QUrl( "https://ssl.google-analytics.com/collect?v=1&tid=UA-6177406-38&cid=3ed87734-ce1c-4613-8255-7933733eef93&uid=3b68b458-02e4-4e18-8b8f-e51e493f0952&av=4.18.13737&an=nickel&sr=1440x1920&ul=en-gb&t=event&ec=ReadingExperience&ea=OpenReadingSettingsMenu&el=Scrubber" )  false 249 
Nov 10 18:09:43 nickel: (   407.808 @ 0x19ea010 / packetdump.warning) dropping stale GA  QUrl( "https://ssl.google-analytics.com/collect?v=1&tid=UA-6177406-38&cid=3ed87734-ce1c-4613-8255-7933733eef93&uid=3b68b458-02e4-4e18-8b8f-e51e493f0952&av=4.18.13737&an=nickel&sr=1440x1920&ul=en-gb&t=event&ec=ReadingExperience&ea=OpenReadingSettingsMenu" )  false 249 
Nov 10 18:09:48 nickel: (   411.953 @ 0x19ea010 / packetdump.warning) dropping stale GA  QUrl( "https://ssl.google-analytics.com/collect?v=1&tid=UA-6177406-38&cid=3ed87734-ce1c-4613-8255-7933733eef93&uid=3b68b458-02e4-4e18-8b8f-e51e493f0952&av=4.18.13737&an=nickel&sr=1440x1920&ul=en-gb&t=event&ec=ReadingExperience&ea=LeaveContent&el=application/x-kobo-epub+zip&ev=54" )  false 249 
[... cut ...] 9d583023f43ae5401', 'file:///mnt/onboard/Books/Barker, JD - The Sixth Wicked Child.kepub.epub', 67, 1021, '', NULL, '2019-11-10T19:36:57.000')" 
Nov 10 19:36:58 nickel: (   957.501 @ 0x19af8d0 / leak.debug) virtual Dialog::~Dialog() Dialog(0x20fb928, name = "ConfirmationDialog") 
Nov 10 19:36:58 nickel: (   957.528 @ 0x19af8d0 / gyro.debug) virtual void KoboOrientationSensor::stop() stop sensor 
Nov 10 19:36:58 nickel: (   957.655 @ 0x19af8d0 / powermanager.debug) lastActivity 1573414618121 
```

jackiew1, 2019-11-10:
Log link sent via MR PM

jackiew1, 2019-11-10:
> @jackiew1, do the [hours](https://doc.qt.io/qt-5/qtime.html#QTime-2) 7, 18, and 5 mean anything to you (possibly shifted, so also consider the intervals in between)?

With the benefit of hindsight I think 7 and 18 are probably hard-coded values for 
sunrise (07:00)
sunset (18:00)

The 5 value is still eluding me.

geek1011, 2019-11-10:
Based on what you said and skimming the logs quickly, here's my guess of how it works:

- The fastest automatic transition is 1 minute.
- The transition starts at sunset (18:00), but not earlier.
- The transition is until bedtime (i.e. bedtime is when the temperature is at it's warmest).
- This one is completely a guess, but maybe the 5 means that the transition back to blue is between 5 and 7am.

geek1011, 2019-11-10:
If you add `- ReplaceInt: {Offset: 0xa49a06, Find: 18, Replace: 23}`, does the light stay blue (assuming it's before 11pm)? If so, that will confirm my theory.

jackiew1, 2019-11-10:
It's already 11:07pm here.

geek1011, 2019-11-10:
What happens if you change the time?

jackiew1, 2019-11-10:
I set the KA1 time back by an hour and changed sunset time from 18 to 23 as per your new patch addition. Bedtime is still set at 15:00 on the KA1.

After the applying the new patch the colour is indeed back at full blue. The prediction is now that at 23:00 the light will auto-adjust itself back to full orange in less than a minute. 

I've now changed the KA1 time again to 22:55 so we'll see what happens in 5 mins.

jackiew1, 2019-11-10:
Well, the prediction was wrong. Nothing happened as the clock passed 23:00.

I'm now changing Bedtime to 23:00 ...

jackiew1, 2019-11-10:
... and the colour has immediately started to crawl it's way from full blue to full orange. I counted 40+ jumps.

As an aside, FYI, when the colour is changed automatically the orange slider never actually reaches 100%. Full blue is 6400 and full orange is 1500 if you slide manually but max orange only gets to 1900 on auto.

geek1011, 2019-11-10:
So, it seems the majority of my guesses are correct. Do you think it would be sufficient to also set the sunset time earlier, or should I see if it is possible to make it transition over a fixed period before bedtime (this may be possible if I can find enough room to put it in)?

One last thing you could try is to change the int at 0x00a49a28 from 5 to 0 (and 0x00a499f2 from 7 to 1) and see if it starts getting cooler again (sunrise)

jackiew1, 2019-11-11:
I applied the 2 new find/replaces with the device time set correctly. I couldn't immediately conclude anything. So I've advanced the time from 00:00 to 00:30, i.e. 30mins until sunrise ... and yes, the colour has moved down from full (almost) orange to about 50% orange. If your theory is true it should reach zero-orange within another 30 mins. I'll let it do this naturally with no more artificial time changes and post again in 30 mins.

jackiew1, 2019-11-11:
... it's now 01:00 on this Kobo and colour is at full blue, i.e zero-orange.

> Do you think it would be sufficient to also set the sunset time earlier, or should I see if it is possible to make it transition over a fixed period before bedtime (this may be possible if I can find enough room to put it in)?

Hmmm. Need to think about this a bit. These are initial personal thoughts, not what I think every other user will necessarily want.

I don't really care very much about what the transition duration from blue to orange is. And I care zero about transition duration from orange to blue because I'm never reading at sunrise. And even if I was I wouldn't care about colour. My personal wish for this patch is for the light to be orange when it actually dark rather than at my bedtime (which is very variable). As such, Kobo's idea of sunset being at 18:00 all year round doesn't work for me. In the UK darkness falls (roughly) anywhere between 15:00 in winter to 21:00 in summer.

I'm very happy to change the on-device Bedtime setting, as required, as the seasons change. Based on today's experiments that suggests we need the patch to
- have an earlier start time than 21:00, but I don't personally need 21 different values at 30min intervals.
- be able to change sunset to something earlier than 18:00 and <= minimum "Bedtime" (or for me minimum dark time)

geek1011, 2019-11-11:
How does this look for the final patch:

```yaml
Change ComfortLight bedtime range:
  - Enabled: no
  - Description: |
      Change the times available in the ComfortLight bedtime dropdown and the
      start/end times for the color change. All values in this patch are
      customizable.

      The default values for this patch give you 21 options between 5PM and 3AM
      at 30 minute intervals, with the colour change starting at 4PM, and the
      change back to blue between 5AM and 7AM. The firmware default is 13 options
      between 9PM-3AM at 30 minute intervals, with the transition starting at 6PM.
    # In an unnamed subroutine two layers into FrontLightPopupController::loadView
    # (find it by going back from QTime::addSecs), the times for the dropdown are
    # generated into a QVector<QPair<QString, QTime>> with a simple loop.
    #
    # The abbreviated structure of the loop:
    #
    #    r0 = new QTime(r1 = 21, 0, 0)
    #    QVector<QPair<QString, QTime>>::reallocData(int, r2 = 13, QFlags<QArrayData::AllocationOption>)   ; https://code.woboq.org/qt5/qtbase/src/corelib/tools/qvector.h.html#_ZN7QVector11reallocDataEii6QFlagsIN10QArrayData16AllocationOptionEE
    #    r5 = 13
    #    do {
    #      QLocale::toString(QTime const&, QLocale::FormatType)
    #      QVector<QPair<QString, QTime>>::append(QPair<QString, QTime> const&)
    #      QTime::addSecs(r1 = 1800)
    #      r5--
    #    } while (r5 > 0)
    #
  # Change the starting hour passed to the QTime constructor (mov r1, #21):
  - ReplaceInt: {Offset: 0xA50E3E, Find: 21, Replace: 17} # 21:00 (9PM) -> 17:00 (5PM)
  # Optional: Change the increment passed to QTime::addSecs (you need to use an
  # ARM assembler to change this value):
  - ReplaceBytes: {Offset: 0xA50EA2, FindH: 4F F4 E1 61, ReplaceH: 4F F4 E1 61} # mov.w r1, #1800: 30m
  # Change the number of increments (movs r5, #13):
  - ReplaceInt: {Offset: 0xA50E76, Find: 13, Replace: 21} # 13 [9PM..3AM]+30m -> 21 [5PM..3AM]+30m
  # Change the number of pre-allocated elements (I don't think this is actually
  # required, as Qt will realloc as necessary, but we might as well do this just
  # in case):
  - ReplaceInt: {Offset: 0xA510DE, Find: 13, Replace: 21} # same as prev replacement
    #
    # The actual color adjustment is done in BedtimeManager::adjustTemperature.
    # The color temperature is based on the current time, the bedtime, the hardcoded
    # sunset, the hardcoded sunrise start, the hardcoded sunrise end, the daytime
    # temperature, and the nighttime temperature.
    #
  # Change the sunset time (the transition to a warmer temperature starts here
  # and transitions until the bedtime chosen) (this should be 1-3 hours before
  # the minimum bedtime from the first replacement above):
  - ReplaceInt: {Offset: 0xA49A06, Find: 18, Replace: 16} # 18:00 (9PM-3=6PM) -> 16:00 (5PM-1=4PM)
  # Optional: Change the sunrise start time (the transition back to a cooler
  # temperature starts here) (this should be sometime after the last bedtime,
  # but before the sunset time):
  - ReplaceInt: {Offset: 0xA49A28, Find: 5, Replace: 5} # 5AM
  # Optional: Change the sunrise end time (the transition back to a cooler
  # temperature ends here) (this should be 1-3 hours after the sunrise start but
  # before the sunset time or bad things will happen with negative values):
  - ReplaceInt: {Offset: 0xA499F2, Find: 7, Replace: 7} # 5AM+2=7AM
```

jackiew1, 2019-11-11:
It can do a job but If you want my honest opinion it's too dense with too much technical detail that detracts from what the average user needs to know to customise it. 

I do understand that a lot of it is there to document what you've done and acts as a permanent reminder of the details of the processes involved. But do the users need to know all that?

In addition:
- Do we really need to expand the drop-down menu from 13 entries to 21 entries?
- Do we really need to do anything at all with Sunrise time and transition from orange back to blue? Has anyone ever mentioned early morning changes?

Also, the patch name needs to be more generic e.g. 
**Customize Comfortlight settings**
as it's now more than just changing the bedtime range.

As your last post arrived I was just about to write to say that for my purposes the patch could be reduced to 2 find/replaces:

```
Customize Comfortlight settings:
  - Enabled: no
  - Description: |
      Change the times available in the ComfortLight bedtime drop-down and the
      start/end times for the colour change from cool to warm.
  # Change the starting hour for the Bedtime menu:
  - ReplaceInt: {Offset: 0xA50E3E, Find: 21, Replace: 15}   # 21:00 (9PM) -> 15:00 (3PM)
  # change Sunset time from 18:00 to 15:00
  - ReplaceInt: {Offset: 0xA49A06, Find: 18, Replace: 15}
```

geek1011, 2019-11-11:
> It can do a job but If you want my honest opinion it's too dense with too much technical detail that detracts from what the average user needs to know to customise it.

I agree with that completely.

> I do understand that a lot of it is there to document what you've done and acts as a permanent reminder of the details of the processes involved. But do the users need to know all that?

Yes, that's why I put it there, but edited out some of the denser parts. I've settled on that style for my patches, especially the complex ones like this one, as it provides enough information for me to quickly recreate it if I need to (unlike things like the 24 line spacing values patch), and it also provides enough information for somebody without much experience to learn how to update it in case I ever stop updating the patches for whatever reason. Before each section of replacements, I describe which function it's in (for context), how to find that function (in case it loses it's function name due to optimization), and what the defining characteristic of it is (in case the code ever gets moved around, like with the new reader stuff). Before each replacement, I describe what it is replacing, the possible side effects, and the user-facing description and constraints. With each replacement, I show the previous and new values along with the assembly instruction (so someone interested doesn't have to decompile each instruction manually to read it -- this was one of the most time consuming parts of learning to update the patches a few years ago). So, yes, quite a bit of this info is extra for users, but this info is the minimum I can put to save a lot of time and effort in the future if it needs to be rewritten or if someone else updates it.

> Do we really need to expand the drop-down menu from 13 entries to 21 entries?

My logic was that there isn't any harm in putting them all there, and that a patch shouldn't remove functionality if possible.

> Do we really need to do anything at all with Sunrise time and transition from orange back to blue? Has anyone ever mentioned early morning changes?

I'd be using it myself once I have a Clara, and it's also there for reference about the 5 and 7 times.

> Also, the patch name needs to be more generic e.g.
Customize Comfortlight settings
as it's now more than just changing the bedtime range.

Yep, will fix that. Thanks!

> As your last post arrived I was just about to write to say that for my purposes the patch could be reduced to 2 find/replaces:

Yes, that should work. And, did you try that to make sure the array reallocation isn't needed (I didn't think it was, but if it turned out to be, it would segfault or cause unusual bugs when building the array).

jackiew1, 2019-11-11:
> Yes, that should work. And, did you try that to make sure the array reallocation isn't needed (I didn't think it was, but if it turned out to be, it would segfault or cause unusual bugs when [...]

I did it this way to try to minimise the mods required (for me). Do you think I may be creating problems for myself that I haven't fully understood?

In theory, it gives me 13 selectable Bedtimes from 16:00 to 21:00 and a constant start transition of 15:00 (sunset). I've now applied it to a Clara (Bedtime=16:00) and a Forma (Bedtime=21:00) for a proper test today. The _theory_ is that:
- both should start to transition at 15:00
- the Clara will transition in a couple of big jumps
- the Forma will be a more gradual transition in more but smaller jumps.

Based on my (rather haphazard and accidental) observations so far I think the following may also be true:
- if you're actually reading normally as the sunset time is reached, no transition at all happens for the first 30mins.
- if there is a sleep/wake, PCconnect, ... etc ... during that first 30mins then there will be a colour transition to the correct level for the time the device "wakes up".
- or more generally, transition jumps happen every 30mins or whenever the device re-awakens - whichever is sooner.

Now, back to the full patch ... I'm not sure what changes you've already made but the following confused me so may confuse others:
- the first line comment of the following find/replace needs humanising
```
# Change the starting hour passed to the QTime constructor (mov r1, #21):
  - ReplaceInt: {Offset: 0xA50E3E, Find: 21, Replace: 17} # 21:00 (9PM) -> 17:00 (5PM)
```
How about  **# Change the starting hour for the Bedtime drop-down menu**

- What is the following find/replace actually for? ReplaceH is the same as FindH and it doesn't look like something where a user can choose their own ReplaceH?
```
  # Optional: Change the increment passed to QTime::addSecs (you need to use an
  # ARM assembler to change this value):
  - ReplaceBytes: {Offset: 0xA50EA2, FindH: 4F F4 E1 61, ReplaceH: 4F F4 E1 61} # mov.w r1, #1800: 30m
```

geek1011, 2019-11-11:
I was thinking you missed this part:

```yaml
  # Change the number of pre-allocated elements (I don't think this is actually
  # required, as Qt will realloc as necessary, but we might as well do this just
  # in case) (mov r1, #21):
  - ReplaceInt: {Offset: 0xA510DE, Find: 13, Replace: 21} # same as prev replacement
```

But now I realize you only wanted to change the time, not the number of values. So yes, your patch is fine.

> or more generally, transition jumps happen every 30mins or whenever the device re-awakens - whichever is sooner.
That looks like I might have to patch the timer code too.

> the first line comment of the following find/replace needs humanising

Yep, missed that one (you can see the current version in my kobopatch-config repo.

> What is the following find/replace actually for? ReplaceH is the same as FindH and it doesn't look like something where a user can choose their own ReplaceH?

That's for optionally changing the increment of 1800s, but if what I said about needing to change the timer is true, I'll probably remove this option.

geek1011, 2019-11-11:
I was right about missing the timer. Thanks for catching that!

Here is one timer. That is probably all you need:

```asm
00a49b1c 0025                   movs       r5, #0x0 ; CODE XREF=_ZN14BedtimeManager17adjustTemperatureEv+564
00a49b1e 1221                   movs       r1, #18
00a49b20 2B46                   mov        r3, r5
00a49b22 2A46                   mov        r2, r5
00a49b24 0095                   str        r5, [sp, #0x70 + var_70]
00a49b26 2046                   mov        r0, r4
00a49b28 61F6F2E2               blx        _ZN5QTimeC1Eiiii@PLT ; QTime::QTime(int, int, int, int)
00a49b2c 2A46                   mov        r2, r5
00a49b2e 2146                   mov        r1, r4
00a49b30 06F12800               add.w      r0, r6, #0x28
00a49b34 82F66AE6               blx        _ZN10PowerTimer6fireAtERK5QTimeN2Qt8TimeSpecE@PLT ; PowerTimer::fireAt(QTime const&, Qt::TimeSpec)
00a49b38 4437                   adds       r7, #0x44
00a49b3a BD46                   mov        sp, r7
00a49b3c BDE8F08F               pop.w      {r4, r5, r6, r7, r8, sb, sl, fp, pc}
```

Add the following:

```yaml
- ReplaceInt: {Offset: 0xA49B1E, Find: 18, Replace: 15} # should match sunset time
```

geek1011, 2019-11-11:
And just as a note for myself, here's the other timer (which is for the next step of the change, and is calculated from the other times):

```asm
00a49bc6 6CF676E0               blx        _ZNK5QTime6minuteEv@PLT ; QTime::minute() const
00a49bca 0546                   mov        r5, r0
00a49bcc 4946                   mov        r1, sb
00a49bce 85FB0889               smull      r8, sb, r5, r8
00a49bd2 EA17                   asrs       r2, r5, #0x1f
00a49bd4 5346                   mov        r3, sl
00a49bd6 CDF800A0               str.w      sl, [sp, #0x70 + var_70]
00a49bda 2046                   mov        r0, r4
00a49bdc 4D44                   add        r5, sb
00a49bde C2EB2515               rsb        r5, r2, r5, asr #4
00a49be2 6A01                   lsls       r2, r5, #0x5
00a49be4 A2EB4502               sub.w      r2, r2, r5, lsl #1
00a49be8 61F692E2               blx        _ZN5QTimeC1Eiiii@PLT ; QTime::QTime(int, int, int, int)
00a49bec 2146                   mov        r1, r4
00a49bee 5846                   mov        r0, fp
00a49bf0 4FF620E6               blx        _ZNK5QTime7msecsToERKS_@PLT ; QTime::msecsTo(QTime const&) const
00a49bf4 011E                   subs       r1, r0, #0x0
00a49bf6 06F12800               add.w      r0, r6, #0x28
00a49bfa BEBF                   ittt       lt
00a49bfc 01F1A461               add.wlt    r1, r1, #0x5200000
00a49c00 01F5CB21               add.wlt    r1, r1, #0x65800
00a49c04 01F58061               add.wlt    r1, r1, #0x400
00a49c08 75F69EE3               blx        _ZN10PowerTimer6fireInEi@PLT ; PowerTimer::fireIn(int)
00a49c0c 4437                   adds       r7, #0x44
00a49c0e BD46                   mov        sp, r7
00a49c10 BDE8F08F               pop.w      {r4, r5, r6, r7, r8, sb, sl, fp, pc}
```

jackiew1, 2019-11-11:
> > or more generally, transition jumps happen every 30mins or whenever the device re-awakens - whichever is sooner.
> 
> That looks like I might have to patch the timer code too.

I wasn't actually reporting it as an error, just an observation 😃 but if you think this new find/replace will improve things, all well and good.
```
- ReplaceInt: {Offset: 0xA49B1E, Find: 18, Replace: 15}   # should match Sunset hour
```
I've now applied it to both my Clara/Forma test devices ready for "Sunset" to happen at 15:00 (45mins from now). 

What different behaviour should I be looking for, exactly, as a result of having added this extra patch line?

geek1011, 2019-11-11:
> What different behaviour should I be looking for, exactly, as a result of having added this extra patch line?

It should start changing automatically at a sunset time before 6PM without needing to put it to sleep and wake it up again.

jackiew1, 2019-11-11:
It did. I was reading the Clara at 15:00 when there was the tiniest shift off zero-orange. The slider hasn't moved again in the last 12mins of reading.

I've now just done a manual sleep/wake cycle and the slider moved quite a lot. To be expected given the full transition should be completed in 1hour start-to-finish.

geek1011, 2019-11-11:
OK, so that one worked. I might need to change the other timer too if it doesn't keep moving by itself.

jackiew1, 2019-11-11:
The Clara moved on its own at 15:30.

geek1011, 2019-11-11:
> The Clara moved on its own at 15:30.

So, it seems to work then. Do you think this patch is ready and bug-free now?

```yaml
Customize ComfortLight settings:
  - Enabled: no
  - Description: |
      Change the times available in the ComfortLight bedtime dropdown and the
      start/end times for the color changes. All values in this patch are
      customizable.

      The default values for this patch give you 21 options between 5PM and 3AM
      at 30 minute intervals, with the colour change starting at 4PM, and the
      change back to blue between 5AM and 7AM. The firmware default is 13 options
      between 9PM-3AM at 30 minute intervals, with the transition starting at 6PM.
    #
    # In an unnamed subroutine two layers into FrontLightPopupController::loadView
    # (find it by going back from QTime::addSecs), the times for the dropdown are
    # generated into a QVector<QPair<QString, QTime>> with a simple loop.
    #
  # Change the initial hour / first dropdown item passed to the QTime constructor
  # (mov r1, #21):
  - ReplaceInt: {Offset: 0xA50E3E, Find: 21, Replace: 17} # 21:00 (9PM) -> 17:00 (5PM)
  # Optional: Change the increment passed to QTime::addSecs (you need to use an
  # ARM assembler to change this value):
  - ReplaceBytes: {Offset: 0xA50EA2, FindH: 4F F4 E1 61, ReplaceH: 4F F4 E1 61} # mov.w r1, #1800: 30m
  # Change the number of increments / dropdown items (movs r5, #13):
  - ReplaceInt: {Offset: 0xA50E76, Find: 13, Replace: 21} # 13 [9PM..3AM]+30m -> 21 [5PM..3AM]+30m
  # Change the number of pre-allocated elements (I don't think this is actually
  # required, as Qt will realloc as necessary, but we might as well do this just
  # in case) (mov r1, #21):
  - ReplaceInt: {Offset: 0xA510DE, Find: 13, Replace: 21} # same as prev replacement
    #
    # The actual color adjustment is done in BedtimeManager::adjustTemperature.
    # The color temperature is based on the current time, the bedtime, the hardcoded
    # sunset, the hardcoded sunrise start, the hardcoded sunrise end, the daytime
    # temperature, and the nighttime temperature.
    #
  # Change the sunset time (the transition to a warmer temperature starts here
  # and transitions until the bedtime chosen) (this should be 1-3 hours before
  # the minimum bedtime from the first replacement above) (mov r1, #18):
  - ReplaceInt: {Offset: 0xA49A06, Find: 18, Replace: 16} # 18:00 (9PM-3=6PM) -> 16:00 (5PM-1=4PM)
  # Change the time for the sunset timer to fire at (this is passed to a QTime
  # constructor which is passed to PowerTime::fireAt) (mov r1, #18):
  - ReplaceInt: {Offset: 0xA49B1E, Find: 18, Replace: 16} # should match sunset time above
  # Optional: Change the sunrise start time (the transition back to a cooler
  # temperature starts here) (this should be sometime after the last bedtime,
  # but before the sunset time) (mov r1, #5):
  - ReplaceInt: {Offset: 0xA49A28, Find: 5, Replace: 5} # 5AM
  # Optional: Change the sunrise end time (the transition back to a cooler
  # temperature ends here) (this should be 1-3 hours after the sunrise start but
  # before the sunset time or bad things will happen with negative values) (mov r1, #7):
  - ReplaceInt: {Offset: 0xA499F2, Find: 7, Replace: 7} # 5AM+2=7AM
```

jackiew1, 2019-11-11:
> So, it seems to work then. Do you think this patch is ready and bug-free now?

The Clara just completed its transition to orange at 16:00 with no manual intervention whilst I was reading normally. So a 1 hour full transition works as expected. There's nothing more to be learned from the Clara today.

I'd say it's working correctly, but I want to let the Forma's transition play out to the end before giving a final opinion. This should have a 6-hour transition which will complete at 21:00.

I need to go and do something (unrelated) now. I'll comment on your revised patch shortly when I come back.

NiLuJe, 2019-11-11:
As a completely useless data point, the Kindle uses geolocation to request sunset/sunrise times at the user's location ;).

jackiew1, 2019-11-11:
Comments:
- Using variable names when the same value is required in multiple places. With my programmer hat on, this seems like a good idea and one I might try elsewhere in my custom patches. Is it documented somewhere? For non-techies, I'm not sure whether it helps or hinders if they haven't seen it before. A bit of education might help, or an extra comment on subsequent lines where a varname is referenced e.g. "# *** Don't change the next line ***"

- To make it clearer what it does, change "first dropdown" to "first Bedtime dropdown" in the comment for
```
  # Change the initial hour / first dropdown item passed to the QTime constructor
  # (mov r1, #21):
  - ReplaceInt: {Offset: 0xA50E3E, Find: 21, Replace: 17} # 21:00 (9PM) -> 17:00 (5PM)
```

- Why is this still here? It still doesn't appear to serve any purpose from a user POV. What are they supposed to do with it?
```
  # Optional: Change the increment passed to QTime::addSecs (you need to use an
  # ARM assembler to change this value):
  - ReplaceBytes: {Offset: 0xA50EA2, FindH: 4F F4 E1 61, ReplaceH: 4F F4 E1 61} # mov.w r1, #1800: 30m
```

- Perhaps make it clearer how much choice users have in changing the number of increments. Is it a free choice? Or should they either leave it at 21 or comment it out completely (like I did).
```
  # Change the number of increments / dropdown items (movs r5, #13):
  - ReplaceInt: {Offset: 0xA50E76, Find: 13, Replace: 21} # 13 [9PM..3AM]+30m -> 21 [5PM..3AM]+30m
  # Change the number of pre-allocated elements (I don't think this is actually
  # required, as Qt will realloc as necessary, but we might as well do this just
  # in case) (mov r1, #21):
  - ReplaceInt: {Offset: 0xA510DE, Find: 13, Replace: 21} # same as prev replacement
```

jackiew1, 2019-11-11:
> As a completely useless data point, the Kindle uses geolocation to request sunset/sunrise times at the user's location ;).

Does that require wi-fi to be on or does it just use the device's timezone setting? Anything which absolutely requires wi-fi to be on is the devil's work as far as I'm concerned. 😃

geek1011, 2019-11-11:
> Using variable names when the same value is required in multiple places. With my programmer hat on, this seems like a good idea and one I might try elsewhere in my custom patches. Is it documented [...]

Yes, although the use is quite limited. The value must have already been defined as a property of another element (i.e. you can't have things like config vars), the variable is global (I'd rather not use them too much), and it's all or nothing (i.e. you can't concatenate strings or that kind of thing). I'd rather not document it, as it's not something I want to officially endorse using for these reasons.

> To make it clearer what it does, change "first dropdown" to "first Bedtime dropdown" in the comment for

:+1: 

> Why is this still here? It still doesn't appear to serve any purpose from a user POV. What are they supposed to do with it?

You can replace it with something like armconverter.com.

> Perhaps make it clearer how much choice users have in changing the number of increments. Is it a free choice? Or should they either leave it at 21 or comment it out completely (like I did).

I'll add some more notes about that.

[7 more replies]

commit a3771cf7 (2019-11-11): Added new "Change ComfortLight settings" patch for 4.18.13737 (closes #36)
closed 2019-11-11 by geek1011