pgaskin/kobopatch-patches #37: Have a different brightness icon when light is off

hub2git, 2019-11-11:
On the Kobo Forma, the brightness icon is the same whatever the  brightness level, whether 0%, 1%... 5%... 50%... 100%. It's an icon of a  sun. 

I'd like to request the Kobo patch team to either:
A) replace the sun icon with a different icon when the light is completely off (for example, a circle that is filled with black) 
or
 B) create something beside the sun icon to indicate the light is completely off (for example,the word "0% light")  

Rationale for this request: occasionally, when I'm in a completely dark  room, I turn on the brightness from 0% to 1%. The next day, in the  daytime, I read on my Kobo Forma forgetting that the brightness is on,  and wading battery because I don't need the light at all in the daytime.

geek1011, 2019-11-11:
I won't be implementing this. Although it is a relatively simple thing to do, it would require modifying the UI and adding resources, which is quite time consuming.

hub2git, 2019-11-11:
Ok. Thanks for your reply. Who can I bring this request to?

geek1011, 2019-11-12:
I don't think there's anybody else who has experience with this and the interest to work on it at the moment.

hub2git, 2019-11-12:
@geek1011, thanks again for your response. Maybe we can create a fund or a target dollar-figure for each request. Then for those that meet the minimum dollar-figure can get worked on. I think this is called "bounty". 
Anyway, I appreciate all your work on the various patches. :)

closed 2022-08-06 by pgaskin