pgaskin/kobopatch-patches #42: Patches for 4.19.14123

geek1011, 2019-12-11:
- [ ] Upload testdata (@geek1011)
- [ ] Update patches (@geek1011)
- [ ] Update 24 line spacing values (@geek1011 or Semwize)
- [ ] Rewrite patches
  - [ ] Swap header/footer? (@geek1011)
  - Anything else? (TODO)
- [ ] Test most patches (@geek1011)
- [ ] Make release (@geek1011)
- [ ] MR post (@geek1011)
- TODO: anything else, if needed.

geek1011, 2019-12-11:
@jackiew1, the header/footer related patches need to be updated/rewritten. If you can't get around to it tonight, I'll probably do them, but comment before starting to work on them.

geek1011, 2019-12-11:
And, I'll probably add this to the repo later, but here's a useful Hopper script to transfer most offsets between versions which I've been using for a while:

```py3
# Transfer the position (symbol+offset) from the first open file to the second, and print offset/instruction info.
old, new = Document.getAllDocuments()
print("> %s -> %s" % (old.getDocumentName(), new.getDocumentName()))

old_seg         = old.getCurrentSegment()
old_addr        = old.getCurrentAddress()
old_proc        = old.getCurrentProcedure()
old_proc_sig    = old_proc.signatureString()
old_proc_base   = old_proc.getEntryPoint()
old_proc_offset = old_addr - old_proc_base
old_inst        = old_seg.readBytes(old_addr, old_seg.getInstructionAtAddress(old_addr).getInstructionLength())

print("0x%X 0x%X+%d %s %s" % (old_addr, old_proc_base, old_proc_offset, old_proc_sig, " ".join("%02X" % ord(c) for c in old_inst)))

new_seg = new.getCurrentSegment()
for i in range(new_seg.getProcedureCount()):
    new_proc        = new_seg.getProcedureAtIndex(i)
    new_proc_sig    = new_proc.signatureString()
    new_proc_base   = new_proc.getEntryPoint()
    new_proc_offset = old_proc_offset
    new_addr        = new_proc_base + new_proc_offset
    
    if new_proc_sig == old_proc_sig:
        new_inst = new_seg.readBytes(new_addr, new_seg.getInstructionAtAddress(new_addr).getInstructionLength())
        print("0x%X 0x%X+%d %s %s" % (new_addr, new_proc_base, new_proc_offset, new_proc_sig, " ".join("%02X" % ord(c) for c in new_inst)))
        new.setCurrentAddress(new_addr)
        break
```

geek1011, 2019-12-11:
The new firmware adds scrollbars to the settings page. I find these slightly off-putting, so I might make a patch to remove them. I see why they were added, though.

jackiew1, 2019-12-11:
> @jackiew1, the header/footer related patches need to be updated/rewritten. If you can't get around to it tonight, I'll probably do them, but comment before starting to work on them.

@geek1011 ,
I won't have time to deal with this properly until tomorrow afternoon. Too much going on tonight to give it the concentration it needs (_... and no I haven't started the Christmas drinking early!_)

geek1011, 2019-12-11:
OK, I'll see if I can fix it easily tonight, and if not, I'll still release the patches, and have those updated with the next update (I'll be making a new patch for then too).

geek1011, 2019-12-11:
A note to myself about the font sizes patch: the list appending was inlined by gcc, so that's why the offsets are different.

geek1011, 2019-12-11:
The `Freedom to advanced fonts control` patch has been completely messed up, and needs to be rewritten. The entire font settings handling seems to have been refactored.

geek1011, 2019-12-11:
I've finished with the libnickel offsets, and just need to look at the nickel and libadobe stuff.

Also, `Remove forgot pin button from lock screen` needs to be rewritten.

geek1011, 2019-12-11:
@jackiew1, I've fixed `Custom new header/footer font`, which needed one additional replacement to split the trilogy header/footer styles and which doesn't need the locale stuff to be removed to make space anymore.

```yaml
        # Don't change the following pair. It exists to split the header/footer styles
        # which were merged in 4.19.14123.
        - Find:    "#caption[newHeader=true][qApp_deviceIsTrilogy=true],\n#caption[newFooter=true][qApp_deviceIsTrilogy=true] {\n  font-size: 13px;\n}"
          Replace: "#caption[newHeader=true][qApp_deviceIsTrilogy=true] {\n  font-size: 13px;\n}\n#caption[newFooter=true][qApp_deviceIsTrilogy=true] {\n  font-size: 13px;\n}"
```

geek1011, 2019-12-11:
@jackiew1, `Custom page navigation scrubber` didn't need the locale stuff to be removed to make space either.

geek1011, 2019-12-11:
@jackiew1, for `Reduce new header/footer height`, the `[newNavHeight]` property isn't used anymore, and the styles are directly applied on `ReadingFooter`. I just needed to change the selector names.

jackiew1, 2019-12-11:
The 'Default ePub monospace font (Courier)' patch in librmsdk.so.1.0.0.yaml will need some work. I think I know what needs to be done but anything to do with fonts usually needs a lot of testing.

On the bright side this change appears to have gone part way to fixing the lack of a monospace font for standard epub. Still no actual monospace font included in the fw, though.

It's possible that the patch will be able to be removed altogether. Will investigate properly when I have more time.

geek1011, 2019-12-11:
I'll see what I can do with that one (I'm actually looking at it right now), but I'll most likely save it for the next update if the fix isn't obvious.

geek1011, 2019-12-11:
I'm going to remove the patch for this release. If another patch is needed, it'll be a completely new one, so there's no point keeping this one around.

geek1011, 2019-12-11:
All that was needed for `Disable orphans/widows avoidance` is an offset update, probably due to the changes for the monospace font.

I'm going to take a short break now, and will finish up with the libnickel stuff after.

jackiew1, 2019-12-11:
> @jackiew1, `Custom page navigation scrubber` didn't need the locale stuff to be removed to make space either.

@geek1011, The locale stuff isn't needed if the patch is enabled as-is, but some of the Replaces in sections 1b & 2b, (which are commented out by default but where the user can be creative with CSS if they want) could possibly cause 'too long' errors.

We can remove them if you prefer. Kobo seem to be doing quite a lot of work with Chinese locales in recent firmwares so patches that reference them keep breaking. It's easy enough to add them back if someone gets really verbose with extra CSS then reports an error. There are probably very few users using lengthy extra CSS.

geek1011, 2019-12-11:
@jackiew1, thanks for clarifying that. It's up to you what you want to do with those patches. Here is what I have right now:

<details>
<pre>
Reduce new header/footer height:
  - Enabled: no
  - Description: |
        Reduce new reading header/footer heights when they are enabled.
        Affects both ePubs and KePubs.
        This patch may be of interest to those who used to enable the old
        "Custom menubar - reduce height by 33%"
  - FindZlib: "ReadingFooter"
  - ReplaceZlibGroup:
      # Adjust the Replace values of min-height/max-height as you wish.
      # Keep min-height=max-height
      Replacements:
        # ReadingFooter Reduce min/max height by 33%
        # Touch/Mini (Trilogy)
      - Find:    "ReadingFooter[qApp_deviceIsTrilogy=true] {\n  min-height: 56;\n  max-height: 56px;\n}"
        Replace: "ReadingFooter[qApp_deviceIsTrilogy=true] {\n  min-height: 37px;\n  max-height: 37px;\n}"
        # Glo/Aura/Aura2E (Phoenix)
      - Find:    "ReadingFooter[qApp_deviceIsPhoenix=true] {\n  min-height: 71px;\n  max-height: 71px;\n}"
        Replace: "ReadingFooter[qApp_deviceIsPhoenix=true] {\n  min-height: 47px;\n  max-height: 47px;\n}"
        # AuraHD/H2O/GloHD/ClaraHD (Dragon)
      - Find:    "ReadingFooter[qApp_deviceIsDragon=true] {\n  min-height: 101px;\n  max-height: 101px;\n}"
        Replace: "ReadingFooter[qApp_deviceIsDragon=true] {\n  min-height: 66px;\n  max-height: 66px;\n}"
        # AuraOne/Forma (Daylight)
      - Find:    "ReadingFooter[qApp_deviceIsDaylight=true] {\n  min-height: 131px;\n  max-height: 131px;\n}"
        Replace: "ReadingFooter[qApp_deviceIsDaylight=true] {\n  min-height: 86px;\n  max-height: 86px;\n}"
        # LibraH2O (Storm)
      - Find:    "ReadingFooter[qApp_deviceIsStorm=true] {\n  min-height: 118px;\n  max-height: 118px;\n}"
        Replace: "ReadingFooter[qApp_deviceIsStorm=true] {\n  min-height: 78px;\n  max-height: 78px;\n}"

Custom new header/footer font:
  - Enabled: no
  - Description: |
        This patch allows you to change font-family and/or font-size in the 
        new header/footer. They will automatically be a matched pair. Full details & screenshots at 
        https://www.mobileread.com/forums/showpost.php?p=3897175&postcount=4
        N.B: This patch is not suitable for Japanese/Chinese locale users

  - FindZlib: "ReadingFooter"
  - ReplaceZlibGroup:
      Replacements:
        # Part 1. Customise font-family (built-in or sideloaded font)
        #         Please see notes at
        #         https://www.mobileread.com/forums/showpost.php?p=3896190&postcount=2
        #         for general considerations whenever you change font-family in the Kobo GUI
        
        # #caption[newHeader=true] and #caption[newFooter=true]
        # N.B: If new font-family contains spaces wrap in 'single quotes', e.g. 'Kobo Nickel'
        
        # Un-comment and edit replacement font-family to change header/footer font
        #- {Find: "font-family: Avenir;", Replace: "font-family:Avenir;"}

        # Part 2. 
        #   2a. Customise font-size for header/footer
        #   2b. (optional) Fine-tune footer text position to avoid cutting bottom progressbar
        #       (optional) Fine-tune header text position to avoid possible shadow from top bezel
        #       May be useful to those who have also enabled patch 'Reduce new header/footer height'
        #       and have customised it to aggressively reduce header/footer height.
        
        # 2a. Change font-size for models other than Touch/Mini (header & footer simultaneously)
        # #caption[newHeader=true] & #caption[newFooter=true]
          # Glo/Aura/Aura2E (Phoenix)
        - Find:    "[qApp_deviceIsPhoenix=true] {\n  font-size: 16px;\n}"
          Replace: "[qApp_deviceIsPhoenix=true] {\n  font-size: 16px;\n}"
          # AuraHD/H2O (Dragon)
        - Find:    "[qApp_deviceIsDragon=true] {\n  font-size: 21px;\n}"
          Replace: "[qApp_deviceIsDragon=true] {\n  font-size: 21px;\n}"
[... 95 lines cut ...]
          Replace: "#page[qApp_deviceIsPhoenix=true] {\n  font-size: 22px;"
          # AuraHD/H2O (Dragon)
        - Find:    "#page[qApp_deviceIsDragon=true] {\n  font-size: 26px;"
          Replace: "#page[qApp_deviceIsDragon=true] {\n  font-size: 26px;"
          # GloHD (Alyssum)
        - Find:    "#page[qApp_deviceIsAlyssum=true] {\n  font-size: 30px;"
          Replace: "#page[qApp_deviceIsAlyssum=true] {\n  font-size: 30px;"
          # ClaraHD (Nova)
        - Find:    "#page[qApp_deviceIsNova=true] {\n  font-size: 30px;"
          Replace: "#page[qApp_deviceIsNova=true] {\n  font-size: 30px;"
          # LibraH2O (Storm)
        - Find:    "#page[qApp_deviceIsStorm=true] {\n  font-size: 30px;"
          Replace: "#page[qApp_deviceIsStorm=true] {\n  font-size: 30px;"
          # AuraOne/Forma (Daylight)
        - Find:    "#page[qApp_deviceIsDaylight=true] {\n  font-size: 34px;"
          Replace: "#page[qApp_deviceIsDaylight=true] {\n  font-size: 34px;"

        # 2b. Customise other font properties of 'Page x of y' label (all models)
        #     e.g. text-transform (uppercase or none)
        #          font-family, font-weight, font-style, etc.
        # Un-comment next 2 lines and change Replace CSS as desired
        #- Find:    "#page {\n  padding-left: 0px;\n}"
        #  Replace: "#page {padding-left: 0px; font-family:Georgia; text-transform:none;}"
</pre>
</details>

jackiew1, 2019-12-11:
> @jackiew1, thanks for clarifying that. It's up to you what you want to do with those patches.

@geek1011, But it's not just up to me, is it? You're in the process of updating my patches. Thanks for the work you've done so far but I'm not sure what the mad rush is. I should be able to give this my full attention when I get home tomorrow mid-afternoon.

geek1011, 2019-12-11:
@jackiew1, I try to release the patches the same day to give them the most exposure for testing and feedback (plus I use them myself). Also, I'm quite busy tomorrow compared to today, so today's the best time for me. I think I'll see if I can get the replacements to fit in, and if not, I'll let you work on it tomorrow and then release the next version within a few days.

geek1011, 2019-12-11:
@jackiew1, the additional replacements do fit in with the new firmware version as-is. Something must have changed about the way the streams are compressed...

geek1011, 2019-12-12:
I've spent the last hour and a bit rewriting the 24 line spacing values patch to only require 5 offsets to update rather than 15 offsets and a whole lot of manually assembled BLX instructions. Now, it should be as easy as any other patch to update.

geek1011, 2019-12-12:
I've spent the last 45 minutes rewriting the `Freedom to advanced fonts control` from scratch. It is now quite a bit simpler and less bug-prone, and should be easier to update in the future.

All I have left to do is test the patches.

geek1011, 2019-12-12:
I'm finished with updating, rewriting and thoroughly testing all of the current patches and will release v56.

Here's what's planned for v57:
- Make more patches multi-version?
- Hide scrollbars? (I've already figured out how to hide them, I just need to put the padding back)
- Monospace font stuff (@jackiew1?)
- Any other miscellaneous bugfixes/improvements

geek1011, 2019-12-12:
MR:t-325572, [v56](https://github.com/geek1011/kobopatch-patches/releases/tag/v56).

closed 2019-12-12 by geek1011

jackiew1, 2019-12-12:
> @jackiew1, the additional replacements do fit in with the new firmware version as-is. Something must have changed about the way the streams are compressed...

Not enough testing. As you can see in the MR forum, your decision to remove the Chinese/Japanese space-savers from patch 'Custom new header/footer font' was wrong. I've done a new commit to put them back with the required corrections for 4.19.

geek1011, 2019-12-13:
I'll publish another release by Sunday (just in case I decide to make other fixes/improvements with other patches or do any requests).

jackiew1, 2019-12-13:
@geek1011, I've re-added a monospace patch to geoffr's  librmsdk.so.1.0.0.yaml because at least one person (2 if you include me 😃) was using it. I've updated Description and Find/Replaces to suit fw 4.19.14123.

However, If you would prefer it wasn't there, just delete it again.

geek1011, 2019-12-13:
I'm going to move the patch to jackie_w.yaml, as it's essentially a new patch.

jackiew1, 2019-12-13:
OK, but it is the first of a set of 4 similar patches in the same yaml. You can move all of them if you want.

jackiew1, 2019-12-14:
@geek1011, I've just installed v57 on a Clara. I noted your offset change to 'Custom font sizes'.

It used to be that if I had both this patch and 'ePub uniform font scale' enabled then font-size in epubs and kepubs were visibly the same when using the stored value in the .conf file. This doesn't seem to be true in v57.

I even thought this was working OK in v56. Now I'm no longer sure. 😕

geek1011, 2019-12-14:
Try 4 and 14 for the scale values (ePub and KePub respectively). JSWolf sent me a PM a while ago about it.

jackiew1, 2019-12-14:
But that's a different patch - 'Set font scale factor'. I've never used that one so have no idea when it's working right, I use 'ePub uniform font scale'.

Extra info, my Forma still has the v56 patches on it. I just re-checked and, my gut feel was right, everything **is** working OK with the same 2 patches enabled. So for me the v56 version of 'Custom font sizes' was "more correct". More confusion???

jackiew1, 2019-12-14:
... I've just put v56 back on the Clara. My gut feel was only partially correct. v56 works on Forma but not Clara. Is the problem with the 'ePub uniform font scale' patch? Does it no longer do what it used to do???

I'll now try what you asked. If 'Set font scale factor' works on both Clara and Forma using v57 offsets then I'll concede and forget about 'ePub uniform font scale' forever.

jackiew1, 2019-12-14:
... hang on. Restart the clock. I've found a problem at my end. I'm starting from scratch...

jackiew1, 2019-12-14:
Right, now I've corrected my base epubs & kepubs ...

With v57 patches:
- **'ePub uniform font scale'** is working OK on Clara/Forma (so I'm sorted I think, sorry for sounding the alarm)
- **'Set font scale factor'** is NOT OK on Clara/Forma. An epub displays massive compared to a kepub at the same font-size when using the settings you specified, i.e. 
`       - ReplaceInt: {Offset: 0x52C522, Find: 14, Replace: 4} # ePub`
`       - ReplaceInt: {Offset: 0xAE9C80, Find: 14, Replace: 14} # KePub`
Are you sure you got the 4 & 14 the right way round? 
I can't call on previous experience with this.

I'm not sure how to test your **'Custom font sizes'** offset tweak without a slow painstaking effort. Do you already know for sure that v57 is correct and v56 wasn't?