pgaskin/kobopatch-patches #44: Set font scale factor should have defaults of 4 and not 8

JSWolf, 2019-12-15:
The Set font scale patch as it is does not keep the same font size in ePub before the patch is applied. Values of 8 do not work well. Values of 4 are the correct defaults. Sure you can change the values if you want, but if you don't want, then 4 is what needs to be put in place.

Set font scale factor:
  - Enabled: yes
  - Description: |
      This patch sets the font scale factor for ePub and KePub readers.
      See also `ePub uniform font scale` and `Custom font sizes` patches.
  # Font scale factor is calculated as (font size slider) / (X+32)
  # Original value X=14, so original scale factor is (font size slider) / 46.
  # Allowed replacement values for X: 0-15 (smaller X --> larger scale factor)
  - ReplaceInt: {Offset: 0x52C522, Find: 14, Replace: 4} # ePub
  - ReplaceInt: {Offset: 0xAE9C80, Find: 14, Replace: 4} # KePub

That is the patch with the correct default values of 4.

geek1011, 2019-12-16:
@jackiew1, do you mind testing this before I merge it?

jackiew1, 2019-12-16:
As you know, normally, I wouldn't mind testing a patch for you. This is what I tried to say yesterday. With patch **'Set font scale factor'** I honestly don't understand what it's for and why anyone might decide to use it. That's despite having read the Description and GeoffR's original thread when he created it. So I've no idea how to test whether it's working properly. I think you're going to need JSWolf to do the test. He'll tell you quick enough if it's not working as expected.

This is no help, but I understand what it's sibling does, **'ePub uniform font scale'** and that one is still working as expected.

geek1011, 2019-12-16:
I don't totally understand either, as I didn't really follow the development of this patch. I'll look into this further later this week, but it'd be helpful if @JSWolf could explain this with a bit more depth.

I don't mind merging this change as long as I understand what the implications of this change are regarding the custom font sizes patch and the uniform font scale patch.

Note that I only read kepubs, so I'm not the best person to test this patch.

JSWolf, 2019-12-16:
The unpatched ePub font size is changed with the patch when the replace values are set to 8. When the values are set to 4, the font size stays the same. So for those who read mostly ePub, this keeps the font size the same. With replace values of 8, the ePub font size is made smaller.

jackiew1, 2019-12-16:
@JSWolf @geek1011 I'm still none the wiser. 😕
Jon's explanation seems to say that **'Set font scale factor'** should be changed to 4 for epub so that font sizes don't change for epub. Wouldn't it make more sense to just disable the patch if you don't want anything to change??? What am I missing here?

JSWolf, 2019-12-16:
@jackiew1 the idea of the patch is to make ePub and KePub font sizes match instead of having KePub be that much smaller. We are changing the font size of KePub in any case. But we don't have to change the size of ePub by changing the replace values to 4. It means that ePub stays the same and KePub matches ePub.

jackiew1, 2019-12-16:
@JSWolf But patch **'ePub uniform font scale'** makes epub and kepub font sizes match. You haven't explained why we need a a different patch to do the same thing? What am I still missing?

JSWolf, 2019-12-16:
@jackiew1 It's not a different patch. It's a modification to the default values in the existing patch. The replace values in the existing patch are 8 and they should be changed to 4.

jackiew1, 2019-12-17:
@JSWolf , @geek1011  
No, you've misunderstood my question. I'll try again. Why does patch **'Set font scale factor'** need to exist at all when patch **'ePub uniform font scale'** already does the job of equalising font-size in epub and kepub? This has been my confusion all along.

JSWolf, 2019-12-17:
@jackiew1 @geek1011 What it looks like to me is that 'ePub uniform font scale' sets both ePub and KePub to be the same size as KePub. What 'Set font scale factor' does is change the font size scaling.

The first patch makes ePub match KePub and the second patch changes the sizing so you can set how large or small you want the font scaling to be. with the second patch when you set the replace values to be 4, you get both ePub and KePub to match the unpatched ePub size.

So one sets ePub & Kepub to the same size as KePub so they match and the other changes the size. So they both go together.

geek1011, 2019-12-17:
Thanks for the explanation @JSWolf! That seems to make sense to me, and I can see why this change is useful.

So, if @jackiew1 could test to ensure that this change keeps the relative epub to kepub sizes the same, even when combined with the uniform font scale and custom font sizes patch, I'll be happy to merge this.

JSWolf, 2019-12-17:
@geek1011 @jackiew1 I was thinking, would it be OK to combine these two patches into one with an a & b section like some other patches? That way they will be just one patch and not so confusing?

Change the replace values from 8 to 4 and combine the two patches and there we go, done.

jackiew1, 2019-12-17:
@JSWolf and @geek1011 Well I learned something today. I thought these 2 patches were either/or not both enabled.

I still don't see (and this may just be me) why I should care whether the actual font-size on the slider is 35 or 50 as long as both kepub and epub look the same.

@geek1011 I'll do the test you asked for with both patches enabled and both ReplaceInt values in 'Set font scale factor' set at 4.

NiLuJe, 2019-12-17:
@jackiew1: Possibly because of how Nickel has no real hard concept of a "global default", and instead uses the last "local" (i.e., per-document) settings when opening a new book (i.e., one that was never opened, it has no per-document setting). In which case, switching from a KePub to an ePub (or vice-versa) would ensure consistent font sizes (if you forget the usual ePub font-sizes inconsistencies depending on ibundled CSS ;p) *without* having to remember that you usually want 35 in an ePub and 50 in  a KePub ;).

jackiew1, 2019-12-17:
> @jackiew1: Possibly because of how Nickel has no real hard concept of a "global default", and instead uses the last "local" (i.e., per-document) settings when opening a new book (i.e., one that was never opened, so it has no document-specific settings). In which case, switching from a KePub to an ePub (or vice-versa) would ensure consistent font sizes (if you forget the usual font-sizes inconsistencies depending on bundled CSS ;p) _without_ having to remember that you usually want 35 in an ePub and 50 in a KePub ;).

@NiLuJe Yes, I understand that, but enabling **'ePub uniform font scale'** on its own already lets me switch between epub and kepub without a care in the world. My question has always been, why would I want to faff around with two patches when one will suffice? especially when that one patch is already multiversion, i.e. no ever-changing absolute offsets.

NiLuJe, 2019-12-17:
Oh, right ;).

AFAICT, while you *can* achieve `ePub uniform font scale`'s effect via `Set font scale factor`, the latter has a much broader scope: you can do whatever you want with it as far as scaling per format is concerned.

Now, *why* you would want to do anything else than make KePub and ePub scaling match is a good question, though ^^.

JSWolf, 2019-12-17:
@NiLuJe I read ePub mostly. So I want to have the ePub font scale remain the same. With the second patch, I get that. I also feel that the font scale for ePub is better then the font scale for KePub especially for someone who wants a much larger font size.

NiLuJe, 2019-12-17:
That said, I'm the absolutely worst person to make any judgement on that front, though, as I never read ePubs, because the insane latency when turning pages induces either profound rage or profound sleepiness ;).

NiLuJe, 2019-12-17:
@JSWolf: What's unclear to me, then, is why you're not happy with simply using `ePub uniform font scale` ;).

(Not that I disagree in making the defaults in `Set font scale factor` more-or-less match that, with the few caveats exposed above).

NiLuJe, 2019-12-17:
Oh, I think I may have answered my own question: because `ePub uniform font scale` affects *both* scales (I was assuming it kept one at the default value).

In which case, it's simply a case of the smaller multiplier leaving you with a more desirable set of effective font sizes, which I can perfectly understand ;).

NiLuJe, 2019-12-17:
Which leads me to another new question: on a device with kludgy ePub/KePub scales, with *both* patches enabled, I'm assuming you'll want both multipliers in `Set font scale factor` to match in order for the scaling to be consistent between KePub and ePub?

In which case, the current defaults *do* make sense... ^^

jackiew1, 2019-12-17:
@geek1011
Tests complete on: 
- ClaraHD - which unpatched **DOES** have a kepub/epub font-size mismatch. Enabling both patches with **Set font scale factor** set at 4/4 does no harm. Epub/kepub look the same. The only "benefit" to me is that my comfortable default font-size in .conf has now changed from 50-->31.
- Forma - which unpatched **DOES NOT** have a kepub/epub font-size mismatch. Same result as the ClaraHD except my default .conf font-size changed from 53-->34.

Conclusions: 
- @JSWolf's request for changes to **Set font scale factor** seems harmless, if **both** patches are enabled.
- I wouldn't recommend enabling **Set font scale factor** on its own because of the device-specific differences in unpatched font-size behaviour. The KA1 & Forma are in one group. The ClaraHD and H2O are in the other group. It's unclear which group the other models are in. I think GeoffR said the Glo behaves like the KA1.
- Speaking only for myself, I plan to continue ignoring **Set font scale factor** completely. If the only benefit is reducing my personal font-size from 50-_something_ to 30-_something_ then I can do without it.
- If anyone on MR asks for advice about customising **Set font scale factor** I'm going to stay out of it, much as I do with Home screen patches.

NiLuJe, 2019-12-17:
@jackiew1: Thanks! That would match with my understanding of how this all works (after misreading both patches twice, at least :D).

JSWolf, 2019-12-17:
> 
> 
> Which leads me to another new question: on a device with kludgy ePub/KePub scales, with _both_ patches enabled, I'm assuming you'll want _both_ multipliers in `Set font scale factor` to match in order for the scaling to be consistent between KePub and ePub?
> 
> In which case, the current defaults _do_ make sense... ^^
> 
> EDIT: Gah. As does your suggestion. (I, again, thought it was 8/4 or 4/8, not 4/4 -_-").

The first patch sets both ePub & KePub to match the font scale used by KePub. The second patch changes the font scale for both ePub & KePub. I see no reason for the second patch to have values that do not match one of the existing (unpatched) font scales. As I'm no good with KePub, I was able to get both font scales to match ePub. If both patches are combined into one, then when it's applied, chances are most people will keep the default scale values of 4 and have ePub stay the same and KePub modified to match ePub.

geek1011, 2019-12-17:
Thanks for looking into this @jackiew1 and @NiLuJe.

---

> I wouldn't recommend enabling Set font scale factor on its own because of the device-specific differences in unpatched font-size behaviour. The KA1 & Forma are in one group. The ClaraHD and H2O are [...]

From looking at the disassembly, I'm pretty sure it's for all [dragon](https://www.mobileread.com/forums/showthread.php?t=325471) devices (AuraHD/H2O/GloHD/H2O2/Clara/Libra). If someone can confirm this, I'll update the patch comments.

> Speaking only for myself, I plan to continue ignoring Set font scale factor completely. If the only benefit is reducing my personal font-size from 50-something to 30-something then I can do without it.

Yes, and for myself, I only read KePubs, so this doesn't make a difference to me.

> If anyone on MR asks for advice about customising Set font scale factor I'm going to stay out of it, much as I do with Home screen patches.

I will too. The most help I'll be giving is the facts we found here.

---

> In which case, the current defaults do make sense... ^^ EDIT: Gah. As does your suggestion. (I, again, thought it was 8/4 or 4/8, not 4/4 -_-").

> @JSWolf's request for changes to Set font scale factor seems harmless, if both patches are enabled.

I agree. I'll merge this change unless anyone has any further objections?

geek1011, 2019-12-17:
Do the above commits (contents and messages) look good?

Note: In the name of compatibility, I'm not backporting this change to older versions.

jackiew1, 2019-12-17:
I don't have any strong feelings on this, but did you particularly want to leave them as 2 separate patches? Combining them, as @JSWolf asked, might prevent someone enabling just the **'Set font scale factor'** on its own.

Smart folk (_like you'n'me_ 😁)  know enough to comment out the bottom 2 scaling lines if we don't want them.

geek1011, 2019-12-17:
I don't have much of an opinion on this either (other than that it will break compatibility with existing overrides, but we're already changing the scale, so why not break the names too...).

@NiLuJe, @JSWolf?

Also, I'll make the patch multi-version.

closed 2019-12-17 by geek1011

jackiew1, 2019-12-17:
@geek1011 Well now, if you're going to make a combined patch 100% multi-version I might even be persuaded to leave the scale 4/4 lines in.

It has to be said, no matter what you do, if a newbie enables these patch(es) and they have some books already open with font-sizes stored in the database then under certain circumstances their old book will be the "wrong size" after patching. Some of those people may complain. If you keep Jon happy, at least he won't be one of them LOL.

geek1011, 2019-12-17:
> Well now, if you're going to make a combined patch 100% multi-version I might even be persuaded to leave the scale 4/4 lines in.

Oh, I see your ulterior motive now ... you're even more impatient than me about patches for new firmware versions, and want to be able to update instantly :rofl:.

> It has to be said, no matter what you do, if a newbie enables these patch(es) and they have some books already open with font-sizes stored in the database then under certain circumstances their old book will be the "wrong size" after patching. Some of those people may complain.

Yes, I know. That's true about the line spacing patches too.

jackiew1, 2019-12-17:
> Oh, I see your ulterior motive now ... you're even more impatient than me about patches for new firmware versions, and want to be able to update instantly 🤣.

Oh, no. I'm only impatient to get the testing started early. 😝 Quality trumps speed. Always. Only sacrificial goat devices actually get early fw updates.

JSWolf, 2019-12-18:
> 
> 
> I don't have much of an opinion on this either (other than that it will break compatibility with existing overrides, but we're already changing the scale, so why not break the names too...).
> 
> @NiLuJe, @JSWolf?
> 
> Also, I'll make the patch multi-version.
> 
> Update: here it is, but it doesn't work with 4.18.13737, so I won't do it for now (I'll do it if it works with the next firmware version):
> 
> ```yaml
>   - FindBaseAddressSymbol: "AdobeStyling::update(QString const&)" # ePub
>   - ReplaceInt:            {Offset: 2965, Find: 14, Replace: 4}
>   - FindBaseAddressSymbol: "KepubBookReader::pageStyleCss(bool)"  # KePub
>   - ReplaceInt:            {Offset:  283, Find: 14, Replace: 4}
> ```

Looks good to me. Thank you.

JSWolf, 2019-12-18:
> 
> 
> > Well now, if you're going to make a combined patch 100% multi-version I might even be persuaded to leave the scale 4/4 lines in.
> 
> Oh, I see your ulterior motive now ... you're even more impatient than me about patches for new firmware versions, and want to be able to update instantly 🤣.
> 
> > It has to be said, no matter what you do, if a newbie enables these patch(es) and they have some books already open with font-sizes stored in the database then under certain circumstances their old book will be the "wrong size" after patching. Some of those people may complain.
> 
> Yes, I know. That's true about the line spacing patches too.

And it may also be true for the font size patch. Just tell everyone that the patches may change the existing settings and if so, just change the settings.

commit 5ef052de (2019-12-17): Changed default font scale factor in "Set font scale factor" to 4 (closes #44)

The previous value of 8 didn't match the unpatched KePub or ePub
scale factor, but 4 matches ePub and modifies KePub to match, which
is arguably more useful as a default, especially when using this
patch with "ePub uniform font scale".

See https://github.com/geek1011/kobopatch-patches/issues/44#issuecomment-566708751.
commit 9e770757 (2019-12-17): Changed default font scale factor in "Set font scale factor" to 4 (closes #44)

The previous value of 8 didn't match the unpatched KePub or ePub
scale factor, but 4 matches ePub and modifies KePub to match, which
is arguably more useful as a default, especially when using this
patch with "ePub uniform font scale".

See https://github.com/geek1011/kobopatch-patches/issues/44#issuecomment-566708751.