pgaskin/kobopatch-patches #46: 'Enable advanced settings for all fonts' not working in epubs

jackiew1, 2020-01-11:
@geek1011 
See 11.1.2020 posts in https://www.mobileread.com/forums/showthread.php?t=325572

It seems you can no longer change font-weight in standard epubs. This seems to be true even in unpatched 4.19 firmware with built-in fonts. The Advanced menu opens, the slider works, but there is no change to the epub's font-weight. 

I'm not sure whether this is fixable, but if anyone can you can.

geek1011, 2020-01-12:
I doubt he'll respond, but: @gtalusan.

And for a bit of context, I'll quote @jackiew1 on MR:post-3940208:

> **jackie_w:** Thanks for checking. I have an old AuraHD on v4.19.14123 which is completely unpatched. Even with the built-in fonts the font bolding doesn't work in epubs. The menu & slider work but it has no effect in the book.

geek1011, 2020-01-12:
I doubt I'll be able to fix this in a patch. Also, I don't read EPUBs (and judging by how long it was until this was noticed, neither do most patch users) and am not too familiar with the intricacies of RMSDK either.

I'll see what I can do later this week.

geek1011, 2020-01-12:
And a post from @JSWolf (MR:post-3940380):

> **JSWolf:** I just had a play around with the Advanced Font Controls and they do sort of work with ePub. The font weight value is being changed with the slider. That does work. But the new value is not being read until a reboot.

Which was then confirmed by @davidfor (MR:post-3940385):

> **davidfor:** Just tested one of my devices, and that's what happened. So, a bug in the firmware.

JSWolf, 2020-01-12:
@geek1011 @NiLuJe @jackiew1 Maybe a way an be found to be able to reboot nickle without rebooting the Kobo.

NiLuJe, 2020-01-12:
We kind of [always have](https://github.com/koreader/koreader/blob/master/platform/kobo/nickel.sh)?

Do heed the comments, this is [*NOT*](https://github.com/koreader/koreader/blob/72fd93c4ddae4542ffc1f34793b52f618a14d9db/platform/kobo/koreader.sh#L94-L96).

As-is, it's designed to be run from something started via on-animator.sh (i.e., KFMon/KSM). That won't necessarily be the case of whatever method you use to get shell access, which means more env variables would need to be siphoned off nickel.

JSWolf, 2020-01-12:
@NiLuJe @geek1011  @jackiew1 Guess a reboot will have to do until the bug is fixed by Kobo.

gtalusan, 2020-01-16:
Fixed in the next beta.

closed 2020-01-16 by geek1011