pgaskin/kobopatch-patches #48: Patches for 4.20.14601

geek1011, 2020-03-02:
- [ ] Upload testdata (@geek1011)
- [ ] Test dictionary stuff (it changed quite a bit) (for dictutil and the dictionary patches) (@geek1011)
- [ ] Find libnickel offsets (@geek1011)
- [ ] Update patches (@geek1011)
- [ ] Update nickel patches if necessary (@jackiew1)
- [ ] Rewrite patches if necessary
  - Anything?
- Test most patches (@geek1011)
- Make release (@geek1011)
- MR post (@geek1011)

geek1011, 2020-03-02:
- The new selection menu is nice. The selection now appears with a solid black background rather than an underline, and the menu is now more compact, with submenus instead of tabs.
- The new dictionary view is more streamlined, but is still just as small (it does have a fullscreen button, though). I haven't checked the functionality or if this is patchable yet. The extra dictionaries bug still remains, though (the one which requires a patch to fix the space splitting).

geek1011, 2020-03-02:
- Image tags in dictionaries with `dict:///` URLs don't segfault anymore, but they don't display either.

jackiew1, 2020-03-02:
@geek1011 
I believe I've got the necessary change for libnickel.so.1.0.0.yaml to make patch `'Dictionary text font-family/font-size/line-height`' work again. However, I haven't had time to apply it to an actual Kobo yet to double-check.

geek1011, 2020-03-02:
@jackiew1: Thanks for looking into that. I'll make the patch folder for 4.20.14601 soon (I haven't started working on the patches yet, as I was working on dictutil: see [v0.0.1-alpha.4](https://github.com/geek1011/dictutil/releases/tag/v0.0.1-alpha.4)).

jackiew1, 2020-03-02:
@geek1011 
Re: nickel.yaml patches ... just based on the ones I use myself my initial gut feel is that very little has changed in this fw.

geek1011, 2020-03-02:
Yep. The main interface changes I've noticed are with the selection menu and dictionary.

I'll need to rewrite the hide browser patch.

geek1011, 2020-03-02:
@jackiew1: None of your nickel patches are broken. I'm planning to release the patches tonight (I'll start working on them in an hour or so), so if you have the `Dictionary text font-family/font-size/line-height` change ready (you can post it in this issue), I'll include it in the initial release.

It also seems like the adobe libs have been updated in this version, so I'll need to fix those patches too.

jackiew1, 2020-03-02:
> @jackiew1: None of your nickel patches are broken. I'm planning to release the patches tonight (I'll start working on them in an hour or so), so if you have the `Dictionary text font-family/font-size/line-height` change ready (you can post it in this issue), I'll include it in the initial release.

I don't see any 4.20 yaml files for me to update if you want me to do them tonight.

Did you also notice that the libnickel.so.1.0.0.yaml `KePub stylesheet additions` patch is broken?

geek1011, 2020-03-02:
> I don't see any 4.20 yaml files for me to update if you want me to do them tonight.
Yes, they're not ready yet. If you post the changes here, I'll commit them for you.

> Did you also notice that the libnickel.so.1.0.0.yaml KePub stylesheet additions patch is broken?
Yes, I noticed that. It should be simple enough to fix.

jackiew1, 2020-03-03:
> > I don't see any 4.20 yaml files for me to update if you want me to do them tonight.
> 
> Yes, they're not ready yet (I'm starting to work on the patches in a few minutes). If you post the changes here, I'll commit them for you later tonight.

Here are the changes to Find/Replace and the help notes.
```
  # Stage 1: Change DictionaryView to remove %variables
  # from:
  #    * { font: %1px %2; line-height: 1.4em; }
  #    div.descriptionFont { font-family: %3; }
  #    ol { font-size: %1px; font-weight: bold; margin-left: %4em; margin-top: 0px; }
  #    ol p { font-size: %1px; font-weight:normal; }
  # to:
  #    * { font-size: %1px; line-height: 1.40em;font-family: Georgia                       ;}
  #    ol {margin-left:1em; margin-top:0}
  #    blockquote {margin:0.3em 0 0.3em 1em} blockquote>blockquote {margin:-0.3em 0 0 2em}
  #
  # ##### N.B. Do not change the next 4 lines #####
  - FindReplaceString:
      Find:    "* { font: %1px %2; line-height: 1.4em; }\ndiv.descriptionFont { font-family: %3; }\nol { font-size: %1px; font-weight: bold; margin-left: %4em; margin-top: 0px; }\nol p { font-size: %1px; font-weight:normal; }\n"
      Replace: "* { font-size: %1px; line-height: 1.40em;font-family: Georgia                       ;}\n\nol {margin-left:1em; margin-top:0}\nblockquote {margin:0.3em 0 0.3em 1em} blockquote>blockquote {margin:-0.3em 0 0 2em}\n"
      MustMatchLength: yes
  # ##### N.B. Do not change anything above this line #####

```

geek1011, 2020-03-03:
`Remove PDF map widget shown during panning` needs a rewrite (and through testing after).

geek1011, 2020-03-03:
`Remove PDF page turn arrows` is not needed anymore.

geek1011, 2020-03-03:
`Keep PDF pan position` won't work anymore AFAICT. I may look into it further if there is demand for it.

jackiew1, 2020-03-03:
@geek1011 
Another dictionary observation _(I don't think it's patch related)_ ...

It looks like it's no longer possible to see the additional label text contained in the SQL 'Dictionary' table 'Name' field, e.g. if I have a database Dictionary 'Name' of `Extra:_cl (Collins)` (Suffix `-cl`) then only the "Extra:_cl" portion of 'Name' is now visible in the dictionary drop-down list. In 4.19 the whole field content was visible.

geek1011, 2020-03-03:
Hmph. It seems like Kobo only half-fixed that bug then. They fixed the display issue, but not the internal one. @gtalusan?

I'll see if I can do anything about that, but I doubt I can. I'll probably add a note to dictutil about it.

Otherwise, I'll be releasing the patches as soon as I actually test them (probably in the next 30 mins).

geek1011, 2020-03-03:
@jackiew1, are you able to help me experiment with the dictionaries right now (just to figure out what's broken and what's fixed)?

geek1011, 2020-03-03:
I'll have to make another attempt at updating the "Remove PDF map widget shown during panning" patch. I think I botched this one. I might just leave this one for the next patch release.

geek1011, 2020-03-03:
OK, so it seems the dictionaries still work fine (minus the custom label, which was an unsupported hack from the beginning) if the patch (`Extra: ` -> `Extra:_`) is installed. I'll leave dictutil as-is for now, and we can continue looking into this later.

geek1011, 2020-03-03:
I just took a through look through `My 24 line spacing values`, and everything seems correct (i.e. the original code is basically identical to the last version, the pointer math adds up right, and there aren't any new references). I've tried applying it by itself, but it still segfaults when a book is opened.

The stack trace is kind of useless (but interestingly, it's nearly identical each time):

<details><pre>OH THE HUMANITY!
pid: 711, tid: 711 (nickel), rev: 30c2f5c348a74f67a162021a5d8cad7d4893b088
  >>> /usr/local/Kobo/nickel <<<
signal 11 (SIGSEGV), code 2 (SEGV_ACCERR), fault addr 2fa74928
 r0 7ee0f51c  r1 2b59a450  r2 851eb852  r3 2e9a4a74
 r4 7ee0f51c  r5 00000000  r6 2bf79cf0  r7 7ee0f518
 r8 00000000  r9 7ee0f5c0  10 019b5290  fp 7ee0f768
 ip 2fa770ac  sp 7ee0f518  lr 2b59a3b3  pc 2fa74928  cpsr 60030010
    #00 sp: 0x7ee0f518 ip: 0x2fa74928  /lib/libstdc++.so.6.0.12: _ZNSbIwSt11char_traitsIwESaIwEEC2IPKwEET_S6_RKS1_+0x70000001
--- --- --- --- --- --- --- --- --- --- --- --- --- --- --- ---
pid: 711, tid: 1066
 r0 00000006  r1 30727d80  r2 00000000  r3 00000000
 r4 00000000  r5 00000020  r6 30727d80  r7 0000008e
 r8 2eb04ca8  r9 00000000  10 01484d38  fp 00000000
 ip 00000000  sp 30727d60  lr 00000000  pc 2fbaa8d2  cpsr 800d0030
    #00 sp: 0x30727d60 ip: 0x2fbaa8d2  /lib/libc-2.11.1.so: __select+0x31
    #01 sp: 0x30727d70 ip: 0x2e8e1bc9  /usr/local/Trolltech/QtEmbedded-4.6.2-arm/lib/libQtCore.so.4.6.2: _ZN9QLockFile6unlockEv+0x17e4
    #02 sp: 0x30727e28 ip: 0x2e7b6789  /usr/local/Trolltech/QtEmbedded-4.6.2-arm/lib/libQtCore.so.4.6.2: _ZN7QThread21setTerminationEnabledEb+0x218
    #03 sp: 0x30727e50 ip: 0x2f9aa473  /lib/libpthread-2.11.1.so: __pthread_get_minstack+0xec6
--- --- --- --- --- --- --- --- --- --- --- --- --- --- --- ---
pid: 711, tid: 1069
 r0 0000000d  r1 31300560  r2 31300770  r3 31300980
 r4 00000000  r5 00000000  r6 0000000d  r7 0000008e
 r8 31300560  r9 31300770  10 31300980  fp 31300468
 ip 00000000  sp 311fec40  lr 00000000  pc 2fbaa8d2  cpsr 800e0030
    #00 sp: 0x311fec40 ip: 0x2fbaa8d2  /lib/libc-2.11.1.so: __select+0x31
    #01 sp: 0x311fec50 ip: 0x2e96365b  /usr/local/Trolltech/QtEmbedded-4.6.2-arm/lib/libQtCore.so.4.6.2: _Z14qt_safe_selectiP6fd_setS0_S0_PK8timespec+0x172
    #02 sp: 0x311feca0 ip: 0x2e964b7d  /usr/local/Trolltech/QtEmbedded-4.6.2-arm/lib/libQtCore.so.4.6.2: _ZN27QEventDispatcherUNIXPrivate8doSelectE6QFlagsIN10QEventLoop17ProcessEventsFlagEEP8timespec+0xdc
    #03 sp: 0x311feda0 ip: 0x2e965029  /usr/local/Trolltech/QtEmbedded-4.6.2-arm/lib/libQtCore.so.4.6.2: _ZN20QEventDispatcherUNIX13processEventsE6QFlagsIN10QEventLoop17ProcessEventsFlagEE+0xe8
    #04 sp: 0x311fedc8 ip: 0x2e92806b  /usr/local/Trolltech/QtEmbedded-4.6.2-arm/lib/libQtCore.so.4.6.2: _ZN10QEventLoop4execE6QFlagsINS_17ProcessEventsFlagEE+0xe6
    #05 sp: 0x311fee00 ip: 0x2e7b3605  /usr/local/Trolltech/QtEmbedded-4.6.2-arm/lib/libQtCore.so.4.6.2: _ZN7QThread4execEv+0x60
    #06 sp: 0x311fee28 ip: 0x2e7b6789  /usr/local/Trolltech/QtEmbedded-4.6.2-arm/lib/libQtCore.so.4.6.2: _ZN7QThread21setTerminationEnabledEb+0x218
    #07 sp: 0x311fee50 ip: 0x2f9aa473  /lib/libpthread-2.11.1.so: __pthread_get_minstack+0xec6
--- --- --- --- --- --- --- --- --- --- --- --- --- --- --- ---
pid: 711, tid: 1079
 r0 0000000c  r1 014b77b0  r2 014b79c0  r3 014b7bd0
 r4 00000000  r5 00000000  r6 0000000c  r7 0000008e
 r8 014b77b0  r9 014b79c0  10 014b7bd0  fp 014acb98
 ip 00000000  sp 31bfec40  lr 00000000  pc 2fbaa8d2  cpsr 80000030
    #00 sp: 0x31bfec40 ip: 0x2fbaa8d2  /lib/libc-2.11.1.so: __select+0x31
    #01 sp: 0x31bfec50 ip: 0x2e96365b  /usr/local/Trolltech/QtEmbedded-4.6.2-arm/lib/libQtCore.so.4.6.2: _Z14qt_safe_selectiP6fd_setS0_S0_PK8timespec+0x172
    #02 sp: 0x31bfeca0 ip: 0x2e964b7d  /usr/local/Trolltech/QtEmbedded-4.6.2-arm/lib/libQtCore.so.4.6.2: _ZN27QEventDispatcherUNIXPrivate8doSelectE6QFlagsIN10QEventLoop17ProcessEventsFlagEEP8timespec+0xdc
    #03 sp: 0x31bfeda0 ip: 0x2e965029  /usr/local/Trolltech/QtEmbedded-4.6.2-arm/lib/libQtCore.so.4.6.2: _ZN20QEventDispatcherUNIX13processEventsE6QFlagsIN10QEventLoop17ProcessEventsFlagEE+0xe8
    #04 sp: 0x31bfedc8 ip: 0x2e92806b  /usr/local/Trolltech/QtEmbedded-4.6.2-arm/lib/libQtCore.so.4.6.2: _ZN10QEventLoop4execE6QFlagsINS_17ProcessEventsFlagEE+0xe6
    #05 sp: 0x31bfee00 ip: 0x2e7b3605  /usr/local/Trolltech/QtEmbedded-4.6.2-arm/lib/libQtCore.so.4.6.2: _ZN7QThread4execEv+0x60
    #06 sp: 0x31bfee28 ip: 0x2e7b6789  /usr/local/Trolltech/QtEmbedded-4.6.2-arm/lib/libQtCore.so.4.6.2: _ZN7QThread21setTerminationEnabledEb+0x218
    #07 sp: 0x31bfee50 ip: 0x2f9aa473  /lib/libpthread-2.11.1.so: __pthread_get_minstack+0xec6
--- --- --- --- --- --- --- --- --- --- --- --- --- --- --- ---
pid: 711, tid: 1130
 r0 0000001a  r1 34e8d9a0  r2 00000000  r3 00000000
 r4 00000000  r5 34e8da20  r6 32701d40  r7 0000008e
 r8 34e8da20  r9 3271359c  10 00000001  fp 3271359c
 ip 00000000  sp 34e8d900  lr 00000000  pc 2fbaa8d2  cpsr 800f0030
    #00 sp: 0x34e8d900 ip: 0x2fbaa8d2  /lib/libc-2.11.1.so: __select+0x31
    #01 sp: 0x34e8d910 ip: 0x326fe657  /usr/local/Kobo/generic/libkevdevtouch.so: _init+0x49f7
    #02 sp: 0x34e8da48 ip: 0x2e953737  /usr/local/Trolltech/QtEmbedded-4.6.2-arm/lib/libQtCore.so.4.6.2: _ZN15QSocketNotifier5eventEP6QEvent+0x1f2
    #03 sp: 0x34e8da68 ip: 0x2e94b2f9  /usr/local/Trolltech/QtEmbedded-4.6.2-arm/lib/libQtCore.so.4.6.2: _ZN7QObject5eventEP6QEvent+0xd0
[... cut ...]
last user operation[4]: "pluggedIn" 
last user operation[3]: "pluggedInBatte
last user operation[2]: "appStart" 
last user operation[1]: "homeWidgetClic
last user operation[0]: "openContentRel
--- --- --- --- --- --- --- --- --- --- --- --- --- --- --- ---</pre></details>

I'm going to remove this patch for now and take another attempt at it for the next release of the patches.

I've also tried using LD_PRELOAD to trace the calls to `QList<double>::append`, and it didn't give me anything useful...

closed 2020-03-03 by geek1011