pgaskin/kobopatch-patches #49: Dictionary changes on 4.20.14601

geek1011, 2020-03-03:
Continuing from #48.

jackiew1, 2020-03-03:
@geek1011 
Better late than never! What did you have in mind for the dictionary testing you asked about 16 hours ago?

geek1011, 2020-03-03:
Specifically how the extra dictionaries are handled differently without the patch enabled.

jackiew1, 2020-03-03:
I have a completely unpatched 4.20 KA1. Extra-type dictionaries are listed in the dict drop-down selection list, e.g.
```
Extra: cl
Extra: co
```

However, if you try to select one of them it just pops up an errmsg `You don't have an English dictionary installed` with a button to the `Manage dictionaries` settings.

geek1011, 2020-03-03:
I don't recall how, but last night, I was able to select a sideloaded dictionary (I think it was an Extra:_ one) without the patches, but I wasn't able to switch away from it.

jackiew1, 2020-03-03:
I just tried again. I can get a sideloaded -en-en dictionary to lookup words but not an Extra.

As with the Extras, the drop-down list label shows as a truncated `English - English` rather than the actual full label of `English - English (OED_ed3)`

taosxx, 2020-03-04:
> but I wasn't able to switch away from it.

I've seen this a couple of times before (starting with firmware 4.19.x). Happened before and after patching. Issue was always gone after a reboot and re-selecting the other dictionary.

> I just tried again. I can get a sideloaded -en-en dictionary to lookup words but not an Extra.

For me, before patching, all side-loaded dictionaries with "known" language codes (those that are mentioned in libnickel) worked, e.g. -fr-de, -da-en, -it-fr, -fr-fr, -de-de, etc. As expected, they were shown in the drop-down menu with their full names: `Français - Deutsch`, `Dansk - English`, etc. I haven't checked if these dictionaries must be added to the database before they will work (as far as I remember this wasn't necessary in older firmwares).

Side-loaded dictionaries with "extra" languages (those that have to be added in Kobo eReader.conf) were shown as e.g `Extra: la` in the drop-down menu. Going to `Manage dictionaries` after the error message `You don't have an English - English dictionary installed`, for the first time ever, actually opened my -en-en dictionary.... After patching, looking up a word worked, of course. 

BTW, I noticed that if I choose to change my system language to one of the (three in my case) extra languages, the first two extra languages are shown as activated in the language settings menu (I have only one translation file installed). Not sure if this is new.

PescheHelfer, 2020-03-07:
Is your new version already based on the newest firmware (4.20.14601)? After installing the newest firmware on my KoboGlo HD, there is no more Dictionary table in KoboReader.sqlite :(

geek1011, 2020-03-07:
^ geek1011/dictutil#3

geek1011, 2020-03-08:
While looking into geek1011/dictutul#3, I found something new!

It seems that you don't need funny tricks to make nickel preserve overwritten dictionaries anymore. Now, all you need to do is mark it read only!

I'll be updating dictutil with a case for this if the firmware version is 4.20.14601+.

geek1011, 2020-03-08:
I've also confirmed geek1011/dictutil#3 is valid! Firmware 4.20.14601 doesn't create the Dictionary table anymore. I guess there wasn't a migration (or it was broken), and that's why devices upgrading from an older version still have that table.

That would also explain the unusual behaviour with the `Extra: ` stuff. Here's my theory (I'll test it later today):

- **Same:** Nickel still generates locale names by default with `Extra: LOCALE`.
- **New:** Nickel doesn't read the dictionary table anymore, so the name in it is ignored. In addition, entries in the table won't change anything even if it is still present.
- **Same:** Nickel still has the bug where the locale splitting is messed up, so the `Extra: LOCALE` names are inherently broken.
- **Same:** The matching can be fixed by replacing `Extra: ` with `Extra:_`.
- **New:** The database doesn't need to be changed anymore in addition to the patch, as the names are generated dynamically using the same string.
- **Therefore:** The steps required to install custom dictionaries are now:
  - Copy the dictzip and mark it read-only.
  - Add it to ExtraLocales if it is not a built-in locale.
  - Replace `Extra: ` in libnickel with any other string (same length or shorter with a null byte at the end), but not containing anything in the Unicode whitespace class.

I'll write some stuff about this for the dictutil docs after I test it.

@jackiew1, it would be nice if you could confirm this as well.

geek1011, 2020-03-08:
And here are the logs (I've replaced `dicthtml.zip` and added `dicthtml-xx.zip`, but made both read-only):

```
Mar  8 14:45:46 nickel: (  2518.512 @ 0x22514b0 / sync.debug) =================== SYNC QUEUE [ SyncClient(0x1836b78) ] ==================== 
Mar  8 14:45:46 nickel: (  2518.512 @ 0x22514b0 / sync.debug)  ----- Queued ------- 
Mar  8 14:45:46 nickel: (  2518.512 @ 0x22514b0 / sync.debug)  [ SyncDictionariesCommand(0x309018a0) ] 
Mar  8 14:45:46 nickel: (  2518.512 @ 0x22514b0 / sync.debug)  [ FrictionlessReadingCommand(0x30903500) ] 
Mar  8 14:45:46 nickel: (  2518.512 @ 0x22514b0 / sync.debug)  [ DownloadQueuedBooksCommand(0x30904720) ] 
Mar  8 14:45:46 nickel: (  2518.512 @ 0x22514b0 / sync.debug)  [ CleanupCommand(0x3092a0b8) ] 
Mar  8 14:45:46 nickel: (  2518.513 @ 0x22514b0 / sync.debug) =================================================== 
Mar  8 14:45:46 nickel: (  2518.513 @ 0x22514b0 / sync.debug)    
Mar  8 14:45:46 nickel: (  2518.513 @ 0x22514b0 / sync.debug) pumping with 4 pending commands and 0 active commands 
Mar  8 14:45:46 nickel: (  2518.513 @ 0x22514b0 / sync.debug) executing SyncDictionariesCommand(0x309018a0) 
Mar  8 14:45:46 nickel: (  2518.513 @ 0x22514b0 / sync.debug) virtual void SyncDictionariesCommand::execute() 
Mar  8 14:45:46 nickel: (  2518.730 @ 0x22514b0 / sync.debug) "dicthtml" marked as read-only.. skipping 
Mar  8 14:45:46 nickel: (  2519.101 @ 0x22514b0 / packetdump.warning) "https://kbdownload1-a.akamaihd.net/ereader/dictionaries/v2/dicthtml-xx.zip" => "Error downloading https://kbdownload1-a.akamaihd.net/ereader/dictionaries/v2/dicthtml-xx.zip - server replied: Not Found" 
Mar  8 14:45:46 nickel: (  2519.102 @ 0x22514b0) void FileSizeGetter::onFinished() FAILED! 203 "Error downloading https://kbdownload1-a.akamaihd.net/ereader/dictionaries/v2/dicthtml-xx.zip - server replied: Not Found" 
Mar  8 14:45:46 nickel: (  2519.102 @ 0x22514b0 / sync.warning) void SyncDictionariesCommand::prepareDownloadList() Get size failed:  203 0 
Mar  8 14:45:46 nickel: (  2519.102 @ 0x22514b0 / sync.debug) SyncDictionariesCommand(0x309018a0) finished 
```

It will still attempt to download the sideloaded dictionaries (due to the default value of IsSynced without the database), but it when it gets to replacing them, it won't due to the file permission check.

In addition, I've noticed that instead of having a migration to delete the dictionary table, they just neutered the existing migrations to add the dictionary stuff, for example:

**4.20.14601:**
```asm
                            _ZN39Migration119_UpdateJapaneseDictionaries7migrateERK6DeviceR12QSqlDatabase:        // Migration119_UpdateJapaneseDictionaries::migrate(Device const&, QSqlDatabase&)
00508b4c 80B4                   push       r7
00508b4e 0120                   movs       r0, #0x1
00508b50 00AF                   add        r7, sp, #0x0
00508b52 BD46                   mov        sp, r7
00508b54 5DF8047B               ldr        r7, [sp, saved_fp], #0x4
00508b58 7047                   bx         lr
```

**4.19.14123:**
```asm
                            _ZN39Migration124_UpdateJapaneseDictionaries7migrateERK6DeviceR12QSqlDatabase:        // Migration124_UpdateJapaneseDictionaries::migrate(Device const&, QSqlDatabase&)
004eec08 2DE9F04F               push.w     {r4, r5, r6, r7, r8, sb, sl, fp, lr}
004eec0c 8046                   mov        r8, r0
004eec0e DFF8F005               ldr.w      r0, =0x933096                        ; 0x4ef200,0x933096
004eec12 95B0                   sub        sp, #0x54
004eec14 02AF                   add        r7, sp, #0x8
004eec16 1546                   mov        r5, r2
004eec18 8A46                   mov        sl, r1
004eec1a 7844                   add        r0, pc                               ; 0xe21cb4
004eec1c 0221                   movs       r1, #0x2
004eec1e 07F13C04               add.w      r4, r7, #0x3c
004eec22 A9F70EE9               blx        _ZN7QString17fromLatin1_helperEPKci@PLT ; QString::fromLatin1_helper(char const*, int)
004eec26 2946                   mov        r1, r5
004eec28 B863                   str        r0, [r7, #0x38]
004eec2a 2046                   mov        r0, r4
004eec2c BDF74AEE               blx        _ZNK12QSqlDatabase12databaseNameEv@PLT ; QSqlDatabase::databaseName() const
004eec30 07F13803               add.w      r3, r7, #0x38
004eec34 07F11000               add.w      r0, r7, #0x10
004eec38 1946                   mov        r1, r3
004eec3a 2246                   mov        r2, r4
004eec3c 3860                   str        r0, [r7]
004eec3e BB60                   str        r3, [r7, #0x8]
004eec40 E2F732EA               blx        _ZN17DictionaryManager26getDictionariesForLanguageERK7QStringS2_@PLT ; DictionaryManager::getDictionariesForLanguage(QString const&, QString const&)
004eec44 F86B                   ldr        r0, [r7, #0x3c]
004eec46 0368                   ldr        r3, [r0]
004eec48 002B                   cmp        r3, #0x0
004eec4a 40F08381               bne.w      loc_4eef54
; --- snip ---
```

jackiew1, 2020-03-08:
> @jackiew1, it would be nice if you could confirm this as well.

I've just run an SQLite `DROP TABLE Dictionary` on my KA1's database, then done a full re-boot. The KA1 already had the Extra: space/underscore patch installed.

I can confirm that all my sideloaded custom dictionaries are still working OK without the Dictionary table. Most of them are Extra types but one is a custom dicthtml-en-en.zip. I don't currently have any with the same filename as a built-in Kobo dictionary.

I have to admit I didn't see this one coming. Installing custom dictionaries should be much simpler now. Well done, Kobo ... _or at least it would have been if they'd documented it somewhere_. I wonder if this is just the beginning of some bigger plan?

geek1011, 2020-03-08:
> I have to admit I didn't see this one coming. Installing custom dictionaries should be much simpler now. Well done, Kobo ... or at least it would have been if they'd documented it somewhere. I wonder if this is just the beginning of some bigger plan?

I think it's both a bit of altruism on their part and part of a plan to simplify the whole situation about double-state and inconsistencies regarding the DB, migrations, and in-memory state.

jackiew1, 2020-03-08:
@geek1011 Odd that they didn't remove the need for the Extra: space/underscore patch at the same time. A 1-character fix must be easier than any of the other improvements/fixes they made. 😄

geek1011, 2020-03-08:
@jackiew1 Maybe they thought it would fix that too? It's kind of an obscure bug.

geek1011, 2020-03-08:
I think we've now figured out and tested all of the relevant dictionary changes, so I'm going to close this issue now. Feel free to comment if you have any additional things, though.

In addition, I've released dictutil v0.2.1 based on the information here.

closed 2020-03-08 by geek1011

mzelyony, 2020-03-23:
Thanks @geek1011, @jackiew1 
I went through the instructions and installed the sideloaded dictionary on 4.20.14622 and it seems to work fine. There are two things which concern me:
1) In Languages and Dictionaries it shows a warning `1 dictionary download pending`
2) Despite the XXXX.zip being marked readonly it was still deleted when I unchecked it in the settings and hit Save

Thanks!

geek1011, 2020-03-23:
1. That's normal, it's just a side-effect of how it's implemented.
2. You shouldn't uncheck it. Being read-only doesn't stop it from being deleted/uninstalled, only overwritten during the automated sync.

mzelyony, 2020-03-23:
Ok, thanks

kaveheng, 2020-04-24:
> I've also confirmed [geek1011/dictutil#3](https://github.com/geek1011/dictutil/issues/3) is valid! Firmware 4.20.14601 doesn't create the Dictionary table anymore. I guess there wasn't a migration (or it was broken), and that's why devices upgrading from an older version still have that table.
> 
> That would also explain the unusual behaviour with the `Extra: ` stuff. Here's my theory (I'll test it later today):
> 
> * **Same:** Nickel still generates locale names by default with `Extra: LOCALE`.
> * **New:** Nickel doesn't read the dictionary table anymore, so the name in it is ignored. In addition, entries in the table won't change anything even if it is still present.
> * **New:** The built-in dictionaries are hard-coded, rather than writing them to the db during migrations and reading from it at runtime.
> * **Same:** Nickel still has the bug where the locale splitting is messed up, so the `Extra: LOCALE` names are inherently broken.
> * **Same:** The matching can be fixed by replacing `Extra: ` with `Extra:_`.
> * **New:** The database doesn't need to be changed anymore in addition to the patch, as the names are generated dynamically using the same string.
> * **Therefore:** If the dictionary table is present, it can safely be removed.
> * **Therefore:** The steps required to install custom dictionaries are now:
>   
>   * Copy the dictzip and mark it read-only.
>   * Add it to ExtraLocales if it is not a built-in locale.
>   * Replace `Extra: ` in libnickel with any other string (same length or shorter with a null byte at the end), but not containing anything in the Unicode whitespace class.
> 
> I'll write some stuff about this for the dictutil docs after I test it.
> 
> @jackiew1, it would be nice if you could confirm this as well.

How should I copy and paste the dictzip file as read only? when I paste the file in dict folder it changes to read and write

geek1011, 2020-05-05:
> How should I copy and paste the dictzip file as read only? when I paste the file in dict folder it changes to read and write

You should be able to do this from the file's properties dialog. If not, try doing it over SSH or telnet and see if that sticks.