pgaskin/kobopatch-patches #50: My 24 line spacing values patch for 4.20.14601

geek1011, 2020-03-03:
Continuing from https://github.com/geek1011/kobopatch-patches/issues/48#issuecomment-593753752.

NiLuJe, 2020-03-03:
I assume the relevant bit of the stacktrace is the main thread? (every other thread seems to be doing its own stuff in a `select`, which sounds normal).

That appears to be a string alloc when demangled?

`std::basic_string<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> >::basic_string<wchar_t const*>(wchar_t const*, wchar_t const*, std::allocator<wchar_t> const&)`

geek1011, 2020-03-03:
Well, the thing is, I suspect the stack trace doesn't even contain any relevant info (where's all the libnickel stuff which is usually in the good ones, for one). I've had issues with the stack trace being outdated before too.

The most useful thing would be if you could try the patch (see the commit which removed it) and see if you can figure out anything useful from it (it will crash when you open any book [including non-epub/kepubs], but not before, so it's relatively safe to test).

NiLuJe, 2020-03-03:
Would capturing an actual coredump instead of whatever black magic (libsegfault?) they're doing help?

geek1011, 2020-03-03:
Yes, certainly.

NiLuJe, 2020-03-03:
I'll try that later this evening then ;). (8AM here).

NiLuJe, 2020-03-03:
Okay, here goes:

```
#0  0x2f9f7928 in vtable for std::moneypunct<char, true> () from /lib/libstdc++.so.6
[Current thread is 1 (LWP 1652)]
(gdb) bt full
#0  0x2f9f7928 in vtable for std::moneypunct<char, true> () from /lib/libstdc++.so.6
No symbol table info available.
#1  0x2b5223b2 in ReadingSettings::lineHeightScalars() const () from /usr/local/Kobo/libnickel.so.1
No symbol table info available.
#2  0x2b522f84 in ReadingSettings::getReadingLineHeight(ContentSettings&, Volume const&) () from /usr/local/Kobo/libnickel.so.1
No symbol table info available.
#3  0x2b90d4e2 in KepubBookReader::pageStyleCss(bool) () from /usr/local/Kobo/libnickel.so.1
No symbol table info available.
#4  0x2b62dfce in KepubBookReaderBase::updateHtmlCss() () from /usr/local/Kobo/libnickel.so.1
No symbol table info available.
#5  0x2b634614 in ?? () from /usr/local/Kobo/libnickel.so.1
No symbol table info available.
#6  0x2b62ebde in ?? () from /usr/local/Kobo/libnickel.so.1
No symbol table info available.
#7  0x2b63063c in KepubBookReaderBase::startChapterLoad(Shortcover const&) () from /usr/local/Kobo/libnickel.so.1
No symbol table info available.
#8  0x2b634ec8 in KepubBookReaderBase::renderShortcover(Shortcover const&, ReadingNavigationType) () from /usr/local/Kobo/libnickel.so.1
No symbol table info available.
#9  0x2b90e1bc in KepubBookReader::renderShortcover(Shortcover const&, ReadingNavigationType) () from /usr/local/Kobo/libnickel.so.1
No symbol table info available.
#10 0x2b632f82 in KepubBookReaderBase::renderVolumeWithLocalBookmark(Volume const&, ReadingNavigationType) () from /usr/local/Kobo/libnickel.so.1
No symbol table info available.
#11 0x2b6331aa in KepubBookReaderBase::renderVolume(Volume const&) () from /usr/local/Kobo/libnickel.so.1
No symbol table info available.
--Type <RET> for more, q to quit, c to continue without paging--
#12 0x2baf154c in ?? () from /usr/local/Kobo/libnickel.so.1
No symbol table info available.
#13 0x2e8a4816 in QMetaObject::activate(QObject*, int, int, void**) () from /usr/local/Qt-5.2.1-arm/lib/libQt5Core.so.5
No symbol table info available.
#14 0x2bb64b9a in ReadingView::renderVolume(Volume const&) () from /usr/local/Kobo/libnickel.so.1
No symbol table info available.
#15 0x2b9064b8 in ReadingView::setVolume(Volume const&, Bookmark const&) () from /usr/local/Kobo/libnickel.so.1
No symbol table info available.
#16 0x2b8f9c2a in ReadingController::setVolume(Volume const&, Bookmark const&) () from /usr/local/Kobo/libnickel.so.1
No symbol table info available.
#17 0x2b9277f2 in N3ReaderOpener::openReader(Bookmark const&) () from /usr/local/Kobo/libnickel.so.1
No symbol table info available.
#18 0x2b620d8c in AbstractReadingViewMixin::showReader(Volume const&, QVector<ContentKey> const&, AbstractLibraryController<Volume>*, Bookmark const&) () from /usr/local/Kobo/libnickel.so.1
No symbol table info available.
#19 0x2b6210b4 in AbstractReadingViewMixin::showReader(Volume const&, AbstractLibraryController<Volume>*, Bookmark const&) () from /usr/local/Kobo/libnickel.so.1
No symbol table info available.
#20 0x2b9275a4 in ReadingViewMixin::showReaderOrInfo(Volume const&, AbstractLibraryController<Volume>*, Bookmark const&) () from /usr/local/Kobo/libnickel.so.1
No symbol table info available.
#21 0x2b7c70ea in SingleBookHomeWidget::tapped() () from /usr/local/Kobo/libnickel.so.1
No symbol table info available.
#22 0x2b7ca7c2 in TappableHomePageWidget::tapGesture(GestureReceiver*, TapGesture*) () from /usr/local/Kobo/libnickel.so.1
No symbol table info available.
#23 0x2b5f5942 in GestureReceiver::sendGestureToDelegate(QGestureEvent*, GestureDelegate*) () from /usr/local/Kobo/libnickel.so.1
No symbol table info available.
--Type <RET> for more, q to quit, c to continue without paging--
#24 0x2dc3fb62 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/local/Qt-5.2.1-arm/lib/libQt5Widgets.so.5
No symbol table info available.
#25 0x2dc45568 in QApplication::notify(QObject*, QEvent*) () from /usr/local/Qt-5.2.1-arm/lib/libQt5Widgets.so.5
No symbol table info available.
#26 0x2b8895d6 in Nickel3Application::notify(QObject*, QEvent*) () from /usr/local/Kobo/libnickel.so.1
No symbol table info available.
#27 0x2e883728 in QCoreApplication::notifyInternal(QObject*, QEvent*) () from /usr/local/Qt-5.2.1-arm/lib/libQt5Core.so.5
No symbol table info available.
#28 0x2dc7b3b4 in ?? () from /usr/local/Qt-5.2.1-arm/lib/libQt5Widgets.so.5
No symbol table info available.
#29 0x2dc7c1f6 in ?? () from /usr/local/Qt-5.2.1-arm/lib/libQt5Widgets.so.5
No symbol table info available.
#30 0x2dc7cad4 in ?? () from /usr/local/Qt-5.2.1-arm/lib/libQt5Widgets.so.5
No symbol table info available.
[... 14 lines cut ...]
#37 0x2dc4499a in QApplication::notify(QObject*, QEvent*) () from /usr/local/Qt-5.2.1-arm/lib/libQt5Widgets.so.5
No symbol table info available.
#38 0x2b8895d6 in Nickel3Application::notify(QObject*, QEvent*) () from /usr/local/Kobo/libnickel.so.1
No symbol table info available.
#39 0x2e883728 in QCoreApplication::notifyInternal(QObject*, QEvent*) () from /usr/local/Qt-5.2.1-arm/lib/libQt5Core.so.5
No symbol table info available.
#40 0x2e40c4ca in QGuiApplicationPrivate::processTouchEvent(QWindowSystemInterfacePrivate::TouchEvent*) () from /usr/local/Qt-5.2.1-arm/lib/libQt5Gui.so.5
No symbol table info available.
#41 0x2e40e148 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) () from /usr/local/Qt-5.2.1-arm/lib/libQt5Gui.so.5
[... cut ...]
#44 0x2e88206a in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/local/Qt-5.2.1-arm/lib/libQt5Core.so.5
No symbol table info available.
#45 0x2e886f48 in QCoreApplication::exec() () from /usr/local/Qt-5.2.1-arm/lib/libQt5Core.so.5
No symbol table info available.
#46 0x0001e01a in main ()
No symbol table info available.
```

I *hope* that's it, because writing the 153MB core dump took a while, and then the device rebooted on its own :?.

(I wasn't following syslogs at the time :s)

NiLuJe, 2020-03-03:
(I mean, at least that sounds on point ;p).

I can provide the coredump if you want to play some more with it? (In a few hours, when I actually have a real internet connection ;p).

geek1011, 2020-03-03:
That's a lot more useful. So somehow, it ended up referencing the moneypunct vtable. I did go through the pointer math manually a few times, so I didn't expect something like that to happen.

Also, if you can get a gdb disassembly at that point for that layer and one above (10 or so instructions is fine), that would greatly aid debugging it.

NiLuJe, 2020-03-03:
Something like that? (I suck at gdb ;p).

```
(gdb) disassemble
Dump of assembler code for function _ZTVSt10moneypunctIcLb1EE:
=> 0x2f9f7928 <+0>:	andeq	r0, r0, r0
   0x2f9f792c <+4>:	svccs	0x009f7cb0
   0x2f9f7930 <+8>:	svccs	0x009a5bc9
   0x2f9f7934 <+12>:	svccs	0x009a5c4d
   0x2f9f7938 <+16>:	svccs	0x009aca75
   0x2f9f793c <+20>:	svccs	0x009aca7d
   0x2f9f7940 <+24>:	svccs	0x009accfd
   0x2f9f7944 <+28>:	svccs	0x009acd15
   0x2f9f7948 <+32>:	svccs	0x009acd2d
   0x2f9f794c <+36>:	svccs	0x009acd45
   0x2f9f7950 <+40>:	svccs	0x009aca85
   0x2f9f7954 <+44>:	svccs	0x009aca8d
   0x2f9f7958 <+48>:	svccs	0x009acabd
End of assembler dump.
(gdb) disassemble 0x2b5223b2
Dump of assembler code for function _ZNK15ReadingSettings17lineHeightScalarsEv:
   0x2b522240 <+0>:	ldr	r3, [pc, #604]	; (0x2b5224a0 <_ZNK15ReadingSettings17lineHeightScalarsEv+608>)
   0x2b522242 <+2>:	stmdb	sp!, {r4, r5, r6, r7, r8, r9, lr}
   0x2b522246 <+6>:	sub	sp, #132	; 0x84
   0x2b522248 <+8>:	ldr	r6, [pc, #600]	; (0x2b5224a4 <_ZNK15ReadingSettings17lineHeightScalarsEv+612>)
   0x2b52224a <+10>:	add	r3, pc
   0x2b52224c <+12>:	add	r7, sp, #0
   0x2b52224e <+14>:	ldr	r2, [r3, #0]
   0x2b522250 <+16>:	mov	r5, r0
   0x2b522252 <+18>:	add	r6, pc
   0x2b522254 <+20>:	lsls	r2, r2, #31
   0x2b522256 <+22>:	bmi.n	0x2b522260 <_ZNK15ReadingSettings17lineHeightScalarsEv+32>
   0x2b522258 <+24>:	mov	r0, r3
   0x2b52225a <+26>:	blx	0x2b2c38b4 <__cxa_guard_acquire@plt>
   0x2b52225e <+30>:	cbnz	r0, 0x2b522276 <_ZNK15ReadingSettings17lineHeightScalarsEv+54>
   0x2b522260 <+32>:	ldr	r1, [pc, #580]	; (0x2b5224a8 <_ZNK15ReadingSettings17lineHeightScalarsEv+616>)
   0x2b522262 <+34>:	mov	r0, r5
   0x2b522264 <+36>:	add	r1, pc
   0x2b522266 <+38>:	adds	r1, #4
   0x2b522268 <+40>:	blx	0x2b2b1d14 <_ZN5QListIdEC1ERKS0_@plt>
   0x2b52226c <+44>:	mov	r0, r5
   0x2b52226e <+46>:	adds	r7, #132	; 0x84
   0x2b522270 <+48>:	mov	sp, r7
   0x2b522272 <+50>:	ldmia.w	sp!, {r4, r5, r6, r7, r8, r9, pc}
--Type <RET> for more, q to quit, c to continue without paging--
   0x2b522276 <+54>:	ldr	r3, [pc, #564]	; (0x2b5224ac <_ZNK15ReadingSettings17lineHeightScalarsEv+620>)
   0x2b522278 <+56>:	mov.w	r9, #0
   0x2b52227c <+60>:	add.w	r1, r7, #128	; 0x80
   0x2b522280 <+64>:	mov.w	r8, #0
   0x2b522284 <+68>:	movt	r9, #16368	; 0x3ff0
   0x2b522288 <+72>:	adds	r4, r7, #4
   0x2b52228a <+74>:	ldr	r3, [r6, r3]
   0x2b52228c <+76>:	mov	r0, r4
   0x2b52228e <+78>:	strd	r8, r9, [r1, #-120]!	; 0x78
   0x2b522292 <+82>:	str	r3, [r7, #4]
   0x2b522294 <+84>:	mov	r9, r5
   0x2b522296 <+86>:	movs	r5, #0
   0x2b522298 <+88>:	lsls	r5, r5, #3
   0x2b52229a <+90>:	mov	r0, r4
   0x2b52229c <+92>:	add	r1, pc, #16	; (adr r1, 0x2b5222b0 <_ZNK15ReadingSettings17lineHeightScalarsEv+112>)
   0x2b52229e <+94>:	add	r1, r5
   0x2b5222a0 <+96>:	blx	0x2b2bd368 <_ZN5QListIdE6appendERKd@plt>
   0x2b5222a4 <+100>:	adds	r5, #8
   0x2b5222a6 <+102>:	cmp	r5, #112	; 0x70
   0x2b5222a8 <+104>:	bne.n	0x2b52229a <_ZNK15ReadingSettings17lineHeightScalarsEv+90>
   0x2b5222aa <+106>:	b.n	0x2b5223a6 <_ZNK15ReadingSettings17lineHeightScalarsEv+358>
   0x2b5222ac <+108>:	nop
   0x2b5222ae <+110>:	nop
   0x2b5222b0 <+112>:	movs	r0, r0
--Type <RET> for more, q to quit, c to continue without paging--
[... 195 lines cut ...]
   0x2b522496 <+598>:	subs	r7, #246	; 0xf6
   0x2b522498 <+600>:	movs	r0, r0
   0x2b52249a <+602>:	movs	r0, r0
   0x2b52249c <+604>:	movs	r0, r0
--Type <RET> for more, q to quit, c to continue without paging--
   0x2b52249e <+606>:	subs	r7, #248	; 0xf8
   0x2b5224a0 <+608>:	stmia	r5!, {r1, r6}
   0x2b5224a2 <+610>:	lsls	r1, r4, #2
   0x2b5224a4 <+612>:			; <UNDEFINED> instruction: 0xfa9a009d
   0x2b5224a8 <+616>:	stmia	r5!, {r3, r5}
   0x2b5224aa <+618>:	lsls	r1, r4, #2
   0x2b5224ac <+620>:	cmp	r0, #240	; 0xf0
   0x2b5224ae <+622>:	movs	r1, r0
   0x2b5224b0 <+624>:	stmia	r3!, {r1, r2, r3, r6, r7}
   0x2b5224b2 <+626>:	lsls	r1, r4, #2
   0x2b5224b4 <+628>:	stmia	r3!, {r2, r6, r7}
   0x2b5224b6 <+630>:	lsls	r1, r4, #2
   0x2b5224b8 <+632>:	bvs.n	0x2b52258c <_ZN15ReadingSettings30getLastReadingLineHeightScalarEv+200>
   0x2b5224ba <+634>:	movs	r1, r0
   0x2b5224bc <+636>:	beq.n	0x2b5224e8 <_ZN15ReadingSettings30getLastReadingLineHeightScalarEv+36>
   0x2b5224be <+638>:	lsls	r7, r3, #2
   0x2b5224c0 <+640>:	stmia	r3!, {r2, r3, r6}
   0x2b5224c2 <+642>:	lsls	r1, r4, #2
End of assembler dump.
```

NiLuJe, 2020-03-03:
And before that:

```
(gdb) disas 0x2b522f84
Dump of assembler code for function _ZN15ReadingSettings20getReadingLineHeightER15ContentSettingsRK6Volume:
   0x2b522f50 <+0>:	stmdb	sp!, {r4, r5, r6, r7, r8, r9, r10, r11, lr}
   0x2b522f54 <+4>:	mov	r4, r2
   0x2b522f56 <+6>:	vpush	{d8-d10}
   0x2b522f5a <+10>:	sub	sp, #28
   0x2b522f5c <+12>:	mov	r5, r0
   0x2b522f5e <+14>:	mov	r8, r1
   0x2b522f60 <+16>:	add	r7, sp, #0
   0x2b522f62 <+18>:	blx	0x2b2b207c <_ZN15ReadingSettings26getReadingLineHeightScalarERK15ContentSettings@plt>
   0x2b522f66 <+22>:	add.w	r6, r7, #8
   0x2b522f6a <+26>:	mov	r2, r4
   0x2b522f6c <+28>:	mov	r1, r8
   0x2b522f6e <+30>:	mov	r0, r5
   0x2b522f70 <+32>:	vmov.f64	d8, d0
   0x2b522f74 <+36>:	blx	0x2b2ac8ac <_ZN15ReadingSettings21calcMinimumLineHeightERK15ContentSettingsRK6Volume@plt>
   0x2b522f78 <+40>:	mov	r1, r5
   0x2b522f7a <+42>:	mov	r0, r6
   0x2b522f7c <+44>:	vmov.f64	d9, d0
   0x2b522f80 <+48>:	blx	0x2b2cb978 <_ZNK15ReadingSettings17lineHeightScalarsEv@plt>
   0x2b522f84 <+52>:	ldr.w	r10, [r7, #8]
   0x2b522f88 <+56>:	ldr.w	r3, [r10, #12]
   0x2b522f8c <+60>:	ldr.w	r4, [r10, #8]
   0x2b522f90 <+64>:	add.w	r9, r3, #4
   0x2b522f94 <+68>:	add.w	r9, r10, r9, lsl #2
--Type <RET> for more, q to quit, c to continue without paging--c
   0x2b522f98 <+72>:	adds	r4, #4
   0x2b522f9a <+74>:	add.w	r4, r10, r4, lsl #2
   0x2b522f9e <+78>:	mov	r3, r9
   0x2b522fa0 <+80>:	b.n	0x2b522fb4 <_ZN15ReadingSettings20getReadingLineHeightER15ContentSettingsRK6Volume+100>
   0x2b522fa2 <+82>:	ldr.w	r2, [r3, #-4]!
   0x2b522fa6 <+86>:	vldr	d16, [r2]
   0x2b522faa <+90>:	vcmp.f64	d8, d16
   0x2b522fae <+94>:	vmrs	APSR_nzcv, fpscr
   0x2b522fb2 <+98>:	beq.n	0x2b523022 <_ZN15ReadingSettings20getReadingLineHeightER15ContentSettingsRK6Volume+210>
   0x2b522fb4 <+100>:	cmp	r4, r3
   0x2b522fb6 <+102>:	bne.n	0x2b522fa2 <_ZN15ReadingSettings20getReadingLineHeightER15ContentSettingsRK6Volume+82>
   0x2b522fb8 <+104>:	mov.w	r11, #0
   0x2b522fbc <+108>:	ldr.w	r3, [r10]
   0x2b522fc0 <+112>:	cbnz	r3, 0x2b522fee <_ZN15ReadingSettings20getReadingLineHeightER15ContentSettingsRK6Volume+158>
   0x2b522fc2 <+114>:	cmp	r4, r9
   0x2b522fc4 <+116>:	beq.n	0x2b522fd2 <_ZN15ReadingSettings20getReadingLineHeightER15ContentSettingsRK6Volume+130>
   0x2b522fc6 <+118>:	ldr.w	r0, [r9, #-4]!
   0x2b522fca <+122>:	blx	0x2b2aa76c <_ZdlPv@plt+4>
   0x2b522fce <+126>:	cmp	r9, r4
   0x2b522fd0 <+128>:	bne.n	0x2b522fc6 <_ZN15ReadingSettings20getReadingLineHeightER15ContentSettingsRK6Volume+118>
   0x2b522fd2 <+130>:	mov	r0, r10
   0x2b522fd4 <+132>:	blx	0x2b2b33ec <_ZN9QListData7disposeEPNS_4DataE@plt+4>
   0x2b522fd8 <+136>:	cmp.w	r11, #0
   0x2b522fdc <+140>:	beq.n	0x2b523028 <_ZN15ReadingSettings20getReadingLineHeightER15ContentSettingsRK6Volume+216>
   0x2b522fde <+142>:	vmul.f64	d0, d9, d8
   0x2b522fe2 <+146>:	adds	r7, #28
   0x2b522fe4 <+148>:	mov	sp, r7
   0x2b522fe6 <+150>:	vpop	{d8-d10}
   0x2b522fea <+154>:	ldmia.w	sp!, {r4, r5, r6, r7, r8, r9, r10, r11, pc}
   0x2b522fee <+158>:	adds	r3, #1
   0x2b522ff0 <+160>:	beq.n	0x2b522fd8 <_ZN15ReadingSettings20getReadingLineHeightER15ContentSettingsRK6Volume+136>
   0x2b522ff2 <+162>:	ldrex	r3, [r10]
   0x2b522ff6 <+166>:	sub.w	r3, r3, #1
   0x2b522ffa <+170>:	strex	r2, r3, [r10]
   0x2b522ffe <+174>:	teq	r2, #0
   0x2b523002 <+178>:	bne.n	0x2b522ff2 <_ZN15ReadingSettings20getReadingLineHeightER15ContentSettingsRK6Volume+162>
   0x2b523004 <+180>:	cmp	r3, #0
   0x2b523006 <+182>:	bne.n	0x2b522fd8 <_ZN15ReadingSettings20getReadingLineHeightER15ContentSettingsRK6Volume+136>
   0x2b523008 <+184>:	ldr.w	r10, [r7, #8]
   0x2b52300c <+188>:	ldr.w	r2, [r10, #8]
   0x2b523010 <+192>:	ldr.w	r3, [r10, #12]
[... 141 lines cut ...]
   0x2b52319e <+590>:	bic.w	r3, r3, #3
   0x2b5231a2 <+594>:	adds	r3, #4
   0x2b5231a4 <+596>:	add	r3, r0
   0x2b5231a6 <+598>:	str	r3, [r7, #16]
   0x2b5231a8 <+600>:	b.n	0x2b5230ca <_ZN15ReadingSettings20getReadingLineHeightER15ContentSettingsRK6Volume+378>
   0x2b5231aa <+602>:	adds	r1, #1
   0x2b5231ac <+604>:	beq.n	0x2b523154 <_ZN15ReadingSettings20getReadingLineHeightER15ContentSettingsRK6Volume+516>
   0x2b5231ae <+606>:	ldrex	r3, [r6]
   0x2b5231b2 <+610>:	sub.w	r3, r3, #1
   0x2b5231b6 <+614>:	strex	r2, r3, [r6]
   0x2b5231ba <+618>:	teq	r2, #0
   0x2b5231be <+622>:	bne.n	0x2b5231ae <_ZN15ReadingSettings20getReadingLineHeightER15ContentSettingsRK6Volume+606>
   0x2b5231c0 <+624>:	cmp	r3, #0
   0x2b5231c2 <+626>:	bne.w	0x2b5230c2 <_ZN15ReadingSettings20getReadingLineHeightER15ContentSettingsRK6Volume+370>
   0x2b5231c6 <+630>:	ldr	r6, [r7, #4]
   0x2b5231c8 <+632>:	b.n	0x2b52309c <_ZN15ReadingSettings20getReadingLineHeightER15ContentSettingsRK6Volume+332>
   0x2b5231ca <+634>:	mov	r0, r4
   0x2b5231cc <+636>:	blx	0x2b2d9948 <_ZN5QListIdED1Ev@plt>
   0x2b5231d0 <+640>:	blx	0x2b2b6420 <__cxa_end_cleanup@plt>
   0x2b5231d4 <+644>:	nop.w
   0x2b5231d8 <+648>:	movs	r0, r0
   0x2b5231da <+650>:			; <UNDEFINED> instruction: 0xffc0ffff
   0x2b5231de <+654>:	rors	r7, r3
End of assembler dump.
```

shermp, 2020-03-03:
Um... should Github be referencing issues and PR's inside code blocks?

geek1011, 2020-03-03:
Um... no it shouldn't... and the markdown parser's been acting quite wonky recently. It's been parsing markdown inside HTML elements, which it didn't use to do.

geek1011, 2020-03-03:
And @NiLuJe: that's quite helpful. I'll look at it in depth later.

geek1011, 2020-03-03:
Can you extra the base address of libnickel's memory map from the coredump?

NiLuJe, 2020-03-03:
@geek1011: How would I go about doing that?

geek1011, 2020-03-03:
@NiLuJe: Actually, it seems that isn't possible without a running process. I'll just have to calculate it from the known pointers, then.

NiLuJe, 2020-03-03:
Does this excerpt from `info files` fit the bill?

```
        0x2f94b134 - 0x2f94b158 is .note.gnu.build-id in /lib/libstdc++.so.6
        0x2f94b158 - 0x2f94fdd4 is .hash in /lib/libstdc++.so.6
        0x2f94fdd4 - 0x2f95f004 is .dynsym in /lib/libstdc++.so.6
        0x2f95f004 - 0x2f986251 is .dynstr in /lib/libstdc++.so.6
        0x2f986252 - 0x2f988098 is .gnu.version in /lib/libstdc++.so.6
        0x2f988098 - 0x2f988514 is .gnu.version_d in /lib/libstdc++.so.6
        0x2f988514 - 0x2f9885c4 is .gnu.version_r in /lib/libstdc++.so.6
        0x2f9885c4 - 0x2f98d7bc is .rel.dyn in /lib/libstdc++.so.6
        0x2f98d7bc - 0x2f98ed3c is .rel.plt in /lib/libstdc++.so.6
        0x2f98ed3c - 0x2f98ed48 is .init in /lib/libstdc++.so.6
        0x2f98ed48 - 0x2f99103c is .plt in /lib/libstdc++.so.6
        0x2f991040 - 0x2f9d84b4 is .text in /lib/libstdc++.so.6
        0x2f9d84b4 - 0x2f9d84bc is .fini in /lib/libstdc++.so.6
        0x2f9d84bc - 0x2f9dd1a8 is .rodata in /lib/libstdc++.so.6
        0x2f9dd1a8 - 0x2f9e2840 is .ARM.extab in /lib/libstdc++.so.6
        0x2f9e2840 - 0x2f9e53e8 is .ARM.exidx in /lib/libstdc++.so.6
        0x2f9e53e8 - 0x2f9e53ec is .eh_frame in /lib/libstdc++.so.6
        0x2f9f5f14 - 0x2f9f5f28 is .tbss in /lib/libstdc++.so.6
        0x2f9f5f14 - 0x2f9f5f34 is .init_array in /lib/libstdc++.so.6
        0x2f9f5f34 - 0x2f9f5f38 is .fini_array in /lib/libstdc++.so.6
        0x2f9f5f38 - 0x2f9f5f3c is .jcr in /lib/libstdc++.so.6
        0x2f9f5f40 - 0x2f9f8ef0 is .data.rel.ro in /lib/libstdc++.so.6
        0x2f9f8ef0 - 0x2f9f9000 is .dynamic in /lib/libstdc++.so.6
        0x2f9f9000 - 0x2f9fa1e8 is .got in /lib/libstdc++.so.6
        0x2f9fa1e8 - 0x2f9fa350 is .data in /lib/libstdc++.so.6
        0x2f9fa350 - 0x2fa00ee0 is .bss in /lib/libstdc++.so.6
```

```
        0x2aded0f4 - 0x2aded118 is .note.gnu.build-id in /usr/local/Kobo/libnickel.so.1
        0x2aded118 - 0x2ae303c8 is .hash in /usr/local/Kobo/libnickel.so.1
        0x2ae303c8 - 0x2af05098 is .dynsym in /usr/local/Kobo/libnickel.so.1
        0x2af05098 - 0x2b19c229 is .dynstr in /usr/local/Kobo/libnickel.so.1
        0x2b19c22a - 0x2b1b6bc4 is .gnu.version in /usr/local/Kobo/libnickel.so.1
        0x2b1b6bc4 - 0x2b1b6d94 is .gnu.version_r in /usr/local/Kobo/libnickel.so.1
        0x2b1b6d94 - 0x2b2799ec is .rel.dyn in /usr/local/Kobo/libnickel.so.1
        0x2b2799ec - 0x2b29ea7c is .rel.plt in /usr/local/Kobo/libnickel.so.1
        0x2b29ea7c - 0x2b29ea88 is .init in /usr/local/Kobo/libnickel.so.1
        0x2b29ea88 - 0x2b2d9c10 is .plt in /usr/local/Kobo/libnickel.so.1
        0x2b2d9c10 - 0x2bc2f104 is .text in /usr/local/Kobo/libnickel.so.1
        0x2bc2f104 - 0x2bc2f10c is .fini in /usr/local/Kobo/libnickel.so.1
        0x2bc2f110 - 0x2bd6c180 is .rodata in /usr/local/Kobo/libnickel.so.1
        0x2bd6c180 - 0x2be4d0c4 is .ARM.extab in /usr/local/Kobo/libnickel.so.1
        0x2be4d0c4 - 0x2be8b73c is .ARM.exidx in /usr/local/Kobo/libnickel.so.1
        0x2be8b73c - 0x2be8b740 is .eh_frame in /usr/local/Kobo/libnickel.so.1
        0x2be9b740 - 0x2be9bccc is .init_array in /usr/local/Kobo/libnickel.so.1
        0x2be9bccc - 0x2be9bcd0 is .fini_array in /usr/local/Kobo/libnickel.so.1
        0x2be9bcd0 - 0x2be9bcd4 is .jcr in /usr/local/Kobo/libnickel.so.1
        0x2be9bcd8 - 0x2bf01ac0 is .data.rel.ro in /usr/local/Kobo/libnickel.so.1
        0x2bf01ac0 - 0x2bf01cf0 is .dynamic in /usr/local/Kobo/libnickel.so.1
        0x2bf01cf0 - 0x2bf1f3f0 is .got in /usr/local/Kobo/libnickel.so.1
        0x2bf1f3f0 - 0x2bf3d188 is .data in /usr/local/Kobo/libnickel.so.1
        0x2bf3d188 - 0x2bf3fdec is .bss in /usr/local/Kobo/libnickel.so.1
```

NiLuJe, 2020-03-03:
(FWIW, I'm back home, if you want the full coredump).

geek1011, 2020-03-03:
OK, that would do the trick. I don't think I'll need the full coredump, but keep it just in case.

geek1011, 2020-03-13:
This patch will probably never be released for 4.20.14601 (4.20.14617 was just released, so I'll most likely be doing it for that instead).

NiLuJe, 2020-03-13:
Ping me when you're ready if you need a new backtrace ;).

geek1011, 2020-03-14:
Note to self:

- ASLR offset for libnickel (14601) in backtrace: 0x2b522240-0x735240 = 0x2ADED000)
- Bad inst offset: 0x2b5223b2-0x2ADED000 = 0x7353B2
  - End of 3rd replacement in patch

geek1011, 2020-03-14:
Might be possible that I just made a typo in an offset (and did it multiple times), as everything seems pretty much identical between working/non-working versions:

4.19.14123:

<details><pre>14123

                            _ZNK15ReadingSettings17lineHeightScalarsEv:        // ReadingSettings::lineHeightScalars() const
0071fc08 974B                   ldr        r3, =0x9e3a5a                        ; dword_71fe68,0x9e3a5a
0071fc0a 2DE9F043               push.w     {r4, r5, r6, r7, r8, sb, lr}
0071fc0e A1B0                   sub        sp, #0x84
0071fc10 964E                   ldr        r6, =0x9a7622                        ; dword_71fe6c,0x9a7622
0071fc12 7B44                   add        r3, pc                               ; dword_1103670
0071fc14 00AF                   add        r7, sp, #0x0
0071fc16 1A68                   ldr        r2, [r3]                             ; dword_1103670
0071fc18 0546                   mov        r5, r0
0071fc1a 7E44                   add        r6, pc                               ; dword_10c7240
0071fc1c D207                   lsls       r2, r2, #0x1f
0071fc1e 03D4                   bmi        loc_71fc28

0071fc20 1846                   mov        r0, r3
0071fc22 9CF5B8EA               blx        __cxa_guard_acquire@PLT              ; __cxa_guard_acquire
0071fc26 50B9                   cbnz       r0, loc_71fc3e

                            loc_71fc28:
0071fc28 9149                   ldr        r1, =0x9e3a40                        ; dword_71fe70,0x9e3a40, CODE XREF=_ZNK15ReadingSettings17lineHeightScalarsEv+22, _ZNK15ReadingSettings17lineHeightScalarsEv+466, _ZNK15ReadingSettings17lineHeightScalarsEv+470, _ZNK15ReadingSettings17lineHeightScalarsEv+494
0071fc2a 2846                   mov        r0, r5
0071fc2c 7944                   add        r1, pc                               ; dword_1103670
0071fc2e 0431                   adds       r1, #0x4                             ; 0x1103674
0071fc30 8AF57EEE               blx        _ZN5QListIdEC1ERKS0_@PLT             ; QList<double>::QList(QList<double> const&)
0071fc34 2846                   mov        r0, r5
0071fc36 8437                   adds       r7, #0x84
0071fc38 BD46                   mov        sp, r7
0071fc3a BDE8F083               pop.w      {r4, r5, r6, r7, r8, sb, pc}
                        ; endp

                            loc_71fc3e:
0071fc3e 8D4B                   ldr        r3, =0x12540                         ; 0x12540,dword_71fe74, CODE XREF=_ZNK15ReadingSettings17lineHeightScalarsEv+30
0071fc40 4FF00009               mov.w      sb, #0x0
0071fc44 07F18001               add.w      r1, r7, #0x80
0071fc48 4FF00008               mov.w      r8, #0x0
0071fc4c C3F6F079               movt       sb, #0x3ff0
0071fc50 3C1D                   adds       r4, r7, #0x4
0071fc52 F358                   ldr        r3, [r6, r3]
0071fc54 2046                   mov        r0, r4
0071fc56 61E91E89               strd       r8, sb, [r1, #-0x78]!
0071fc5a 7B60                   str        r3, [r7, #0x4]
0071fc5c 96F5A2E8               blx        _ZN5QListIdE6appendERKd@PLT          ; QList<double>::append(double const&)
0071fc60 6DA3                   adr        r3, #0x71fe18
0071fc62 D3E90023               ldrd       r2, r3, [r3]
0071fc66 07F18001               add.w      r1, r7, #0x80
0071fc6a 2046                   mov        r0, r4
0071fc6c 61E91C23               strd       r2, r3, [r1, #-0x70]!
0071fc70 96F598E8               blx        _ZN5QListIdE6appendERKd@PLT          ; QList<double>::append(double const&)
0071fc74 6AA3                   adr        r3, #0x71fe20
0071fc76 D3E90023               ldrd       r2, r3, [r3]
0071fc7a 07F18001               add.w      r1, r7, #0x80
0071fc7e 2046                   mov        r0, r4
0071fc80 61E91A23               strd       r2, r3, [r1, #-0x68]!
0071fc84 96F58EE8               blx        _ZN5QListIdE6appendERKd@PLT          ; QList<double>::append(double const&)
0071fc88 67A3                   adr        r3, #0x71fe28
0071fc8a D3E90023               ldrd       r2, r3, [r3]
0071fc8e 07F18001               add.w      r1, r7, #0x80
0071fc92 2046                   mov        r0, r4
0071fc94 61E91823               strd       r2, r3, [r1, #-0x60]!
0071fc98 96F584E8               blx        _ZN5QListIdE6appendERKd@PLT          ; QList<double>::append(double const&)
0071fc9c 43F23333               movw       r3, #0x3333
0071fca0 4FF03332               mov.w      r2, #0x33333333
0071fca4 C3F6F373               movt       r3, #0x3ff3
0071fca8 07F18001               add.w      r1, r7, #0x80
0071fcac 2046                   mov        r0, r4
[... 418 lines cut ...]
00735496 0440                   ands       r4, r0
00735498 6666                   str        r6, [r4, #0x64]                      ; DATA XREF=_ZNK15ReadingSettings17lineHeightScalarsEv+336
0073549a 6666                   str        r6, [r4, #0x64]
0073549c 6666                   str        r6, [r4, #0x64]
0073549e 0640                   ands       r6, r0
                            dword_7354a0:
007354a0                        dd         0x00a1c542                           ; DATA XREF=_ZNK15ReadingSettings17lineHeightScalarsEv
                            dword_7354a4:
007354a4                        dd         0x009dfa9a                           ; DATA XREF=_ZNK15ReadingSettings17lineHeightScalarsEv+8
                            dword_7354a8:
[... cut ...]
007354b8                        dd         0x0001d668                           ; DATA XREF=_ZNK15ReadingSettings17lineHeightScalarsEv+400
                            dword_7354bc:
007354bc                        dd         0x009fd014                           ; DATA XREF=_ZNK15ReadingSettings17lineHeightScalarsEv+404
                            dword_7354c0:
007354c0                        dd         0x00a1c34c                           ; DATA XREF=_ZNK15ReadingSettings17lineHeightScalarsEv+510

</pre></details>

geek1011, 2020-03-14:
:man_facepalming:

Decoding the BLX `96 F5 16 E8` at 0x7353AE is: `0x7353AE: BLX #0x4CB3DC`, and 0x4CB3DC is:

[image]

Which branches to the address at ip+0x104, and ip would likely be the one from the PLT call to lineHeightScalars:

[image]

Making it call moneypunct most of the time (depending on the load order)...

---

In short, I forgot to update the line `  - ReplaceBytes: {Offset: 10, FindH: 80 01 20 46,       ReplaceH: 96 F5 16 E8}       # blx QList<double>::append()`, and I also forgot to add an automated check like the one near the beginning of the patch (`ReplaceBLX: {Offset: 12, Find: 0x4D0368, Replace: 0x4D0368}`), but when I was updating it, I thought I already replaced it ... resulting in the wrong branch instruction being used. And, nickel's stack traces being slightly different was due to the way the load order affects this.

I've added a check for that replacement so I don't make this mistake again, and I've also improved the comments for the patch to make it more obvious. I'll push it in a few minutes once I update it for all the 4.20.* versions.

geek1011, 2020-03-14:
Another note to self (slightly off-topic, but may be useful to know in the future):

```asm
00b28fae B6F5C2E6               blx        _ZNK15ReadingSettings17lineHeightScalarsEv@PLT ; ReadingSettings::lineHeightScalars() const
00b28fb2 2A68                   ldr        r2, [r5]
00b28fb4 3046                   mov        r0, r6
00b28fb6 0021                   movs       r1, #0x0
00b28fb8 D368                   ldr        r3, [r2, #12]
00b28fba 9268                   ldr        r2, [r2, #8]
00b28fbc 9A1A                   subs       r2, r3, r2
00b28fbe 013A                   subs       r2, #0x1
00b28fc0 B2F500E5               blx        _ZN15QAbstractSlider8setRangeEii@PLT ; QAbstractSlider::setRange(int, int)
00b28fc4 2E68                   ldr        r6, [r5]
00b28fc6 3368                   ldr        r3, [r6]
00b28fc8 002B                   cmp        r3, #0x0
00b28fca 40F04181               bne.w      loc_b29250
```

```cpp
struct Q_CORE_EXPORT QListData {
    struct Data {
        QBasicAtomicInt ref;
        int alloc, begin, end;
        uint sharable : 1;
        void *array[1];
    };
    enum { DataHeaderSize = sizeof(Data) - sizeof(void *) };

    Data *detach(); // remove in 5.0
    Data *detach2(); // remove in 5.0
    Data *detach3();
    void realloc(int alloc);
    static Data shared_null;
    Data *d;
    void **erase(void **xi);
    void **append();
    void **append(const QListData &l);
    void **append2(const QListData &l); // remove in 5.0
    void **prepend();
    void **insert(int i);
    void remove(int i);
    void remove(int i, int n);
    void move(int from, int to);
    inline int size() const { return d->end - d->begin; }
    inline bool isEmpty() const { return d->end  == d->begin; }
    inline void **at(int i) const { return d->array + d->begin + i; }
    inline void **begin() const { return d->array + d->begin; }
    inline void **end() const { return d->array + d->end; }
};
```

i.e. `QAbstractSlider::setRange(0, ReadingSettings::lineHeightScalars().size()-1)`

commit 262e45cd (2020-03-14): Updated "My 24 line spacing values" for 4.20.* (closes #50)
closed 2020-03-14 by geek1011