pgaskin/kobopatch-patches #52: Series tab CSS patches

geek1011, 2020-03-05:
MR:post-3960493

jackiew1, 2020-03-05:
Added beta to MR:post-3960493 for Dragon devices only for now. 

Unfortunately it's difficult to optimise patch to suit all Dragon devices. Maximising ClaraHD cover thumbnails results in 5th (bottom) cover being truncated on H2O. I got around this by adding a second beta patch to reduce the height of the page navigation arrows at the bottom of the Series list page (which also affects other booklist pages).

There's a similar issue/workaround when trying to choose best values for Daylight devices. Maximising for Forma causes truncation on KA1.

The other option would be to make the thumbnails small enough so they don't truncate even when the nav arrows are full height. Kind of counter-productive to the purpose of the patch, though.

Only going to include this in standard pack if there are enough volunteers to test settings for Trilogy, Phoenix, LibraH20. Given issues stated above can't really guess without active co-operation from other users.

geek1011, 2020-03-08:
@jackiew1, have you decided if you're going to include this yet? I can test it on my Mini (trilogy) and Aura2v1 (phoenix).

NiLuJe, 2020-03-08:
I can also test on a H2O (which doesn't help :D), and a Forma (which probably does, since it's using the scales of a KA1 with a different screen size).

jackiew1, 2020-03-08:
@geek1011 
I didn't realise you could cover the Trilogy/Phoenix device testing. I can do the KA1/Forma. That will still leave the LibraH20 missing. I thought following the original request and the Dragon beta I posted that other users might have pushed for the missing devices, but so far they haven't so it can't be urgent.

Ideally I'd like to defer this for another week or two. I'm still really struggling with the tendinitis/RSI. Even the fairly limited work I did on the 4.20 release has made it flare up worse than ever.

geek1011, 2020-03-08:
OK, sounds good. I'll leave it for v60, then (v59 will be released as soon as I get around to fixing the 24 line spacing values patch).

rtiangha, 2020-03-10:
To be fair, part of the reason could also be that Series tab support only showed up in Calibre on Friday so maybe people haven't had time to upgrade yet and/or sync to be able to play with the new feature or to notice the difference in size.

But I appreciate the patch. Feel better!!

jackiew1, 2020-03-15:
@geek1011 @NiLuJe 
I've done some more work on this. I blew the cobwebs off my Glo & Aurav1, so now I think I've tested Glo, Aura6"v1, AuraHD, H2O, ClaraHD, KA1, Forma. Confirmation by second opinion would be helpful from both of you. 😄 

@geek1011 
This is only as urgent as you want it to be. Please will you test on any devices you can get hold of. The `Trilogy` (and LibraH2O) values in the patches below are pure guesswork. This is an outline of the process I used for testing the other models:
- To begin with only enable patch `beta Series list increase cover thumbnails`.
- Set the Part 2 `height` replacement high at first. Reduce it gradually until there is no truncation of the bottom edge of the top 4 thumbnails on each page. The bottom of each thumbnail should be sitting on the horizontal gridline below it. If it's not, the `height` value is too low. (_Finding books in my library where it's obvious when the bottom is truncated was time-consuming for me_).
- At this point the 5th thumbnail may still be truncated, but leave the `height` value alone now.
- Enable the 2nd patch `beta Reduce height of page navigator arrows` which should stop the 5th thumbnail truncating. If it doesn't, reduce the min/max-height values more until the truncation problem goes away.

Patch 1:
```
beta Series list increase cover thumbnails:
  - Enabled: no
  - Description: |
        Series list view (not cover view). Enlarge cover thumbnails.
        NB: After enabling this patch, if you find that the 
            last Series thumbnail on each page is truncated along its bottom edge
            then you should also enable the following nickel patch:
            'Reduce height of page navigator arrows'
  - FindZlib: "#seriesCoverPack"
  - ReplaceZlibGroup:
      Replacements:
        # Part 1: #seriesWidgetContainer - Reduce top/bottom margins
        #   NB: The smaller the topMargin the closer the Series name
        #       will be to the horizontal gridline above it
          # Touch/Mini
        - {Find: "topMargin: 9px;",     Replace: "topMargin: 3px;"}
        - {Find: "bottomMargin: 9px;",  Replace: "bottomMargin: 0px;"}
          # Glo/Aura6"/Aurav2
        - {Find: "topMargin: 12px;",    Replace: "topMargin: 4px;"}
        - {Find: "bottomMargin: 12px;", Replace: "bottomMargin: 0px;"}
          # AuraHD/H2O/H2Ov2/GloHD/ClaraHD
        - {Find: "topMargin: 14px;",    Replace: "topMargin: 5px;"}
        - {Find: "bottomMargin: 14px;", Replace: "bottomMargin: 0px;"}
          # AuraOne/Forma
        - {Find: "topMargin: 21px;",    Replace: "topMargin: 7px;"}
        - {Find: "bottomMargin: 21px;", Replace: "bottomMargin: 0px;"}
          # LibraH2O
        - {Find: "topMargin: 17px;",    Replace: "topMargin: 6px;"}
        - {Find: "bottomMargin: 17px;", Replace: "bottomMargin: 0px;"}
        #
        # Part 2: #seriesCoverPack - Increase cover thumbnail size
          # Touch/Mini (pure guesswork)
        - {Find: "width: 60px;",   Replace: "width: 85px;"}
        - {Find: "height: 94px;",  Replace: "height: 105px;"}
          # Glo/Aura6"/Aurav2
        - {Find: "width: 70px;",   Replace: "width: 100px;"}
        - {Find: "height: 115px;", Replace: "height: 140px;"}
          # AuraHD/H2O/H2Ov2/GloHD/ClaraHD
        - {Find: "width: 108px;",  Replace: "width: 141px;"}
        - {Find: "height: 175px;", Replace: "height: 200px;"}
          # AuraOne/Forma
        - {Find: "width: 140px;",  Replace: "width: 190px;"}
        - {Find: "height: 222px;", Replace: "height: 265px;"}
          # LibraH2O (pure guesswork)
        - {Find: "width: 126px;",  Replace: "width: 165px;"}
        - {Find: "height: 205px;", Replace: "height: 232px;"}
        #
        # Part 3: All models: #hBooks - Move bookcount up & away from horizontal gridline
        #         Users of lower-res models (Touch/Mini/Glo/Aura6"/Aurav2) may prefer to 
        #         decrease the Replace value of 20px for padding-bottom
        - Find:    "#hBooks {\n  padding-left: 0px;\n  padding-bottom: 0px;\n  font-style: normal;\n}"
          Replace: "#hBooks {\n  padding-left: 0px;\n  padding-bottom:20px;\n  font-style: normal;\n}"
```
Patch 2:
```
beta Reduce height of page navigator arrows:
  - Enabled: no
  - Description: |
        Reduce page navigator footer height (Book list, Series list)
        This patch may be helpful to users who also enable the nickel patch
        'Series list increase cover thumbnails'
        It should prevent the white background of the page footer/navigation arrow 
        overlapping the bottom edge of the last Serieslist thumbnail on each page.
  - FindZlib: "QWidget[footerFixedHeight=true][qApp_deviceIsTrilogy=true]"
  - ReplaceZlibGroup:
      Replacements:
          # Touch/Mini (pure guesswork)
        - Find:    "QWidget[footerFixedHeight=true][qApp_deviceIsTrilogy=true] {\n  min-height: 70px;\n  max-height: 70px;\n}"
          Replace: "QWidget[footerFixedHeight=true][qApp_deviceIsTrilogy=true] {\n  min-height: 54px;\n  max-height: 54px;\n}"
          # Glo/Aura6"/Aurav2
        - Find:    "QWidget[footerFixedHeight=true][qApp_deviceIsPhoenix=true] {\n  min-height: 82px;\n  max-height: 82px;\n}"
          Replace: "QWidget[footerFixedHeight=true][qApp_deviceIsPhoenix=true] {\n  min-height: 70px;\n  max-height: 70px;\n}"
          # AuraHD/H20/H20v2/GloHD/ClaraHD
        - Find:    "QWidget[footerFixedHeight=true][qApp_deviceIsDragon=true] {\n  min-height: 120px;\n  max-height: 120px;\n}"
[... cut ...]
        - Find:    "QWidget[footerFixedHeight=true][qApp_deviceIsDaylight=true] {\n  min-height: 156px;\n  max-height: 156px;\n}"
          Replace: "QWidget[footerFixedHeight=true][qApp_deviceIsDaylight=true] {\n  min-height: 120px;\n  max-height: 120px;\n}"
          # LibraH2O (pure guesswork)
        - Find:    "QWidget[footerFixedHeight=true][qApp_deviceIsStorm=true] {\n  min-height: 138px;\n  max-height: 138px;\n}"
          Replace: "QWidget[footerFixedHeight=true][qApp_deviceIsStorm=true] {\n  min-height: 105px;\n  max-height: 105px;\n}"
```

NiLuJe, 2020-03-15:
Thanks, I'll give it a go on H2O & Forma (probably tomorrow ;)).

NiLuJe, 2020-03-16:
@jackiew1: Are the replacement values meant to be pixel-perfect? (is it actually realistically achievable?). 

Especially the second one: stock values, on a H2O, make the navigation arrows fit *just* right.

To replicate that with patched values, I had to play with it a bit more. Just asking if I should actually bother, or if that's a compromise to make it work mostly okay across the full Dragon range, for instance ;).

NiLuJe, 2020-03-16:
Specifically, because my first educated guess turned out to be right (:D):

```
          # AuraHD/H20/H20v2/GloHD/ClaraHD
        - Find:    "QWidget[footerFixedHeight=true][qApp_deviceIsDragon=true] {\n  min-height: 120px;\n  max-height: 120px;\n}"
          Replace: "QWidget[footerFixedHeight=true][qApp_deviceIsDragon=true] {\n  min-height: 96px;\n  max-height: 96px;\n}"
```

I've also toned down Part 3 to 15px, it's much closer to the stock behavior (+/- 1px):

```
        #
        # Part 3: All models: #hBooks - Move bookcount up & away from horizontal gridline
        #         Users of lower-res models (Touch/Mini/Glo/Aura6"/Aurav2) may prefer to
        #         decrease the Replace value of 20px for padding-bottom
        - Find:    "#hBooks {\n  padding-left: 0px;\n  padding-bottom: 0px;\n  font-style: normal;\n}"
          Replace: "#hBooks {\n  padding-left: 0px;\n  padding-bottom:15px;\n  font-style: normal;\n}"
```

NiLuJe, 2020-03-16:
As for Part 1, am I right to assume that the top margin applies to *both* the thumbnail and the text, which is why we have to live with the text closer to the previous line, just in order to make room for taller thumbnails?

NiLuJe, 2020-03-16:
Oh, just realized I don't even need to make an educated guess, just measure the distance between the bottom thumbnail's gray margin and the bottom edge of the screen (which happens to be exactly 96px on the H2O).

Note that the H2O might be tricky to measure if you rely on the stock screenshot feature (I haven't checked if it was fixed, but it used to be very dumb things with the H2O viewport: the top 11px are dead, and shouldn't be captured. So far, so good, but the last time I checked (which, granted, was a couple years ago, it was *rescaling* the snapshot to a size that included those 11px).

TL;DR: Use fbgrab.

jackiew1, 2020-03-16:
> @jackiew1: Are the replacement values meant to be pixel-perfect? (is it actually realistically achievable?).
>
> Especially the second one: stock values, on a H2O, make the navigation arrows fit _just_ [...]

There is nothing "pixel-perfect" about either of the patches. 😄  Patch 2 particularly. The aim for patch 2 is to reduce the height of the Nav arrows (the whole footer, actually) "enough" (_vague term_) to remove the potential bottom thumbnail truncation problem. I've prioritised thumbnail size over footer height

And, yes, Dragon devices are a particular PITA because the same values will be used for AuraHD, H2O, GloHD, ClaraHD. There are no options for Alyssum/Nova The discerning user (_i.e. you_) can, of course, customise and make pixel-perfect whatever is eventually used in the official nickel.yaml.

That said, any suggestions you make to improve what I posted as the first stab I can test on the other Dragon devices that I have access to. My GloHD is out-of-bounds because my husband is using it with an old fw (_if it ain't broke etc, etc_). I'm hoping the GloHD behaves like the ClaraHD but I can't be sure. I thought the AuraHD and H2O would be the same but they're not, they're merely similar.

NiLuJe, 2020-03-16:
Things are... funnier on the Forma.

The arrows are gigantic, but we knew that already, and despite that, there's still vertical space left.
Nonetheless, growing the thumbnails too much *will* crop them, even before hitting the arrows...

Currently playing with values :).

Note that there, for Part 3, 20px is perfect, though ;).

NiLuJe, 2020-03-16:
Yep, can't improve on the defaults for the Forma ;).

(i.e., even growing height by 1px to 266 will crop it).

Which means the second patch isn't strictly necessary, but I still kind of like it with a light -10px to the stock values:

```
          # AuraOne/Forma
        - Find:    "QWidget[footerFixedHeight=true][qApp_deviceIsDaylight=true] {\n  min-height: 156px;\n  max-height: 156px;\n}"
          Replace: "QWidget[footerFixedHeight=true][qApp_deviceIsDaylight=true] {\n  min-height: 146px;\n  max-height: 146px;\n}"
```

That's probably highly-Forma specific because of the discrepancies between the two device's actual resolution.

(Conflating the Forma with the KA1 in nickel makes no goddamn sense :/).

NiLuJe, 2020-03-16:
So to recap:

H2O:

I like 15px instead of 20px for Part 3 of the first patch. It's close to the stock behavior, +/- 1px. I expect that to hold across almost all the Dragon line, more or less.

The arrows can be tweaked to 96px for pixel-perfect results, matching the stock behavior. That might not hold across all the Dragon line, though, but it's "easy" to compute with a fbgrab snapshot and a quick ruler check in an image editor (edge of the screen to gray border of the bottom thumbnail == max height).

----

Forma:

Defaults are perfect. Second patch is not strictly necessary, but I like it when set to 146px because those arrows are goddamn gigantic ;).

jackiew1, 2020-03-16:
@NiLuJe Re: Patch 1. The aim for Parts 1, 2, 3 were:

**Part 1:** I'd would have preferred to reduce both top/bottom margins to zero, but then the Series Title is butting the horiz gridline above - ugly. So I set top margin to approx 0.5mm for each device option. e.g. Dragon CSS settings usually equate to 10px=1mm, Daylight 13px=1mm

**Part 2:** Make thumbnails as big as possible. I believe the settings I used for tested devices will have increased the stock distance between horiz gridlines (maybe by 1mm???). This wasn't a precise scientific exercise. If you can improve them using Fbgrab, let's have them. 😄 

**Part 3:** The zero bottom margin in Part1 results in the bookcount sitting on the gridline below - ugly again. so I added padding. One size fits all models, unfortunately. I made no attempt to replicate stock fw. The "correct" value would be the original top/bottom margin from Part 1. My thinking is, there's plenty of room available, so a bit too big is better than a bit too small.

NiLuJe, 2020-03-16:
P1: Yeah, that's what I figured given the "container" class/id whatever (I suck at CSS :D).

P2: I hadn't looked at the actual height of an entry, but, yeah, it does (extreeeemly slightly on the H2O, like, 203px to 205px ^^. 264px to 272px on the Forma). I didn't play with the values on the H2O, but given the results on the Forma, I trust that we can't really do any better there, :+1; ;).

P3: Agreed, I'm just weird about vertical alignment, so it immediately felt a tiny bit off to me ;p.

jackiew1, 2020-03-16:
@NiLuJe I don't believe Patch 2 is required for Forma or Glo, but it was for every other device in my test list. I could have avoided the Nav arrow problem altogether by reducing the patched thumbnail size but that would be counter-productive to the purpose of the patch.

The original requester of Patch 1 also said he didn't need Patch 2 but decided he preferred the Nav arrows to be shorter than stock fw.

The Daylight value on a KA1 could be a few pixels bigger than the 120px I set but IMHO I don't think precision is particularly important for these arrow heights. All you need to be able to do is to hit them reliably - so avoid making them _too_ small. 

At the end of the day users can decide for themselves whether they want to customise Nav arrows to precise heights.

jackiew1, 2020-03-19:
`beta Series list increase cover thumbnails` updated after LibraH2O testing. Just Trilogy testing left to do.

NiLuJe, 2020-03-19:
Err, updated *where*? ;o).

jackiew1, 2020-03-19:
> Err, updated _where_? ;o).

Well I thought I'd edited it 2 hours ago further back up the thread where I posted the original. Don't know what happened to that, but it's done now. 😕 

Only height value changed by 2px, so not a bad original guess. Patch 2 not neeed for LibraH2O.

jackiew1, 2020-03-19:
@NiLuJe You're the aesthetics guy. What do you think about having the Patch 1 bookcount right-aligned like the Author list?

NiLuJe, 2020-03-19:
Hmm, might be nice, if that and the Series/Author can be vertically centered (essentially like the Books list).

jackiew1, 2020-03-20:
The default use of available space is fairly poor in this Series list. 

`qproperty-alignment:'AlignRight|AlignVCenter'` doesn't work for `#hBooks` and I had no luck adding top padding to the Series Title when I tried a few days ago.

I'll just leave it as-is and stop trying to be clever.

jackiew1, 2020-05-01:
@geek1011 (_bolting the door after the horse has gone_) ...

Patch 1 & 2 above were supposed to get added to nickel.yaml before the next release of kobopatch. I hadn't added them yet because I'm still waiting for you to test on your Trilogy Mini.

geek1011, 2020-05-01:
Oops (it's been a while). I thought I posted about it working on the mini a few weeks ago, but I guess not.

I guess it'll be in the next release, which will be in a month and a half, or the next firmware release, whichever is earliest. Sorry about that.

jackiew1, 2020-05-02:
@geek1011  ... for the Mini was Patch 1 sufficient on its own or did it need Patch 2 in order to avoid bottom left navigation arrow overwriting the bottom thumbnail?

geek1011, 2020-05-13:
@jackiew1: Patch 1 was sufficient on its own.

geek1011, 2020-05-13:
@jackiew1, are the patches finished (have you pushed the final version)?

jackiew1, 2020-05-13:
@geek1011 , I'm not sure what you mean by 'pushed'. 

If you're referring to the 2 commits I made about 3 hours ago `src/versions/4.20.14622/nickel.yaml/jackie_w.yaml` for the 2 new patches... As far as I know those contain the patch code which ought to be included in the official nickel.yaml - whenever you decide the next release is.

Is there something else I should have done?

geek1011, 2020-05-13:
> I'm not sure what you mean by 'pushed'
> Is there something else I should have done?

I meant the commits (if you had made them locally, you would have needed to `git push` them to GitHub). And no, that was all. I was just checking to make sure everything was ready.

The next release will probably be when the next firmware version is out, or at the end of June, whichever comes first.

closed 2020-05-13 by geek1011