pgaskin/kobopatch-patches #53: "Change dicthtml strings to micthtml" doesn't work completely on 4.20

rtiangha, 2020-03-06:
I posted about my experience [here](https://www.mobileread.com/forums/showpost.php?p=3960208&postcount=14), but basically, the patch applies without errors, but the full list of dictionaries doesn't appear in the drop down box; just a small subset. And the Settings menu thinks there are no dictionaries installed, despite the Installed and IsSynced columns in the database all being set to true and the micthtml files all being there. And you can still do searches on the dictionaries that are selectable, meaning that changing the file name does work since the Kobo can still access them.

Remove the patch and rename everything to dicthtml format, and everything works as expected again. Patch worked fine in 4.19 and earlier.

geek1011, 2020-03-06:
I'm most likely going to remove the patch, as it's quite old, and it's not necessary (I think).

Can someone else confirm that setting IsSynced to false on an overwritten installed built-in dictionary prevents it being overwritten during a sync?

geek1011, 2020-03-08:
This patch isn't necessary anymore: https://github.com/geek1011/kobopatch-patches/issues/49#issuecomment-596228361.

commit b809caae (2020-03-08): Removed "Change dicthtml strings to micthtml" for 4.20.14601 (fixes #53)

see #49
closed 2020-03-08 by geek1011