pgaskin/kobopatch-patches #54: Patches for 4.20.14617

geek1011, 2020-03-13:
I won't be starting these until tomorrow or the day after. I'll update this issue with a TODO list when I start.

geek1011, 2020-03-13:
#50 has been moved to this version.

geek1011, 2020-03-13:
This firmware release is mainly for bugfixes:
- The highlighting bug when moving vertically (introduced in 4.20.14601) has been fixed.
- The PDF zoom level bug has been fixed.
- Other fixes?

rtiangha, 2020-03-13:
According to [DNSB](https://www.mobileread.com/forums/showthread.php?t=328025):

- PIN lock skip when powering on
- CYMK jpgs not displaying

geek1011, 2020-03-13:
And, @jackiew1, there aren't any CSS changes in this release:

```
patrick@dpc01:~/kp$ diff --width=$COLUMNS --suppress-common-lines -y <(sort -k7 ./14601/cssextract.out.css) <(sort -k7 ./14617/cssextract.out.css)
/* zlib stream: offset_hex(0x13FDAF7) offset_int32(20962039) len_int32(1425) sha1(01a7784fe062e554fa95a2c67ea4ff93a0a12136) */	      |	/* zlib stream: offset_hex(0x145056C) offset_int32(21300588) len_int32(1425) sha1(01a7784fe062e554fa95a2c67ea4ff93a0a12136) */
/* zlib stream: offset_hex(0x13FC58E) offset_int32(20956558) len_int32(2947) sha1(01c187c801add2dc7aa1b1199ba27300816a74ba) */	      |	/* zlib stream: offset_hex(0x144416B) offset_int32(21250411) len_int32(2947) sha1(01c187c801add2dc7aa1b1199ba27300816a74ba) */
/* zlib stream: offset_hex(0x13F896A) offset_int32(20941162) len_int32(1238) sha1(028aa78bb03f41d89c665e9255ed2cf51ffe41d6) */	      |	/* zlib stream: offset_hex(0x14524FA) offset_int32(21308666) len_int32(1238) sha1(028aa78bb03f41d89c665e9255ed2cf51ffe41d6) */
/* zlib stream: offset_hex(0x13EB1D2) offset_int32(20885970) len_int32(1229) sha1(074921af3ce609b8dadd450e2201d574a5664d84) */	      |	/* zlib stream: offset_hex(0x1446A26) offset_int32(21260838) len_int32(1229) sha1(074921af3ce609b8dadd450e2201d574a5664d84) */
/* zlib stream: offset_hex(0x13EEE26) offset_int32(20901414) len_int32(1532) sha1(08483873663e3aef868cb87904dd8eced534b442) */	      |	/* zlib stream: offset_hex(0x144227F) offset_int32(21242495) len_int32(1532) sha1(08483873663e3aef868cb87904dd8eced534b442) */
/* zlib stream: offset_hex(0x13EA03D) offset_int32(20881469) len_int32(6973) sha1(0dc5712016c24b79c74c5e04f23b273c83383c12) */	      |	/* zlib stream: offset_hex(0x144DB74) offset_int32(21289844) len_int32(6973) sha1(0dc5712016c24b79c74c5e04f23b273c83383c12) */
/* zlib stream: offset_hex(0x13FDC6C) offset_int32(20962412) len_int32(6183) sha1(0e68541549435b7b6737aba9678976be4cf62a1a) */	      |	/* zlib stream: offset_hex(0x1446273) offset_int32(21258867) len_int32(6183) sha1(0e68541549435b7b6737aba9678976be4cf62a1a) */
/* zlib stream: offset_hex(0x13EBBDE) offset_int32(20888542) len_int32(1415) sha1(0ec0849942386e73da9d40a5b873206b7720a6ff) */	      |	/* zlib stream: offset_hex(0x144035B) offset_int32(21234523) len_int32(1415) sha1(0ec0849942386e73da9d40a5b873206b7720a6ff) */
/* zlib stream: offset_hex(0x13FD71F) offset_int32(20961055) len_int32(1415) sha1(0ec0849942386e73da9d40a5b873206b7720a6ff) */	      |	/* zlib stream: offset_hex(0x1445718) offset_int32(21255960) len_int32(1415) sha1(0ec0849942386e73da9d40a5b873206b7720a6ff) */
/* zlib stream: offset_hex(0x13FC76B) offset_int32(20957035) len_int32(872) sha1(0f5274e8cca32cf5b5f121b5193283c33e0fbc38) */	      |	/* zlib stream: offset_hex(0x144244C) offset_int32(21242956) len_int32(872) sha1(0f5274e8cca32cf5b5f121b5193283c33e0fbc38) */
/* zlib stream: offset_hex(0x13F9659) offset_int32(20944473) len_int32(2150) sha1(1236c8097391071de71585457cd0f171e74da98d) */	      |	/* zlib stream: offset_hex(0x144884D) offset_int32(21268557) len_int32(2150) sha1(1236c8097391071de71585457cd0f171e74da98d) */
/* zlib stream: offset_hex(0x13EA95A) offset_int32(20883802) len_int32(1739) sha1(130e31da0c5133c010b4cb7a4dfcb16c745da593) */	      |	/* zlib stream: offset_hex(0x143D7A5) offset_int32(21223333) len_int32(1739) sha1(130e31da0c5133c010b4cb7a4dfcb16c745da593) */
/* zlib stream: offset_hex(0x13F57A5) offset_int32(20928421) len_int32(1242) sha1(1348fd1a6a777e6d6c1f6aac557b3e756774cfae) */	      |	/* zlib stream: offset_hex(0x144E0EC) offset_int32(21291244) len_int32(1242) sha1(1348fd1a6a777e6d6c1f6aac557b3e756774cfae) */
/* zlib stream: offset_hex(0x13E8DCD) offset_int32(20876749) len_int32(1539) sha1(141a46e4bc04882f8f1fcd756b03e90906320795) */	      |	/* zlib stream: offset_hex(0x14490D7) offset_int32(21270743) len_int32(1539) sha1(141a46e4bc04882f8f1fcd756b03e90906320795) */
/* zlib stream: offset_hex(0x13EFED4) offset_int32(20905684) len_int32(3257) sha1(14398b1a9deb3083583c9158a87079c28cb25217) */	      |	/* zlib stream: offset_hex(0x1448A81) offset_int32(21269121) len_int32(3257) sha1(14398b1a9deb3083583c9158a87079c28cb25217) */
/* zlib stream: offset_hex(0x13F8249) offset_int32(20939337) len_int32(7744) sha1(171604fa932bc16d0c21f6444000fce2ca208185) */	      |	/* zlib stream: offset_hex(0x14465B5) offset_int32(21259701) len_int32(7744) sha1(171604fa932bc16d0c21f6444000fce2ca208185) */
/* zlib stream: offset_hex(0x13F153F) offset_int32(20911423) len_int32(4122) sha1(182d39043364478b1e6607fc607a6dccadf01efa) */	      |	/* zlib stream: offset_hex(0x144F0CE) offset_int32(21295310) len_int32(4122) sha1(182d39043364478b1e6607fc607a6dccadf01efa) */
/* zlib stream: offset_hex(0x13EF10C) offset_int32(20902156) len_int32(4646) sha1(182e3dc69c4d7e91340b1e03cf901a94a6ced20e) */	      |	/* zlib stream: offset_hex(0x143B9A0) offset_int32(21215648) len_int32(4646) sha1(182e3dc69c4d7e91340b1e03cf901a94a6ced20e) */
/* zlib stream: offset_hex(0x13FDFAE) offset_int32(20963246) len_int32(7201) sha1(1ba1a59bdb0ba565a954106f4b3eea6d8b651e03) */	      |	/* zlib stream: offset_hex(0x1452FFD) offset_int32(21311485) len_int32(7201) sha1(1ba1a59bdb0ba565a954106f4b3eea6d8b651e03) */
/* zlib stream: offset_hex(0x13FD1D9) offset_int32(20959705) len_int32(2005) sha1(1ed42ea0520277e3376da01957db6de5056f0fcd) */	      |	/* zlib stream: offset_hex(0x144545F) offset_int32(21255263) len_int32(2005) sha1(1ed42ea0520277e3376da01957db6de5056f0fcd) */
[... cut ...]
/* zlib stream: offset_hex(0x13EA6F2) offset_int32(20883186) len_int32(4722) sha1(fb0fc32e137a96d14c0dda09b17447dbbe0cdc7b) */	      |	/* zlib stream: offset_hex(0x1447B40) offset_int32(21265216) len_int32(4722) sha1(fb0fc32e137a96d14c0dda09b17447dbbe0cdc7b) */
/* zlib stream: offset_hex(0x13F6782) offset_int32(20932482) len_int32(1403) sha1(fcda4476088252d8dddb4b78d38648d2f02ad907) */	      |	/* zlib stream: offset_hex(0x144CC3F) offset_int32(21285951) len_int32(1403) sha1(fcda4476088252d8dddb4b78d38648d2f02ad907) */
```

jackiew1, 2020-03-13:
> And, @jackiew1, there aren't any CSS changes in this release:

Actually, Pedants'R'Us says there are 4 CSS streams which have changed (1 old, 3 new in 4.20.14617). They are all nozlib streams but none of them are used in existing patches.

FYI The 'old' stream change indicates that Developer Options has a new entry `'Show webkit layout rects'`. If enabled it's 'rects' are visible inside a kepub.

rtiangha, 2020-03-13:
Just as well that you waited since 4.20.14622 just came out.

geek1011, 2020-03-14:
> FYI The 'old' stream change indicates that Developer Options has a new entry 'Show webkit layout rects'. If enabled it's 'rects' are visible inside a kepub.

This option is actually pretty useful!

closed 2020-03-14 by geek1011