pgaskin/kobopatch-patches #55: Footnote patches in 4.20.xx

jackiew1, 2020-03-13:
@geek1011 

I don't think your official nickel patch `Change pop-up footnote font-family` is doing anything anymore. If the aim is to make sure pop-up footnotes use Georgia (rather than Avenir) then I think the patch can be removed. However, if the aim is to give font-family full customisation to the user then I think a new pop-up footnote code investigation is needed. Second opinions welcome 😄 

As a follow on ... I've previously posted an unofficial patch to control font-size of footnote text. This no longer works either. As with many parts of the GUI (IMO) default footnote font-size on a KA1/Forma is huge. Other models seem OK. 
```
test Pop-up footnote main text font-size:
  - Enabled: no
  - Description: Change font-size of pop-up footnote main text.
  - FindZlib: "RegularTouchLabel {\n  font-family"
  - ReplaceZlibGroup:
      Replacements:
        # RegularTouchLabel Popup footnote main text
        # Change Replace font-size as required for your Kobo model
          #Touch/Mini
        - Find:    "RegularTouchLabel[qApp_deviceIsTrilogy=true] {\n  font-size: 19px;\n}"
          Replace: "RegularTouchLabel[qApp_deviceIsTrilogy=true] {\n  font-size: 19px;\n}"
          #Glo/Aura/Aura2
        - Find:    "RegularTouchLabel[qApp_deviceIsPhoenix=true] {\n  font-size: 23px;\n}"
          Replace: "RegularTouchLabel[qApp_deviceIsPhoenix=true] {\n  font-size: 23px;\n}"
          #AuraHD/H2O/H2O2
        - Find:    "RegularTouchLabel[qApp_deviceIsDragon=true] {\n  font-size: 29px;\n}"
          Replace: "RegularTouchLabel[qApp_deviceIsDragon=true] {\n  font-size: 29px;\n}"
          #GloHD
        - Find:    "RegularTouchLabel[qApp_deviceIsAlyssum=true] {\n  font-size: 32px;\n}"
          Replace: "RegularTouchLabel[qApp_deviceIsAlyssum=true] {\n  font-size: 32px;\n}"
          #ClaraHD
        - Find:    "RegularTouchLabel[qApp_deviceIsNova=true] {\n  font-size: 32px;\n}"
          Replace: "RegularTouchLabel[qApp_deviceIsNova=true] {\n  font-size: 32px;\n}"
          #LibraH2O
        - Find:    "RegularTouchLabel[qApp_deviceIsStorm=true] {\n  font-size: 34px;\n}"
          Replace: "RegularTouchLabel[qApp_deviceIsStorm=true] {\n  font-size: 34px;\n}"
          #AuraOne/Forma
        - Find:    "RegularTouchLabel[qApp_deviceIsDaylight=true] {\n  font-size: 37px;\n}"
          Replace: "RegularTouchLabel[qApp_deviceIsDaylight=true] {\n  font-size: 34px;\n}"
```

The only workaround I've found so far is a new libnickel patch, but it's not ideal because it doesn't allow a different font-size per model.
```
test Pop-up footnote main text font-size:
  - Enabled: no
  - Description: |
        Change pop-up footnote main text font-size
        fw 4.20.14601 CSS moved to libnickel.so.1.0.0 from nickel???
        Daylight: %1 = 42 ???
  - FindReplaceString:
      Find:    "p { font-size: %1px; line-height: 1.4em; }"
      Replace: "p { font-size: 34px; line-height: 1.4em; }"
      MustMatchLength: yes
```

If we knew where the `%1px` value was being read from it might be a help.

geek1011, 2020-03-13:
I'm most likely going to remove it, then (I'm didn't originally create that one, I don't use it, and I'm not familiar enough with it).

jackiew1, 2020-03-14:
> I'm most likely going to remove it, then (I'm didn't originally create that one, I don't use it, and I'm not familiar enough with it).

Hmm, as I remember, you created the footnote font-family patch when I asked you whether it could be done. You were the one who told me that it was "hidden" in the CSS stream containing `#ConfirmationDialog`. I'm not convinced this stream is still the right one. Hence my original question.

jackiew1, 2020-03-14:
Oops, fat fingers!

geek1011, 2020-03-14:
I'd have to look at my notes. I honestly don't remember doing it, but what you said rings a bell for me. Sorry about my forgetfulness.

geek1011, 2020-03-14:
OK, I guess I was thinking of a different footnote patch earlier which wasn't mine.

I just had a look at this, and it does seem to have changed. Your new patch is correct, as confirmed by:

[image]

My patch wasn't intended to do anything other than change the font-family, so it doesn't seem like it is necessary anymore (the font is now serif by default).

I'm going to remove this patch, but if anyone ends up wanting to change the font-size, I'll include the one you posted above.

closed 2020-03-14 by geek1011

jackiew1, 2020-03-14:
I missed your response until just now. Thanks for looking. 

My Assembler is non-existent but your code snippet `DictionaryView::fontSize() const` seems to suggest that footnote pop-up font-size is using the same value as dictionary pop-up font-size. This matches my observations on KA1/Forma which both default to 42px.

I've still no idea where these font-size constants are being set, though. Does that snippet provide any clues to you? I investigated by quick&dirty patching a few occurrences of Daylight/font-size:42px in the nickel CSS but none of them did me any good.

fromhz, 2021-04-08:
Hello!
I commented on the kobopatch request thread on mobileread a while back:
https://www.mobileread.com/forums/showpost.php?p=4000645&postcount=509
https://www.mobileread.com/forums/showpost.php?p=4000755&postcount=511
I would like to use this patch to display footnotes properly for Chinese books. Would you by any chance have a look at this?
Thanks a lot!

commit 0633e147 (2020-03-14): Removed "Change pop-up footnote font-family" for 4.20.* (closes #55)

* It's not necessary anymore, as it is now serif by default.
* Either way, the location of the CSS used has changed, so it
  wouldn't work as-is either way.