pgaskin/kobopatch-patches #58: Re-jig patch 'Dictionary pop-up - increase available text area'

jackiew1, 2020-03-17:
Kobo added a new CSS stream (nickel nozlib `DictionaryViewFooter`) to control footer height of dictionary pop-up. I'm going to roll this into existing nickel patch `Dictionary pop-up - increase available text area` rather than add an extra one.

@geek1011 @NiLuJe 
Some of the dictionary pop-up variables in nickel don't have specific `qApp_deviceIsStorm` values. In this situation does the LibraH2O use Dragon or Daylight values?

geek1011, 2020-03-17:
See here: MR:thread-325471

[image]

Kobo uses a three-layered approach for device codenames.

jackiew1, 2020-03-17:
Thanks, so it's Dragon.

jackiew1, 2020-03-17:
@geek1011 Update commited, but I'll leave you to decide when to close it. It's been tested with Glo, Aura6", AuraHD, H2O, ClaraHD, KA1, Forma.

If you want to test with your Mini Trilogy device then all you need to do is enable as-is and check that the dictionary pop-up with the built-in English dictionary doesn't look terrible.

geek1011, 2020-03-17:
Looks good!

closed 2020-03-17 by geek1011