pgaskin/kobopatch-patches #59: New kobopatch instructions from v0.15.0

geek1011, 2020-03-17:
This issue is for any questions/comments about the new instructions introduced in v0.15.0 (see geek1011/kobopatch#32, e8c64f59ce032274f27a25994eda074c399d572f, and [v0.15.0](https://github.com/geek1011/kobopatch/releases/v0.15.0)).

jackiew1, 2020-03-17:
I just tried to download the v15.0 Windows .exe files. Windows Defender flags `koboptch-windows.exe` and `cssextract-windows.exe` as Severe threats and immediately removes them.

These are the details for `koboptch-windows.exe` `cssextract-windows.exe` is similar.
```
Threat detected: Trojan:Win32/Fuery.C!cl
Alert level: Severe
Category: Trojan
Details: This program is dangerous and executes commands from an attacker
```
and
```
Affected items:
webfile: D:\sysfiles\Downloads\koboptch-windows.exe|https://github-production-release-asset-2e65be.s3.amazonaws.com/130938106/bb33a480-67ca-11ea-9b06-7c46f4d06f0b?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20200317%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20200317T202526Z&X-Amz-Expires=300&X-Amz-Signature=2f3b40b60eb98272e989763f7cc46c38faddb4fe596f652ba6d2e7e276e1642f&X-Amz-SignedHeaders=host&actor_id=55494768&response-content-disposition=attachment%3B%20filename%3Dkoboptch-windows.exe&response-content-type=application%2Foctet-stream|pid:14116,ProcessStart:132289498495669039
```

geek1011, 2020-03-17:
Oh, that's annoying... I guess it's due to the fact that I have code which reads symbols and emulates ARM instructions.

I'll see what I can do about it. Thanks for the heads-up.

geek1011, 2020-03-17:
Does symdump trigger it?

geek1011, 2020-03-17:
OK, I just ran it through VirusTotal, and it doesn't seem to be detected by Defender that way: https://www.virustotal.com/gui/file/21254f046662846dd01a1679aaa846846c777b185b0a489961bc9e9fe53a3b61/detection.

jackiew1, 2020-03-17:
> Does symdump trigger it?

No. Neither does `koboptch-apply-windows.exe` although I don't know what these 2 do.

geek1011, 2020-03-17:
I've submitted it to Microsoft as a false positive for reclassification.

jackiew1, 2020-03-17:
> OK, I just ran it through VirusTotal, and it doesn't seem to be detected by Defender that way: https://www.virustotal.com/gui/file/21254f046662846dd01a1679aaa846846c777b185b0a489961bc9e9fe53a3b61/detection.

I suspect VirusTotal results won't carry as much weight as default Windows' own virus checking with your average MR Windows user. Getting MS to pass it as clean would be a good idea. Is that easy to do? Sounds time-consuming, not that I've any experience with such things.

geek1011, 2020-03-17:
[image]

geek1011, 2020-03-18:
[image]

Fixed.

jackiew1, 2020-03-18:
@geek1011 
Did you also report `cssextract-windows.exe` to MS? Because this is still getting zapped by Windows as a Severe threat when trying to download.

`koboptch-windows.exe` now downloads without complaint.

geek1011, 2020-03-18:
No, not yet. What is it being classified as?

jackiew1, 2020-03-18:
As I said in my original post, `cssextract-windows.exe` and `koboptch-windows.exe` originally showed the same Trojan.

geek1011, 2020-03-18:
I've submitted it to MS: https://www.microsoft.com/en-us/wdsi/submission/3a34f341-5409-471f-a0de-97235c1592a3.

geek1011, 2020-03-18:
It should be fine now (update your definitions and try again).

jackiew1, 2020-03-18:
All OK now. 👍

geek1011, 2020-05-01:
Released in v60.

closed 2020-05-01 by geek1011