pgaskin/kobopatch-patches #60: [Question] Change system font with custom patches

babygau, 2020-05-05:
hi, 
i found some patches that allow KOBO to change its default system fonts from [here](https://github.com/jcn363/kobopatch/blob/master/src/versions/4.12.12111/nickel-363.yaml#L240) and [there](https://github.com/jcn363/kobopatch/blob/master/src/versions/4.12.12111/libnickel-PROGRESIVE.yaml#L168). Following the instruction but fail to get it compiled.
```
Applying overrides
    ENABLE  `Font-family and font-size bulk change (part 2 of 2)`
Error: could not apply patches: invalid patch file 'src/font-patch-libnickel.yaml': patch "Font-family and font-size bulk change (part 2 of 2)": line 9: inst 3: FindReplaceString: replacement string 5 chars too short

Applying overrides
    ENABLE  `Font-family and font-size bulk change (part 1 of 2)`
Error: could not apply patches: invalid patch file 'src/font-patch-libnickel.yaml': patch "Font-family and font-size bulk change (part 1 of 2)": line 23: inst 3: FindReplaceString: replacement string 5 chars too short
```
would you mind telling me what I did wrong? 

Here is mine: [font-patches-for-kobo](https://gist.github.com/babygau/0c25f1eca4e45c0ee40e967554ebea80)

many thanks

geek1011, 2020-05-05:
You've changed a patch which still has the older requirement of needing a replacement string exactly the length of the original. You'll need to add enough spaces somewhere harmless (i.e. after the `:` in CSS properties) to make them match.

babygau, 2020-05-06:
many thanks ❤️ , the problem is solved!

closed 2020-05-06 by babygau