pgaskin/kobopatch-patches #62: Patches for 4.21.15015

geek1011, 2020-05-19:
- [ ] Upload testdata (@geek1011)
- [ ] Find libnickel offsets where necessary (@geek1011)
- [ ] Update patches (@geek1011)
- [ ] Test most patches (@geek1011)
- [ ] Make release (with #56) (@geek1011)
- [ ] MR post (@geek1011)

geek1011, 2020-05-19:
It seems my work in #59 was a success. I have significantly fewer patches to update in this version (almost 15x less offsets to find), and most of the complex ones still work!

<details><pre>
$ go run ./scripts/test 4.21.15015
### scan versions
--- provided on command line, skipping scan
### test patches
>>> test 4.21.15015
--- scanning sources
--- generating config
--- running kobopatch
kobopatch unknown
https://github.com/geek1011/kobopatch

Loading configuration from -
Reading input firmware testdata tarball

Patching libadobe.so
 ✔  Remove PDF map widget shown during panning

Patching libnickel.so.1.0.0
 ✔  Custom navigation menu page number text
 ✔  Dictionary text font-family/font-size/line-height
 ✔  Bulgarian Phonetic Keyboard (GloHD/ClaraHD/AuraOne/H2O2/Forma/Libra)
 ✔  Cyrillic Keyboard (GloHD/ClaraHD/AuraOne/H2O2)
 ✔  Greek Keyboard (GloHD/ClaraHD/AuraOne/H2O2)
 ✕  Don't grab exclusive access to event0
 ✔  Allow USB storage even when device locked
 ✕  Allow rotation on all devices
 ✔  Allow showing info panel on random screensaver
 ✔  Always show confirmation dialog before upgrading
 ✔  Both page turn buttons go next
 ✕  Both page turn sides go next
 ✕  Change Browse Kobo home screen link target - Activity
 ✕  Change Browse Kobo home screen link target - Articles
 ✔  Custom header/footer page number text
 ✕  Customize ComfortLight settings
 ✕  DeveloperSettings - AutoUsbGadget
 ✔  Disable all tutorial dialogs
 ✔  Don't uppercase header/footer text
 ✔  Don't uppercase header/footer text and change page number text
 ✔  Enable advanced settings for all fonts
 ✔  FeatureSettings - BookSpecificStats
 ✔  FeatureSettings - ExportHighlights
 ✔  FeatureSettings - FullScreenBrowser
 ✔  FeatureSettings - MyWords
 ✔  FeatureSettings - ShowFacebookShare
 ✕  Hide browser from beta features
 ✕  Hide top-left menu item - Activity
 ✕  Hide top-left menu item - Dropbox
 ✕  Hide top-left menu item - Pocket
 ✕  Hide top-left menu item - Recommended
 ✕  Hide top-left menu item - Store
 ✕  Hide top-left menu item - Wishlist
 ✔  Increase TOC level indentation
 ✔  Increase TOC level indentation and fix extra indentation bug
 ✔  Larger Sleep/Power-off timeouts
 ✕  Never show Kobo Plus, wishlist, and points SmartLinks
 ✔  Only show Pocket SmartLink
 ✔  Only show stats SmartLink
 ✕  PowerSettings - UnlockEnabled
 ✔  Remove beta features not supported text
 ✔  Remove recommendations (row1col2) from home screen
 ✕  Remove title from reading header/footer
 ✕  Rename new home screen footer
 ✔  Replace adobe page numbers toggle with invert screen
 ✔  Set visible SmartLink
 ✔  Swap reading header/footer
 ✔  Allow searches on Extra dictionaries
 ✔  Block WiFi firmware upgrade
[... 64 lines cut ...]
  /home/patrick/src/kobopatch-patches/src/versions/4.21.15015/libnickel.so.1.0.0.yaml/geek1011.yaml: Never show Kobo Plus, wishlist, and points SmartLinks: could not apply patch "Never show Kobo Plus, wishlist, and points SmartLinks": line 271: inst 7: ReplaceBytes: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.21.15015/libnickel.so.1.0.0.yaml/geek1011.yaml: PowerSettings - UnlockEnabled: could not apply patch "PowerSettings - UnlockEnabled": line 676: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.21.15015/libnickel.so.1.0.0.yaml/geek1011.yaml: Allow rotation on all devices: could not apply patch "Allow rotation on all devices": line 482: inst 3: ReplaceBytes: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.21.15015/libnickel.so.1.0.0.yaml/geek1011.yaml: Both page turn sides go next: could not apply patch "Both page turn sides go next": line 17: inst 4: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.21.15015/libnickel.so.1.0.0.yaml/geek1011.yaml: Hide top-left menu item - Wishlist: could not apply patch "Hide top-left menu item - Wishlist": line 335: inst 4: ReplaceBytes: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.21.15015/libnickel.so.1.0.0.yaml/geek1011.yaml: Remove title from reading header/footer: could not apply patch "Remove title from reading header/footer": line 415: inst 4: ReplaceBytes: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.21.15015/libnickel.so.1.0.0.yaml/geek1011.yaml: Change Browse Kobo home screen link target - Articles: could not apply patch "Change Browse Kobo home screen link target - Articles": line 191: inst 4: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.21.15015/libnickel.so.1.0.0.yaml/geek1011.yaml: Customize ComfortLight settings: could not apply patch "Customize ComfortLight settings": line 601: inst 3: ReplaceInt: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.21.15015/libnickel.so.1.0.0.yaml/geek1011.yaml: Change Browse Kobo home screen link target - Activity: could not apply patch "Change Browse Kobo home screen link target - Activity": line 176: inst 4: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.21.15015/libnickel.so.1.0.0.yaml/geek1011.yaml: Hide top-left menu item - Store: could not apply patch "Hide top-left menu item - Store": line 313: inst 4: ReplaceBytes: could not find specified bytes at offset
[... cut ...]
  /home/patrick/src/kobopatch-patches/src/versions/4.21.15015/libnickel.so.1.0.0.yaml/geoffr.yaml: Set KePub hyphenation: could not apply patch "Set KePub hyphenation": line 319: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.21.15015/libnickel.so.1.0.0.yaml/geoffr.yaml: Set font scale factor: could not apply patch "Set font scale factor": line 535: inst 3: ReplaceInt: could not find specified bytes at offset
Tests for 4.21.15015 failed (exit status 1)! Stopping.
exit status 1
</pre></details>

geek1011, 2020-05-19:
Some random notes:

- The "Browse Kobo" bottom-left link now has the option to show an "OverDrive" link instead.
- The evdev input device path is now parameterized (/dev/input/event0 is now %1).
- TODO

closed 2020-05-19 by geek1011

jackiew1, 2020-05-19:
@geek1011, re: changes to patch `Custom collection/author header title font`

I don't like the changes you've made to my patch because of the absolute hex locations. Previously all my nickel patches have been multi-version.

We now seem to have a 3rd CSS stream where font-size is set for `#header #headerTitle`. Do you know which part of the GUI this 3rd stream is controlling?

The 1st (and original) stream controls the header size in the Author, Series and Collections booklists.

The 2nd looks as if it has something to do with Dropbox. I was never really sure why you added this, but because I haven't enabled Dropbox perhaps there's something I don't see on my own devices. If it needs to be left in I can find a unique string rather than your hex location.

What is this new 3rd stream (BaseAddress: 0x1455C16) for? Are you sure it needs to be included in this patch? If so I'll look for a unique string.

geek1011, 2020-05-19:
And a symbol diff (left is removed, right is added):

<details><pre>diff -y <(cat kp/14622/symdump.out.json | jq -r '.[] | .Name' | sort) <(cat kp/15015/symdump.out.json | jq -r '.[] | .Name' | sort) | grep -e '<' -e '>'
ATTRIBUTE_LOVE_ELIGIBLE					      <
ATTRIBUTE_LOVE_POINTS_PRICE				      <
							      >	ERROR_PATRON_NOT_FOUND
							      >	ERROR_TITLE_NOT_FOUND
							      >	getpid
							      >	kill
ONE_STORE_MAGAZINE_DETAIL_PAGE_URL			      <
ONE_STORE_MAGAZINE_LANDING_PAGE_URL			      <
							      >	OVERDRIVE_CHECKOUT_BOOK_EXPECTED_DATE
							      >	OVERDRIVE_CHECKOUT_BOOK_EXPIRE_DATE
							      >	OVERDRIVE_CHECKOUT_BOOK_IS_AVAILABLE
							      >	OVERDRIVE_CHECKOUT_BOOK_IS_SUSPENDED
							      >	OVERDRIVE_CHECKOUT_BOOK_PLACED_DATE
							      >	OVERDRIVE_HOLD_AVAILABLE_TRUE
							      >	OVERDRIVE_HOLDS_TAB_TYPE
							      >	_ZGVZN7HasNullI11BookPricingE4nullEvE7nullObj
							      >	_ZGVZN7HasNullI19OverDriveHeldVolumeE4nullEvE7nullObj
_ZGVZN9QtPrivate15ConnectionTypesINS_4ListIIP7QWidgetEEELb1EE <
_ZGVZN9QtPrivate26MetaTypeSmartPointerHelperI8QPointerI16Inte <
_ZGVZN9QtPrivate26MetaTypeSmartPointerHelperI8QPointerI7QObje <
							      >	_ZlsI7QString4QMapIS0_S0_EE6QDebugS3_RKS1_IT_T0_E
							      >	_ZN10FilterBaseI19OverDriveHeldVolumeE6filterERKS0_
							      >	_ZN10FilterBaseI19OverDriveHeldVolumeED0Ev
							      >	_ZN10FilterBaseI19OverDriveHeldVolumeED1Ev
							      >	_ZN10FilterBaseI19OverDriveHeldVolumeED2Ev
							      >	_ZN10Repository12refreshCacheI21OverDriveCheckoutBookEEP5Cach
							      >	_ZN10SyncClient22overDriveAvailableHoldERK7QStringS2_b
							      >	_ZN10WebkitView21setAutoHighlightStyleEb
							      >	_ZN11BookPricing12setLovePriceERK5Price
							      >	_ZN11BookPricing15setLoveEligibleEb
							      >	_ZN11BookPricing18setLovePointsPriceERK7QString
							      >	_ZN11BookPricing8setPriceERK5Price
							      >	_ZN11BookPricingC1ERK5PriceS2_RK7QStringb
							      >	_ZN11BookPricingC1ERKS_
							      >	_ZN11BookPricingC1Ev
							      >	_ZN11BookPricingC2ERK5PriceS2_RK7QStringb
							      >	_ZN11BookPricingC2ERKS_
							      >	_ZN11BookPricingC2Ev
							      >	_ZN11DhcpcdProxy13EnableNetworkERK7QStringi
							      >	_ZN11DhcpcdProxy14DisableNetworkERK7QStringi
							      >	_ZN11DhcpcdProxy14SignalStrengthE4QMapI7QStringS0_IS1_S1_EE
							      >	_ZN11Entitlement17setPhasesFromListERK5QListI8QVariantE
							      >	_ZN11Entitlement18setPhasesFromBytesERK10QByteArray
_ZN11FickelProxy11qt_metacallEN11QMetaObject4CallEiPPv	      <
_ZN11FickelProxy11qt_metacastEPKc			      <
_ZN11FickelProxy14SignalStrengthEd			      <
_ZN11FickelProxy16staticMetaObjectE			      <
_ZN11FickelProxyC1ERK7QStringS2_RK15QDBusConnectionP7QObject  <
_ZN11FickelProxyC2ERK7QStringS2_RK15QDBusConnectionP7QObject  <
_ZN11FickelProxyD0Ev					      <
_ZN11FickelProxyD1Ev					      <
_ZN11FickelProxyD2Ev					      <
_ZN11ReaderProxy11currentPageEv				      <
_ZN11ReaderProxy14clearSelectionEv			      <
_ZN11ReaderProxy8nextPageEv				      <
							      >	_ZN11SubtitleKeyI19OverDriveHeldVolumeED0Ev
							      >	_ZN11SubtitleKeyI19OverDriveHeldVolumeED1Ev
							      >	_ZN11SubtitleKeyI19OverDriveHeldVolumeED2Ev
							      >	_ZN11TitleSorterI19OverDriveHeldVolumeED0Ev
							      >	_ZN11TitleSorterI19OverDriveHeldVolumeED1Ev
							      >	_ZN11TitleSorterI19OverDriveHeldVolumeED2Ev
							      >	_ZN12AuthorSorterI19OverDriveHeldVolumeED0Ev
							      >	_ZN12AuthorSorterI19OverDriveHeldVolumeED1Ev
							      >	_ZN12AuthorSorterI19OverDriveHeldVolumeED2Ev
							      >	_ZN12LibKmodMixin14driverBasePathEv
							      >	_ZN12LibKmodMixin6insertERK7QStringS2_
							      >	_ZN12LibKmodMixin6removeERK7QString
[... 1999 lines cut ...]
							      >	_ZTV9KeySorterI19OverDriveHeldVolume14FirstAuthorKeyIS0_E21Nu
							      >	_ZTV9KeySorterI19OverDriveHeldVolume14ThirdAuthorKeyIS0_E21Nu
							      >	_ZTV9KeySorterI19OverDriveHeldVolume15SecondAuthorKeyIS0_E21N
							      >	_ZTV9KeySorterI19OverDriveHeldVolume15SeriesNumberKeyIS0_E8Co
							      >	_ZTV9KeySorterI19OverDriveHeldVolume15SubtitleKanaKeyIS0_E18K
							      >	_ZTV9KeySorterI19OverDriveHeldVolume18FirstAuthorKanaKeyIS0_E
							      >	_ZTV9KeySorterI19OverDriveHeldVolume18ThirdAuthorKanaKeyIS0_E
							      >	_ZTV9KeySorterI19OverDriveHeldVolume19SecondAuthorKanaKeyIS0_
							      >	_ZTV9KeySorterI19OverDriveHeldVolume21OverDriveDateAddedKeyIS
							      >	_ZTV9KeySorterI19OverDriveHeldVolume22OverDriveExpireDateKeyI
							      >	_ZTV9KeySorterI19OverDriveHeldVolume23OverDriveIsAvailableKey
							      >	_ZTV9KeySorterI19OverDriveHeldVolume34OverDriveEstimatedAvail
							      >	_ZTV9KeySorterI19OverDriveHeldVolume8TitleKeyIS0_E21NumericSt
							      >	_ZTV9KeySorterI19OverDriveHeldVolume9SeriesKeyIS0_E8CollatorE
							      >	_ZTV9SeriesKeyI19OverDriveHeldVolumeE
_ZZN11QMetaTypeIdI8QPointerI16InternetProviderEE14qt_metatype <
							      >	_ZZN7HasNullI11BookPricingE4nullEvE7nullObj
							      >	_ZZN7HasNullI19OverDriveHeldVolumeE4nullEvE7nullObj
_ZZN9QtPrivate15ConnectionTypesINS_4ListIIP7QWidgetEEELb1EE5t <
							      >	_ZZN9QtPrivate15ConnectionTypesINS_4ListIIRK11QStringListEEEL
							      >	_ZZN9QtPrivate15ConnectionTypesINS_4ListIIRK7QStringRK11QStri
_ZZN9QtPrivate26MetaTypeSmartPointerHelperI8QPointerI16Intern <
_ZZN9QtPrivate26MetaTypeSmartPointerHelperI8QPointerI7QObject <
_ZZN9QtPrivate29SharedPointerMetaTypeIdHelperI8QPointerI7QObj <
							      >	_ZZNK13BaseAuthorKeyI19OverDriveHeldVolumeE3keyERKS0_E13unkno</pre></details>

geek1011, 2020-05-19:
> I don't like the changes you've made to my patch because of the absolute hex locations. Previously all my nickel patches have been multi-version.

> What is this new 3rd stream (BaseAddress: 0x1455C16) for? Are you sure it needs to be included in this patch? If so I'll look for a unique string.

Those are necessary due to some duplicated streams (see the forgot pin patch for another example). I'd need to make some changes in kobopatch itself to deal with this properly.

> We now seem to have a 3rd CSS stream where font-size is set for #header #headerTitle. Do you know which part of the GUI this 3rd stream is controlling?

I'll take a look.

> The 1st (and original) stream controls the header size in the Author, Series and Collections booklists.

geek1011, 2020-05-19:
> BaseAddress: 0x1455C16

Oh, that is the Dropbox setup screen. And, it seems I mixed up the offsets when updating that patch. I meant to use `0x1446850` instead...

jackiew1, 2020-05-19:
Now you've got me completely baffled. Have you tested any of this on an actual device before you released it?

geek1011, 2020-05-19:
> Now you've got me completely baffled.

Scratch what I said about the whole offset thing. I mixed up my cssextract files and misinterpreted my notes when updating...

> Have you tested any of this on an actual device before you released it?

I only tested the libnickel ones.

jackiew1, 2020-05-19:
In what way is it fixed? It still contains a hex address and It's no longer equivalent to the 4.20 version of the same patch.

Part 1 `FindZlib: "#header,\n#backButton,\n#moreLink {"` works on it's own on my devices. 

Part 2 in the 4.20.x patch affected the stream containing #dropboxContainer and you've now changed this to point at a new 3rd stream `BaseAddress: 0x1452DC9`,  (which didn't exist in 4.20.x) which doesn't appear to have anything to do with dropbox. Why do you think that the dropbox stream patching is no longer required but this new 3rd one is?

As I've already said, I never really understood why you added the extra dropbox part in the first place, but I can't really argue facts because I've never even seen any of the dropbox GUI screens.

geek1011, 2020-05-19:
> Why do you think that the dropbox stream patching is no longer required but this new 3rd one is?

Looking back, that Dropbox one was indeed unnecessary. The new stream is related to the library view in a similar way to the first one.

jackiew1, 2020-05-19:
> The new stream is related to the library view in a similar way to the first one.

Please elaborate. Why does this extra stream need patching at all? If everything from `# Second set (do the same thing as above):` onwards is deleted my own device looks as if it's patched correctly as advertised.

geek1011, 2020-05-19:
> Why does this extra stream need patching at all?

See my edit above.

jackiew1, 2020-05-19:
So, finally we're in agreement. Who'd have thought it?