pgaskin/kobopatch-patches #64: Request: Under each version's release, please attach only the binary with that same version, rather than the current practice of attaching all asset versions.

hub2git, 2020-05-25:
Hello @geek1011 , 

In https://github.com/geek1011/kobopatch-patches/releases, may I suggest  attaching only the binary connected to the version in question, and not all the versions.

For example,  in https://github.com/geek1011/kobopatch-patches/releases/tag/v61, currently there are 24 assets:

> 
> kobopatch_4.10.11591.zip
> kobopatch_4.10.11655.zip
> kobopatch_4.11.11911.zip
> kobopatch_4.11.11976.zip
> kobopatch_4.11.11980.zip
> kobopatch_4.11.11982.zip
> kobopatch_4.11.12019.zip
> kobopatch_4.12.12111.zip
> kobopatch_4.13.12638.zip
> kobopatch_4.14.12777.zip
> kobopatch_4.15.12920.zip
> kobopatch_4.16.13162.zip
> kobopatch_4.17.13651.zip
> kobopatch_4.17.13694.zip
> kobopatch_4.18.13737.zip
> kobopatch_4.19.14123.zip
> kobopatch_4.20.14601.zip
> kobopatch_4.20.14617.zip
> kobopatch_4.20.14622.zip
> kobopatch_4.21.15015.zip
> kobopatch_4.9.11311.zip
> kobopatch_4.9.11314.zip
> Source code (zip)
> Source code (tar.gz) 

I think there should just be 3:

> kobopatch_4.21.15015.zip
> Source code (zip)
> Source code (tar.gz)

geek1011, 2020-05-25:
What would be the purpose of this? I often provide updates to the patches which go back many versions, and there are still many people using older versions. Also, doing this simplifies linking to the latest release for a version.

hub2git, 2020-05-25:
Oh, I didn't realize that you provide updates to patches for older versions.

> What would be the purpose of this?

It is a cognitive hurdle to see which is the latest version. There are  two
factors behind this:
1. Version number (e.g. v61) doesn't match the latest zip file's version
(e.g. kobopatch_4.21.15015.zip)
2. The assets are not listed on version order (as you say, sort order is
broken).

geek1011, 2020-05-25:
> 1. Version number (e.g. v61) doesn't match the latest zip file's version (e.g. kobopatch_4.21.15015.zip)

But how does this make it harder? You never need to look at the version number in most cases 

> 2. The assets are not listed on version order (as you say, sort order is broken).

Only for 4.9.*.

geek1011, 2020-06-23:
As there hasn't been any response, I'm going to close this issue for now.

hub2git, 2020-06-23:
Hi geek, I've been following this thread and appreciate your comments to it. But I just did not have anything else to say.

hub2git, 2020-06-23:
From your comments, I learned that some people download older versions.

closed 2022-08-06 by pgaskin