pgaskin/kobopatch-patches #65: Patches for 4.22.15190

geek1011, 2020-06-23:
- [ ] Upload testdata (@geek1011)
- [ ] Find libnickel offsets where necessary (@geek1011)
- [ ] Update patches (@geek1011)
- [ ] Test nickel patches (@jackiew1)
- [ ] Test most patches (@geek1011)
- [ ] Make release (@geek1011)
- [ ] MR post (@geek1011)

geek1011, 2020-06-23:
Only a bit more (home screen font, forgot pin, top-left menu item) than the usual is required for this version:

```
Errors:
  /home/patrick/src/kobopatch-patches/src/versions/4.22.15190/nickel.yaml/oren64.yaml: New home screen subtitle custom font: could not apply patch "New home screen subtitle custom font": line 30: inst 3: BaseAddress: offset greater than length of buf
  /home/patrick/src/kobopatch-patches/src/versions/4.22.15190/libnickel.so.1.0.0.yaml/geoffr.yaml: Set KePub hyphenation: could not apply patch "Set KePub hyphenation": line 320: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.22.15190/libnickel.so.1.0.0.yaml/geoffr.yaml: Set font scale factor: could not apply patch "Set font scale factor": line 536: inst 3: ReplaceInt: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.22.15190/libnickel.so.1.0.0.yaml/geek1011.yaml: Hide top-left menu item - Dropbox: could not apply patch "Hide top-left menu item - Dropbox": line 323: inst 7: ReplaceBytes: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.22.15190/libnickel.so.1.0.0.yaml/geek1011.yaml: Hide top-left menu item - Recommended: could not apply patch "Hide top-left menu item - Recommended": line 350: inst 4: ReplaceBytes: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.22.15190/libnickel.so.1.0.0.yaml/geek1011.yaml: Hide top-left menu item - Wishlist: could not apply patch "Hide top-left menu item - Wishlist": line 364: inst 4: ReplaceBytes: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.22.15190/libnickel.so.1.0.0.yaml/geek1011.yaml: Customize ComfortLight settings: could not apply patch "Customize ComfortLight settings": line 633: inst 4: ReplaceInt: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.22.15190/libnickel.so.1.0.0.yaml/geek1011.yaml: Hide top-left menu item - Activity: could not apply patch "Hide top-left menu item - Activity": line 374: inst 4: ReplaceBytes: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.22.15190/libnickel.so.1.0.0.yaml/geek1011.yaml: Allow rotation on all devices: could not apply patch "Allow rotation on all devices": line 513: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.22.15190/libnickel.so.1.0.0.yaml/geek1011.yaml: DeveloperSettings - AutoUsbGadget: could not apply patch "DeveloperSettings - AutoUsbGadget": line 704: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.22.15190/nickel.yaml/geek1011.yaml: Remove forgot pin button from lock screen: could not apply patch "Remove forgot pin button from lock screen": line 18: inst 3: BaseAddress: offset greater than length of buf
```

jackiew1, 2020-06-23:
That's good. I just need to finish off what I'm doing and then I'll start my standard new fw routine. Should be within the hour.

geek1011, 2020-06-23:
A few notes:

- This update is quite a bit smaller than the last one (in terms of binary size), presumably due to cleaning up old code and/or resources.
- More?

geek1011, 2020-06-23:
```
patrick@patrick-tp:~/kp$ diff -y <(cat 15015/symdump.out.json | jq -r '.[] | .Name' | sort) <(cat 15190/symdump.out.json | jq -r '.[] | .Name' | sort) | grep -e '<' -e '>'
							      >	CATEGORY_FEATUREDLIST_BROWSE_TYPE
							      >	ONE_STORE_PRIVACY_CENTRE_URL
							      >	_Z27qRegisterNormalizedMetaTypeI4QSetI7QStringEEiRK10QByteArr
							      >	_Z3wpav
							      >	_ZGVZN7HasNullI8LanguageE4nullEvE7nullObj
							      >	_ZGVZN9QtPrivate15ConnectionTypesINS_4ListIIbP20SearchLanguag
							      >	_ZGVZN9QtPrivate15ConnectionTypesINS_4ListIIRK4QSetI7QStringE
							      >	_ZGVZN9QtPrivate19ValueTypeIsMetaTypeI4QSetI7QStringELb1EE17r
							      >	_ZN10N3WebStore9showEventEP10QShowEvent
							      >	_ZN10NameSorterI8LanguageED0Ev
							      >	_ZN10NameSorterI8LanguageED1Ev
							      >	_ZN10NameSorterI8LanguageED2Ev
							      >	_ZN10SyncClient18userProfileUpdatedEv
							      >	_ZN11EventEngine11setUserSlotERK4User
							      >	_ZN11EventEngine15fireSignInEventEv
							      >	_ZN11EventEngine17setCreatedAccountEb
							      >	_ZN11EventEngine21setCreatedAccountSlotEb
							      >	_ZN11EventEngine7setUserERK4User
							      >	_ZN11QMetaTypeIdI4QSetI7QStringEE14qt_metatype_idEv
_ZN11QMetaTypeIdI5QListI9QSslErrorEE14qt_metatype_idEv	      <
							      >	_ZN12AdobeStyling3bufEv
							      >	_ZN12LanguageUtil13sortLanguagesER7QVectorI8LanguageE
							      >	_ZN13EventInfoImpl12acsmReminderEv
_ZN13EventInfoImpl12categoryListEv			      <
_ZN13EventInfoImpl12sideloadBookEv			      <
_ZN13EventInfoImpl15categoryLandingEv			      <
							      >	_ZN13EventInfoImpl15readHourChangedEv
							      >	_ZN13EventInfoImpl16browseCategoriesEv
_ZN13EventInfoImpl17dropboxGetStartedEv			      <
							      >	_ZN13EventInfoImpl18searchLanguageListEv
							      >	_ZN13EventInfoImpl24overDriveViewCuratedListEv
_ZN13EventInfoImpl9shareBookEv				      <
							      >	_ZN13N3BrowserView12loadFinishedEb
_ZN13SyncQueueItemC1ERKS_				      <
_ZN13SyncQueueItemC2ERKS_				      <
							      >	_ZN14QSharedPointerI17LibraryDataSourceI8LanguageEE5derefEPN1
							      >	_ZN14QSharedPointerI19LibraryDataProviderI8LanguageEE5derefEP
							      >	_ZN14StatisticsUtil15replaceOutliersIjEE5QListIT_ERKS3_
							      >	_ZN14StatisticsUtil8quartileIjEET_RK5QListIS1_Ei
							      >	_ZN15AddDeviceParser11qt_metacallEN11QMetaObject4CallEiPPv
							      >	_ZN15AddDeviceParser11qt_metacastEPKc
							      >	_ZN15AddDeviceParser15addDeviceParsedERK7QStringS2_S2_S2_
							      >	_ZN15AddDeviceParser16staticMetaObjectE
							      >	_ZN15AddDeviceParser9parseJSONEv
							      >	_ZN15AddDeviceParserC1EP7QObject
							      >	_ZN15AddDeviceParserC2EP7QObject
							      >	_ZN15AddDeviceParserD0Ev
							      >	_ZN15AddDeviceParserD1Ev
							      >	_ZN15AddDeviceParserD2Ev
							      >	_ZN15QtSharedPointer33ExternalRefCountWithCustomDeleterI18Dic
							      >	_ZN15QtSharedPointer33ExternalRefCountWithCustomDeleterI19Sto
							      >	_ZN15QtSharedPointer33ExternalRefCountWithCustomDeleterI23OnD
							      >	_ZN15SyncManagerBase18userProfileUpdatedEv
							      >	_ZN16AddDeviceCommand11qt_metacallEN11QMetaObject4CallEiPPv
							      >	_ZN16AddDeviceCommand11qt_metacastEPKc
							      >	_ZN16AddDeviceCommand15addDeviceParsedERK7QStringS2_S2_S2_
							      >	_ZN16AddDeviceCommand16staticMetaObjectE
							      >	_ZN16AddDeviceCommand17migrationProgressEi
							      >	_ZN16AddDeviceCommand20authenticatedExecuteEv
							      >	_ZN16AddDeviceCommandC1EP9SyncStateP7QObjectbS3_
							      >	_ZN16AddDeviceCommandC2EP9SyncStateP7QObjectbS3_
							      >	_ZN16AddDeviceCommandD0Ev
							      >	_ZN16AddDeviceCommandD1Ev
							      >	_ZN16AddDeviceCommandD2Ev
							      >	_ZN16AddDeviceRequestC1ERK6Device
							      >	_ZN16AddDeviceRequestC2ERK6Device
							      >	_ZN16AddDeviceRequestD0Ev
							      >	_ZN16AddDeviceRequestD1Ev
[... 484 lines cut ...]
							      >	_ZTV19LibraryDataProviderI8LanguageE
_ZTV19UserPlatformCommand				      <
_ZTV19UserPlatformRequest				      <
							      >	_ZTV20InMemoryDataProviderI8LanguageE
							      >	_ZTV20SearchLanguageWidget
_ZTV21MenuSectionButtonItem				      <
							      >	_ZTV21SearchLanguageBuilder
							      >	_ZTV22Grid2LibraryControllerI8LanguageE
							      >	_ZTV23LinearLibraryDataSourceI8LanguageE
							      >	_ZTV24SearchLanguageController
							      >	_ZTV25AbstractLibraryControllerI8LanguageE
							      >	_ZTV38Migration160_AddUserPrivacyPermissions
							      >	_ZTV3KeyI7QString8LanguageE
							      >	_ZTV6SorterI8LanguageE
							      >	_ZTV7NameKeyI8LanguageE
							      >	_ZTV8Language
							      >	_ZTV9KeySorterI8Language7NameKeyIS0_E21NumericStringCollatorE
							      >	_ZZN11QMetaTypeIdI4QSetI7QStringEE14qt_metatype_idEvE11metaty
							      >	_ZZN18QMetaTypeIdQObjectIP20SearchLanguageWidgetLb1EE14qt_met
							      >	_ZZN7HasNullI8LanguageE4nullEvE7nullObj
							      >	_ZZN9QtPrivate15ConnectionTypesINS_4ListIIbP20SearchLanguageW
							      >	_ZZN9QtPrivate15ConnectionTypesINS_4ListIIRK4QSetI7QStringEEE
							      >	_ZZN9QtPrivate15ConnectionTypesINS_4ListIIRK7QStringS4_S4_S4_
							      >	_ZZN9QtPrivate19ValueTypeIsMetaTypeI4QSetI7QStringELb1EE17reg
```

jackiew1, 2020-06-23:
@geek1011  As far as I can tell upfront none of my nickel patches will need to be changed. 

There have been changes to the CSS stream used by patch `Dictionary pop-up - increase available text area` but nothing that should adversely affect it.

I haven't installed on & patched an actual device yet. More later if I discover anything new & interesting.

geek1011, 2020-06-23:
> I haven't installed on & patched an actual device yet. More later if I discover anything new & interesting.

I'm just about to do that, then I'll make the release after testing the patches. See my post above for a few interesting things about this version.

geek1011, 2020-06-23:
Released in [v62](https://github.com/geek1011/kobopatch-patches/releases/tag/v62) ([MR post](https://www.mobileread.com/forums/showthread.php?t=330996)).

closed 2020-06-23 by geek1011

jackiew1, 2020-06-23:
@geek1011 I've installed & patched a KA1. Only one of my standard patches is disabled until the official release (`Custom font sizes`). The rest are multi-version. Everything seems to working OK on the KA1.

There are very few changes at all in nickel CSS. Not sure how interesting any of them are (_meh_). 
- new zlib CSS at `0x1403830` [#addToFiltersContainer]
- new nozlib CSS at `0x140aff6` [SearchLanguageWidget]
- new piece of nozlib HTML at `0x112c628` which closely resembles a similar bit of HTML at `0x112caa4`. Something to do with frontlight. Both reference image `qrc:/images/light/rgbfrontlight_logo.svg`

jackiew1, 2020-06-23:
Correction to my previous comment about patch `Custom font sizes`. I see it's now multi-version, too. That's great, no more absolute hex locations in any of my enabled patches anymore. 👍

jackiew1, 2020-06-25:
@geek1011 

Re: oren64's nickel patch `New home screen subtitle custom font` ...

I just noticed that you've converted this from a single multi-version FindZlib to absolute 2 x BaseAddress. I was wondering why you needed to do this? What have you discovered that I don't know?

Despite my reticence to get publicly involved with Home screen patches, I've been using a personal customisation of this patch for years and didn't need to change anything for 4.22.15190.

If all you're trying to do is change colour/font-size of the 'x HOURS TO GO' labels on the Home screen then, as far as I know, only the first half of the patch to `MetaDataLabel` at `0x1400C98` is required and it can be identified with `FindZlib: "#homeWidgetTopContainer"`. 

I've no idea what the second "duplicate" half at `0x14093E8` is changing.

geek1011, 2020-06-25:
I haven't actually looked into that one myself, tbh. I don't use that patch myself, and my main focus for actual testing is on libnickel patches, as they are more complex and problems with them tend to be more serious.

I'll look into what the other one is actually from next week using my resources scripts.

jackiew1, 2020-06-25:
> Also, I wasn't involved in the creation of that patch, so I'm not too familiar with how it's intended to b used.

It's a pretty trivial cosmetic patch. My previous description 'change colour/font-size of the 'x HOURS TO GO' labels' is just about all there is to it. 😄 Change text from grey to black and adjust font-size to suit.

Based on your one-off report on .qss files from a couple of weeks ago, I think it is `SubcategoryCarouselWidget.qss`, whatever that's supposed to be.