pgaskin/kobopatch-patches #67: Patches for 4.22.15268

pgaskin, 2020-07-15:
- [ ] Upload testdata (@geek1011)
- [ ] Find libnickel offsets where necessary (@geek1011)
- [ ] Update patches (@geek1011)
- [ ] Look at Nia stuff (pgaskin/kobo-plugin-experiments#11) (@geek1011)
- [ ] Test nickel patches (@jackiew1)
- [ ] Test most patches (@geek1011)
- [ ] Make release (@geek1011)
- [ ] MR post (@geek1011)

pgaskin, 2020-07-15:
Not too much to update for the patches, but the first error is quite unusual...

```
  /home/patrick/src/kobopatch-patches/src/versions/4.22.15268/nickel.yaml/pgaskin.yaml: Remove forgot pin button from lock screen: could not apply patch "Remove forgot pin button from lock screen": line 20: inst 4: ReplaceZlib: not a zlib stream
  /home/patrick/src/kobopatch-patches/src/versions/4.22.15268/libnickel.so.1.0.0.yaml/pgaskin.yaml: Customize ComfortLight settings: could not apply patch "Customize ComfortLight settings": line 641: inst 4: ReplaceInt: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.22.15268/libnickel.so.1.0.0.yaml/pgaskin.yaml: DeveloperSettings - AutoUsbGadget: could not apply patch "DeveloperSettings - AutoUsbGadget": line 712: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.22.15268/libnickel.so.1.0.0.yaml/pgaskin.yaml: Allow rotation on all devices: could not apply patch "Allow rotation on all devices": line 521: inst 3: ReplaceBytes: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.22.15268/libnickel.so.1.0.0.yaml/geoffr.yaml: Set KePub hyphenation: could not apply patch "Set KePub hyphenation": line 320: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.22.15268/libnickel.so.1.0.0.yaml/geoffr.yaml: Set font scale factor: could not apply patch "Set font scale factor": line 537: inst 3: ReplaceInt: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.22.15268/nickel.yaml/oren64.yaml: New home screen subtitle custom font: could not apply patch "New home screen subtitle custom font": line 31: inst 4: ReplaceZlib: not a zlib stream
```

pgaskin, 2020-07-15:
So, this release only adds Nia-specific stuff and possible some bugfixes, but almost certainly not new features. And, the Nia is Luna.

```
$ diff -y <(cat 15190/symdump.out.json | jq -r '.[] | .Name' | sort) <(cat 15268/symdump.out.json | jq -r '.[] | .Name' | sort) | grep -e '<' -e '>'
_ZN13EventInfoImpl12acsmReminderEv			      <
_ZN13EventInfoImpl18searchLanguageListEv		      <
							      >	_ZN13EventInfoImpl19searchLanguageCountEv
							      >	_ZN13EventInfoImpl20signedInWithFacebookEv
_ZN14ReviewWorkflow12getSyncStateEv			      <
							      >	_ZN16QCoreApplication11aboutToQuitENS_14QPrivateSignalE
							      >	_ZN16QCoreApplication16staticMetaObjectE
_ZN5QListIiEaSERKS0_					      <
							      >	_ZN7QLocale16languageToStringENS_8LanguageE
							      >	_ZN7QVectorIiE5clearEv
							      >	_ZN7QVectorIiE6appendERKi
							      >	_ZN7QVectorIiE6detachEv
							      >	_ZN7QVectorIiEaSERKS0_
							      >	_ZN7QVectorIiED1Ev
							      >	_ZN7QVectorIiED2Ev
							      >	_ZNK6Device6isLunaEv
```

pgaskin, 2020-07-15:
Nia stuff (pgaskin/kobo-plugin-experiments#11):

Device ID: 482
Hardware: kobo7
Processor: i.MX6ULL
PCB: E60U2
USB ID: 0x4320
Codename 1: Phoenix
Codename 2: Luna
Codename 3: N/A
Same cover size as Kraken, Star.
Same touch tolerance as Star (30px).
No exclusive firmware features.
Most hardware-related quirks should be identical to Star.

pgaskin, 2020-07-15:
There isn't anything new in the CSS other than doing the same thing as the other newer devices with the games.

```
$ diff <(sort 15190/cssextract.out.css) <(sort 15268/cssextract.out.css)
5564a5565
> }
6116a6118
> #boggleContainer[devModeOn=false][qApp_deviceIsLuna=true],
18438a18441
>   qproperty-visible: false;
19182a19186
> #rushHourContainer[devModeOn=false][qApp_deviceIsLuna=true],
19571a19576
> #sketchPadContainer[devModeOn=false][qApp_deviceIsLuna=true],
19710a19716
> #solitaireContainer[devModeOn=false][qApp_deviceIsLuna=true],
19725a19732
> #spacerWidget[devModeOn=false][qApp_deviceIsLuna=true] {
19969a19977
> #sudokuContainer[devModeOn=false][qApp_deviceIsLuna=true],
21046,21184c21054,21192
< /* zlib stream: offset_hex(0x13F6E72) offset_int32(20934258) len_int32(1721) sha1(b692bf3f0b61437c3826a456f8c13a49790377ba) */
< /* zlib stream: offset_hex(0x13F703B) offset_int32(20934715) len_int32(5293) sha1(c6bffd1ac4f79934e0a4943446326268b5afaaea) */
< /* zlib stream: offset_hex(0x13F77F5) offset_int32(20936693) len_int32(3142) sha1(7c1345e6979de6854cf9f5eddc9f1907923e8e58) */
< /* zlib stream: offset_hex(0x13F7D0B) offset_int32(20937995) len_int32(600) sha1(4aa031bca11cb057ea734c4e7ac25998210f0cf7) */
< /* zlib stream: offset_hex(0x13F7DBC) offset_int32(20938172) len_int32(1225) sha1(dffbd5aa73d437b25d583e050b9f7a5d7e11eb5b) */
< /* zlib stream: offset_hex(0x13F8083) offset_int32(20938883) len_int32(3111) sha1(e0a7d230c70b1ed2ef86bcd13842d79712cd77ce) */
< /* zlib stream: offset_hex(0x13F8290) offset_int32(20939408) len_int32(1574) sha1(e6ea4e5492e3985ebf80094bf9916e20860c2a0e) */
< /* zlib stream: offset_hex(0x13F87A0) offset_int32(20940704) len_int32(878) sha1(31d66ceb0f1533ae4f18badfc979e81d5aaa16a7) */
< /* zlib stream: offset_hex(0x13F8B4B) offset_int32(20941643) len_int32(644) sha1(7b8c995acc9bc6c3995dad260fc39a5686715091) */
< /* zlib stream: offset_hex(0x13F8C09) offset_int32(20941833) len_int32(4518) sha1(621813b96fc0b901b58bb000f82599a146c58e97) */
< /* zlib stream: offset_hex(0x13F90C3) offset_int32(20943043) len_int32(1882) sha1(71061566e7dd8496fb60b3e9a5453b7b5b3b6e78) */
< /* zlib stream: offset_hex(0x13F939C) offset_int32(20943772) len_int32(2022) sha1(be61292509323cbd094a36a0b502add1d110ab2e) */
< /* zlib stream: offset_hex(0x13F956F) offset_int32(20944239) len_int32(806) sha1(a01b83011d67a4bd5df400760bbec556ba600cc6) */
< /* zlib stream: offset_hex(0x13F9638) offset_int32(20944440) len_int32(4282) sha1(fe5f756b45c03caa7b407f80280b2a4cfa14b218) */
< /* zlib stream: offset_hex(0x13F9941) offset_int32(20945217) len_int32(1826) sha1(40081e0f8bfe1a75ca902530476d0e2e57299e0b) */
< /* zlib stream: offset_hex(0x13F9AB1) offset_int32(20945585) len_int32(1403) sha1(fcda4476088252d8dddb4b78d38648d2f02ad907) */
< /* zlib stream: offset_hex(0x13F9BF7) offset_int32(20945911) len_int32(6935) sha1(844a059d2e6cd081923931d58ce3074d4fffa3e7) */
< /* zlib stream: offset_hex(0x13FA01B) offset_int32(20946971) len_int32(3420) sha1(50cb3395d4ab84910a36e464a694ce3035e72bb1) */
< /* zlib stream: offset_hex(0x13FA2A3) offset_int32(20947619) len_int32(2821) sha1(be5d28feb431d1b70fce28221ca14731fc8e6adb) */
< /* zlib stream: offset_hex(0x13FA4C9) offset_int32(20948169) len_int32(1155) sha1(3af580725df180d41a70b3bf7ca509e7194ff065) */
< /* zlib stream: offset_hex(0x13FA60D) offset_int32(20948493) len_int32(1486) sha1(1298739b28a95e138a7cc190d39ab29f183fcf0e) */
< /* zlib stream: offset_hex(0x13FA758) offset_int32(20948824) len_int32(5427) sha1(faa474813b2e68b148fcb79b99a4a9d4a9efcefe) */
< /* zlib stream: offset_hex(0x13FAABA) offset_int32(20949690) len_int32(2601) sha1(683a4bc712a73e5327197c5324d34f6627295c3a) */
< /* zlib stream: offset_hex(0x13FB155) offset_int32(20951381) len_int32(7201) sha1(1ba1a59bdb0ba565a954106f4b3eea6d8b651e03) */
< /* zlib stream: offset_hex(0x13FB869) offset_int32(20953193) len_int32(1487) sha1(c34a42ecc0e9a450c19fb29d997662c7eb85ca6f) */
< /* zlib stream: offset_hex(0x13FBBD1) offset_int32(20954065) len_int32(5686) sha1(754171734e70d8497cc82aab67781e934f6e2289) */
< /* zlib stream: offset_hex(0x13FBF43) offset_int32(20954947) len_int32(2635) sha1(afa8d8fc7822825acc49bd11dd0c943d0fc56a47) */
< /* zlib stream: offset_hex(0x13FC1BA) offset_int32(20955578) len_int32(872) sha1(0f5274e8cca32cf5b5f121b5193283c33e0fbc38) */
< /* zlib stream: offset_hex(0x13FC28F) offset_int32(20955791) len_int32(959) sha1(7cc0dd862d89cb85d3d11bd513deddebda9ecc93) */
< /* zlib stream: offset_hex(0x13FC375) offset_int32(20956021) len_int32(3305) sha1(4bd31a4d27b76d63b360af850013d166d656c7df) */
< /* zlib stream: offset_hex(0x13FC5D2) offset_int32(20956626) len_int32(9615) sha1(47f2a18c2a5ac2dee9018ab7096d0063684e6f22) */
< /* zlib stream: offset_hex(0x13FCB01) offset_int32(20957953) len_int32(5367) sha1(cc9e528b9743e08a209b1bdf3dc98fb38e9c1940) */
< /* zlib stream: offset_hex(0x13FCE9E) offset_int32(20958878) len_int32(2088) sha1(27bacba66560d9edd5c011f8e29ddf3c3b79079f) */
< /* zlib stream: offset_hex(0x13FD02C) offset_int32(20959276) len_int32(1133) sha1(2f0c2cc225365ac23ca0444e3798a9ca940c7f38) */
< /* zlib stream: offset_hex(0x13FD156) offset_int32(20959574) len_int32(4346) sha1(a4f2f40d496af56f04b26fa787850a868c4f966f) */
< /* zlib stream: offset_hex(0x13FD5C8) offset_int32(20960712) len_int32(11724) sha1(aba09b7ed39dd3359d6e54b95369ac6844fcb012) */
< /* zlib stream: offset_hex(0x13FDE03) offset_int32(20962819) len_int32(6776) sha1(c24603904c97e71832b2f7829c0b184dfe46db4e) */
[... cut ...]
#spacerWidget[devModeOn=false][qApp_deviceIsPika=true] {
  qproperty-visible: false;
}
#boggleContainer[devModeOn=false][qApp_deviceIsLuna=true],
#solitaireContainer[devModeOn=false][qApp_deviceIsLuna=true],
#rushHourContainer[devModeOn=false][qApp_deviceIsLuna=true],
#sudokuContainer[devModeOn=false][qApp_deviceIsLuna=true],
#sketchPadContainer[devModeOn=false][qApp_deviceIsLuna=true],
#spacerWidget[devModeOn=false][qApp_deviceIsLuna=true] {
  qproperty-visible: false;
}
#boggleContainer[devModeOn=false][qApp_deviceIsAlyssum=true],
```

pgaskin, 2020-07-15:
I'm going to start working on the patches themselves now.

pgaskin, 2020-07-15:
This was probably the simplest update so far:

```diff
diff --git a/src/versions/4.22.15190/libnickel.so.1.0.0.yaml/geoffr.yaml b/src/versions/4.22.15268/libnickel.so.1.0.0.yaml/geoffr.yaml
index ea3f795..bfd5815 100644
--- a/src/versions/4.22.15190/libnickel.so.1.0.0.yaml/geoffr.yaml
+++ b/src/versions/4.22.15268/libnickel.so.1.0.0.yaml/geoffr.yaml
@@ -317,8 +317,8 @@ Set KePub hyphenation:
       will always be turned on, regardless of justification button setting.
       
       (The publisher can still turn hyphenation off/on in the book's stylesheet.)
-  - ReplaceBytes: {Offset: 0xB7FEEA, FindInstBLX: {SymPLT: "QString::operator==(QLatin1String) const"}, ReplaceH: 01 20 01 20} # Alternative 1: Always turn KePub hyphenation on
-# - ReplaceBytes: {Offset: 0xB7FEEA, FindInstBLX: {SymPLT: "QString::operator==(QLatin1String) const"}, ReplaceH: 00 20 00 20} # Alternative 2: Never turn KePub hyphenation on
+  - ReplaceBytes: {Offset: 0xB801D2, FindInstBLX: {SymPLT: "QString::operator==(QLatin1String) const"}, ReplaceH: 01 20 01 20} # Alternative 1: Always turn KePub hyphenation on
+# - ReplaceBytes: {Offset: 0xB801D2, FindInstBLX: {SymPLT: "QString::operator==(QLatin1String) const"}, ReplaceH: 00 20 00 20} # Alternative 2: Never turn KePub hyphenation on
 
 Force user line spacing in KePubs:
   - Enabled: no
@@ -534,5 +534,5 @@ Set font scale factor:
   # Original value X=14, so original scale factor is (font size slider) / 46.
   # Allowed replacement values for X: 0-15 (smaller X --> larger scale factor)
   # Note: vmov.f32 s1, #15.0 before getReadingFontSizeScaleFactor
-  - ReplaceInt: {Offset: 0x5773AC, Find: 14, Replace: 4} # ePub
-  - ReplaceInt: {Offset: 0xB7FAC0, Find: 14, Replace: 4} # KePub
+  - ReplaceInt: {Offset: 0x577624, Find: 14, Replace: 4} # ePub
+  - ReplaceInt: {Offset: 0xB7FDA8, Find: 14, Replace: 4} # KePub
diff --git a/src/versions/4.22.15190/libnickel.so.1.0.0.yaml/pgaskin.yaml b/src/versions/4.22.15268/libnickel.so.1.0.0.yaml/pgaskin.yaml
index 6da7e6d..369560f 100644
--- a/src/versions/4.22.15190/libnickel.so.1.0.0.yaml/pgaskin.yaml
+++ b/src/versions/4.22.15268/libnickel.so.1.0.0.yaml/pgaskin.yaml
@@ -518,13 +518,13 @@ Allow rotation on all devices:
       Unlike only the DeveloperSettings ForceAllowLandscape option, this patch
       also makes the rotation icon show on the new reader as well.
   # Always return true for ForceAllowLandscape:
-  - ReplaceBytes: {Offset: 0x76745E, FindInstBLX: {SymPLT: "QVariant::toBool() const"}, ReplaceH: 4F F0 01 00}
+  - ReplaceBytes: {Offset: 0x76783E, FindInstBLX: {SymPLT: "QVariant::toBool() const"}, ReplaceH: 4F F0 01 00}
   # In ReadingMenuView::ReadingMenuView, replace the value of Device::hasOrientationSensor,
   # which is passed to a function to hide/show the rotate icon:
-  - ReplaceBytes: {Offset: 0xB96E72, FindInstBLX: {SymPLT: "Device::hasOrientationSensor() const"}, ReplaceH: 4F F0 01 00}
+  - ReplaceBytes: {Offset: 0xB971BA, FindInstBLX: {SymPLT: "Device::hasOrientationSensor() const"}, ReplaceH: 4F F0 01 00}
   # Also fix sizing of popup by doing the same in RotatePopup::RotatePopup, but
   # note that this will make the Automatic option show too (and do nothing):
-  - ReplaceBytes: {Offset: 0xBB30CE, FindInstBLX: {SymPLT: "Device::hasOrientationSensor() const"}, ReplaceH: 4F F0 01 00}
+  - ReplaceBytes: {Offset: 0xBB33DE, FindInstBLX: {SymPLT: "Device::hasOrientationSensor() const"}, ReplaceH: 4F F0 01 00}
 
 Don't uppercase header/footer text:
   - Enabled: no
@@ -634,7 +634,7 @@ Customize ComfortLight settings:
   # In an unnamed subroutine two layers into FrontLightPopupController::loadView
   # (find it by going back from QTime::addSecs), the times for the dropdown are
   # generated into a QVector<QPair<QString, QTime>> with a simple loop.
-  - BaseAddress: 0xADA8B0 # find by the x-ref to _ZN5QTimeC1Eiiii from an unnamed sub
+  - BaseAddress: 0xADAF98 # find the base of the unnamed subroutine with the x-ref to _ZN5QTimeC1Eiiii
   #
   # Change the initial hour / first bedtime dropdown item passed to the QTime
   # constructor (mov r1, #21):
@@ -709,7 +709,7 @@ FeatureSettings - ExportHighlights:
 DeveloperSettings - AutoUsbGadget:
   - Enabled: no
   - Description: Automatically enable USB Storage mode when connected.
-  - ReplaceBytes: {Offset: 0x7674FE, FindInstBLX: {SymPLT: "QVariant::toBool() const"}, ReplaceH: 4F F0 01 00}
+  - ReplaceBytes: {Offset: 0x7678DE, FindInstBLX: {SymPLT: "QVariant::toBool() const"}, ReplaceH: 4F F0 01 00}
 
 PowerSettings - UnlockEnabled:
   - Enabled: no
diff --git a/src/versions/4.22.15190/nickel.yaml/oren64.yaml b/src/versions/4.22.15268/nickel.yaml/oren64.yaml
index c8d0cee..fb50526 100644
--- a/src/versions/4.22.15190/nickel.yaml/oren64.yaml
+++ b/src/versions/4.22.15268/nickel.yaml/oren64.yaml
@@ -27,7 +27,7 @@ New home screen subtitle custom font:
[... 11 lines cut ...]
   # duplicate
-  - BaseAddress: 0x14093E8
+  - BaseAddress: 0x148FAF2
   - ReplaceZlib: {Find: "MetaDataLabel {\n  color: #666666;\n  qproperty-indent: 0;\n}", Replace: "MetaDataLabel {\n  color: #000000;\n  qproperty-indent: 0;\n}"}
   - ReplaceZlib:
       Find:    "MetaDataLabel[qApp_deviceIsTrilogy=true] {\n  font-size: 13px;\n}"
diff --git a/src/versions/4.22.15190/nickel.yaml/pgaskin.yaml b/src/versions/4.22.15268/nickel.yaml/pgaskin.yaml
index 18caf6e..25a7bc3 100644
--- a/src/versions/4.22.15190/nickel.yaml/pgaskin.yaml
+++ b/src/versions/4.22.15268/nickel.yaml/pgaskin.yaml
[... cut ...]
+  - BaseAddress: 0x1484AE1
   - ReplaceZlib:
       Find: "#lblForgotPin[qApp_deviceIsDragon=true],\n#lblSignOut[qApp_deviceIsDragon=true] {\n  font-size: 26px;\n}"
       Replace: "#lblForgotPin,#lblSignOut{qproperty-visible:false;}"
-  - BaseAddress: 0x1409CDC
+  - BaseAddress: 0x14939B8
   - ReplaceZlib:
       Find: "#lblForgotPin[qApp_deviceIsDragon=true],\n#lblSignOut[qApp_deviceIsDragon=true] {\n  font-size: 26px;\n}"
       Replace: "#lblForgotPin,#lblSignOut{qproperty-visible:false;}"
```

pgaskin, 2020-07-15:
When booting for the first time after the update, there's a pop-up about a free 30-day Kobo+ trial.

pgaskin, 2020-07-15:
I'm not sure if this is new or not (it's the first time I've seen it), but the top-right of the homescreen with <2 books is showing the current author rather than recommendations for me right now.

pgaskin, 2020-07-15:
The bug where the PIN entry would change sizes after entering the first digit has been fixed.

pgaskin, 2020-07-15:
Kobo Plus also now has it's own item in the menu.

pgaskin, 2020-07-15:
Released in [v64](https://github.com/pgaskin/kobopatch-patches/releases/tag/v64).

closed 2020-07-15 by pgaskin

jackiew1, 2020-07-16:
@pgaskin 
Sorry, I've only just seen the new fw release as I've been offline all day. 

Reading this post it doesn't look as if I have much (or even anything???) to do but I'll do my standard new fw routine tomorrow, but it probably won't be until the afternoon.

pgaskin, 2020-07-16:
> Reading this post it doesn't look as if I have much (or even anything???) to do

Yes, that's correct.

jackiew1, 2020-07-16:
> There isn't anything new in the CSS other than doing the same thing with the games as the other newer devices.
> 
> ```
> $ grep -C3 deviceIsLuna 15268/cssextract.out.css
> #spacerWidget[devModeOn=false][qApp_deviceIsPika=true] {
>   qproperty-visible: false;
> }
> #boggleContainer[devModeOn=false][qApp_deviceIsLuna=true],
> #solitaireContainer[devModeOn=false][qApp_deviceIsLuna=true],
> #rushHourContainer[devModeOn=false][qApp_deviceIsLuna=true],
> #sudokuContainer[devModeOn=false][qApp_deviceIsLuna=true],
> #sketchPadContainer[devModeOn=false][qApp_deviceIsLuna=true],
> #spacerWidget[devModeOn=false][qApp_deviceIsLuna=true] {
>   qproperty-visible: false;
> }
> #boggleContainer[devModeOn=false][qApp_deviceIsAlyssum=true],
> ```

This is the only change of any kind I can see in the whole 193 css streams (zlib and nozlib) in nickel. 

Nothing to see here, move along. I suppose there may be another fw update when the Nia actually ships.