pgaskin/kobopatch-patches #69: Patches for 4.23.15505

pgaskin, 2020-08-10:
- [ ] Upload testdata (@pgaskin)
- [ ] Update [pgaskin/NickelMenu](https://github.com/pgaskin/NickelMenu) symbol tests (@pgaskin)
- [ ] Find libnickel offsets where necessary (@pgaskin)
- [ ] Update patches (@pgaskin)
- TBD: More stuff about 4.23.15505
- [ ] Test nickel patches (@jackiew1)
- [ ] Test most patches (@pgaskin)
- [ ] Make release (@pgaskin)
- [ ] MR post (@pgaskin)

jackiew1, 2020-08-10:
@pgaskin Seen your new message early for once.

pgaskin, 2020-08-10:
I hope to release the patches tomorrow night (EDT) or the next day, but NickelMenu will take priority for me unless the patch updates are mostly trivial and the CSS ones don't need to be rewritten.

jackiew1, 2020-08-10:
6 new nickel CSS streams (4 zlib, 2 nozlib). More info as and when.

jackiew1, 2020-08-10:
Early warning for my nickel patches. These 2 no longer working:
- `Custom collection/author header title font`
- `Custom new header/footer font`

Will look at details later.

pgaskin, 2020-08-10:
reserved

pgaskin, 2020-08-10:
Sorted zlib CSS diff:

<details><summary>

**$ diff -yW300 <(sort 15268/cssextract.out.css) <(sort 15505/cssextract.out.css) | grep -B1 -e '<' -e '>'**

</summary>

```

																		     >
																		     >
																		     >
																		     >
																		     >
																		     >
																		     >
																		     >
--
}																			}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
																		     >	}
[... 2439 lines cut ...]
#upgradeBannerContainer[qApp_deviceIsDragon=true] {													#upgradeBannerContainer[qApp_deviceIsDragon=true] {
																		     >	#upgradeBannerContainer[qApp_deviceIsDragon=true] {
																		     >	#upgradeBannerContainer[qApp_deviceIsPhoenix=true] {
--
#upgradeBannerContainer[qApp_deviceIsStorm=true] {													#upgradeBannerContainer[qApp_deviceIsStorm=true] {
																		     >	#upgradeBannerContainer[qApp_deviceIsStorm=true] {
																		     >	#upgradeBannerContainer[qApp_deviceIsTrilogy=true] {
--
#upgradeLbl[localeName="zh-TW"] {															#upgradeLbl[localeName="zh-TW"] {
#upgradeLbl[qApp_deviceIsAlyssum=true] {													     <
--
#upgradeLbl[qApp_deviceIsDragon=true] {															#upgradeLbl[qApp_deviceIsDragon=true] {
#upgradeLbl[qApp_deviceIsNova=true] {														     <
--
#WirelessQuickSettingsPopup {																#WirelessQuickSettingsPopup {
#WirelessQuickSettingsPopup > widget {															#WirelessQuickSettingsPopup > widget {
--
/* zlib stream: offset_hex(0x9FF336) offset_int32(10482486) len_int32(78810) sha1(b3ccd2d78fe76ba1e839654b35f833729ba416d7) */			     |	/* zlib stream: offset_hex(0x14C43D4) offset_int32(21775316) len_int32(1644) sha1(7f8e40a0082972f3f416b8bdd127497a3f76f6ae) */
																		     >	/* zlib stream: offset_hex(0x285F2) offset_int32(165362) len_int32(3744) sha1(2ff51c25daf4c135c9aeda24e110677892616a6a) */
																		     >	/* zlib stream: offset_hex(0x83E41D) offset_int32(8643613) len_int32(79838) sha1(a89addf9292797a18e2ac1b1677326ba274d3f14) */
																		     >	/* zlib stream: offset_hex(0x91889) offset_int32(596105) len_int32(5429) sha1(78b1dc5005c5d97476629900f7016ec1506de111) */
																		     >	/* zlib stream: offset_hex(0x91E34) offset_int32(597556) len_int32(17588) sha1(a0e7230b9214a010f0cd94bd11ba5ce54a1dc907) */
```

</details>

pgaskin, 2020-08-10:
Symbols:

<details><summary>

**$ diff -y <(cat 15268/symdump.out.json | jq -r '.[] | .Name' | sort) <(cat 15505/symdump.out.json | jq -r '.[] | .Name' | sort) | grep -e '<' -e '>'**

</summary>

```
							      >	__ctype_b_loc
							      >	__ctype_tolower_loc
_Z27qRegisterNormalizedMetaTypeI4QSetI7QStringEEiRK10QByteArr <
_ZGVZN9QtPrivate15ConnectionTypesINS_4ListII5QListIiEEEELb1EE <
							      >	_ZGVZN9QtPrivate15ConnectionTypesINS_4ListII7QVectorIiEEEELb1
							      >	_ZGVZN9QtPrivate19ValueTypeIsMetaTypeI7QVectorIiELb1EE17regis
							      >	_ZGVZN9QtPrivate30AssociativeValueTypeIsMetaTypeI4QMapI15QDBu
							      >	_ZGVZN9QtPrivate30AssociativeValueTypeIsMetaTypeI4QMapI15QDBu
							      >	_ZN10IconButton10tapGestureEP15GestureReceiverP10TapGesture
							      >	_ZN10IconButton11qt_metacallEN11QMetaObject4CallEiPPv
							      >	_ZN10IconButton11qt_metacastEPKc
							      >	_ZN10IconButton13defaultColorsEv
							      >	_ZN10IconButton13reverseColorsEv
							      >	_ZN10IconButton16staticMetaObjectE
							      >	_ZN10IconButton5eventEP6QEvent
							      >	_ZN10IconButton6tappedEv
							      >	_ZN10IconButton7setTextERK7QString
							      >	_ZN10IconButton9setPixmapERK7QPixmap
							      >	_ZN10IconButtonC1EP7QWidget
							      >	_ZN10IconButtonC2EP7QWidget
							      >	_ZN10IconButtonD0Ev
							      >	_ZN10IconButtonD1Ev
							      >	_ZN10IconButtonD2Ev
_ZN10N3WebStore12titleChangedERK7QString		      <
_ZN10PowerTimer13onAlarmWakeupEv			      <
							      >	_ZN10PowerTimer5alarmEv
							      >	_ZN10QScrollBar10wheelEventEP11QWheelEvent
							      >	_ZN10QScrollBar11qt_metacallEN11QMetaObject4CallEiPPv
							      >	_ZN10QScrollBar11qt_metacastEPKc
							      >	_ZN10QScrollBar12sliderChangeEN15QAbstractSlider12SliderChang
							      >	_ZN10QScrollBar14mouseMoveEventEP11QMouseEvent
							      >	_ZN10QScrollBar15mousePressEventEP11QMouseEvent
							      >	_ZN10QScrollBar16contextMenuEventEP17QContextMenuEvent
							      >	_ZN10QScrollBar16staticMetaObjectE
							      >	_ZN10QScrollBar17mouseReleaseEventEP11QMouseEvent
							      >	_ZN10QScrollBar5eventEP6QEvent
							      >	_ZN10QScrollBar9hideEventEP10QHideEvent
							      >	_ZN10QScrollBarC2EP7QWidget
							      >	_ZN10QScrollBarD2Ev
_ZN11DropboxView14hideBackButtonEv			      <
_ZN11DropboxView22showBackButtonWithTextERK7QString	      <
							      >	_ZN11MainNavView11qt_metacallEN11QMetaObject4CallEiPPv
							      >	_ZN11MainNavView11qt_metacastEPKc
							      >	_ZN11MainNavView13setHomeActiveEb
							      >	_ZN11MainNavView13setMoreActiveEb
							      >	_ZN11MainNavView14setBooksActiveEb
							      >	_ZN11MainNavView16staticMetaObjectE
							      >	_ZN11MainNavView17setDiscoverActiveEb
							      >	_ZN11MainNavView20updateVisibleButtonsEv
							      >	_ZN11MainNavView4homeEv
							      >	_ZN11MainNavView4moreEv
							      >	_ZN11MainNavView5booksEv
							      >	_ZN11MainNavView8discoverEv
							      >	_ZN11MainNavViewC1EP7QWidget
							      >	_ZN11MainNavViewC2EP7QWidget
							      >	_ZN11MainNavViewD0Ev
							      >	_ZN11MainNavViewD1Ev
							      >	_ZN11MainNavViewD2Ev
							      >	_ZN11N3TabWidget8hideLineEv
_ZN11QMetaTypeIdI5QListIiEE14qt_metatype_idEv		      <
							      >	_ZN11QMetaTypeIdI7QVectorIiEE14qt_metatype_idEv
[... 1195 lines cut ...]
_ZTV21NewHomePageController				      <
							      >	_ZTV22AbstractSliderDelegate
							      >	_ZTV22ParentalControlManager
							      >	_ZTV23LibraryEmptyStateWidget
							      >	_ZTV23ScrollableContentWidgetI11QGridLayoutE
							      >	_ZTV23ScrollableContentWidgetI11QVBoxLayoutE
_ZTV23StatusBarMenuController				      <
							      >	_ZTV25AbstractSliderTouchFilter
							      >	_ZTV27ScrollableContentWidgetBase
							      >	_ZTV8MoreView
							      >	_ZTV9TimeLabel
							      >	_ZTVN22N3PowerWorkflowManager23OffAlarmControlDelegateE
							      >	_ZTVN22N3PowerWorkflowManager25SleepAlarmControlDelegateE
							      >	_ZZN11QMetaTypeIdI15QDBusObjectPathE14qt_metatype_idEvE11meta
							      >	_ZZN11QMetaTypeIdI4QMapI15QDBusObjectPathS0_I7QString8QVarian
							      >	_ZZN11QMetaTypeIdI4QMapI15QDBusObjectPathS0_I7QStringS0_IS2_8
							      >	_ZZN11QMetaTypeIdI7QVectorIiEE14qt_metatype_idEvE11metatype_i
_ZZN9QtPrivate15ConnectionTypesINS_4ListII5QListIiEEEELb1EE5t <
							      >	_ZZN9QtPrivate15ConnectionTypesINS_4ListII7QVectorIiEEEELb1EE
							      >	_ZZN9QtPrivate19ValueTypeIsMetaTypeI7QVectorIiELb1EE17registe
							      >	_ZZN9QtPrivate30AssociativeValueTypeIsMetaTypeI4QMapI15QDBusO
							      >	_ZZN9QtPrivate30AssociativeValueTypeIsMetaTypeI4QMapI15QDBusO
```

</details>

pgaskin, 2020-08-10:
A lot more breakage than usual:

```
Errors:
  /home/patrick/src/kobopatch-patches/src/versions/4.23.15505/libnickel.so.1.0.0.yaml/pgaskin.yaml: Hide top-left menu item - Kobo Plus: could not apply patch "Hide top-left menu item - Kobo Plus": line 393: inst 3: BaseAddress: resolve address (kobopatch.BaseAddress{Offset:(*int32)(nil), Sym:(*string)(0xc0043c7e20), SymPLT:(*string)(nil), SymPLTTail:(*string)(nil), Inline:false, Rel:(*int32)(0xc004506f98)}): ResolveSym("StatusBarMenuController::updateParentControlSensitiveMenuItems()"): no such symbol "StatusBarMenuController::updateParentControlSensitiveMenuItems()"
  /home/patrick/src/kobopatch-patches/src/versions/4.23.15505/libnickel.so.1.0.0.yaml/pgaskin.yaml: Hide top-left menu item - Wishlist: could not apply patch "Hide top-left menu item - Wishlist": line 371: inst 3: BaseAddress: resolve address (kobopatch.BaseAddress{Offset:(*int32)(nil), Sym:(*string)(0xc00a26c7b0), SymPLT:(*string)(nil), SymPLTTail:(*string)(nil), Inline:false, Rel:(*int32)(0xc004374d88)}): ResolveSym("StatusBarMenuController::updateParentControlSensitiveMenuItems()"): no such symbol "StatusBarMenuController::updateParentControlSensitiveMenuItems()"
  /home/patrick/src/kobopatch-patches/src/versions/4.23.15505/libnickel.so.1.0.0.yaml/pgaskin.yaml: Hide top-left menu item - Pocket: could not apply patch "Hide top-left menu item - Pocket": line 337: inst 3: BaseAddress: resolve address (kobopatch.BaseAddress{Offset:(*int32)(nil), Sym:(*string)(0xc00439ac30), SymPLT:(*string)(nil), SymPLTTail:(*string)(nil), Inline:false, Rel:(*int32)(0xc0043752b8)}): ResolveSym("StatusBarMenuController::updateParentControlSensitiveMenuItems()"): no such symbol "StatusBarMenuController::updateParentControlSensitiveMenuItems()"
  /home/patrick/src/kobopatch-patches/src/versions/4.23.15505/libnickel.so.1.0.0.yaml/pgaskin.yaml: DeveloperSettings - AutoUsbGadget: could not apply patch "DeveloperSettings - AutoUsbGadget": line 720: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.23.15505/libnickel.so.1.0.0.yaml/pgaskin.yaml: Hide top-left menu item - Dropbox: could not apply patch "Hide top-left menu item - Dropbox": line 327: inst 3: BaseAddress: resolve address (kobopatch.BaseAddress{Offset:(*int32)(nil), Sym:(*string)(0xc0043c7420), SymPLT:(*string)(nil), SymPLTTail:(*string)(nil), Inline:false, Rel:(*int32)(0xc004506e60)}): ResolveSym("StatusBarMenuController::updateParentControlSensitiveMenuItems()"): no such symbol "StatusBarMenuController::updateParentControlSensitiveMenuItems()"
  /home/patrick/src/kobopatch-patches/src/versions/4.23.15505/libnickel.so.1.0.0.yaml/pgaskin.yaml: Hide top-left menu item - Recommended: could not apply patch "Hide top-left menu item - Recommended": line 357: inst 3: BaseAddress: resolve address (kobopatch.BaseAddress{Offset:(*int32)(nil), Sym:(*string)(0xc0045aa7d0), SymPLT:(*string)(nil), SymPLTTail:(*string)(nil), Inline:false, Rel:(*int32)(0xc004506760)}): ResolveSym("StatusBarMenuController::updateParentControlSensitiveMenuItems()"): no such symbol "StatusBarMenuController::updateParentControlSensitiveMenuItems()"
  /home/patrick/src/kobopatch-patches/src/versions/4.23.15505/libnickel.so.1.0.0.yaml/pgaskin.yaml: Swap reading header/footer: could not apply patch "Swap reading header/footer": line 604: inst 5: ReplaceBytes: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.23.15505/libnickel.so.1.0.0.yaml/pgaskin.yaml: Customize ComfortLight settings: could not apply patch "Customize ComfortLight settings": line 649: inst 4: ReplaceInt: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.23.15505/libnickel.so.1.0.0.yaml/pgaskin.yaml: Change Browse Kobo home screen link target - Articles: could not apply patch "Change Browse Kobo home screen link target - Articles": line 213: inst 4: ReplaceBytes: expand ReplaceInstB=kobopatch.FlexAbsOffset{Offset:(*int32)(nil), Sym:(*string)(nil), SymPLT:(*string)(nil), SymPLTTail:(*string)(0xc00a26dc70), Inline:false, Rel:(*int32)(nil)}: ResolveSymPLTTail("LibraryNavMixin::showPocketLibrary()") = &patchlib.dynsym{Name:"_ZN15LibraryNavMixin17showPocketLibraryEv", Offset:0xb115a4, Index:0x90a0, Type:elf.STT_FUNC, OffsetGOT:0x11cec1c, OffsetPLT:0x5160b0, OffsetPLTTail:0x0, Demangled:"LibraryNavMixin::showPocketLibrary()"}: no tail stub before PLT entry
  /home/patrick/src/kobopatch-patches/src/versions/4.23.15505/libnickel.so.1.0.0.yaml/pgaskin.yaml: Hide top-left menu item - Store: could not apply patch "Hide top-left menu item - Store": line 347: inst 3: BaseAddress: resolve address (kobopatch.BaseAddress{Offset:(*int32)(nil), Sym:(*string)(0xc000057dd0), SymPLT:(*string)(nil), SymPLTTail:(*string)(nil), Inline:false, Rel:(*int32)(0xc004374c50)}): ResolveSym("StatusBarMenuController::updateParentControlSensitiveMenuItems()"): no such symbol "StatusBarMenuController::updateParentControlSensitiveMenuItems()"
  /home/patrick/src/kobopatch-patches/src/versions/4.23.15505/libnickel.so.1.0.0.yaml/pgaskin.yaml: Larger Sleep/Power-off timeouts: could not apply patch "Larger Sleep/Power-off timeouts": line 491: inst 4: ReplaceBytes: expand Base=kobopatch.FlexAbsOffset{Offset:(*int32)(nil), Sym:(*string)(0xc00440ccf0), SymPLT:(*string)(nil), SymPLTTail:(*string)(nil), Inline:true, Rel:(*int32)(nil)}: ResolveSym("N3PowerWorkflowManager::configureWakeup(AlarmControl*, int, char const*)"): no such [... cut ...] ndBaseAddress: could not find bytes
Tests for 4.23.15505 failed (exit status 1)! Stopping.
```

But, most are for the home screen and old main menu. The home screen patches will probably need to be rewritten. The menu patches will be removed entirely, since they aren't really needed anymore (the items in the More menu are unobtrusive).

@NiLuJe, if you don't get to `Don't grab exclusive access to event0: could not apply patch "Don't grab exclusive access to event0` first, I'll take a look later.

pgaskin, 2020-08-10:
Changes to `_ZN22N3PowerWorkflowManager15configureWakeupEP12AlarmControliPKc` mean the issue with timeouts larger than one day might be fixed:

Old:

```c
int _ZN22N3PowerWorkflowManager15configureWakeupEP12AlarmControliPKc(void * arg0, int arg1, int * arg2) {
    stack[-4] = r4;
    stack[-8] = r5;
    stack[-12] = r6;
    stack[-16] = r7;
    stack[-20] = r8;
    stack[-24] = r9;
    stack[-28] = lr;
    r7 = (sp - 0x28) + 0x0;
    r4 = r7 + 0x8;
    r0 = PowerManager::sharedInstance();
    r0 = PowerManager::timeLastUsed();
    r2 = arg0;
    r8 = r0 + 0xea60 * arg2;
    r3 = r3;
    r9 = arg1 + 0xea60 * arg2 / 0x80000000 + CARRY(FLAGS);
    r0 = arg1;
    asm { strd       r8, sb, [r4, #-0x8]! };
    r1 = r4;
    r0 = AlarmControl::setWakeupTimeUTC();
    sp = r7 + 0xc;
    return r0;
}
```

New:

```c
int _ZN22N3PowerWorkflowManager15configureWakeupEP12AlarmControliP20AlarmControlDelegate(void * arg0, int arg1, void * arg2) {
    stack[-4] = r4;
    stack[-8] = r5;
    stack[-12] = r6;
    stack[-16] = r7;
    stack[-20] = r8;
    stack[-24] = r9;
    stack[-28] = lr;
    r7 = (sp - 0x28) + 0x0;
    r0 = PowerManager::sharedInstance();
    r4 = PowerManager::timeLastUsed() + 0xea60 * arg2;
    r5 = arg1 + 0xea60 * arg2 / 0x80000000 + CARRY(FLAGS);
    r4 = r4 - QDateTime::currentMSecsSinceEpoch();
    r5 = r5 - arg1 + !CARRY(FLAGS);
    r1 = r7 + 0x8;
    r3 = r3;
    r2 = arg0;
    r0 = arg1;
    asm { strd       r4, r5, [r1, #-0x8]! };
    r0 = AlarmControl::addWakeupInterval();
    sp = r7 + 0xc;
    return r0;
}
```

pgaskin, 2020-08-10:
Larger timeouts patch changes:

<details><summary>

**$ diff docs/lgrpwroff-1{4,5}\***

</summary>

```diff
7c7
<     <title>Larger Sleep/Power-off timeouts patch customizer (Firmware 4.20.14601+)</title>
---
>     <title>Larger Sleep/Power-off timeouts patch customizer (Firmware 4.23.15505+)</title>
200c200
<       few weeks). To customize this patch, see https://pgaskin.net/kobopatch-patches/lgrpwroff-14601.html .
---
>       few weeks). To customize this patch, see https://pgaskin.net/kobopatch-patches/lgrpwroff-15505.html .
217,221c217,224
<   # Multiplier in N3PowerWorkflowManager::configureWakeup
<   - ReplaceBytesAtSymbol: {Symbol: "N3PowerWorkflowManager::configureWakeup(AlarmControl*, int, char const*)", Offset: 11, FindH: 04 FB 05 F5, ReplaceH: <b><u>{{ data[cur].insts["lsl r5, r5"] }}</u></b>} # mul r5(dest), r4(multiplier), r5(mins)  -> lsl r5, r5, <b><u>#{{ cur }}</u></b>
<   # Multiplier in N3PowerWorkflowManager::pollBattery
<   - ReplaceBytesAtSymbol: {Symbol: "N3PowerWorkflowManager::pollBattery()", Offset: 105, FindH: 08 FB 00 F8, ReplaceH: <b><u>{{ data[cur].insts["lsl r8, r0"] }}</u></b>} # mul r8(dest), r8(multiplier), r0(mins)  -> lsl r8, r0, <b><u>#{{ cur }}</u></b>
<   - ReplaceBytesAtSymbol: {Symbol: "N3PowerWorkflowManager::pollBattery()", Offset: 155, FindH: 02 FB 00 F2, ReplaceH: <b><u>{{ data[cur].insts["lsl r2, r0"] }}</u></b>} # mul r2(dest), r2(multiplier), r0(mins)  -> lsl r2, r0, <b><u>#{{ cur }}</u></b>
---
>   # Multipliers
>   - ReplaceBytes:
>       Base: "N3PowerWorkflowManager::configureWakeup(AlarmControl*, int, char const*)"
>       Offset: 28
>       FindH: 03 FB 04 F2    # MUL r2(dest), r3(multiplier), r4(mins)
>       ReplaceH: <b><u>{{ data[cur].insts["lsl r2, r4"] }}</u></b> # LSL r2, r4, <b><u>#{{ cur }}</u></b>
>   - ReplaceBytes: {Base: "N3PowerWorkflowManager::pollBattery()", Offset: 106, FindH: 08 FB 00 F8, ReplaceH: <b><u>{{ data[cur].insts["lsl r8, r0"] }}</u></b>} # MUL r8(dest), r8(multiplier), r0(mins) -> LSL r8, r0, <b><u>#{{ cur }}</u></b>
>   - ReplaceBytes: {Base: "N3PowerWorkflowManager::pollBattery()", Offset: 156, FindH: 02 FB 00 F2, ReplaceH: <b><u>{{ data[cur].insts["lsl r2, r0"] }}</u></b>} # MUL r2(dest), r2(multiplier), r0(mins) -> LSL r2, r0, <b><u>#{{ cur }}</u></b>
255c258
<                 data: [{"insts":{"lsl r2, r0":"4F EA 00 02","lsl r5, r5":"4F EA 05 05","lsl r8, r0":"4F EA 00 08"},"values":["","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",""]},{"insts":{"lsl r2, r0":"4F EA 40 02","lsl r5, r5":"4F EA 45 05","lsl r8, r0":"4F EA 40 08"},"values":["","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",""]},{"insts":{"lsl r2, r0":"4F EA 80 02","lsl r5, r5":"4F EA 85 05","lsl r8, r0":"4F EA 80 08"},"values":["","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",""]},{"insts":{"lsl r2, r0":"4F EA C0 02","lsl r5, r5":"4F EA C5 05","lsl r8, r0":"4F EA C0 08"},"values":["","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","" [... cut ...]
>                     "4FEA8472", // lsl r2, r4, #30
>                     "4FEAC472", // lsl r2, r4, #31
369a374
>                 // for d in {0..31}; do echo "\"$(kstool thumb "lsl r8, r0, #$d" | cut -d '[' -f2 | cut -d ']' -f1 | tr -d ' ' | tr '[:lower:]' '[:upper:]')\", // lsl r8, r0, #$d"; done
403a409
>                 // for d in {0..31}; do echo "\"$(kstool thumb "lsl r2, r0, #$d" | cut -d '[' -f2 | cut -d ']' -f1 | tr -d ' ' | tr '[:lower:]' '[:upper:]')\", // lsl r2, r0, #$d"; done
```

</details>

pgaskin, 2020-08-10:
@NiLuJe, it was just a change from `%0:keymap=keys/device.qmap:grab=0` to `%0::keymap=keys/device.qmap::grab=`.

pgaskin, 2020-08-10:
@jackiew1, you have two patches which need updating. In addition, can you test the new sizes for `Increase home screen cover size` on your devices?

pgaskin, 2020-08-10:
I've updated most of the patches, but it hasn't been tested at all yet.

jackiew1, 2020-08-10:
> @jackiew1, you have two patches which need updating. In addition, can you test the new sizes for `Increase home screen cover size` on your devices?

`Increase home screen cover size` isn't one of mine so I'm not very familiar with it. If you want it tested on several different devices that's going to take me quite a long time to bring the devices up to current firmware, so it won't be tonight. The earliest I can start will be tomorrow late afternoon as I have to go out.

Update on the 2 patches of mine ...
> * `Custom collection/author header title font`
> * `Custom new header/footer font`

I think I've resuscitated these. I'll commit to my GH nickel.yaml file soon-ish.

jackiew1, 2020-08-10:
re: nickel patches `Series list increase cover thumbnails` & `Reduce height of page navigator arrows`

There's no indication so far that these aren't working but given the amount of faffing about I had to do originally to stop the bottom Series cover thumbnail being overlapped by the (now non-existent) navigation arrows I'm probably going to need to give them a closer look once I've got all my old devices updated to 4.23.15505. The 2nd patch may not be needed anymore.

pgaskin, 2020-08-10:
> The 2nd patch may not be needed anymore.

That's probably right. The new nav bar has replaced them everywhere AFAICT.

jackiew1, 2020-08-11:
> re: nickel patches `Series list increase cover thumbnails` & `Reduce height of page navigator arrows`
>
> There's no indication so far that these aren't working but given the amount of faffing about I [...]

Sigh... Based on initial testing on a KA1 I'm pretty sure I'm going to need to re-work all the cover thumbnail sizes in patch ``Series list increase cover thumbnails` as the bottom edge of each one has a few pixels shaved off. Deep joy.

pgaskin, 2020-08-11:
> I'm pretty sure I'm going to need to re-work all the cover thumbnail sizes in patch `Series list increase cover thumbnails`

Do you think you'll get to that today/tomorrow, or do you want me to leave it (as-is or removed) for the next version of the patches. I'll probably be doing the next version of the patches a week or two later, since I'm planning on making a few patches to clean up the new UI (maybe one to remove the tab bar line if it looks good that way, probably one to increase the clock size for people who have a hard time reading it, and one to tweak the padding on various views to make space for more content).

jackiew1, 2020-08-11:
> Do you think you'll get to that today/tomorrow, or do you want me to leave it (as-is or removed) for the next version of the patches. I'll probably be doing the next version of the patches a week or [...]

Ask me again later. I've already started the work and am testing it at the same time as the Home screen cover size patch you asked about. So far the Series list patch is painfully slow going.

None of this is helped by the fact it's about 90degF today and Brits with no air-conditioning are not designed for these conditions. Tetchy doesn't begin to describe my current mood.

jackiew1, 2020-08-11:
> @jackiew1, ... can you test the new sizes for `Increase home screen cover size` on your devices?

Tested on Glo, Aura6", AuraHD, H2O, ClaraHD, KA1, Forma. As far as I can see there are no cut-offs or overlaps with your updated patch.

pgaskin, 2020-08-11:
> Tested on Glo, Aura6", AuraHD, H2O, ClaraHD, KA1, Forma. As far as I can see there are no cut-offs or overlaps with your updated patch.

OK, thanks.

I've also added a new patch to remove the line from the bottom tab bar, and it really makes things look less cluttered.

> So far the Series list patch is painfully slow going.

Do you want to leave that for the next version?

pgaskin, 2020-08-11:
I have finished testing some of the nickel patches (mainly the ones I use, plus the ones which needed to be updated) and most of the libnickel patches. @jackiew1, I'm ready to release it whenever you are.

jackiew1, 2020-08-11:
> > So far the Series list patch is painfully slow going.
> 
> Do you want to leave that for the next version?

No, it's OK, I've just committed the updated patch. 

Please will you test on your Trilogy device before releasing? This time you're checking to see whether the top 4 thumbnails have had their bottom edge shaved off. The 5th/bottom thumbnail has more room than the other 4 so I don't think it will be shaved.

Tested by me on Glo, Aura6", AuraHD, H2O, ClaraHD, KA1, Forma. I'm going to risk that the new LibraH2O values will be OK. I'll correct it if/when someone reports a problem.

Given the puny changes I eventually made, that took a disproportionate amount of time. 😄

pgaskin, 2020-08-11:
> Please will you test on your Trilogy device before releasing

OK, I'll do that after I finish organizing the changes for NickelMenu.

> Given the puny changes I eventually made, that took a disproportionate amount of time.

This entire firmware update has taken a disproportionate amount of time ... it's almost purely cosmetic changes and a few hundred new symbols, yet it's taken me almost as long as one of the big versions from last September (but to be fair, that was because I needed to figure out stuff related to the menus and touch input from scratch for NickelMenu, plus the crazy bug with the nickel_open action and LibraryNavMixin). At least what I learned about touch input will be useful for future mods (it's been a mostly unsolved thing until now).

pgaskin, 2020-08-11:
Released in [v65](https://github.com/pgaskin/kobopatch-patches/releases/tag/v65).

closed 2020-08-11 by pgaskin