pgaskin/kobopatch-patches #72: Patches for 4.23.15548

pgaskin, 2020-08-19:
- [ ] Upload testdata (@pgaskin)
- [ ] Extract resources for @jackiew1 (@pgaskin)
- [ ] Find libnickel offsets where necessary (@pgaskin)
- [ ] Update patches (@pgaskin)
- [ ] Update nickel patches if necessary (@jackiew1)
- [ ] Test most patches (@pgaskin)
- [ ] Make release (@pgaskin)
- [ ] MR post (@pgaskin)

pgaskin, 2020-08-19:
<details><summary>

`$ diff -y <(cat 15505/symdump.out.json | jq -r '.[] | .Name' | sort) <(cat 15548/symdump.out.json | jq -r '.[] | .Name' | sort) | grep -e '<' -e '>'`

</summary>

```
							      >	_ZN19ScrollBarTouchLabel10paintEventEP11QPaintEvent
							      >	_ZN19ScrollBarTouchLabel11qt_metacallEN11QMetaObject4CallEiPP
							      >	_ZN19ScrollBarTouchLabel11qt_metacastEPKc
							      >	_ZN19ScrollBarTouchLabel16staticMetaObjectE
							      >	_ZN19ScrollBarTouchLabel9setGreyedEb
							      >	_ZN19ScrollBarTouchLabelC1EP7QWidget
							      >	_ZN19ScrollBarTouchLabelC2EP7QWidget
							      >	_ZN19ScrollBarTouchLabelD0Ev
							      >	_ZN19ScrollBarTouchLabelD1Ev
							      >	_ZN19ScrollBarTouchLabelD2Ev
							      >	_ZN27ScrollableContentWidgetBase12goToNextPageEv
							      >	_ZN27ScrollableContentWidgetBase16goToPreviousPageEv
							      >	_ZN9QtPrivate11QSlotObjectIM27ScrollableContentWidgetBaseFvvE
							      >	_ZN9QtPrivate11QSlotObjectIM27ScrollableContentWidgetBaseFvvE
							      >	_ZNK19ScrollBarTouchLabel10metaObjectEv
							      >	_ZNK27ScrollableContentWidgetBase14getCurrentPageEv
_ZThn124_N16N3BrowserWebView17webViewTapHandledEb	      <
_ZThn124_N16N3BrowserWebViewD0Ev			      <
_ZThn124_N16N3BrowserWebViewD1Ev			      <
_ZThn132_N10N3WebStore17webViewTapHandledEb		      <
_ZThn132_N10N3WebStoreD0Ev				      <
_ZThn132_N10N3WebStoreD1Ev				      <
							      >	_ZThn148_N16N3BrowserWebView17webViewTapHandledEb
							      >	_ZThn148_N16N3BrowserWebViewD0Ev
							      >	_ZThn148_N16N3BrowserWebViewD1Ev
							      >	_ZThn156_N10N3WebStore17webViewTapHandledEb
							      >	_ZThn156_N10N3WebStoreD0Ev
							      >	_ZThn156_N10N3WebStoreD1Ev
_ZThn164_N10QImageViewD0Ev				      <
_ZThn164_N10QImageViewD1Ev				      <
_ZThn164_N19WebElementImageViewD0Ev			      <
_ZThn164_N19WebElementImageViewD1Ev			      <
_ZThn164_N9ImageView5eventEP6QEvent			      <
_ZThn164_N9ImageViewD0Ev				      <
_ZThn164_N9ImageViewD1Ev				      <
_ZThn176_N19WebElementImageView14allowDoubleTapEP15GestureRec <
_ZThn176_N9ImageView10tapGestureEP15GestureReceiverP10TapGest <
_ZThn176_N9ImageView11dragGestureEP15GestureReceiverP11DragGe <
_ZThn176_N9ImageView12pinchGestureEP15GestureReceiverP12Pinch <
							      >	_ZThn188_N10QImageViewD0Ev
							      >	_ZThn188_N10QImageViewD1Ev
							      >	_ZThn188_N19WebElementImageViewD0Ev
							      >	_ZThn188_N19WebElementImageViewD1Ev
							      >	_ZThn188_N9ImageView5eventEP6QEvent
							      >	_ZThn188_N9ImageViewD0Ev
							      >	_ZThn188_N9ImageViewD1Ev
							      >	_ZThn200_N19WebElementImageView14allowDoubleTapEP15GestureRec
							      >	_ZThn200_N9ImageView10tapGestureEP15GestureReceiverP10TapGest
							      >	_ZThn200_N9ImageView11dragGestureEP15GestureReceiverP11DragGe
							      >	_ZThn200_N9ImageView12pinchGestureEP15GestureReceiverP12Pinch
							      >	_ZThn32_N19ScrollBarTouchLabelD0Ev
							      >	_ZThn32_N19ScrollBarTouchLabelD1Ev
							      >	_ZThn76_N19ScrollBarTouchLabelD0Ev
							      >	_ZThn76_N19ScrollBarTouchLabelD1Ev
							      >	_ZThn8_N19ScrollBarTouchLabelD0Ev
							      >	_ZThn8_N19ScrollBarTouchLabelD1Ev
							      >	_ZTI19ScrollBarTouchLabel
							      >	_ZTS19ScrollBarTouchLabel
							      >	_ZTV19ScrollBarTouchLabel
```

</details>

jackiew1, 2020-08-19:
@pgaskin I've taken a quick look. There are a few changes to nickel CSS (in default.qss, ScrollableContentWidget.qss) but, as far as I can tell, nothing that was currently being patched.

pgaskin, 2020-08-19:
```
Errors:
  /home/patrick/src/kobopatch-patches/src/versions/4.23.15548/libnickel.so.1.0.0.yaml/pgaskin.yaml: DeveloperSettings - AutoUsbGadget: could not apply patch "DeveloperSettings - AutoUsbGadget": line 658: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.23.15548/libnickel.so.1.0.0.yaml/pgaskin.yaml: Customize ComfortLight settings: could not apply patch "Customize ComfortLight settings": line 587: inst 4: ReplaceInt: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.23.15548/libnickel.so.1.0.0.yaml/pgaskin.yaml: Allow rotation on all devices: could not apply patch "Allow rotation on all devices": line 453: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.23.15548/libnickel.so.1.0.0.yaml/geoffr.yaml: Set font scale factor: could not apply patch "Set font scale factor": line 537: inst 3: ReplaceInt: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.23.15548/libnickel.so.1.0.0.yaml/geoffr.yaml: Set KePub hyphenation: could not apply patch "Set KePub hyphenation": line 320: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.23.15548/nickel.yaml/pgaskin.yaml: Remove forgot pin button from lock screen: could not apply patch "Remove forgot pin button from lock screen": line 20: inst 4: ReplaceZlib: not a zlib stream
  /home/patrick/src/kobopatch-patches/src/versions/4.23.15548/nickel.yaml/oren64.yaml: New home screen subtitle custom font: could not apply patch "New home screen subtitle custom font": line 31: inst 4: ReplaceZlib: not a zlib stream
```

Nothing more than the usual updates for the ones with fixed offsets.

pgaskin, 2020-08-19:
Resource offsets:

```
EntryPoint_3
1 0x115b390 0x115af40 0x26608

EntryPoint_4
1 0x14bb880 0x14bb5f8 0x115b730

EntryPoint_5
1 0x14d7f38 0x14d5990 0x14bb998

EntryPoint_6
1 0x14ff288 0x14fe238 0x14d89e8

qInitResources_resources
1 0x115b390 0x115af40 0x26608

qInitResources_translations
1 0x14bb880 0x14bb5f8 0x115b730

qInitResources_styles
1 0x14d7f38 0x14d5990 0x14bb998

qInitResources_certificates
1 0x14ff288 0x14fe238 0x14d89e8
```

pgaskin, 2020-08-19:
<details><summary>

`$ git diff --no-index src/versions/4.23.155{05,48}`

</summary>

```diff
diff --git a/src/versions/4.23.15505/libnickel.so.1.0.0.yaml/geoffr.yaml b/src/versions/4.23.15548/libnickel.so.1.0.0.yaml/geoffr.yaml
index 8642af6..44ca920 100644
--- a/src/versions/4.23.15505/libnickel.so.1.0.0.yaml/geoffr.yaml
+++ b/src/versions/4.23.15548/libnickel.so.1.0.0.yaml/geoffr.yaml
@@ -317,8 +317,8 @@ Set KePub hyphenation:
       will always be turned on, regardless of justification button setting.
 
       (The publisher can still turn hyphenation off/on in the book's stylesheet.)
-  - ReplaceBytes: {Offset: 0xB8F15E, FindInstBLX: {SymPLT: "QString::operator==(QLatin1String) const"}, ReplaceH: 01 20 01 20} # Alternative 1: Always turn KePub hyphenation on
-# - ReplaceBytes: {Offset: 0xB8F15E, FindInstBLX: {SymPLT: "QString::operator==(QLatin1String) const"}, ReplaceH: 00 20 00 20} # Alternative 2: Never turn KePub hyphenation on
+  - ReplaceBytes: {Offset: 0xB8FB8E, FindInstBLX: {SymPLT: "QString::operator==(QLatin1String) const"}, ReplaceH: 01 20 01 20} # Alternative 1: Always turn KePub hyphenation on
+# - ReplaceBytes: {Offset: 0xB8FB8E, FindInstBLX: {SymPLT: "QString::operator==(QLatin1String) const"}, ReplaceH: 00 20 00 20} # Alternative 2: Never turn KePub hyphenation on
 
 Force user line spacing in KePubs:
   - Enabled: no
@@ -534,5 +534,5 @@ Set font scale factor:
   # Original value X=14, so original scale factor is (font size slider) / 46.
   # Allowed replacement values for X: 0-15 (smaller X --> larger scale factor)
   # Note: vmov.f32 s1, #15.0 before getReadingFontSizeScaleFactor
-  - ReplaceInt: {Offset: 0x586A7C, Find: 14, Replace: 4} # ePub
-  - ReplaceInt: {Offset: 0xB8ED34, Find: 14, Replace: 4} # KePub
+  - ReplaceInt: {Offset: 0x5874AC, Find: 14, Replace: 4} # ePub
+  - ReplaceInt: {Offset: 0xB8F764, Find: 14, Replace: 4} # KePub
diff --git a/src/versions/4.23.15505/libnickel.so.1.0.0.yaml/pgaskin.yaml b/src/versions/4.23.15548/libnickel.so.1.0.0.yaml/pgaskin.yaml
index dde0138..b2720f8 100644
--- a/src/versions/4.23.15505/libnickel.so.1.0.0.yaml/pgaskin.yaml
+++ b/src/versions/4.23.15548/libnickel.so.1.0.0.yaml/pgaskin.yaml
@@ -450,13 +450,13 @@ Allow rotation on all devices:
       Unlike only the DeveloperSettings ForceAllowLandscape option, this patch
       also makes the rotation icon show on the new reader as well.
   # Always return true for ForceAllowLandscape:
-  - ReplaceBytes: {Offset: 0x777E4E, FindInstBLX: {SymPLT: "QVariant::toBool() const"}, ReplaceH: 4F F0 01 00}
+  - ReplaceBytes: {Offset: 0x77887E, FindInstBLX: {SymPLT: "QVariant::toBool() const"}, ReplaceH: 4F F0 01 00}
   # In ReadingMenuView::ReadingMenuView, replace the value of Device::hasOrientationSensor,
   # which is passed to a function to hide/show the rotate icon:
-  - ReplaceBytes: {Offset: 0xBA60FC, FindInstBLX: {SymPLT: "Device::hasOrientationSensor() const"}, ReplaceH: 4F F0 01 00}
+  - ReplaceBytes: {Offset: 0xBA6B2C, FindInstBLX: {SymPLT: "Device::hasOrientationSensor() const"}, ReplaceH: 4F F0 01 00}
   # Also fix sizing of popup by doing the same in RotatePopup::RotatePopup, but
   # note that this will make the Automatic option show too (and do nothing):
-  - ReplaceBytes: {Offset: 0xBC29BE, FindInstBLX: {SymPLT: "Device::hasOrientationSensor() const"}, ReplaceH: 4F F0 01 00}
+  - ReplaceBytes: {Offset: 0xBC340E, FindInstBLX: {SymPLT: "Device::hasOrientationSensor() const"}, ReplaceH: 4F F0 01 00}
 
 Don't uppercase header/footer text:
   - Enabled: no
@@ -580,7 +580,7 @@ Customize ComfortLight settings:
   # In an unnamed subroutine two layers into FrontLightPopupController::loadView
   # (find it by going back from QTime::addSecs), the times for the dropdown are
   # generated into a QVector<QPair<QString, QTime>> with a simple loop.
-  - BaseAddress: 0xAE9B90 # find the base of the unnamed subroutine with the x-ref to _ZN5QTimeC1Eiiii
+  - BaseAddress: 0xAEA5C0 # find the base of the unnamed subroutine with the x-ref to _ZN5QTimeC1Eiiii
   #
   # Change the initial hour / first bedtime dropdown item passed to the QTime
   # constructor (mov r1, #21):
@@ -655,7 +655,7 @@ FeatureSettings - ExportHighlights:
 DeveloperSettings - AutoUsbGadget:
   - Enabled: no
   - Description: Automatically enable USB Storage mode when connected.
-  - ReplaceBytes: {Offset: 0x777EEE, FindInstBLX: {SymPLT: "QVariant::toBool() const"}, ReplaceH: 4F F0 01 00}
+  - ReplaceBytes: {Offset: 0x77891E, FindInstBLX: {SymPLT: "QVariant::toBool() const"}, ReplaceH: 4F F0 01 00}
 
 PowerSettings - UnlockEnabled:
   - Enabled: no
diff --git a/src/versions/4.23.15505/nickel.yaml/oren64.yaml b/src/versions/4.23.15548/nickel.yaml/oren64.yaml
[... 17 lines cut ...]
+  - BaseAddress: 0x14C3975
   - ReplaceZlib: {Find: "MetaDataLabel {\n  color: #666666;\n  qproperty-indent: 0;\n}", Replace: "MetaDataLabel {\n  color: #000000;\n  qproperty-indent: 0;\n}"}
   - ReplaceZlib:
       Find:    "MetaDataLabel[qApp_deviceIsTrilogy=true] {\n  font-size: 13px;\n}"
diff --git a/src/versions/4.23.15505/nickel.yaml/pgaskin.yaml b/src/versions/4.23.15548/nickel.yaml/pgaskin.yaml
index 68f2054..34b1b32 100644
--- a/src/versions/4.23.15505/nickel.yaml/pgaskin.yaml
+++ b/src/versions/4.23.15548/nickel.yaml/pgaskin.yaml
@@ -15,11 +15,11 @@ Remove forgot pin button from lock screen:
       Removes the Forgot PIN -> Sign Out button from the lock screen.
       If this is enabled and you forget your pin, you will need to
[... cut ...]
+  - BaseAddress: 0x14AE879
   - ReplaceZlib:
       Find: "#lblForgotPin[qApp_deviceIsDragon=true],\n#lblSignOut[qApp_deviceIsDragon=true] {\n  font-size: 26px;\n}"
       Replace: "#lblForgotPin,#lblSignOut{qproperty-visible:false;}"
-  - BaseAddress: 0x14BB84A
+  - BaseAddress: 0x14B5543
   - ReplaceZlib:
       Find: "#lblForgotPin[qApp_deviceIsDragon=true],\n#lblSignOut[qApp_deviceIsDragon=true] {\n  font-size: 26px;\n}"
       Replace: "#lblForgotPin,#lblSignOut{qproperty-visible:false;}"
```

</details>

pgaskin, 2020-08-19:
<details><summary>

`$ kobo-versionextract ~/src/kobopatch-config/src/kobo7-update-4.23.15548.zip`

</summary>

```
Revision: c8e55ad06855e989bc1a7aa552c6530e140a6a68
 Version: 4.23.15548
    Date: August 2020
```

</details>

pgaskin, 2020-08-19:
Done: [v66](https://github.com/pgaskin/kobopatch-patches/releases/v66), MR:thread-332469.

closed 2020-08-19 by pgaskin