pgaskin/kobopatch-patches #76: Patches for 4.24.15672

pgaskin, 2020-09-23:
- [ ] Upload testdata (@pgaskin)
- [ ] Extract resources for @jackiew1 (@pgaskin)
- [ ] Find libnickel offsets where necessary (@pgaskin)
- [ ] Look at new dictionary stuff (@pgaskin)
- [ ] Look at TOC/annotation view stuff (@pgaskin)
- [ ] Update patches (@pgaskin)
- [ ] Update nickel patches if necessary (@jackiew1)
- [ ] Test most patches (@pgaskin)
- [ ] Make release (@pgaskin)
- [ ] MR post (@pgaskin)

pgaskin, 2020-09-23:
- The TOC/annotations now shows in the same dialog with tabs at the top:
    - It now uses the same scrollbar/pagination as the library.
    - The new annotations view wastes quite a bit of space for the padding around it.
    - The new TOC view is much better.
        - The current item is bolded rather than having a large checkmark beside it.
        - Nested items are now indented more (i.e. what the `Increase TOC level indentation` patch was for).
- The new dictionaries are Oxford rather than MW.
  - I haven't looked at the internals yet.
- TODO

pgaskin, 2020-09-23:
<details><summary>

**$ diff -y <(cat 15548/symdump.out.json | jq -r '.[] | .Name' | sort) <(cat 15672/symdump.out.json | jq -r '.[] | .Name' | sort) | grep -e '<' -e '>'**

</summary>

```
							      >	DOGEAR_DIV_CLASS
							      >	HIGHLIGHT_DIV_CLASS
							      >	NOTE_DIV_CLASS
							      >	strchr
							      >	_Z25qSharedPointerDynamicCastI18ManageBooksBuilder15AbstractB
							      >	_Z5qHashRK10Dictionary
							      >	zip_get_archive_comment
							      >	_Zls6QDebugRK10Dictionary
							      >	_ZN10Dictionary7setPathERK8QVariant
_ZN10DictionaryC1ERK7QStringS2_bxb			      <
_ZN11ReaderProxy12hasSelectionEv			      <
							      >	_ZN11ReaderProxy15updateSelectionEiibbb
_ZN11ReaderProxy9errorListEv				      <
							      >	_ZN11ReadingView21updateBookmarkHistoryEv
							      >	_ZN11ReadingView22bookmarkHistoryUpdatedEv
							      >	_ZN12QWeakPointerI16DictionaryParserED1Ev
							      >	_ZN12QWeakPointerI16DictionaryParserED2Ev
							      >	_ZN12QWeakPointerI21DictionaryFileManagerED1Ev
							      >	_ZN12QWeakPointerI21DictionaryFileManagerED2Ev
							      >	_ZN13Battery1Proxy11qt_metacallEN11QMetaObject4CallEiPPv
							      >	_ZN13Battery1Proxy11qt_metacastEPKc
							      >	_ZN13Battery1Proxy16staticMetaObjectE
							      >	_ZN13Battery1ProxyC1ERK7QStringS2_RK15QDBusConnectionP7QObjec
							      >	_ZN13Battery1ProxyC2ERK7QStringS2_RK15QDBusConnectionP7QObjec
							      >	_ZN13Battery1ProxyD0Ev
							      >	_ZN13Battery1ProxyD1Ev
							      >	_ZN13Battery1ProxyD2Ev
							      >	_ZN13EventInfoImpl14showAnnotationEv
							      >	_ZN13N3SyncManager10updateSyncEv
							      >	_ZN13N3SyncManager12isUpdateSyncEv
							      >	_ZN13SearchEditBox9showEventEP10QShowEvent
_ZN13StoreListView13addEmptyStateEv			      <
_ZN14AddBookCommand14bookmarkFailedERK6Volume		      <
_ZN14QScopedPointerI17MarisaTrieAdapter21QScopedPointerDelete <
_ZN14QScopedPointerI17MarisaTrieAdapter21QScopedPointerDelete <
							      >	_ZN14QSharedPointerI16DictionaryParserE5derefEPN15QtSharedPoi
							      >	_ZN14QSharedPointerI21DictionaryFileManagerE5derefEPN15QtShar
							      >	_ZN15KepubBookReader17refreshAnnotationERK8Bookmark
							      >	_ZN15QtSharedPointer33ExternalRefCountWithCustomDeleterI16Dic
							      >	_ZN15QtSharedPointer33ExternalRefCountWithCustomDeleterI21Dic
_ZN16DictionaryParser16activeDictionaryEv		      <
							      >	_ZN16DictionaryParser16dictionarySourceEv
							      >	_ZN16DictionaryParser16staticMetaObjectE
_ZN16DictionaryParser20initializeFromVolumeERK6VolumeRK15Cont <
_ZN16DictionaryParser23searchWordMultipleCasesERK7QStringS2_  <
_ZN16DictionaryParser25findAllDefinitionsForWordERK7QStringS2 <
_ZN16DictionaryParser27searchForJapaneseWordInHtmlERK7QString <
_ZN16DictionaryParser30findDefinitionsForKanjiVariantERK7QStr <
_ZN16DictionaryParser30queryHtmlDefinitionForJapaneseERK7QStr <
_ZN16DictionaryParser30requestedDictionaryWithoutTrimEv	      <
_ZN16DictionaryParser31searchForJapaneseWordExactMatchERK7QSt <
_ZN16DictionaryParser7getHtmlERK7QString		      <
							      >	_ZN16DictionaryParser9_INSTANCEE
							      >	_ZN16DictionaryParserC1Ev
							      >	_ZN16DictionaryParserC2Ev
							      >	_ZN16DictionaryParserD0Ev
							      >	_ZN16DictionaryParserD1Ev
							      >	_ZN16DictionaryParserD2Ev
							      >	_ZN16ReadingViewMixin11readingViewEv
							      >	_ZN16ReadingViewMixin15showAnnotationsERK6Volume
							      >	_ZN16ReadingViewMixin19openTableOfContentsERK6Volume
							      >	_ZN16ReadingViewMixin23showAnnotationsListViewERK6Volume
							      >	_ZN16ReadingViewMixin27showAnnotationsNotebookViewERK6Volume
[... 262 lines cut ...]
_ZThn8_N31OverDriveSearchHeaderController8loadViewEv	      <
_ZThn8_N32AnnotationSearchHeaderController8loadViewEv	      <
_ZThn8_N32DictionarySearchHeaderController8loadViewEv	      <
							      >	_ZTI13Battery1Proxy
							      >	_ZTI16DictionaryParser
_ZTI18GenericLibraryViewI10ShortcoverE			      <
							      >	_ZTI20BlueAlsaRFCOMM1Proxy
							      >	_ZTI25NewAnnotationNotebookView
							      >	_ZTI36Migration161_AddBookmarkContextField
							      >	_ZTS13Battery1Proxy
							      >	_ZTS16DictionaryParser
_ZTS18GenericLibraryViewI10ShortcoverE			      <
							      >	_ZTS20BlueAlsaRFCOMM1Proxy
							      >	_ZTS25NewAnnotationNotebookView
							      >	_ZTS36Migration161_AddBookmarkContextField
							      >	_ZTV13Battery1Proxy
							      >	_ZTV16DictionaryParser
_ZTV18GenericLibraryViewI10ShortcoverE			      <
							      >	_ZTV20BlueAlsaRFCOMM1Proxy
							      >	_ZTV25NewAnnotationNotebookView
							      >	_ZTV36Migration161_AddBookmarkContextField
							      >	_ZZN11QMetaTypeIdI23QDBusUnixFileDescriptorE14qt_metatype_idE
```

</details>

pgaskin, 2020-09-23:
Resource extraction:

```bash
./qrc2zip --output "nickel.15672.qInitResources_resources.zip"    --recursive --verbose "nickel" 1 $((0x115dfe8 - 0x0010000)) $((0x0026948 - 0x0010000)) $((0x115db98 - 0x0010000))
./qrc2zip --output "nickel.15672.qInitResources_translations.zip" --recursive --verbose "nickel" 1 $((0x14b8e68 - 0x0010000)) $((0x115e388 - 0x0010000)) $((0x14b8be0 - 0x0010000))
./qrc2zip --output "nickel.15672.qInitResources_styles.zip"       --recursive --verbose "nickel" 1 $((0x14d58f0 - 0x0010000)) $((0x14b8f80 - 0x0010000)) $((0x14d3310 - 0x0010000))
./qrc2zip --output "nickel.15672.qInitResources_certificates.zip" --recursive --verbose "nickel" 1 $((0x14fcc48 - 0x0010000)) $((0x14d63a8 - 0x0010000)) $((0x14fbbf8 - 0x0010000))
```

jackiew1, 2020-09-23:
@pgaskin I'm a bit slow seeing this tonight, but better late than never. Looks interesting.

Did you produce the resource map we discussed last time? Here's the first few lines from the previous fw as a reminder.
```
nickel@15548/qInitResources_resources
DIR     "qss" (0x114B71E + 14)
FILE    "qss/TouchMenu.qss" (0x114ADA7 + 189)
DIR     "html" (0x114B702 + 28)
FILE    "html/RGBFrontLightAboutView.html" (0x114A45B + 1144)
FILE    "html/RGBFrontLightAboutViewSnow.html" (0x114A8D7 + 1228)
DIR     "keys" (0x114B6F4 + 14)
FILE    "keys/device.qmap" (0x114A3E7 + 112)
FILE    "omfTemplate.html" (0x94A2C1 + 231)
DIR     "pika.rcc" (0x410D + 14)
```

pgaskin, 2020-09-23:
@jackiew1, here it is:

<details>

```
nickel@15672/qInitResources_resources
DIR     "qss" (0x114E376 + 14)
FILE    "qss/TouchMenu.qss" (0x114D03E + 189)
DIR     "html" (0x114E35A + 28)
FILE    "html/RGBFrontLightAboutView.html" (0x114D71A + 1144)
FILE    "html/RGBFrontLightAboutViewSnow.html" (0x114D24A + 1228)
DIR     "keys" (0x114E34C + 14)
FILE    "keys/device.qmap" (0x114D0FF + 112)
FILE    "omfTemplate.html" (0x1098716 + 231)
DIR     "pika.rcc" (0x410D + 14)
DIR     "pika.rcc/images" (0x411B + 42)
DIR     "pika.rcc/images/fte" (0x4161 + 14)
FILE    "pika.rcc/images/fte/quicktour_diagram.png" (0x110D + 3841)
DIR     "pika.rcc/images/home" (0x4153 + 14)
FILE    "pika.rcc/images/home/quicktour_device.png" (0x18 + 4337)
DIR     "pika.rcc/images/about" (0x4145 + 14)
FILE    "pika.rcc/images/about/global_certification.png" (0x2012 + 8221)
FILE    "verifier.pub" (0x1147E35 + 451)
DIR     "daylight.rcc" (0xC03E7 + 14)
DIR     "daylight.rcc/images" (0xC03F5 + 322)
DIR     "daylight.rcc/images/fte" (0xC1885 + 280)
FILE    "daylight.rcc/images/fte/missingbooks_factoryreset.png" (0xA073D + 9065)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg4_adjust_brightness.png" (0x67C3E + 24664)
FILE    "daylight.rcc/images/fte/missingbooks_account.png" (0x5CBDC + 22560)
FILE    "daylight.rcc/images/fte/missingbooks_repair.png" (0x73656 + 20737)
FILE    "daylight.rcc/images/fte/quicktour_nightmode.png" => "daylight.rcc/images/fte/quicktour_nightmode[language!Japanese].png" (0x56B92 + 24646)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg2.png" => "daylight.rcc/images/fte/quicktour_reading_pg2[language!Japanese].png" (0x7875B + 23892)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg2.png" (0x7E4B3 + 23662)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg3.png" => "daylight.rcc/images/fte/quicktour_reading_pg3[language!Japanese].png" (0x89C50 + 23716)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg3.png" (0x8F8F8 + 23519)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg5.png" (0x97540 + 19735)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg1.png" (0x6DD3C + 22806)
FILE    "daylight.rcc/images/fte/missingbooks_sync.png" (0x62400 + 6061)
FILE    "daylight.rcc/images/fte/fte_oob_connected.png" => "daylight.rcc/images/fte/fte_oob_connected[language!Japanese].png" (0x64ACB + 9157)
FILE    "daylight.rcc/images/fte/fte_oob_plugin.svg" (0x66E94 + 3494)
FILE    "daylight.rcc/images/fte/pocket_logo_large.png" (0x63BB1 + 3862)
FILE    "daylight.rcc/images/fte/quicktour_texticon_search.png" (0x6DC9A + 158)
FILE    "daylight.rcc/images/fte/quicktour_diagram.png" (0x954DB + 8289)
FILE    "daylight.rcc/images/fte/missingbooks_help.png" (0x84125 + 23335)
FILE    "daylight.rcc/images/fte/overflow_small.png" (0x56761 + 1069)
FILE    "daylight.rcc/images/fte/missingbooks_intro.png" (0x9C25B + 17630)
DIR     "daylight.rcc/images/book" (0xC185B + 42)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_highlight.png" (0x283 + 447)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_dogear.png" (0xE6 + 409)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_note.png" (0x18 + 202)
DIR     "daylight.rcc/images/help" (0xC1831 + 42)
FILE    "daylight.rcc/images/help/help_icon.png" (0x1F02B + 1022)
FILE    "daylight.rcc/images/help/kobopluslogo_help_sp.png" (0x20E3A + 7449)
FILE    "daylight.rcc/images/help/kobopluslogo_help_version.png" (0x1F42D + 6665)
DIR     "daylight.rcc/images/home" (0xC1743 + 238)
FILE    "daylight.rcc/images/home/main_nav_books.png" (0xA41EA + 637)
FILE    "daylight.rcc/images/home/annotations_bookmark.png" (0xA3B24 + 409)
FILE    "daylight.rcc/images/home/pocket_favourite_star.png" (0xA409E + 328)
FILE    "daylight.rcc/images/home/main_nav_discover.png" (0xA446B + 1106)
FILE    "daylight.rcc/images/home/main_nav_discover_active.png" (0xA3CC1 + 860)
FILE    "daylight.rcc/images/home/annotations_note.png" (0xA3663 + 202)
FILE    "daylight.rcc/images/home/quicktour_device.png" (0xA48C1 + 42215)
FILE    "daylight.rcc/images/home/annotations_highlight.png" (0xA38BF + 447)
FILE    "daylight.rcc/images/home/main_nav_more_active.png" (0xA4021 + 121)
FILE    "daylight.rcc/images/home/main_nav_books_active.png" (0xA3731 + 394)
FILE    "daylight.rcc/images/home/main_nav_home.png" (0xAF47A + 353)
FILE    "daylight.rcc/images/home/first_purchase_tile.png" (0xB0ACA + 5761)
FILE    "daylight.rcc/images/home/main_nav_more.png" (0xA3A82 + 158)
FILE    "daylight.rcc/images/home/home_emptystate_hit.png" (0xA3143 + 1308)
FILE    "daylight.rcc/images/home/main_nav_home_active.png" (0xAEDAC + 285)
[... 2373 lines cut ...]
FILE    "certificates/48a195d8.der" (0x14C7B49 + 1525)
FILE    "certificates/583d0756.der" (0x14E0734 + 1519)
FILE    "certificates/3d441de8.der" (0x14E356A + 1486)
FILE    "certificates/33ee480d.der" (0x14D6B78 + 1505)
FILE    "certificates/12d55845.der" (0x14C6D4E + 846)
FILE    "certificates/985c1f52.der" (0x14EAA86 + 1415)
FILE    "certificates/1eb37bdf.der" (0x14CFAEC + 1875)
FILE    "certificates/9d6523ce.der" (0x14DBADB + 1460)
FILE    "certificates/81b9768f.der" (0x14C949D + 969)
FILE    "certificates/6187b673.der" (0x14D1044 + 1391)
FILE    "certificates/69105f4f.der" (0x14D5489 + 955)
FILE    "certificates/e8651083.der" (0x14D715D + 1038)
FILE    "certificates/91739615.der" (0x14D9EC5 + 1031)
FILE    "certificates/5cf9d536.der" (0x14D6178 + 1492)
FILE    "certificates/7a7c655d.der" (0x14D5848 + 442)
FILE    "certificates/7999be0d.der" (0x14D1ECF + 856)
FILE    "certificates/1dcd6f4c.der" (0x14C8F23 + 1398)
FILE    "certificates/d18e9066.der" (0x14E112F + 1380)
FILE    "certificates/c2c1704e.der" (0x14D6750 + 1060)
FILE    "certificates/9f533518.der" (0x14DE4AD + 1869)
FILE    "certificates/3a3b02ce.der" (0x14DEF26 + 1013)
FILE    "certificates/21855f49.der" (0x14D15B7 + 1389)
```

</details>

pgaskin, 2020-09-23:
Patch test results:

```
Errors:
  /home/patrick/src/kobopatch-patches/src/versions/4.24.15672/libnickel.so.1.0.0.yaml/geoffr.yaml: Set KePub hyphenation: could not apply patch "Set KePub hyphenation": line 320: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.24.15672/libnickel.so.1.0.0.yaml/geoffr.yaml: Set font scale factor: could not apply patch "Set font scale factor": line 537: inst 3: ReplaceInt: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.24.15672/nickel.yaml/pgaskin.yaml: Remove forgot pin button from lock screen: could not apply patch "Remove forgot pin button from lock screen": line 20: inst 4: ReplaceZlib: not a zlib stream
  /home/patrick/src/kobopatch-patches/src/versions/4.24.15672/nickel.yaml/oren64.yaml: New home screen subtitle custom font: could not apply patch "New home screen subtitle custom font": line 31: inst 4: ReplaceZlib: not a zlib stream
  /home/patrick/src/kobopatch-patches/src/versions/4.24.15672/libnickel.so.1.0.0.yaml/pgaskin.yaml: Increase TOC level indentation: could not apply patch "Increase TOC level indentation": line 329: inst 4: ReplaceBytes: expand Base=kobopatch.FlexAbsOffset{Offset:(*int32)(nil), Sym:(*string)(0xc00451d4b0), SymPLT:(*string)(nil), SymPLTTail:(*string)(nil), Inline:false, Rel:(*int32)(nil)}: ResolveSym("N3TableOfContentsWidget::setContent(Shortcover const&)"): no such symbol "N3TableOfContentsWidget::setContent(Shortcover const&)"
  /home/patrick/src/kobopatch-patches/src/versions/4.24.15672/libnickel.so.1.0.0.yaml/pgaskin.yaml: Increase page navigation history: could not apply patch "Increase page navigation history": line 33: inst 4: ReplaceInt: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.24.15672/libnickel.so.1.0.0.yaml/pgaskin.yaml: Allow rotation on all devices: could not apply patch "Allow rotation on all devices": line 453: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.24.15672/libnickel.so.1.0.0.yaml/pgaskin.yaml: Customize ComfortLight settings: could not apply patch "Customize ComfortLight settings": line 587: inst 4: ReplaceInt: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.24.15672/libnickel.so.1.0.0.yaml/pgaskin.yaml: Increase TOC level indentation and fix extra indentation bug: could not apply patch "Increase TOC level indentation and fix extra indentation bug": line 341: inst 4: BaseAddress: resolve address (kobopatch.BaseAddress{Offset:(*int32)(nil), Sym:(*string)(0xc00457f8c0), SymPLT:(*string)(nil), SymPLTTail:(*string)(nil), Inline:false, Rel:(*int32)(0xc00460b040)}): ResolveSym("N3TableOfContentsWidget::setContent(Shortcover const&)"): no such symbol "N3TableOfContentsWidget::setContent(Shortcover const&)"
  /home/patrick/src/kobopatch-patches/src/versions/4.24.15672/libnickel.so.1.0.0.yaml/pgaskin.yaml: DeveloperSettings - AutoUsbGadget: could not apply patch "DeveloperSettings - AutoUsbGadget": line 658: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.24.15672/libnickel.so.1.0.0.yaml/jackie_w.yaml: Dictionary text font-family/font-size/line-height: could not apply patch "Dictionary text font-family/font-size/line-height": line 38: inst 3: FindReplaceString: FindBaseAddressString: FindBaseAddress: could not find bytes
```

Not too much worse than usual. The only ones which will need additional changes to apply are `Dictionary text font-family/font-size/line-height`, `Increase TOC level indentation`, `Increase page navigation history`, and `Increase TOC level indentation and fix extra indentation bug`.

jackiew1, 2020-09-23:
Thanks for the map. 😄 

If patch `Dictionary text font-family/font-size/line-height` needs changing it may not be trivial. If it's not a minor matter I'll need to look at it closer tomorrow.

pgaskin, 2020-09-24:
I'm going to start working on the patches now.

pgaskin, 2020-09-24:
I just finished looking at the TOC indent patches, and I've decided to remove them since the fixes have been integrated into the original firmware.

pgaskin, 2020-09-24:
I probably will not be able to update `Increase page navigation history` for this version. I might consider rewriting it in the future, though, as I use this patch a lot when reading non-fiction.

pgaskin, 2020-09-24:
@jackiew1, any update on the nickel patches?

jackiew1, 2020-09-24:
I think all existing nickel patches should run without error. However, I've been meaning to get around to updating these 2 to make better use of available space in the MyBooks lists
- `Increase library cover size`
- `Series list increase cover thumbnails`

I'm planning to fix them tonight. Here's the background if you're interested ... 

Since the intro of the scrollbar down the right-hand side of MyBooks there is now less horizontal room available for useful details (book title, author etc). It's usually worse for kepubs than epubs because of the longer `KOBO EPUB - xxx KB` label. When increasing cover thumbnail size, the above 2 patches do not currently maintain the original aspect ratio when setting the new patch values for width/height. I think this is wasting some horizontal real estate.

pgaskin, 2020-09-24:
@jackiew1, are you ready for a release within the next hour or two? If I don't, I may not have time to do so until the weekend.

I'll probably be making another within the next few weeks.

jackiew1, 2020-09-24:
> @jackiew1, are you ready for a release within the next hour or two? If I don't, I may not have time to do so until the weekend.
> 
> I'll probably be making another within the next few weeks.

I don't think I can update and test libnickel `Dictionary text font-family/font-size/line-height` until tomorrow.

`Series list increase cover thumbnails` will be done in the next few minutes.

pgaskin, 2020-09-24:
I've finished with looking at dictionary changes: pgaskin/dictutil#14. I still need to test a few things with the dictzip v3 `prefix_exceptions`, but I think I have all the other relevant stuff.

pgaskin, 2020-09-24:
Slightly off topic, but related: NickelMenu, NickelSeries, and my other tools/mods will work fine with this firmware version.

jackiew1, 2020-09-24:
@pgaskin 
I think my committed changes to `Dictionary text font-family/font-size/line-height` are OK for now but I've no idea whether anything I've done will upset Kobo's new Oct dicts. I won't be surprised if I need to re-visit this patch after their release.

Please will you add a note in your patch release post on MR that:
- this may be only a temporary fix
- any saved versions of a previously customised `Dictionary text font-family/font-size/line-height` need to be updated to accommodate the changes I've had to make.

pgaskin, 2020-09-24:
Released in [v67](https://github.com/pgaskin/kobopatch-patches/releases/v67) and MR:thread-333424.

closed 2020-09-24 by pgaskin

pgaskin, 2020-09-24:
<details><summary>

**$ git diff --no-index src/versions/4.23.15548 src/versions/4.24.15672**

</summary>

```diff
diff --git a/src/versions/4.23.15548/libnickel.so.1.0.0.yaml/geoffr.yaml b/src/versions/4.24.15672/libnickel.so.1.0.0.yaml/geoffr.yaml
index 44ca920..a5ef6e2 100644
--- a/src/versions/4.23.15548/libnickel.so.1.0.0.yaml/geoffr.yaml
+++ b/src/versions/4.24.15672/libnickel.so.1.0.0.yaml/geoffr.yaml
@@ -317,8 +317,8 @@ Set KePub hyphenation:
       will always be turned on, regardless of justification button setting.
 
       (The publisher can still turn hyphenation off/on in the book's stylesheet.)
-  - ReplaceBytes: {Offset: 0xB8FB8E, FindInstBLX: {SymPLT: "QString::operator==(QLatin1String) const"}, ReplaceH: 01 20 01 20} # Alternative 1: Always turn KePub hyphenation on
-# - ReplaceBytes: {Offset: 0xB8FB8E, FindInstBLX: {SymPLT: "QString::operator==(QLatin1String) const"}, ReplaceH: 00 20 00 20} # Alternative 2: Never turn KePub hyphenation on
+  - ReplaceBytes: {Offset: 0xB90F62, FindInstBLX: {SymPLT: "QString::operator==(QLatin1String) const"}, ReplaceH: 01 20 01 20} # Alternative 1: Always turn KePub hyphenation on
+# - ReplaceBytes: {Offset: 0xB90F62, FindInstBLX: {SymPLT: "QString::operator==(QLatin1String) const"}, ReplaceH: 00 20 00 20} # Alternative 2: Never turn KePub hyphenation on
 
 Force user line spacing in KePubs:
   - Enabled: no
@@ -499,12 +499,6 @@ KePub stylesheet additions:
     # 1.5 times the font size of KePubs by default. With this enabled they match.
 # - FindReplaceString: {Find: "/******************************************/", Replace: "#book-inner{font-size:1.5em                }", MustMatchLength: yes}
 
-Allow searches on Extra dictionaries:
-  - Enabled: no
-    # See https://pgaskin.net/dictutil/dicthtml/install
-  - FindBaseAddressString: "\0Extra:\x20"
-  - ReplaceString: {Offset: 7, Find: "\x20", Replace: "_"}
-
 Ignore .otf fonts:
   - Enabled: no
   - Description:
@@ -534,5 +528,5 @@ Set font scale factor:
   # Original value X=14, so original scale factor is (font size slider) / 46.
   # Allowed replacement values for X: 0-15 (smaller X --> larger scale factor)
   # Note: vmov.f32 s1, #15.0 before getReadingFontSizeScaleFactor
-  - ReplaceInt: {Offset: 0x5874AC, Find: 14, Replace: 4} # ePub
-  - ReplaceInt: {Offset: 0xB8F764, Find: 14, Replace: 4} # KePub
+  - ReplaceInt: {Offset: 0x58953C, Find: 14, Replace: 4} # ePub
+  - ReplaceInt: {Offset: 0xB90B34, Find: 14, Replace: 4} # KePub
diff --git a/src/versions/4.23.15548/libnickel.so.1.0.0.yaml/jackie_w.yaml b/src/versions/4.24.15672/libnickel.so.1.0.0.yaml/jackie_w.yaml
index 5e21e4c..3740799 100644
--- a/src/versions/4.23.15548/libnickel.so.1.0.0.yaml/jackie_w.yaml
+++ b/src/versions/4.24.15672/libnickel.so.1.0.0.yaml/jackie_w.yaml
@@ -7,7 +7,9 @@ Dictionary text font-family/font-size/line-height:
       in the pop-up and full-screen dictionary widgets.
       4.17.13651: improved to also reduce the blank lines displayed in many
                   sideloaded custom dictionaries which contain <blockquote>.
-      .
+      4.24.15672: enforced update to accommodate Kobo changes related to new
+                  dictionaries being released Oct 1st 2020
+      #
       See https://www.mobileread.com/forums/showpost.php?p=3521137&postcount=48
       for screenshots
       .
@@ -15,58 +17,69 @@ Dictionary text font-family/font-size/line-height:
       - font-family
       - font-size
       - line-height
-      .
+      #
       These are the Kobo defaults
                 Glo - Georgia  23px  1.4em
                 H2O - Georgia  29px  1.4em
               GloHD - Georgia  32px  1.4em
             AuraONE - Georgia  42px  1.4em
-  #
[... 302 lines cut ...]
       - Find:    "#chapterSizes[qApp_deviceIsTrilogy=true] {\n  max-height: 56px;\n  min-height: 56px;\n}"
         Replace: "#chapterSizes[qApp_deviceIsTrilogy=true] {\n  max-height: 90px;\n  min-height: 90px;\n  min-width: 385px;\n  max-width: 385px;\n}"
@@ -40,3 +40,19 @@ Increase size of kepub chapter progress chart:
         Replace: "#chapterSizes[qApp_deviceIsDragon=true] {\n  max-height: 170px;\n  min-height: 170px;\n  min-width: 678px;\n  max-width: 678px;\n}"
       - Find:    "#chapterSizes[qApp_deviceIsDaylight=true] {\n  max-height: 130px;\n  min-height: 130px;\n}"
         Replace: "#chapterSizes[qApp_deviceIsDaylight=true] {\n  max-height: 215px;\n  min-height: 215px;\n  min-width: 865px;\n  max-width: 865px;\n}"
+
+Change TOC level indentation:
+  - Enabled: no
+  - Description: |
+      Changes the size of indentation for each level in the TOC. This patch
+      replaces the old "Increase TOC level indentation" patch from firmware
+      versions before 4.24.15672 (see the note in libnickel.so.1.0.0.yaml).
+  - FindZlib: "qproperty-indentUnit"
+  - ReplaceZlibGroup:
+      Replacements:
+      # Change the replacement values to your desired width (I've made it half the default width as an example).
+      - {Find: "qproperty-indentUnit: 30;", Replace: "qproperty-indentUnit: 15;"} # trilogy
+      - {Find: "qproperty-indentUnit: 38;", Replace: "qproperty-indentUnit: 19;"} # phoenix
+      - {Find: "qproperty-indentUnit: 58;", Replace: "qproperty-indentUnit: 29;"} # dragon
+      - {Find: "qproperty-indentUnit: 72;", Replace: "qproperty-indentUnit: 36;"} # storm
+      - {Find: "qproperty-indentUnit: 76;", Replace: "qproperty-indentUnit: 38;"} # daylight
```

</details>