pgaskin/kobopatch-patches #78: Patches for 4.24.15676

pgaskin, 2020-09-25:
- [ ] Upload testdata (@pgaskin)
- [ ] Find libnickel offsets where necessary (@pgaskin)
- [ ] Update most patches (@pgaskin)
- [ ] Make release (@pgaskin)
- [ ] MR post (@pgaskin)

This checklist is shorter than usual since I doubt much changed in this release. I'll put the other items in if it turns out to be larger than I think it'll be.

pgaskin, 2020-09-25:
```
Errors:
  /home/patrick/src/kobopatch-patches/src/versions/4.24.15676/libnickel.so.1.0.0.yaml/pgaskin.yaml: Customize ComfortLight settings: could not apply patch "Customize ComfortLight settings": line 558: inst 4: ReplaceInt: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.24.15676/libnickel.so.1.0.0.yaml/pgaskin.yaml: Allow rotation on all devices: could not apply patch "Allow rotation on all devices": line 424: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.24.15676/libnickel.so.1.0.0.yaml/pgaskin.yaml: DeveloperSettings - AutoUsbGadget: could not apply patch "DeveloperSettings - AutoUsbGadget": line 629: inst 3: ReplaceBytes: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.24.15676/libnickel.so.1.0.0.yaml/geoffr.yaml: Set KePub hyphenation: could not apply patch "Set KePub hyphenation": line 320: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.24.15676/libnickel.so.1.0.0.yaml/geoffr.yaml: Set font scale factor: could not apply patch "Set font scale factor": line 531: inst 3: ReplaceInt: could not find specified bytes at offset
```

As I expected.

---

Also, I seem to have fat fingers today...

pgaskin, 2020-09-25:
No symbol differences, no relevant string differences.

pgaskin, 2020-09-25:
No rootfs differences other than nickel/libnickel/revinfo.

pgaskin, 2020-09-25:
Done in [v68](https://github.com/pgaskin/kobopatch-patches/releases/tag/v68) and MR:thread-333428.

pgaskin, 2020-09-25:
<details><summary>

**$ git diff --no-index src/versions/4.24.15672 src/versions/4.24.15676**

</summary>

```diff
diff --git a/src/versions/4.24.15672/libnickel.so.1.0.0.yaml/geoffr.yaml b/src/versions/4.24.15676/libnickel.so.1.0.0.yaml/geoffr.yaml
index a5ef6e2..c625638 100644
--- a/src/versions/4.24.15672/libnickel.so.1.0.0.yaml/geoffr.yaml
+++ b/src/versions/4.24.15676/libnickel.so.1.0.0.yaml/geoffr.yaml
@@ -317,8 +317,8 @@ Set KePub hyphenation:
       will always be turned on, regardless of justification button setting.
 
       (The publisher can still turn hyphenation off/on in the book's stylesheet.)
-  - ReplaceBytes: {Offset: 0xB90F62, FindInstBLX: {SymPLT: "QString::operator==(QLatin1String) const"}, ReplaceH: 01 20 01 20} # Alternative 1: Always turn KePub hyphenation on
-# - ReplaceBytes: {Offset: 0xB90F62, FindInstBLX: {SymPLT: "QString::operator==(QLatin1String) const"}, ReplaceH: 00 20 00 20} # Alternative 2: Never turn KePub hyphenation on
+  - ReplaceBytes: {Offset: 0xB911A2, FindInstBLX: {SymPLT: "QString::operator==(QLatin1String) const"}, ReplaceH: 01 20 01 20} # Alternative 1: Always turn KePub hyphenation on
+# - ReplaceBytes: {Offset: 0xB911A2, FindInstBLX: {SymPLT: "QString::operator==(QLatin1String) const"}, ReplaceH: 00 20 00 20} # Alternative 2: Never turn KePub hyphenation on
 
 Force user line spacing in KePubs:
   - Enabled: no
@@ -528,5 +528,5 @@ Set font scale factor:
   # Original value X=14, so original scale factor is (font size slider) / 46.
   # Allowed replacement values for X: 0-15 (smaller X --> larger scale factor)
   # Note: vmov.f32 s1, #15.0 before getReadingFontSizeScaleFactor
-  - ReplaceInt: {Offset: 0x58953C, Find: 14, Replace: 4} # ePub
-  - ReplaceInt: {Offset: 0xB90B34, Find: 14, Replace: 4} # KePub
+  - ReplaceInt: {Offset: 0x589544, Find: 14, Replace: 4} # ePub
+  - ReplaceInt: {Offset: 0xB90D74, Find: 14, Replace: 4} # KePub
diff --git a/src/versions/4.24.15672/libnickel.so.1.0.0.yaml/pgaskin.yaml b/src/versions/4.24.15676/libnickel.so.1.0.0.yaml/pgaskin.yaml
index 1938c18..a9ddbf9 100644
--- a/src/versions/4.24.15672/libnickel.so.1.0.0.yaml/pgaskin.yaml
+++ b/src/versions/4.24.15676/libnickel.so.1.0.0.yaml/pgaskin.yaml
@@ -421,13 +421,13 @@ Allow rotation on all devices:
       Unlike only the DeveloperSettings ForceAllowLandscape option, this patch
       also makes the rotation icon show on the new reader as well.
   # Always return true for ForceAllowLandscape:
-  - ReplaceBytes: {Offset: 0x77B3AE, FindInstBLX: {SymPLT: "QVariant::toBool() const"}, ReplaceH: 4F F0 01 00}
+  - ReplaceBytes: {Offset: 0x77B5EE, FindInstBLX: {SymPLT: "QVariant::toBool() const"}, ReplaceH: 4F F0 01 00}
   # In ReadingMenuView::ReadingMenuView, replace the value of Device::hasOrientationSensor,
   # which is passed to a function to hide/show the rotate icon:
-  - ReplaceBytes: {Offset: 0xBA8324, FindInstBLX: {SymPLT: "Device::hasOrientationSensor() const"}, ReplaceH: 4F F0 01 00}
+  - ReplaceBytes: {Offset: 0xBA8564, FindInstBLX: {SymPLT: "Device::hasOrientationSensor() const"}, ReplaceH: 4F F0 01 00}
   # Also fix sizing of popup by doing the same in RotatePopup::RotatePopup, but
   # note that this will make the Automatic option show too (and do nothing):
-  - ReplaceBytes: {Offset: 0xBC54E6, FindInstBLX: {SymPLT: "Device::hasOrientationSensor() const"}, ReplaceH: 4F F0 01 00}
+  - ReplaceBytes: {Offset: 0xBC5726, FindInstBLX: {SymPLT: "Device::hasOrientationSensor() const"}, ReplaceH: 4F F0 01 00}
 
 Don't uppercase header/footer text:
   - Enabled: no
@@ -551,7 +551,7 @@ Customize ComfortLight settings:
   # In an unnamed subroutine two layers into FrontLightPopupController::loadView
   # (find it by going back from QTime::addSecs), the times for the dropdown are
   # generated into a QVector<QPair<QString, QTime>> with a simple loop.
-  - BaseAddress: 0xAEC020 # find the base of the unnamed subroutine with the x-ref to _ZN5QTimeC1Eiiii
+  - BaseAddress: 0xAEC260 # find the base of the unnamed subroutine with the x-ref to _ZN5QTimeC1Eiiii
   #
   # Change the initial hour / first bedtime dropdown item passed to the QTime
   # constructor (mov r1, #21):
@@ -626,7 +626,7 @@ FeatureSettings - ExportHighlights:
 DeveloperSettings - AutoUsbGadget:
   - Enabled: no
   - Description: Automatically enable USB Storage mode when connected.
-  - ReplaceBytes: {Offset: 0x77B44E, FindInstBLX: {SymPLT: "QVariant::toBool() const"}, ReplaceH: 4F F0 01 00}
+  - ReplaceBytes: {Offset: 0x77B68E, FindInstBLX: {SymPLT: "QVariant::toBool() const"}, ReplaceH: 4F F0 01 00}
 
 PowerSettings - UnlockEnabled:
   - Enabled: no
```

</details>

closed 2020-09-25 by pgaskin

pgaskin, 2020-09-25:
Update: @davidfor (thanks!) pointed out that this release was solely to improve the handling of the v3 dictionary prefix_exceptions. Based on a quick glance, the most important part is at 0x5a61ac in 15672 and 0x5a643c in 15676. I'll look at this more later this week.

pgaskin, 2020-09-25:
Fixed symbol diff:

```diff
< _ZN17MarisaTrieAdapter15lookupExceptionERK7QString, MarisaTrieAdapter::lookupException(QString const&)  
---                                                                                                       
> _ZN17MarisaTrieAdapter16lookupExceptionsERK7QString, MarisaTrieAdapter::lookupExceptions(QString const&)38884,38885c38884,38885                                                                                   
< _ZNK16DictionaryParser19htmlForWordJapaneseERK7QString, DictionaryParser::htmlForWordJapanese(QString const&) const                                                                                               
< _ZNK16DictionaryParser22htmlForWordNonJapaneseERK7QString, DictionaryParser::htmlForWordNonJapanese(QString const&) const                                                                                         
---                                                                                                       
> _ZNK16DictionaryParser19htmlForWordJapaneseERK7QStringS0_, DictionaryParser::htmlForWordJapanese(QString const&, QString) const                                                                                   
> _ZNK16DictionaryParser22htmlForWordNonJapaneseERK7QStringS0_, DictionaryParser::htmlForWordNonJapanese(QString const&, QString) const 
```

jackiew1, 2020-09-26:
@pgaskin I've committed a slightly improved `Dictionary text font-family/font-size/line-height`. 

No need for a new release (unless someone reports problems with the previous version) because I still think more changes might be necessary once we've seen the new dictionaries.