pgaskin/kobopatch-patches #79: Patches for 4.25.15875

pgaskin, 2020-11-10:
- [ ] Upload testdata (@pgaskin)
- [ ] Find libnickel offsets where necessary (@pgaskin)
- [ ] Update most patches (@pgaskin)
- [ ] Update nickel patches (@jackiew1)
- [ ] Test patches (@pgaskin)
- [ ] Test nickel patches (@pgaskin)
- [ ] Make release (@pgaskin)
- [ ] MR post (@pgaskin)

jackiew1, 2020-11-10:
@pgaskin I'm ready for the new resource map file as soon as you're ready.

pgaskin, 2020-11-10:
**Resource extraction:**

```bash
echo "nickel@15875/qInitResources_resources";    ./qrc2zip --output "nickel.15875.qInitResources_resources.zip"    --recursive --verbose "nickel" 1 $((0x115f4e8 - 0x0010000)) $((0x0026848 - 0x0010000)) $((0x115f078 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
echo "nickel@15875/qInitResources_translations"; ./qrc2zip --output "nickel.15875.qInitResources_translations.zip" --recursive --verbose "nickel" 1 $((0x14bc6a0 - 0x0010000)) $((0x115f898 - 0x0010000)) $((0x14bc418 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
echo "nickel@15875/qInitResources_styles";       ./qrc2zip --output "nickel.15875.qInitResources_styles.zip"       --recursive --verbose "nickel" 1 $((0x14d9468 - 0x0010000)) $((0x14bc7b8 - 0x0010000)) $((0x14d6de8 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
echo "nickel@15875/qInitResources_certificates"; ./qrc2zip --output "nickel.15875.qInitResources_certificates.zip" --recursive --verbose "nickel" 1 $((0x15007f0 - 0x0010000)) $((0x14d9f50 - 0x0010000)) $((0x14ff7a0 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
```

<details><summary><b>Output</b></summary>

```
nickel@15875/qInitResources_resources
DIR     "qss" (0x114F884 + 14)
FILE    "qss/TouchMenu.qss" (0x114EE6E + 189)
DIR     "html" (0x114F868 + 28)
FILE    "html/RGBFrontLightAboutView.html" (0x114E522 + 1144)
FILE    "html/RGBFrontLightAboutViewSnow.html" (0x114E99E + 1228)
DIR     "keys" (0x114F85A + 14)
FILE    "keys/device.qmap" (0x114F006 + 112)
FILE    "omfTemplate.html" (0xF80B00 + 231)
DIR     "pika.rcc" (0x410D + 14)
DIR     "pika.rcc/images" (0x411B + 42)
DIR     "pika.rcc/images/fte" (0x4161 + 14)
FILE    "pika.rcc/images/fte/quicktour_diagram.png" (0x110D + 3841)
DIR     "pika.rcc/images/home" (0x4153 + 14)
FILE    "pika.rcc/images/home/quicktour_device.png" (0x18 + 4337)
DIR     "pika.rcc/images/about" (0x4145 + 14)
FILE    "pika.rcc/images/about/global_certification.png" (0x2012 + 8221)
FILE    "verifier.pub" (0x47084 + 451)
DIR     "daylight.rcc" (0xC06A9 + 14)
DIR     "daylight.rcc/images" (0xC06B7 + 322)
DIR     "daylight.rcc/images/fte" (0xC1B63 + 280)
FILE    "daylight.rcc/images/fte/missingbooks_factoryreset.png" (0x37DB6 + 9065)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg4_adjust_brightness.png" (0x31D5A + 24664)
FILE    "daylight.rcc/images/fte/missingbooks_account.png" (0x3A123 + 22560)
FILE    "daylight.rcc/images/fte/missingbooks_repair.png" (0x23A58 + 20737)
FILE    "daylight.rcc/images/fte/quicktour_nightmode.png" => "daylight.rcc/images/fte/quicktour_nightmode[language!Japanese].png" (0x48281 + 24646)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg2.png" => "daylight.rcc/images/fte/quicktour_reading_pg2[language!Japanese].png" (0xC803 + 23892)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg2.png" (0x1255B + 23662)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg3.png" => "daylight.rcc/images/fte/quicktour_reading_pg3[language!Japanese].png" (0x181CD + 23716)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg3.png" (0x1DE75 + 23519)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg5.png" (0x28B5D + 19735)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg1.png" (0x6EE9 + 22806)
FILE    "daylight.rcc/images/fte/missingbooks_sync.png" (0x4E2CB + 6061)
FILE    "daylight.rcc/images/fte/fte_oob_connected.png" => "daylight.rcc/images/fte/fte_oob_connected[language!Japanese].png" (0x4FB1E + 9157)
FILE    "daylight.rcc/images/fte/fte_oob_plugin.svg" (0x45472 + 3494)
FILE    "daylight.rcc/images/fte/pocket_logo_large.png" (0x5B9E + 3862)
FILE    "daylight.rcc/images/fte/quicktour_texticon_search.png" (0x4FA7C + 158)
FILE    "daylight.rcc/images/fte/quicktour_diagram.png" (0x4621C + 8289)
FILE    "daylight.rcc/images/fte/missingbooks_help.png" (0x3F947 + 23335)
FILE    "daylight.rcc/images/fte/overflow_small.png" (0x6AB8 + 1069)
FILE    "daylight.rcc/images/fte/missingbooks_intro.png" (0x2D878 + 17630)
DIR     "daylight.rcc/images/book" (0xC1B1D + 70)
FILE    "daylight.rcc/images/book/dustbin_delete_icon_inverted.png" (0xADDDA + 257)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_highlight.png" (0xADFAD + 447)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_dogear.png" (0xADAFA + 409)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_note.png" (0xADEDF + 202)
FILE    "daylight.rcc/images/book/dustbin_delete_icon.png" (0xADC97 + 319)
DIR     "daylight.rcc/images/help" (0xC1AF3 + 42)
FILE    "daylight.rcc/images/help/help_icon.png" (0x4896 + 1022)
FILE    "daylight.rcc/images/help/kobopluslogo_help_sp.png" (0x116C + 7449)
FILE    "daylight.rcc/images/help/kobopluslogo_help_version.png" (0x2E89 + 6665)
DIR     "daylight.rcc/images/home" (0xC1A05 + 238)
FILE    "daylight.rcc/images/home/main_nav_books.png" (0xACACD + 637)
FILE    "daylight.rcc/images/home/annotations_bookmark.png" (0xABE4C + 409)
FILE    "daylight.rcc/images/home/pocket_favourite_star.png" (0xABFE9 + 328)
FILE    "daylight.rcc/images/home/main_nav_discover.png" (0xAC135 + 1106)
FILE    "daylight.rcc/images/home/main_nav_discover_active.png" (0xAC608 + 860)
FILE    "daylight.rcc/images/home/annotations_note.png" (0xABD7E + 202)
[... 2733 lines cut ...]
new file mode 100644
index 0000000..cbccb43
Binary files /dev/null and b/15875/certificates/9282e51c.der differ
diff --git a/15875/certificates/9339512a.der b/15875/certificates/9339512a.der
new file mode 100644
[... cut ...]
Binary files /dev/null and b/15875/certificates/9576d26b.der differ
diff --git a/15875/certificates/95aff9e3.der b/15875/certificates/95aff9e3.der
new file mode 100644
index 0000000..cf29ccc
Binary files /dev/null and b/15875/certificates/95aff9e3.der differ
diff --git a/15875/certificates/9685a493.der b/15875/certificates/9685a493.der
new file mode 100644
index 0000000..19f91c2
Binary files /dev/null and b/15875/certificates/9685a493.der differ
diff --git a/15875/certificates/9772ca32.der b/15875/certificates/9772ca32.der
new file mo

jackiew1, 2020-11-10:
My libnickel patch `Dictionary text font-family/font-size/line-height` will fail your tests because Kobo have changed the dictionary code again.

I've had a quick look at what's changed and I don't think there should be too much trouble fixing it.

pgaskin, 2020-11-10:
KoboRoot changes:

- New udev rule: `KERNEL=="input*" RUN+="/usr/local/Kobo/udev/input"`
- TODO

jackiew1, 2020-11-10:
Initial nickel CSS analysis is that very little has changed, so nickel.yaml should be OK as-is.

3 brand new qss resources: 
`AutoCompleteClear.qss`
`N3SettingsPhysicalKeyView.qss`
`NotePreviewDialog.css`

1 resource with minor changes, but I don't believe any existing patches use it:
`AddToCollectionDialog.qss`

NiLuJe, 2020-11-10:
>     * New `pickel-imx` executable.

pickel is now a shell frontend... WHich currently calls pickel-imx on all currently known platforms. That's... not necessarily a good news, might be a hint that the next device will not be using an i.MX SoC, which would be... unfun.

> 
>     * Shutdown inittab target is now `/etc/init.d/rcK` (which SIGSTOPs nickel twice, unmounts all filesystems, then fscks the user data partition) instead of `/bin/umount -a -r` (cc @NiLuJe, @frostschutz, @baskerville).

That's nickel & sickel, not nickel twice ;).

pgaskin, 2020-11-10:
> That's nickel & sickel, not nickel twice ;).

My eyes are tired :wink:.

---

I'm going to start working on the patches now.

pgaskin, 2020-11-10:
<details><summary><b>$ diff &lt;(cat 15676/symdump.out.json | jq -r '.[] | .Name + ", " + .Demangled' | sort) &lt;(cat 15875/symdump.out.json | jq -r '.[] | .Name + ", " + .Demangled' | sort)</b>

```diff
1055d1054
< _ZN10BatchSaverI8BookmarkE5flushEv, BatchSaver<Bookmark>::flush()
1298,1301c1297,1304
< _ZN10LedManager11qt_metacallEN11QMetaObject4CallEiPPv, LedManager::qt_metacall(QMetaObject::Call, int, void**)
< _ZN10LedManager11qt_metacastEPKc, LedManager::qt_metacast(char const*)
< _ZN10LedManager14sharedInstanceEv, LedManager::sharedInstance()
< _ZN10LedManager16staticMetaObjectE, LedManager::staticMetaObject
---
> _ZN10LedManager3offEv, LedManager::off()
> _ZN10LedManager5redOnEv, LedManager::redOn()
> _ZN10LedManager6blueOnEv, LedManager::blueOn()
> _ZN10LedManager6redOffEv, LedManager::redOff()
> _ZN10LedManager7blueOffEv, LedManager::blueOff()
> _ZN10LedManager7greenOnEv, LedManager::greenOn()
> _ZN10LedManager8greenOffEv, LedManager::greenOff()
> _ZN10LedManager8orangeOnEv, LedManager::orangeOn()
1530d1532
< _ZN10Repository4findI5ShelfSt8functionIFbRKS1_EES1_EET_RK7QStringT0_T1_, Shelf Repository::find<Shelf, std::function<bool (Shelf const&)>, Shelf>(QString const&, std::function<bool (Shelf const&)>, Shelf)
1545d1546
< _ZN10Repository8getMutexI6VolumeEEP6QMutexv, QMutex* Repository::getMutex<Volume>()
1617a1619,1620
> _ZN10SketchNote4redoEv, SketchNote::redo()
> _ZN10SketchNote4undoEv, SketchNote::undo()
1893a1897
> _ZN10WebkitView11noteOffsetsEv, WebkitView::noteOffsets()
1898a1903
> _ZN10WebkitView12notesUpdatedEv, WebkitView::notesUpdated()
1901a1907
> _ZN10WebkitView12viewportRectEv, WebkitView::viewportRect()
1944a1951
> _ZN10WebkitView17updateNoteOffsetsEv, WebkitView::updateNoteOffsets()
2498a2506
> _ZN11MyWordsView12onViewInBookERK8WordList, MyWordsView::onViewInBook(WordList const&)
2778c2786
< _ZN11ReaderProxy11naturalSizeERiS0_, ReaderProxy::naturalSize(int&, int&)
---
> _ZN11ReaderProxy11currentPageEv, ReaderProxy::currentPage()
2782c2790
< _ZN11ReaderProxy11setViewportEii, ReaderProxy::setViewport(int, int)
---
> _ZN11ReaderProxy12hasSelectionEv, ReaderProxy::hasSelection()
2784d2791
< _ZN11ReaderProxy13addHighlightsERK5QListI4QMapI7QString8QVariantEE, ReaderProxy::addHighlights(QList<QMap<QString, QVariant> > const&)
2785a2793
> _ZN11ReaderProxy14clearSelectionEv, ReaderProxy::clearSelection()
2790a2799
> _ZN11ReaderProxy18selectionEndPointsEv, ReaderProxy::selectionEndPoints()
2793c2802
< _ZN11ReaderProxy21handleBookmarkAtPointEii, ReaderProxy::handleBookmarkAtPoint(int, int)
---
> _ZN11ReaderProxy22bookmarkForCurrentPageEv, ReaderProxy::bookmarkForCurrentPage()
2794a2804
> _ZN11ReaderProxy23selectionIntersectsSpanERK5QListI4QMapI7QString8QVariantEERKS2_, ReaderProxy::selectionIntersectsSpan(QList<QMap<QString, QVariant> > const&, QString const&)
2797a2808
> _ZN11ReaderProxy6hasDRMEv, ReaderProxy::hasDRM()
2798a2810
> _ZN11ReaderProxy7canEditEv, ReaderProxy::canEdit()
2800a2813
> _ZN11ReaderProxy9errorListEv, ReaderProxy::errorList()
2885a2899
> _ZN11ReadingView16canSaveHighlightEv, ReadingView::canSaveHighlight()
2919a2934
> _ZN11ReadingView18readingSessionDataER4QMapI7QString8QVariantE, ReadingView::readingSessionData(QMap<QString, QVariant>&)
3716a3732,3736
> _ZN12QPainterPath6lineToERK7QPointF, QPainterPath::lineTo(QPointF const&)
> _ZN12QPainterPath6moveToERK7QPointF, QPainterPath::moveTo(QPointF const&)
> _ZN12QPainterPath7cubicToERK7QPointFS2_S2_, QPainterPath::cubicTo(QPointF const&, QPointF const&, QPointF const&)
[... 526 lines cut ...]
52974a53212
> _ZTS25N3SettingsPhysicalKeyView, typeinfo name for N3SettingsPhysicalKeyView
53128a53367
> _ZTS27NotePreviewDialogController, typeinfo name for NotePreviewDialogController
53343a53583
> _ZTS31N3SettingsPhysicalKeyController, typeinfo name for N3SettingsPhysicalKeyController
54722a54963
> _ZTV17AutoCompleteClear, vtable for AutoCompleteClear
54790a55032
> _ZTV17NotePreviewDialog, vtable for NotePreviewDialog
54973a55216
> _ZTV18TouchEventRecorder, vtable for TouchEventRecorder
55087a55331
> _ZTV19NoteIndicatorWidget, vtable for NoteIndicatorWidget
55243a55488
> _ZTV20NoteIndicatorManager, vtable for NoteIndicatorManager
55828a56074
> _ZTV25N3SettingsPhysicalKeyView, vtable for N3SettingsPhysicalKeyView
55979a56226
> _ZTV27NotePreviewDialogController, vtable for NotePreviewDialogController
56192a56440
> _ZTV31N3SettingsPhysicalKeyController, vtable for N3SettingsPhysicalKeyController
```

</details>

pgaskin, 2020-11-10:
Random notes:

- My words bug fixes.
- TODO

pgaskin, 2020-11-10:
Slightly off-topic status update for dictutil: I really need to get around to pushing the v3 dictionary support. It's nearly finished, but I've been short on time, and it also needs quite a bit of testing.

pgaskin, 2020-11-10:
**Patch test results:**

```
Errors:
  /home/patrick/src/kobopatch-patches/src/versions/4.25.15875/libnickel.so.1.0.0.yaml/geoffr.yaml: Set KePub hyphenation: could not apply patch "Set KePub hyphenation": line 320: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.25.15875/libnickel.so.1.0.0.yaml/geoffr.yaml: Set font scale factor: could not apply patch "Set font scale factor": line 531: inst 3: ReplaceInt: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.25.15875/nickel.yaml/oren64.yaml: New home screen subtitle custom font: could not apply patch "New home screen subtitle custom font": line 31: inst 4: ReplaceZlib: not a zlib stream
  /home/patrick/src/kobopatch-patches/src/versions/4.25.15875/nickel.yaml/pgaskin.yaml: Remove forgot pin button from lock screen: could not apply patch "Remove forgot pin button from lock screen": line 20: inst 4: ReplaceZlib: not a zlib stream
  /home/patrick/src/kobopatch-patches/src/versions/4.25.15875/libnickel.so.1.0.0.yaml/pgaskin.yaml: DeveloperSettings - AutoUsbGadget: could not apply patch "DeveloperSettings - AutoUsbGadget": line 629: inst 3: ReplaceBytes: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.25.15875/libnickel.so.1.0.0.yaml/pgaskin.yaml: Both page turn sides go next: could not apply patch "Both page turn sides go next": line 17: inst 4: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.25.15875/libnickel.so.1.0.0.yaml/pgaskin.yaml: Enable advanced settings for all fonts: could not apply patch "Enable advanced settings for all fonts": line 529: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.25.15875/libnickel.so.1.0.0.yaml/pgaskin.yaml: Customize ComfortLight settings: could not apply patch "Customize ComfortLight settings": line 558: inst 4: ReplaceInt: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.25.15875/libnickel.so.1.0.0.yaml/pgaskin.yaml: Allow rotation on all devices: could not apply patch "Allow rotation on all devices": line 424: inst 3: ReplaceBytes: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.25.15875/libnickel.so.1.0.0.yaml/jackie_w.yaml: Dictionary text font-family/font-size/line-height: could not apply patch "Dictionary text font-family/font-size/line-height": line 52: inst 3: FindReplaceString: FindBaseAddressString: FindBaseAddress: could not find bytes
```

Everything's normal this time.

I'll work on the libnickel patches in an hour.

pgaskin, 2020-11-10:
@jackiew1, how are your nickel patches?

jackiew1, 2020-11-11:
> @jackiew1, how are your nickel patches?

As I said 2 hours ago in my 3rd post, very little has changed in nickel CSS and none of the current patches use the little which has changed.

I'm just about to test the updated (not committed yet) `Dictionary text font-family/font-size/line-height` in libnickel, but I need to actually install the fw to do that. 

More later.

pgaskin, 2020-11-11:
> As I said 2 hours ago in my 3rd post

Looks like I overlooked that post (I think GH is having caching issues again...). I'm going to do the other updates now, and I should have them committed (and lightly tested) within the next half hour.

pgaskin, 2020-11-11:
@jackiew1, it's ready for you to commit your changes now. I'll do a bit more testing with the other patches now, then once you're ready, I'll release them.

I might also look into some of the patch requests if I have time.

NickelMenu, NickelSeries, NickelDBus, and the other related mods work fine with this firmware version.

jackiew1, 2020-11-11:
> @jackiew1, it's ready for you to commit your changes now. I'll do a bit more testing with the other patches now, then once you're ready, I'll release them.

I've just committed the change to `Dictionary text font-family/font-size/line-height`.

pgaskin, 2020-11-11:
I've tested everything. I'll make a release in a few minutes.

pgaskin, 2020-11-11:
Based on the new dictionary, I found something interesting.

If you look up `gaijitest`, you'll see a list of characters. `#x0493` is inserted as a GIF image in the dictzip. Previously, this would cause a blank display, but it shows correctly. The interesting part is that the GIF is gzip-compressed. I'll need to investigate this further and see if this is new (I think this is the case) or if I just missed that image files should be gzipped before.

pgaskin, 2020-11-11:
Released: [`v69`](https://github.com/pgaskin/kobopatch-patches/releases/v69), MR:thread-334722.

closed 2020-11-11 by pgaskin

pgaskin, 2020-11-11:
<details><summary><b>$ git diff --no-index src/versions/{4.24.15676,4.25.15875}</b></summary>

```diff
diff --git a/src/versions/4.24.15676/libnickel.so.1.0.0.yaml/geoffr.yaml b/src/versions/4.25.15875/libnickel.so.1.0.0.yaml/geoffr.yaml
index c625638..d43d14c 100644
--- a/src/versions/4.24.15676/libnickel.so.1.0.0.yaml/geoffr.yaml
+++ b/src/versions/4.25.15875/libnickel.so.1.0.0.yaml/geoffr.yaml
@@ -317,8 +317,8 @@ Set KePub hyphenation:
       will always be turned on, regardless of justification button setting.
 
       (The publisher can still turn hyphenation off/on in the book's stylesheet.)
-  - ReplaceBytes: {Offset: 0xB911A2, FindInstBLX: {SymPLT: "QString::operator==(QLatin1String) const"}, ReplaceH: 01 20 01 20} # Alternative 1: Always turn KePub hyphenation on
-# - ReplaceBytes: {Offset: 0xB911A2, FindInstBLX: {SymPLT: "QString::operator==(QLatin1String) const"}, ReplaceH: 00 20 00 20} # Alternative 2: Never turn KePub hyphenation on
+  - ReplaceBytes: {Offset: 0xB8E556, FindInstBLX: {SymPLT: "QString::operator==(QLatin1String) const"}, ReplaceH: 01 20 01 20} # Alternative 1: Always turn KePub hyphenation on
+# - ReplaceBytes: {Offset: 0xB8E556, FindInstBLX: {SymPLT: "QString::operator==(QLatin1String) const"}, ReplaceH: 00 20 00 20} # Alternative 2: Never turn KePub hyphenation on
 
 Force user line spacing in KePubs:
   - Enabled: no
@@ -528,5 +528,5 @@ Set font scale factor:
   # Original value X=14, so original scale factor is (font size slider) / 46.
   # Allowed replacement values for X: 0-15 (smaller X --> larger scale factor)
   # Note: vmov.f32 s1, #15.0 before getReadingFontSizeScaleFactor
-  - ReplaceInt: {Offset: 0x589544, Find: 14, Replace: 4} # ePub
-  - ReplaceInt: {Offset: 0xB90D74, Find: 14, Replace: 4} # KePub
+  - ReplaceInt: {Offset: 0x58F3F4, Find: 14, Replace: 4} # ePub
+  - ReplaceInt: {Offset: 0xB8E128, Find: 14, Replace: 4} # KePub
diff --git a/src/versions/4.24.15676/libnickel.so.1.0.0.yaml/jackie_w.yaml b/src/versions/4.25.15875/libnickel.so.1.0.0.yaml/jackie_w.yaml
index 7a70dbf..e1546a7 100644
--- a/src/versions/4.24.15676/libnickel.so.1.0.0.yaml/jackie_w.yaml
+++ b/src/versions/4.25.15875/libnickel.so.1.0.0.yaml/jackie_w.yaml
@@ -25,18 +25,18 @@ Dictionary text font-family/font-size/line-height:
 
   # Stage 1: Change DictionaryView to remove %variables
   # from:
-  #   body { padding-left: %5px; }
-  #   body { font: %1px %2; line-height: 1.4em; }
-  #   span.word { font-weight: bold; font-size: 130%; margin-left: -%5px; }
-  #   div.descriptionFont { font-family: %3; }
-  #   ol { font-size: %1px; margin-left: %4em; margin-top: 0px; }
+  #   body { padding-left: %3px; }
+  #   body { font: %1px serif; line-height: 1.4em; }
+  #   span.word { font-weight: bold; font-size: 130%; margin-left: -%3px; }
+  #   div.descriptionFont { font-family: serif; }
+  #   ol { font-size: %1px; margin-left: %2em; margin-top: 0px; }
   #   ol p { font-size: %1px; }
   #   i, i * { font-style: italic; }
   #   b, b * { font-weight: bold; }
   #   .sc, .sc * { font-variant: small-caps; }
   #   .block, .border { border-radius: 3px; padding: 0em .2em; font-size: 90%; }
   # to:
-  #   body {font-size:%1px; line-height:1.40em; font-family:Georgia                   ;}
+  #   body {font-size:%1px; line-height:1.40em; font-family:Georgia                         ;}
   #   body {padding-left:0.5em}
   #   span.word {font-weight:bold; font-size:130%; margin-left:-0.3em}
   #   ol {margin-left:1em; margin-top:0}
@@ -49,8 +49,8 @@ Dictionary text font-family/font-size/line-height:
 
   # ##### N.B. Do not change the next 4 lines #####
   - FindReplaceString:
-      Find:    "body { padding-left: %5px; }\nbody { font: %1px %2; line-height: 1.4em; }\nspan.word { font-weight: bold; font-size: 130%; margin-left: -%5px; }\ndiv.descriptionFont { font-family: %3; }\nol { font-size: %1px; margin-left: %4em; margin-top: 0px; }\nol p { font-size: %1px; }\ni, i * { font-style: italic; }\nb, b * { font-weight: bold; }\n.sc, .sc * { font-variant: small-caps; }\n.block, .border { border-radius: 3px; padding: 0em .2em; font-size: 90%; }\n"
-      Replace: "body {font-size:%1px; line-height:1.40em; font-family:Georgia                   ;}\nbody {padding-left:0.5em}\nspan.word {font-weight:bold; font-size:130%; margin-left:-0.3em}\nol {margin-left:1em; margin-top:0}\nblockquote {margin:.3em 0 .3em 1em}\nblockquote>blockquote {margin:-.3em 0 0 2em}\ni, i * {font-style:italic}\nb, b * {font-weight:bold}\n.sc, .sc * {font-variant:small-caps}\n.block, .border {border-radius:3px; padding:0 .2em; font-size:90%}\n"
+      Find:    "body { padding-left: %3px; }\nbody { font: %1px serif; line-height: 1.4em; }\nspan.word { font-weight: bold; font-size: 130%; margin-left: -%3px; }\ndiv.descriptionFont { font-family: serif; }\nol { font-size: %1px; margin-left: %2em; margin-top: 0px; }\nol p { font-size: %1px; }\ni, i * { font-style: italic; }\nb, b * { font-weight: bold; }\n.sc, .sc * { font-variant: small-caps; }\n.block, .border { border-radius: 3px; padding: 0em .2em; font-size: 90%; }\n"
+      Replace: "body {font-size:%1px; line-height:1.40em; font-family:Georgia                         ;}\nbody {padding-left:0.5em}\nspan.word {font-weight:bold; font-size:130%; margin-left:-0.3em}\nol {margin-left:1em; margin-top:0}\nblockquote {margin:.3em 0 .3em 1em}\nblockquote>blockquote {margin:-.3em 0 0 2em}\ni, i * {font-style:italic}\nb, b * {font-weight:bold}\n.sc, .sc * {font-variant:small-caps}\n.block, .border [... cut ...]
+  - BaseAddress: 0x14B715F
   - ReplaceZlib:
       Find: "#lblForgotPin[qApp_deviceIsDragon=true],\n#lblSignOut[qApp_deviceIsDragon=true] {\n  font-size: 26px;\n}"
       Replace: "#lblForgotPin,#lblSignOut{qproperty-visible:false;}"
-  - BaseAddress: 0x14AB23F
+  - BaseAddress: 0x14B96D7
   - ReplaceZlib:
       Find: "#lblForgotPin[qApp_deviceIsDragon=true],\n#lblSignOut[qApp_deviceIsDragon=true] {\n  font-size: 26px;\n}"
       Replace: "#lblForgotPin,#lblSignOut{qproperty-visible:false;}"
```

</details>

NiLuJe, 2020-12-15:
Fun fact: the latest Kindle FW release dropped another [rather large hint](https://github.com/NiLuJe/KindleTool/commit/ea4caa362a9e19d953abac3f8a2a721c439736a4) of a move off i.MX devices... towards MediaTek ones, in their case.

(The Kindle FW runs X, which means we have a nice driver name swicth, from `imx` to `mtk`...).