pgaskin/kobopatch-patches #80: Patch for removing Discover Tab

Terisa, 2020-11-14:
Hi,

  It would be nice a patch to remove the Discover Tab in the bottom bar of the Home Page.

zmanji, 2021-01-08:
@pgaskin I was able to get this to work on the latest firmware on a Libra:

```
Hide Discover Button:
  - Enabled: Yes
  - BaseAddress: 0x14B90C2
  - FindReplaceString:
      Find:    "#MainNavView[qApp_deviceIsPhoenix=true] {\n  max-height: 76px;\n  min-height: 76px;\n}"
      Replace: "#discoverButton {\n qproperty-visible: false; \n}                                    "
      MustMatchLength: yes
```

The associated qcss file is `MainNavView.qss` (0x14B90C2 is the offset for this resource). Unfortunately in the latest firmware it doesn't appear to be zlib compressed, so I believe this forces us to just use `FindReplaceString` (please let me know if this is not true).

As a basic hack, I break the CSS for Pheonix devices and replace it with the CSS required to hide the button, it appears to work fine without issue.

A proper patch I feel would append `#discoverButton {\n qproperty-visible: false; \n}` to the CSS, minify the CSS and replace the resource entirely but I don't know how to do that. I think this would require changes to `kobopatch` itself, where it would be smarter about css resources.

pgaskin, 2021-01-08:
We still haven't decided if we're going to implement this patch yet.

> Unfortunately in the latest firmware it doesn't appear to be zlib compressed, so I believe this forces us to just use FindReplaceString (please let me know if this is not true).

Yes, that's currently correct. Note that you don't need BaseAddress if you're using FindReplaceString with a unique string.

> I think this would require changes to kobopatch itself, where it would be smarter about css resources.

That's a planned feature. It will be implemented using [pgaskin/qrc](https://github.com/pgaskin/qrc) and will have much better CSS modification support. The main blocker for this is implementing the [heuristics](https://github.com/pgaskin/kobo-plugin-experiments/issues/8#issuecomment-643569680) for detecting Qt resource streams.

zmanji, 2021-01-08:
Thanks for the update, once `kobopatch` can be enhanced to have better support for resources, patches like this (and many other ui tweaks) would become much easier. If qrc was the underlying library, I could even see patches changing the bootscreen svgs and other ui elements.

NiLuJe, 2021-01-08:
Sidebar: the bootscreen is not an SVG, it's a raw fb dump gunzipped in a loop and piped to pickel.

Anything that already patches [on-animator](https://github.com/NiLuJe/kfmon/blob/17138df4ada980a0be55f94f7df13d46ea787503/scripts/on-animator.sh#L73-L107) can replace it (e.g., [KFMon](https://github.com/NiLuJe/KFMon#things-to-watch-out-for).

pgaskin, 2021-01-12:
I'm going to close this for now for the reasons discussed in https://github.com/pgaskin/kobopatch-patches/pull/82#issuecomment-757924564.

closed 2021-01-12 by pgaskin