pgaskin/kobopatch-patches #81: Add nickel Header/Footer option to increase caption width

jackiew1, 2020-12-15:
See post #10 onwards of https://www.mobileread.com/forums/showthread.php?t=334722

jackiew1, 2020-12-15:
@pgaskin Please can you provide some details about the following. 

nickel qss resources are quite variable about which models they assign specific values for, e.g. sometimes Alyssum/Nova are handled by Dragon and sometimes they have their own values.

ReadingFooter.qss is rare in that `ReadingFooter { qproperty-footerMargin ...}` only specifies 3 different values:
- default (presumably Trilogy)
- Dragon
- Daylight

I need to make sure each of the 3 patched default values can accommodate the model with the widest `dogear.png`. There are 5 different `dogear.png` resources:
- Trilogy: 57x54
- Phoenix: 64x61
- Dragon: 102x97
- Storm: 116x110
- Daylight: 133x126

This is my best guess about which models will be assigned to each of the 3 `ReadingFooter { qproperty-footerMargin}. Please will you correct any which you think are wrong?

- default: KoboTouch, Mini, **Glo, Aura6", AuraEd2, Nia**
- Dragon: AuraHD, H2O, H2Ov2, GloHD, ClaraHD, **LibraH2O**
- Daylight: AuraOne, Forma

The 5 **bolded** models are the ones I'm least confident about.

pgaskin, 2020-12-15:
I made this a while ago based on the work @shermp, @NiLuJe, and I were doing for koboutils:

[image]

jackiew1, 2020-12-16:
> I made this a while ago based on the work @shermp, @NiLuJe, and I were doing for koboutils:

I don't really think that answers my question, though. Let me spell it out. 

There isn't a `qApp_deviceIsPhoenix` setting for `ReadingFooter { qproperty-footerMargin}` so will Phoenix-like devices (Glo, Aura6", AuraEd2, Nia) use the default (Trilogy) setting of 105 or the Dragon setting of 170? I was hoping you could check the actual code unless you already know the answer.

Similarly, there is no `qApp_deviceIsStorm` setting for so will the LibraH2O use the Dragon setting of 170 or the Daylight setting of 221. From your table above I'm guessing the former, but I'd rather not have to guess.

jackiew1, 2020-12-17:
@pgaskin Any chance of an answer to my question?

pgaskin, 2020-12-17:
Yep, I'll look into it at the same time as I work on NM and kepubify this weekend.

pgaskin, 2020-12-23:
It appears that the default is 25, but I haven't looked into it closely.

```asm
                            loc_cdd7da:
00cdd7da 0022                   movs       r2, #0x0 ; CODE XREF=_ZN13ReadingFooterC1EP7QWidget+934, _ZN13ReadingFooterC1EP7QWidget+958
00cdd7dc 1923                   movs       r3, #25
00cdd7de 606A                   ldr        r0, [r4, #0x24]
00cdd7e0 1946                   mov        r1, r3
00cdd7e2 0092                   str        r2, [sp, #0xa8 + var_A8]
00cdd7e4 22F4E2E0               blx        _ZN7QLayout18setContentsMarginsEiiii@PLT ; QLayout::setContentsMargins(int, int, int, int)
```

```asm
                            _ZN13ReadingFooter15setFooterMarginEi:        // ReadingFooter::setFooterMargin(int)
00cde170 80B5                   push       {r7, lr}
00cde172 0022                   movs       r2, #0x0
00cde174 82B0                   sub        sp, #0x8
00cde176 0B46                   mov        r3, r1
00cde178 02AF                   add        r7, sp, #0x8
00cde17a 406A                   ldr        r0, [r0, #0x24]
00cde17c 0092                   str        r2, [sp, #0x10 + var_10]
00cde17e 21F416E4               blx        _ZN7QLayout18setContentsMarginsEiiii@PLT ; QLayout::setContentsMargins(int, int, int, int)
00cde182 BD46                   mov        sp, r7
00cde184 80BD                   pop        {r7, pc}
```

jackiew1, 2020-12-23:
> It appears that the default is 25, but I haven't looked into it closely.
> 
> 15875

Default of what is 25? 25 bears no resemblance to any of the 3 `qproperty-footerMargin` values listed in ReadingFooter.qss, i.e. 105, 170, 221.

All I'm asking you to confirm is what value does a Phoenix-type model use on the rare occasion when there is no `qApp_deviceIsPhoenix` value specified? 25px is only about 3-4mm on a Glo/Aura6"

pgaskin, 2020-12-23:
25 is the left/right margins set by default for the `footerMargin` qproperty and should be the one which applies if it isn't explicitly set elsewhere. What happens if you patch out the `qproperty-footerMargin` from the CSS?

jackiew1, 2020-12-23:
I've never had the nerve to try removing any variable completely. I'll try it. I'm pretty sure no recent model is actually using a value as small as 25. 

I don't have a Trilogy device, which I think would use the smallest value of 105 to accommodate a `dogear.png` bookmark of 57x54, i.e. almost twice the necessary size.

jackiew1, 2020-12-23:
> 25 is the left/right margin set by default for the `footerMargin` qproperty and should be the one which applies if it isn't explicitly set elsewhere. What happens if you patch out the `qproperty-footerMargin` from the CSS?

You're right. If the `footerMargin` is completely patched out the left/right margin is approx. 2mm on a KA1 which would fit with the 25 value you speak of. The dogear is almost completely obscured using this value. This confirms my opinion that no model is actually using 25 or user complaints would have occurred by now.

If you've no other insight into what's really happening I'll have to assume that Phoenix devices also use the `ReadingFooter.qss` default value of 105, same as Trilogy devices.

pgaskin, 2020-12-30:
Sorry about the late response. I haven't had much time recently.

I took a look at this from the beginning, and it appears I missed something earlier. I initially assumed it was in the code based on your description of the issue, so I overlooked the CSS itself. 

The plain `ReadingFooter {` block with a margin of `105` essentially becomes the default value, and the other two blocks override it for dragon and daylight devices.

For the future, remember that CSS cascades by design, and more specific or later selectors will override other ones (i.e. CSS selectors aren't exclusive to each other; they overlap).

NiLuJe, 2020-12-30:
I think @jackiew1 was concerned specifically about *which* devices falls into that category (i.e., whether that's strictly speaking the two devices in question, or whether some "similar" devices get batched with it , and which ones (e.g., the Forma has no dedicated CSS handling, and is generally using Aura One codepaths)).

pgaskin, 2020-12-30:
> whether that's strictly speaking the two devices in question, or whether some "similar" devices get batched with it , and which ones

That's in the diagram.

jackiew1, 2020-12-30:
In other words, my guess from the original post was correct. I'll commit the replacement patch shortly.

closed 2020-12-30 by jackiew1

jackiew1, 2021-01-10:
If only I knew how to spell "caption" ... Sorry about that!

commit f7a04aa8 (2020-12-30): New patch 'Custom header/footer captions'

replaces 'Custom new header/footer font' closes #81