pgaskin/kobopatch-patches #85: Patches for 4.26.16704

pgaskin, 2021-02-16:
- [ ] Upload testdata (@pgaskin)
- [ ] Extract resources (@pgaskin)
- [ ] Find libnickel offsets where necessary (@pgaskin)
- [ ] Update most patches (@pgaskin)
- [ ] Update nickel patches (@jackiew1)
- [ ] Test patches (@pgaskin)
- [ ] Test nickel patches (@jackiew1)
- [ ] Make release (@pgaskin)
- [ ] MR post (@pgaskin)

pgaskin, 2021-02-16:
The same as usual:

```
Errors:
  /home/patrick/src/kobopatch-patches/src/versions/4.26.16704/nickel.yaml/pgaskin.yaml: Remove forgot pin button from lock screen: could not apply patch "Remove forgot pin button from lock screen": line 20: inst 4: ReplaceZlib: not a zlib stream
  /home/patrick/src/kobopatch-patches/src/versions/4.26.16704/libnickel.so.1.0.0.yaml/geoffr.yaml: Set font scale factor: could not apply patch "Set font scale factor": line 531: inst 3: ReplaceInt: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.26.16704/libnickel.so.1.0.0.yaml/geoffr.yaml: Set KePub hyphenation: could not apply patch "Set KePub hyphenation": line 320: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.26.16704/libnickel.so.1.0.0.yaml/pgaskin.yaml: Customize ComfortLight settings: could not apply patch "Customize ComfortLight settings": line 566: inst 4: ReplaceInt: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.26.16704/libnickel.so.1.0.0.yaml/pgaskin.yaml: Hide browser from beta features: could not apply patch "Hide browser from beta features": line 80: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.26.16704/libnickel.so.1.0.0.yaml/pgaskin.yaml: Allow rotation on all devices: could not apply patch "Allow rotation on all devices": line 432: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.26.16704/libnickel.so.1.0.0.yaml/pgaskin.yaml: DeveloperSettings - AutoUsbGadget: could not apply patch "DeveloperSettings - AutoUsbGadget": line 637: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.26.16704/nickel.yaml/oren64.yaml: New home screen subtitle custom font: could not apply patch "New home screen subtitle custom font": line 31: inst 4: ReplaceZlib: not a zlib stream
```

pgaskin, 2021-02-16:
**Resource extraction:**

```bash
echo "nickel@16704/qInitResources_resources";    ./qrc2zip --output "nickel.16704.qInitResources_resources.zip"    --recursive --verbose "nickel" 1 $((0x11698e0 - 0x0010000)) $((0x0026818 - 0x0010000)) $((0x1169470 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
echo "nickel@16704/qInitResources_translations"; ./qrc2zip --output "nickel.16704.qInitResources_translations.zip" --recursive --verbose "nickel" 1 $((0x14c7488 - 0x0010000)) $((0x1169c90 - 0x0010000)) $((0x14c7200 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
echo "nickel@16704/qInitResources_styles";       ./qrc2zip --output "nickel.16704.qInitResources_styles.zip"       --recursive --verbose "nickel" 1 $((0x14e4218 - 0x0010000)) $((0x14c75a0 - 0x0010000)) $((0x14e1b98 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
echo "nickel@16704/qInitResources_certificates"; ./qrc2zip --output "nickel.16704.qInitResources_certificates.zip" --recursive --verbose "nickel" 1 $((0x150b5a0 - 0x0010000)) $((0x14e4d00 - 0x0010000)) $((0x150a550 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
```

<details><summary><b>Output</b></summary>

```
nickel@16704/qInitResources_resources
DIR     "qss" (0x1159C7C + 14)
FILE    "qss/TouchMenu.qss" (0x1158915 + 189)
DIR     "html" (0x1159C60 + 28)
FILE    "html/RGBFrontLightAboutView.html" (0x1158B21 + 1144)
FILE    "html/RGBFrontLightAboutViewSnow.html" (0x1158F9D + 1228)
DIR     "keys" (0x1159C52 + 14)
FILE    "keys/device.qmap" (0x1158AAD + 112)
FILE    "omfTemplate.html" (0x10DADA9 + 231)
DIR     "pika.rcc" (0x410D + 14)
DIR     "pika.rcc/images" (0x411B + 42)
DIR     "pika.rcc/images/fte" (0x4161 + 14)
FILE    "pika.rcc/images/fte/quicktour_diagram.png" (0x18 + 3841)
DIR     "pika.rcc/images/home" (0x4153 + 14)
FILE    "pika.rcc/images/home/quicktour_device.png" (0x2F3E + 4337)
DIR     "pika.rcc/images/about" (0x4145 + 14)
FILE    "pika.rcc/images/about/global_certification.png" (0xF1D + 8221)
FILE    "verifier.pub" (0x815D67 + 451)
DIR     "daylight.rcc" (0xC314C + 14)
DIR     "daylight.rcc/images" (0xC315A + 322)
DIR     "daylight.rcc/images/fte" (0xC4614 + 280)
FILE    "daylight.rcc/images/fte/missingbooks_factoryreset.png" (0x8C544 + 9065)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg4_adjust_brightness.png" (0xAEF42 + 24664)
FILE    "daylight.rcc/images/fte/missingbooks_account.png" (0xA3BF3 + 22560)
FILE    "daylight.rcc/images/fte/missingbooks_repair.png" (0x976B7 + 20737)
FILE    "daylight.rcc/images/fte/quicktour_nightmode.png" => "daylight.rcc/images/fte/quicktour_nightmode[language!Japanese].png" (0x9DBA9 + 24646)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg2.png" => "daylight.rcc/images/fte/quicktour_reading_pg2[language!Japanese].png" (0x6E56F + 23892)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg2.png" (0x742C7 + 23662)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg3.png" (0x81C46 + 23519)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg3.png" => "daylight.rcc/images/fte/quicktour_reading_pg3[language!Japanese].png" (0x7BF9E + 23716)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg5.png" (0x87829 + 19735)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg1.png" (0x68C55 + 22806)
FILE    "daylight.rcc/images/fte/missingbooks_sync.png" (0x8E8B1 + 6061)
FILE    "daylight.rcc/images/fte/fte_oob_connected.png" => "daylight.rcc/images/fte/fte_oob_connected[language!Japanese].png" (0x90E0C + 9157)
FILE    "daylight.rcc/images/fte/fte_oob_plugin.svg" (0x90062 + 3494)
FILE    "daylight.rcc/images/fte/pocket_logo_large.png" (0x9C7BC + 3862)
FILE    "daylight.rcc/images/fte/quicktour_texticon_search.png" (0x9DB07 + 158)
FILE    "daylight.rcc/images/fte/quicktour_diagram.png" (0x79F39 + 8289)
FILE    "daylight.rcc/images/fte/missingbooks_help.png" (0xA9417 + 23335)
FILE    "daylight.rcc/images/fte/overflow_small.png" (0x9D6D6 + 1069)
FILE    "daylight.rcc/images/fte/missingbooks_intro.png" (0x931D5 + 17630)
DIR     "daylight.rcc/images/book" (0xC45CE + 70)
FILE    "daylight.rcc/images/book/dustbin_delete_icon_inverted.png" (0xB550F + 257)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_highlight.png" (0xB534C + 447)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_dogear.png" (0xB4F9E + 409)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_note.png" (0xB513B + 202)
FILE    "daylight.rcc/images/book/dustbin_delete_icon.png" (0xB5209 + 319)
DIR     "daylight.rcc/images/help" (0xC4596 + 56)
FILE    "daylight.rcc/images/help/help_icon.png" (0xB8C44 + 1022)
FILE    "daylight.rcc/images/help/kobopluslogo_help_sp.png" (0xBAA53 + 7449)
FILE    "daylight.rcc/images/help/kobopluslogo_eLeya.png" (0xB61D3 + 10861)
FILE    "daylight.rcc/images/help/kobopluslogo_help_version.png" (0xB9046 + 6665)
DIR     "daylight.rcc/images/home" (0xC44A8 + 238)
FILE    "daylight.rcc/images/home/main_nav_books.png" (0x546EC + 637)
FILE    "daylight.rcc/images/home/annotations_bookmark.png" (0x53ACF + 409)
FILE    "daylight.rcc/images/home/pocket_favourite_star.png" (0x523F6 + 328)
FILE    "daylight.rcc/images/home/main_nav_discover.png" (0x54296 + 1106)
FILE    "daylight.rcc/images/home/main_nav_discover_active.png" (0x60E6F + 860)
[... 2686 lines cut ...]
index 17417bc..cb5c824 100644
Binary files a/15875/translations/trans_nl_NL.qm and b/16704/translations/trans_nl_NL.qm differ
diff --git a/15875/translations/trans_pt_BR.qm b/16704/translations/trans_pt_BR.qm
[... cut ...]
new file mode 100644
index 0000000..719850e
Binary files /dev/null and b/16704/trilogy.rcc/images/help/kobopluslogo_eLeya.png differ
```

</details>

---

https://temp.s3.geek1011.net/kobo-resources/nickel.16704.qInitResources_certificates.zip
https://temp.s3.geek1011.net/kobo-resources/nickel.16704.qInitResources_resources.zip
https://temp.s3.geek1011.net/kobo-resources/nickel.16704.qInitResources_styles.zip
https://temp.s3.geek1011.net/kobo-resources/nickel.16704.qInitResources_translations.zip

pgaskin, 2021-02-16:
**Random notes:**

- Kobo Plus e_leYa appears to be another variant of Kobo Plus for `bol.com`, but I haven't looked into it in detail yet.
- TODO

pgaskin, 2021-02-16:
<details><summary><b>$ diff <(cat 15875/symdump.out.json | jq -r '.[] | .Name + ", " + .Demangled' | sort) <(cat 16704/symdump.out.json | jq -r '.[] | .Name + ", " + .Demangled' | sort)</b></summary>

```diff
15a16,17
> AES_encrypt, 
> AES_set_encrypt_key, 
225a228
> calloc, 
230a234,235
> chmod, 
> clearerr, 
279a285,287
> deflate, 
> deflateEnd, 
> deflateInit2_, 
365a374
> EVP_sha1, 
371a381,382
> fdopen, 
> ferror, 
373a385
> fileno, 
387a400
> fseeko, 
388a402
> ftello, 
389a404
> __fxstat, 
416a432,436
> HMAC_CTX_cleanup, 
> HMAC_CTX_init, 
> HMAC_Final, 
> HMAC_Init_ex, 
> HMAC_Update, 
417a438,440
> inflate, 
> inflateEnd, 
> inflateInit2_, 
472a496
> localtime_r, 
475a500
> memchr, 
485d509
< MIME_TYPE_HELP, 
492d515
< MIME_TYPE_RSS, 
495a519
> mktime, 
520d543
< ONE_STORE_ANNOTATIONS_HOST, 
552a576
> ONE_STORE_READING_SERVICES_HOST, 
649a674
> random, 
654a680
> remove, 
655a682
> rename, 
678a706,707
> sprintf, 
> srandom, 
687a717
> strerror, 
839a870
> _Z7networkv, network()
864a896
> zError, 
968d999
< _ZGVZN9QtPrivate15ConnectionTypesINS_4ListIIRK7QVectorI7QStringERK10QByteArraybEEELb1EE5typesEvE1t, guard variable for QtPrivate::ConnectionTypes<QtPrivate::List<QVector<QString> const&, QByteArray const&, bool>, true>::types()::t
978d1008
[... 787 lines cut ...]
56274a56343
> _ZTV28GetUpdatedAnnotationsCommand, vtable for GetUpdatedAnnotationsCommand
56621a56691
> _ZTV40Migration162_AddAnnotationsMigrationFlag, vtable for Migration162_AddAnnotationsMigrationFlag
57045d57114
< _ZZN11QMetaTypeIdI15QDBusObjectPathE14qt_metatype_idEvE11metatype_id, QMetaTypeId<QDBusObjectPath>::qt_metatype_id()::metatype_id
57050d57118
< _ZZN11QMetaTypeIdI23QDBusUnixFileDescriptorE14qt_metatype_idEvE11metatype_id, QMetaTypeId<QDBusUnixFileDescriptor>::qt_metatype_id()::metatype_id
57052,57053d57119
< _ZZN11QMetaTypeIdI4QMapI15QDBusObjectPathS0_I7QString8QVariantEEE14qt_metatype_idEvE11metatype_id, QMetaTypeId<QMap<QDBusObjectPath, QMap<QString, QVariant> > >::qt_metatype_id()::metatype_id
< _ZZN11QMetaTypeIdI4QMapI15QDBusObjectPathS0_I7QStringS0_IS2_8QVariantEEEE14qt_metatype_idEvE11metatype_id, QMetaTypeId<QMap<QDBusObjectPath, QMap<QString, QMap<QString, QVariant> > > >::qt_metatype_id()::metatype_id
57070d57135
< _ZZN11QMetaTypeIdI7QVectorI7QStringEE14qt_metatype_idEvE11metatype_id, QMetaTypeId<QVector<QString> >::qt_metatype_id()::metatype_id
57304a57370
> _ZZN9QtPrivate15ConnectionTypesINS_4ListIIRK7QStringRK10QByteArrayEEELb1EE5typesEvE1t, QtPrivate::ConnectionTypes<QtPrivate::List<QString const&, QByteArray const&>, true>::types()::t
57317d57382
< _ZZN9QtPrivate15ConnectionTypesINS_4ListIIRK7QVectorI7QStringERK10QByteArraybEEELb1EE5typesEvE1t, QtPrivate::ConnectionTypes<QtPrivate::List<QVector<QString> const&, QByteArray const&, bool>, true>::types()::t
57330d57394
< _ZZN9QtPrivate19ValueTypeIsMetaTypeI7QVectorI7QStringELb1EE17registerConverterEiE1f, QtPrivate::ValueTypeIsMetaType<QVector<QString>, true>::registerConverter(int)::f
57332,57333d57395
< _ZZN9QtPrivate30AssociativeValueTypeIsMetaTypeI4QMapI15QDBusObjectPathS1_I7QString8QVariantEELb1EE17registerConverterEiE1f, QtPrivate::AssociativeValueTypeIsMetaType<QMap<QDBusObjectPath, QMap<QString, QVariant> >, true>::registerConverter(int)::f
< _ZZN9QtPrivate30AssociativeValueTypeIsMetaTypeI4QMapI15QDBusObjectPathS1_I7QStringS1_IS3_8QVariantEEELb1EE17registerConverterEiE1f, QtPrivate::AssociativeValueTypeIsMetaType<QMap<QDBusObjectPath, QMap<QString, QMap<QString, QVariant> > >, true>::registerConverter(int)::f
```

</details>

jackiew1, 2021-02-16:
@pgaskin 
There are only 6 changed nickel .qss resources. Only the first one has an existing patch in nickel.yaml but I don't think it needs any changes.
- N3SettingsExtrasView.qss  (`Show all games`)
- ArticleCoverWidget.qss
- DropboxView.qss
- HtmlArticle.qss
- MyWordsView.qss
- SubscriptionPromoWidget.qss

There are no added or removed .qss resources.

At first glance a bit of a non-event for my workload. 😄

pgaskin, 2021-02-16:
**Random notes:**

- Kobo Plus e_leYa appears to be another variant of Kobo Plus for `bol.com`, but I haven't looked into it in detail yet.
- There doesn't appear to be any major CSS changes.
- The bluealsa references aren't there anymore... were they put under a build condition, or will the next Kobo not support audiobooks after all?
- TODO

pgaskin, 2021-02-16:
Thanks, @jackiew1. I'll make a release sometime today after I get a chance to test the patches.

jackiew1, 2021-02-16:
> ```Errors:
>   /home/patrick/src/kobopatch-patches/src/versions/4.26.16704/nickel.yaml/oren64.yaml: New home screen subtitle custom font: could not apply patch "New home screen subtitle custom font": line 31: inst 4: ReplaceZlib: not a zlib stream
> ```

@pgaskin I've been meaning to say this for a while. The above patch really doesn't need to use `BaseAddress`. It's currently patching both `GenericHomeWidget.qss` and `SubcategoryCarouselWidget.qss` when only the former is necessary.

You could change the `BaseAddress` to `FindZlib: "#homeWidgetTopContainer"` to make it multi-version. Unless you know something about `SubcategoryCarouselWidget.qss` that I don't.

pgaskin, 2021-02-16:
Released: [`v70`](https://github.com/pgaskin/kobopatch-patches/releases/v70), MR:thread-337427.

pgaskin, 2021-02-16:
<details><summary><b>$ git diff --no-index src/versions/{4.24.15676,4.25.15875}</b></summary>

```diff
diff --git a/src/versions/4.24.15676/libnickel.so.1.0.0.yaml/geoffr.yaml b/src/versions/4.25.15875/libnickel.so.1.0.0.yaml/geoffr.yaml
index c625638..d43d14c 100644
--- a/src/versions/4.24.15676/libnickel.so.1.0.0.yaml/geoffr.yaml
+++ b/src/versions/4.25.15875/libnickel.so.1.0.0.yaml/geoffr.yaml
@@ -317,8 +317,8 @@ Set KePub hyphenation:
       will always be turned on, regardless of justification button setting.
 
       (The publisher can still turn hyphenation off/on in the book's stylesheet.)
-  - ReplaceBytes: {Offset: 0xB911A2, FindInstBLX: {SymPLT: "QString::operator==(QLatin1String) const"}, ReplaceH: 01 20 01 20} # Alternative 1: Always turn KePub hyphenation on
-# - ReplaceBytes: {Offset: 0xB911A2, FindInstBLX: {SymPLT: "QString::operator==(QLatin1String) const"}, ReplaceH: 00 20 00 20} # Alternative 2: Never turn KePub hyphenation on
+  - ReplaceBytes: {Offset: 0xB8E556, FindInstBLX: {SymPLT: "QString::operator==(QLatin1String) const"}, ReplaceH: 01 20 01 20} # Alternative 1: Always turn KePub hyphenation on
+# - ReplaceBytes: {Offset: 0xB8E556, FindInstBLX: {SymPLT: "QString::operator==(QLatin1String) const"}, ReplaceH: 00 20 00 20} # Alternative 2: Never turn KePub hyphenation on
 
 Force user line spacing in KePubs:
   - Enabled: no
@@ -528,5 +528,5 @@ Set font scale factor:
   # Original value X=14, so original scale factor is (font size slider) / 46.
   # Allowed replacement values for X: 0-15 (smaller X --> larger scale factor)
   # Note: vmov.f32 s1, #15.0 before getReadingFontSizeScaleFactor
-  - ReplaceInt: {Offset: 0x589544, Find: 14, Replace: 4} # ePub
-  - ReplaceInt: {Offset: 0xB90D74, Find: 14, Replace: 4} # KePub
+  - ReplaceInt: {Offset: 0x58F3F4, Find: 14, Replace: 4} # ePub
+  - ReplaceInt: {Offset: 0xB8E128, Find: 14, Replace: 4} # KePub
diff --git a/src/versions/4.24.15676/libnickel.so.1.0.0.yaml/jackie_w.yaml b/src/versions/4.25.15875/libnickel.so.1.0.0.yaml/jackie_w.yaml
index 7a70dbf..e1546a7 100644
--- a/src/versions/4.24.15676/libnickel.so.1.0.0.yaml/jackie_w.yaml
+++ b/src/versions/4.25.15875/libnickel.so.1.0.0.yaml/jackie_w.yaml
@@ -25,18 +25,18 @@ Dictionary text font-family/font-size/line-height:
 
   # Stage 1: Change DictionaryView to remove %variables
   # from:
-  #   body { padding-left: %5px; }
-  #   body { font: %1px %2; line-height: 1.4em; }
-  #   span.word { font-weight: bold; font-size: 130%; margin-left: -%5px; }
-  #   div.descriptionFont { font-family: %3; }
-  #   ol { font-size: %1px; margin-left: %4em; margin-top: 0px; }
+  #   body { padding-left: %3px; }
+  #   body { font: %1px serif; line-height: 1.4em; }
+  #   span.word { font-weight: bold; font-size: 130%; margin-left: -%3px; }
+  #   div.descriptionFont { font-family: serif; }
+  #   ol { font-size: %1px; margin-left: %2em; margin-top: 0px; }
   #   ol p { font-size: %1px; }
   #   i, i * { font-style: italic; }
   #   b, b * { font-weight: bold; }
   #   .sc, .sc * { font-variant: small-caps; }
   #   .block, .border { border-radius: 3px; padding: 0em .2em; font-size: 90%; }
   # to:
-  #   body {font-size:%1px; line-height:1.40em; font-family:Georgia                   ;}
+  #   body {font-size:%1px; line-height:1.40em; font-family:Georgia                         ;}
   #   body {padding-left:0.5em}
   #   span.word {font-weight:bold; font-size:130%; margin-left:-0.3em}
   #   ol {margin-left:1em; margin-top:0}
@@ -49,8 +49,8 @@ Dictionary text font-family/font-size/line-height:
 
   # ##### N.B. Do not change the next 4 lines #####
   - FindReplaceString:
-      Find:    "body { padding-left: %5px; }\nbody { font: %1px %2; line-height: 1.4em; }\nspan.word { font-weight: bold; font-size: 130%; margin-left: -%5px; }\ndiv.descriptionFont { font-family: %3; }\nol { font-size: %1px; margin-left: %4em; margin-top: 0px; }\nol p { font-size: %1px; }\ni, i * { font-style: italic; }\nb, b * { font-weight: bold; }\n.sc, .sc * { font-variant: small-caps; }\n.block, .border { border-radius: 3px; padding: 0em .2em; font-size: 90%; }\n"
-      Replace: "body {font-size:%1px; line-height:1.40em; font-family:Georgia                   ;}\nbody {padding-left:0.5em}\nspan.word {font-weight:bold; font-size:130%; margin-left:-0.3em}\nol {margin-left:1em; margin-top:0}\nblockquote {margin:.3em 0 .3em 1em}\nblockquote>blockquote {margin:-.3em 0 0 2em}\ni, i * {font-style:italic}\nb, b * {font-weight:bold}\n.sc, .sc * {font-variant:small-caps}\n.block, .border {border-radius:3px; padding:0 .2em; font-size:90%}\n"
+      Find:    "body { padding-left: %3px; }\nbody { font: %1px serif; line-height: 1.4em; }\nspan.word { font-weight: bold; font-size: 130%; margin-left: -%3px; }\ndiv.descriptionFont { font-family: serif; }\nol { font-size: %1px; margin-left: %2em; margin-top: 0px; }\nol p { font-size: %1px; }\ni, i * { font-style: italic; }\nb, b * { font-weight: bold; }\n.sc, .sc * { font-variant: small-caps; }\n.block, .border { border-radius: 3px; padding: 0em .2em; font-size: 90%; }\n"
+      Replace: "body {font-size:%1px; line-height:1.40em; font-family:Georgia                         ;}\nbody {padding-left:0.5em}\nspan.word {font-weight:bold; font-size:130%; margin-left:-0.3em}\nol {margin-left:1em; margin-top:0}\nblockquote {margin:.3em 0 .3em 1em}\nblockquote>blockquote {margin:-.3em 0 0 2em}\ni, i * {font-style:italic}\nb, b * {font-weight:bold}\n.sc, .sc * {font-variant:small-caps}\n.block, .border [... cut ...]
+  - BaseAddress: 0x14B715F
   - ReplaceZlib:
       Find: "#lblForgotPin[qApp_deviceIsDragon=true],\n#lblSignOut[qApp_deviceIsDragon=true] {\n  font-size: 26px;\n}"
       Replace: "#lblForgotPin,#lblSignOut{qproperty-visible:false;}"
-  - BaseAddress: 0x14AB23F
+  - BaseAddress: 0x14B96D7
   - ReplaceZlib:
       Find: "#lblForgotPin[qApp_deviceIsDragon=true],\n#lblSignOut[qApp_deviceIsDragon=true] {\n  font-size: 26px;\n}"
       Replace: "#lblForgotPin,#lblSignOut{qproperty-visible:false;}"
```

</details>

pgaskin, 2021-02-16:
<details><summary><b>$ git diff --no-index src/versions/{4.25.15875,4.26.16704}</b></summary>

```diff
diff --git a/src/versions/4.25.15875/libnickel.so.1.0.0.yaml/geoffr.yaml b/src/versions/4.26.16704/libnickel.so.1.0.0.yaml/geoffr.yaml
index d43d14c..fe1cafe 100644
--- a/src/versions/4.25.15875/libnickel.so.1.0.0.yaml/geoffr.yaml
+++ b/src/versions/4.26.16704/libnickel.so.1.0.0.yaml/geoffr.yaml
@@ -317,8 +317,8 @@ Set KePub hyphenation:
       will always be turned on, regardless of justification button setting.
 
       (The publisher can still turn hyphenation off/on in the book's stylesheet.)
-  - ReplaceBytes: {Offset: 0xB8E556, FindInstBLX: {SymPLT: "QString::operator==(QLatin1String) const"}, ReplaceH: 01 20 01 20} # Alternative 1: Always turn KePub hyphenation on
-# - ReplaceBytes: {Offset: 0xB8E556, FindInstBLX: {SymPLT: "QString::operator==(QLatin1String) const"}, ReplaceH: 00 20 00 20} # Alternative 2: Never turn KePub hyphenation on
+  - ReplaceBytes: {Offset: 0xB8AD96, FindInstBLX: {SymPLT: "QString::operator==(QLatin1String) const"}, ReplaceH: 01 20 01 20} # Alternative 1: Always turn KePub hyphenation on
+# - ReplaceBytes: {Offset: 0xB8AD96, FindInstBLX: {SymPLT: "QString::operator==(QLatin1String) const"}, ReplaceH: 00 20 00 20} # Alternative 2: Never turn KePub hyphenation on
 
 Force user line spacing in KePubs:
   - Enabled: no
@@ -528,5 +528,5 @@ Set font scale factor:
   # Original value X=14, so original scale factor is (font size slider) / 46.
   # Allowed replacement values for X: 0-15 (smaller X --> larger scale factor)
   # Note: vmov.f32 s1, #15.0 before getReadingFontSizeScaleFactor
-  - ReplaceInt: {Offset: 0x58F3F4, Find: 14, Replace: 4} # ePub
-  - ReplaceInt: {Offset: 0xB8E128, Find: 14, Replace: 4} # KePub
+  - ReplaceInt: {Offset: 0x58D63C, Find: 14, Replace: 4} # ePub
+  - ReplaceInt: {Offset: 0xB8A968, Find: 14, Replace: 4} # KePub
diff --git a/src/versions/4.25.15875/libnickel.so.1.0.0.yaml/pgaskin.yaml b/src/versions/4.26.16704/libnickel.so.1.0.0.yaml/pgaskin.yaml
index d487756..a7b12c4 100644
--- a/src/versions/4.25.15875/libnickel.so.1.0.0.yaml/pgaskin.yaml
+++ b/src/versions/4.26.16704/libnickel.so.1.0.0.yaml/pgaskin.yaml
@@ -78,7 +78,7 @@ Hide browser from beta features:
   - Description: Hides the built-in browser from beta features.
   - ReplaceBytes:
       Base:        "N3SettingsExtrasView::N3SettingsExtrasView(QWidget*)"
-      Offset:      1220
+      Offset:      1426
       FindInstBLX: {SymPLT: "Device::isParentalControlEnabled() const"}
       ReplaceH:    4F F0 01 00
   - ReplaceBytes:
@@ -429,13 +429,13 @@ Allow rotation on all devices:
       Unlike only the DeveloperSettings ForceAllowLandscape option, this patch
       also makes the rotation icon show on the new reader as well.
   # Always return true for ForceAllowLandscape:
-  - ReplaceBytes: {Offset: 0x77B396, FindInstBLX: {SymPLT: "QVariant::toBool() const"}, ReplaceH: 4F F0 01 00}
+  - ReplaceBytes: {Offset: 0x7784EE, FindInstBLX: {SymPLT: "QVariant::toBool() const"}, ReplaceH: 4F F0 01 00}
   # In ReadingMenuView::ReadingMenuView, replace the value of Device::hasOrientationSensor,
   # which is passed to a function to hide/show the rotate icon:
-  - ReplaceBytes: {Offset: 0xBA55E4, FindInstBLX: {SymPLT: "Device::hasOrientationSensor() const"}, ReplaceH: 4F F0 01 00}
+  - ReplaceBytes: {Offset: 0xBA2BE4, FindInstBLX: {SymPLT: "Device::hasOrientationSensor() const"}, ReplaceH: 4F F0 01 00}
   # Also fix sizing of popup by doing the same in RotatePopup::RotatePopup, but
   # note that this will make the Automatic option show too (and do nothing):
-  - ReplaceBytes: {Offset: 0xBC2DDE, FindInstBLX: {SymPLT: "Device::hasOrientationSensor() const"}, ReplaceH: 4F F0 01 00}
+  - ReplaceBytes: {Offset: 0xBC03DE, FindInstBLX: {SymPLT: "Device::hasOrientationSensor() const"}, ReplaceH: 4F F0 01 00}
 
 Don't uppercase header/footer text:
   - Enabled: no
@@ -559,7 +559,7 @@ Customize ComfortLight settings:
   # In an unnamed subroutine two layers into FrontLightPopupController::loadView
   # (find it by going back from QTime::addSecs), the times for the dropdown are
   # generated into a QVector<QPair<QString, QTime>> with a simple loop.
-  - BaseAddress: 0xAE7708 # find the base of the unnamed subroutine with the x-ref to _ZN5QTimeC1Eiiii
+  - BaseAddress: 0xAE4C40 # find the base of the unnamed subroutine with the x-ref to _ZN5QTimeC1Eiiii
   #
   # Change the initial hour / first bedtime dropdown item passed to the QTime
   # constructor (mov r1, #21):
@@ -634,7 +634,7 @@ FeatureSettings - ExportHighlights:
 DeveloperSettings - AutoUsbGadget:
   - Enabled: no
   - Description: Automatically enable USB Storage mode when connected.
-  - ReplaceBytes: {Offset: 0x77B436, FindInstBLX: {SymPLT: "QVariant::toBool() const"}, ReplaceH: 4F F0 01 00}
[... 22 lines cut ...]
+  - BaseAddress: 0x14CE2B4
   - ReplaceZlib: {Find: "MetaDataLabel {\n  color: #666666;\n  qproperty-indent: 0;\n}", Replace: "MetaDataLabel {\n  color: #000000;\n  qproperty-indent: 0;\n}"}
   - ReplaceZlib:
       Find:    "MetaDataLabel[qApp_deviceIsTrilogy=true] {\n  font-size: 13px;\n}"
diff --git a/src/versions/4.25.15875/nickel.yaml/pgaskin.yaml b/src/versions/4.26.16704/nickel.yaml/pgaskin.yaml
index a408f39..898a9fc 100644
--- a/src/versions/4.25.15875/nickel.yaml/pgaskin.yaml
+++ b/src/versions/4.26.16704/nickel.yaml/pgaskin.yaml
@@ -15,11 +15,11 @@ Remove forgot pin button from lock screen:
[... cut ...]
+  - BaseAddress: 0x14CE181
   - ReplaceZlib:
       Find: "#lblForgotPin[qApp_deviceIsDragon=true],\n#lblSignOut[qApp_deviceIsDragon=true] {\n  font-size: 26px;\n}"
       Replace: "#lblForgotPin,#lblSignOut{qproperty-visible:false;}"
-  - BaseAddress: 0x14B96D7
+  - BaseAddress: 0x14CE4D1
   - ReplaceZlib:
       Find: "#lblForgotPin[qApp_deviceIsDragon=true],\n#lblSignOut[qApp_deviceIsDragon=true] {\n  font-size: 26px;\n}"
       Replace: "#lblForgotPin,#lblSignOut{qproperty-visible:false;}"
```

</details>

pgaskin, 2021-02-16:
> when only the former is necessary

Yes, that's correct.

> You could change the BaseAddress to FindZlib: "#homeWidgetTopContainer" to make it multi-version

Hmm. I generally prefer to use find strings directly related to the patch in question, but I guess this quite close. I'll have to think about it.

Either way, this kind of thing won't be necessary anymore once I implement the qrc heuristics and integrate it into kobopatch.

closed 2021-02-16 by pgaskin