pgaskin/kobopatch-patches #86: Patches for 4.27.17623

pgaskin, 2021-06-10:
I'm going to take a look at the firmware first, then I'll update this post.

jackiew1, 2021-06-10:
I'm a bit confused. Are there 2 new firmwares 4.27 and 4.28, or just the latter for the Elipsa?

pgaskin, 2021-06-10:
Nope; I made a typo in the title the first time. There isn't a 4.27 AFAIK, unless that's the factory Elipsa release.

pgaskin, 2021-06-10:
All required symbols are present on this version. It'd be nice if someone with an Elipsa could test NM to ensure things work properly.

jackiew1, 2021-06-10:
A few observations (no details as yet)
- 12 new CSS streams in nickel
- libnickel patch `KePub stylesheet additions` will need updating. I'll take a closer look.

I think all my published nickel patches are still working, i.e. no kobopatch errors. I've obviously not tested on a real device.

pgaskin, 2021-06-10:
```
Errors:
  /home/patrick/src/kobopatch-patches/src/versions/4.28.17623/libnickel.so.1.0.0.yaml/pgaskin.yaml: Both page turn sides go next: could not apply patch "Both page turn sides go next": line 17: inst 4: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.28.17623/libnickel.so.1.0.0.yaml/pgaskin.yaml: Allow rotation on all devices: could not apply patch "Allow rotation on all devices": line 432: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.28.17623/libnickel.so.1.0.0.yaml/pgaskin.yaml: Customize ComfortLight settings: could not apply patch "Customize ComfortLight settings": line 566: inst 4: ReplaceInt: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.28.17623/libnickel.so.1.0.0.yaml/pgaskin.yaml: DeveloperSettings - AutoUsbGadget: could not apply patch "DeveloperSettings - AutoUsbGadget": line 637: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.28.17623/libnickel.so.1.0.0.yaml/pgaskin.yaml: Hide browser from beta features: could not apply patch "Hide browser from beta features": line 80: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.28.17623/libnickel.so.1.0.0.yaml/pgaskin.yaml: Replace adobe page numbers toggle with invert screen: could not apply patch "Replace adobe page numbers toggle with invert screen": line 42: inst 4: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.28.17623/libnickel.so.1.0.0.yaml/pgaskin.yaml: Remove line from bottom tab bar: could not apply patch "Remove line from bottom tab bar": line 165: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.28.17623/libnickel.so.1.0.0.yaml/geoffr.yaml: Set KePub hyphenation: could not apply patch "Set KePub hyphenation": line 320: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.28.17623/libnickel.so.1.0.0.yaml/geoffr.yaml: Set font scale factor: could not apply patch "Set font scale factor": line 531: inst 3: ReplaceInt: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.28.17623/libnickel.so.1.0.0.yaml/geoffr.yaml: KePub stylesheet additions: could not apply patch "KePub stylesheet additions": line 479: inst 4: ReplaceString: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.28.17623/nickel.yaml/oren64.yaml: New home screen subtitle custom font: could not apply patch "New home screen subtitle custom font": line 31: inst 4: ReplaceZlib: not a zlib stream
  /home/patrick/src/kobopatch-patches/src/versions/4.28.17623/nickel.yaml/pgaskin.yaml: Remove forgot pin button from lock screen: could not apply patch "Remove forgot pin button from lock screen": line 20: inst 4: ReplaceZlib: not a zlib stream
  /home/patrick/src/kobopatch-patches/src/versions/4.28.17623/libadobe.so.yaml/pgaskin.yaml: Remove PDF map widget shown during panning: could not apply patch "Remove PDF map widget shown during panning": line 8: inst 5: ReplaceBytes: could not find specified bytes
```

pgaskin, 2021-06-11:
Yep, I'll do that tomorrow.

pgaskin, 2021-06-14:
**Resource extraction:**

```bash
echo "nickel@17623/qInitResources_resources";    ./qrc2zip --output "nickel.17623.qInitResources_resources.zip"    --recursive --verbose "nickel" 1 $((0x1216f78 - 0x0010000)) $((0x00272a0 - 0x0010000)) $((0x1216a88 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
echo "nickel@17623/qInitResources_translations"; ./qrc2zip --output "nickel.17623.qInitResources_translations.zip" --recursive --verbose "nickel" 1 $((0x15cc308 - 0x0010000)) $((0x1217360 - 0x0010000)) $((0x15cc080 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
echo "nickel@17623/qInitResources_styles";       ./qrc2zip --output "nickel.17623.qInitResources_styles.zip"       --recursive --verbose "nickel" 1 $((0x15ea8d0 - 0x0010000)) $((0x15cc420 - 0x0010000)) $((0x15e8060 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
echo "nickel@17623/qInitResources_certificates"; ./qrc2zip --output "nickel.17623.qInitResources_certificates.zip" --recursive --verbose "nickel" 1 $((0x1611ce0 - 0x0010000)) $((0x15eb440 - 0x0010000)) $((0x1610c90 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
```

<details><summary><b>Output</b></summary>

```
nickel@16704/qInitResources_resources
DIR     "qss" (0x1159C7C + 14)
FILE    "qss/TouchMenu.qss" (0x1158915 + 189)
DIR     "html" (0x1159C60 + 28)
FILE    "html/RGBFrontLightAboutView.html" (0x1158B21 + 1144)
FILE    "html/RGBFrontLightAboutViewSnow.html" (0x1158F9D + 1228)
DIR     "keys" (0x1159C52 + 14)
FILE    "keys/device.qmap" (0x1158AAD + 112)
FILE    "omfTemplate.html" (0x10DADA9 + 231)
DIR     "pika.rcc" (0x410D + 14)
DIR     "pika.rcc/images" (0x411B + 42)
DIR     "pika.rcc/images/fte" (0x4161 + 14)
FILE    "pika.rcc/images/fte/quicktour_diagram.png" (0x18 + 3841)
DIR     "pika.rcc/images/home" (0x4153 + 14)
FILE    "pika.rcc/images/home/quicktour_device.png" (0x2F3E + 4337)
DIR     "pika.rcc/images/about" (0x4145 + 14)
FILE    "pika.rcc/images/about/global_certification.png" (0xF1D + 8221)
FILE    "verifier.pub" (0x815D67 + 451)
DIR     "daylight.rcc" (0xC314C + 14)
DIR     "daylight.rcc/images" (0xC315A + 322)
DIR     "daylight.rcc/images/fte" (0xC4614 + 280)
FILE    "daylight.rcc/images/fte/missingbooks_factoryreset.png" (0x8C544 + 9065)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg4_adjust_brightness.png" (0xAEF42 + 24664)
FILE    "daylight.rcc/images/fte/missingbooks_account.png" (0xA3BF3 + 22560)
FILE    "daylight.rcc/images/fte/missingbooks_repair.png" (0x976B7 + 20737)
FILE    "daylight.rcc/images/fte/quicktour_nightmode.png" => "daylight.rcc/images/fte/quicktour_nightmode[language!Japanese].png" (0x9DBA9 + 24646)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg2.png" => "daylight.rcc/images/fte/quicktour_reading_pg2[language!Japanese].png" (0x6E56F + 23892)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg2.png" (0x742C7 + 23662)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg3.png" (0x81C46 + 23519)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg3.png" => "daylight.rcc/images/fte/quicktour_reading_pg3[language!Japanese].png" (0x7BF9E + 23716)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg5.png" (0x87829 + 19735)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg1.png" (0x68C55 + 22806)
FILE    "daylight.rcc/images/fte/missingbooks_sync.png" (0x8E8B1 + 6061)
FILE    "daylight.rcc/images/fte/fte_oob_connected.png" => "daylight.rcc/images/fte/fte_oob_connected[language!Japanese].png" (0x90E0C + 9157)
FILE    "daylight.rcc/images/fte/fte_oob_plugin.svg" (0x90062 + 3494)
FILE    "daylight.rcc/images/fte/pocket_logo_large.png" (0x9C7BC + 3862)
FILE    "daylight.rcc/images/fte/quicktour_texticon_search.png" (0x9DB07 + 158)
FILE    "daylight.rcc/images/fte/quicktour_diagram.png" (0x79F39 + 8289)
FILE    "daylight.rcc/images/fte/missingbooks_help.png" (0xA9417 + 23335)
FILE    "daylight.rcc/images/fte/overflow_small.png" (0x9D6D6 + 1069)
FILE    "daylight.rcc/images/fte/missingbooks_intro.png" (0x931D5 + 17630)
DIR     "daylight.rcc/images/book" (0xC45CE + 70)
FILE    "daylight.rcc/images/book/dustbin_delete_icon_inverted.png" (0xB550F + 257)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_highlight.png" (0xB534C + 447)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_dogear.png" (0xB4F9E + 409)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_note.png" (0xB513B + 202)
FILE    "daylight.rcc/images/book/dustbin_delete_icon.png" (0xB5209 + 319)
DIR     "daylight.rcc/images/help" (0xC4596 + 56)
FILE    "daylight.rcc/images/help/help_icon.png" (0xB8C44 + 1022)
FILE    "daylight.rcc/images/help/kobopluslogo_help_sp.png" (0xBAA53 + 7449)
FILE    "daylight.rcc/images/help/kobopluslogo_eLeya.png" (0xB61D3 + 10861)
FILE    "daylight.rcc/images/help/kobopluslogo_help_version.png" (0xB9046 + 6665)
DIR     "daylight.rcc/images/home" (0xC44A8 + 238)
FILE    "daylight.rcc/images/home/main_nav_books.png" (0x546EC + 637)
FILE    "daylight.rcc/images/home/annotations_bookmark.png" (0x53ACF + 409)
FILE    "daylight.rcc/images/home/pocket_favourite_star.png" (0x523F6 + 328)
FILE    "daylight.rcc/images/home/main_nav_discover.png" (0x54296 + 1106)
FILE    "daylight.rcc/images/home/main_nav_discover_active.png" (0x60E6F + 860)
[... 2418 lines cut ...]
FILE    "certificates/5cf9d536.der" (0x14D83C8 + 1492)
FILE    "certificates/7a7c655d.der" (0x14D7670 + 442)
FILE    "certificates/7999be0d.der" (0x14F98A2 + 856)
FILE    "certificates/1dcd6f4c.der" (0x14F0594 + 1398)
[... cut ...]
<details><summary><b>$ git diff --no-index 16704 17623</b></summary>

https://gist.github.com/pgaskin/ac186e2aceb39ccc8a1c2d0590a9d147

</details>

---

https://temp.s3.geek1011.net/kobo-resources/nickel.17623.qInitResources_certificates.zip
https://temp.s3.geek1011.net/kobo-resources/nickel.17623.qInitResources_resources.zip
https://temp.s3.geek1011.net/kobo-resources/nickel.17623.qInitResources_styles.zip
https://temp.s3.geek1011.net/kobo-resources/nickel.17623.qInitResources_translations.zip

jackiew1, 2021-06-14:
@pgaskin 
Have you posted the wrong file? The data I need is what you list under `Output` above. However this seems to be binary equal to the data from 4.26.16704.

pgaskin, 2021-06-14:
Are you sure? I just checked the zips, and I see the files for `europa` (the Elipsa).

NiLuJe, 2021-06-14:
The inlined "Output" block is from the old FW, though ;).

pgaskin, 2021-06-14:
Fixed. It appears the entire comment got reverted when I tried to paste the diff in and it was too long.

jackiew1, 2021-06-15:
@pgaskin Thanks, that's OK now. I'm busy for the next few days so I won't be able to look at the nickel changes in any detail until the backend of the week. I don't think existing nickel patches are affected though.

pgaskin, 2021-07-18:
> I don't think existing nickel patches are affected though.

I just took a quick look through the nickel patches for ones which references resources containing `deviceIsEuropa`, and I didn't find anything, so I think you're right.

pgaskin, 2021-07-18:
Also, I've removed `KePub stylesheet additions`.

pgaskin, 2021-07-20:
Released: [`v71`](https://github.com/pgaskin/kobopatch-patches/releases/v71), MR:thread-340738.

pgaskin, 2021-07-20:
<details><summary><b>$ git diff --no-index src/versions/{4.26.16704,4.28.17623}</b></summary>

```diff
diff --git a/src/versions/4.26.16704/libadobe.so.yaml/pgaskin.yaml b/src/versions/4.28.17623/libadobe.so.yaml/pgaskin.yaml
index a2811d6..290a39b 100644
--- a/src/versions/4.26.16704/libadobe.so.yaml/pgaskin.yaml
+++ b/src/versions/4.28.17623/libadobe.so.yaml/pgaskin.yaml
@@ -5,4 +5,4 @@ Remove PDF map widget shown during panning:
   - Description: Removes the PDF map widget shown during panning and zooming.
   - BaseAddress:  {Sym: "N3AdobeReader::showMapWidget()"}
   - ReplaceBytes: {Offset:  80, FindInstBW: {SymPLTTail: "N3AdobeReader::updatePanningMap()"}, ReplaceInstBW: {SymPLTTail: "N3AdobeReader::hideMapWidget()"}}
-  - ReplaceBytes: {Offset: 190, FindInstBW: {SymPLTTail: "N3AdobeReader::updatePanningMap()"}, ReplaceInstBW: {SymPLTTail: "N3AdobeReader::hideMapWidget()"}}
+  - ReplaceBytes: {Offset: 192, FindInstBW: {SymPLTTail: "N3AdobeReader::updatePanningMap()"}, ReplaceInstBW: {SymPLTTail: "N3AdobeReader::hideMapWidget()"}}
diff --git a/src/versions/4.26.16704/libnickel.so.1.0.0.yaml/geoffr.yaml b/src/versions/4.28.17623/libnickel.so.1.0.0.yaml/geoffr.yaml
index fe1cafe..16365d8 100644
--- a/src/versions/4.26.16704/libnickel.so.1.0.0.yaml/geoffr.yaml
+++ b/src/versions/4.28.17623/libnickel.so.1.0.0.yaml/geoffr.yaml
@@ -317,8 +317,9 @@ Set KePub hyphenation:
       will always be turned on, regardless of justification button setting.
 
       (The publisher can still turn hyphenation off/on in the book's stylesheet.)
-  - ReplaceBytes: {Offset: 0xB8AD96, FindInstBLX: {SymPLT: "QString::operator==(QLatin1String) const"}, ReplaceH: 01 20 01 20} # Alternative 1: Always turn KePub hyphenation on
-# - ReplaceBytes: {Offset: 0xB8AD96, FindInstBLX: {SymPLT: "QString::operator==(QLatin1String) const"}, ReplaceH: 00 20 00 20} # Alternative 2: Never turn KePub hyphenation on
+    # == "justify"
+  - ReplaceBytes: {Base: "KepubBookReader::pageStyleCss(bool)", Offset: 1432, FindInstBLX: {SymPLT: "QString::operator==(QLatin1String) const"}, ReplaceH: 01 20 01 20} # Alternative 1: Always turn KePub hyphenation on
+# - ReplaceBytes: {Base: "KepubBookReader::pageStyleCss(bool)", Offset: 1432, FindInstBLX: {SymPLT: "QString::operator==(QLatin1String) const"}, ReplaceH: 00 20 00 20} # Alternative 2: Never turn KePub hyphenation on
 
 Force user line spacing in KePubs:
   - Enabled: no
@@ -471,34 +472,6 @@ Un-force link decoration in KePubs:
   - ReplaceString: {Offset: 0x27, Find: "b", Replace: "_"} # Disable border-bottom style
   - ReplaceString: {Offset: 0x53, Find: "c", Replace: "_"} # Disable color style
 
-KePub stylesheet additions:
-  - Enabled: no
-  - Description: Makes room to add some additional CSS to the default KePub stylesheet.
-    # See replacement CSS below for examples.
-  - FindBaseAddressString: ".KBHighlighting, .KBSearchResult {"
-  - ReplaceString: {Offset:   0, Find: ".KBHighlighting, .KBSearchResult { background-color: #C6C6C6 !important; } \t",                                                                         Replace: ".KBHighlighting,.KBSearchResult{background-color:#C6C6C6!important}.KBSearch", MustMatchLength: yes}
-  - ReplaceString: {Offset:  76, Find: ".KBSearchResult, .KBAnnotation, .KBHighlighting { font-size: 100% !important; -webkit-text-combine: inherit !important; } \t",                          Replace: "Result,.KBAnnotation,.KBHighlighting{font-size:100%!important;-webkit-text-combine:inherit!important}.KBAnnotation[writingM", MustMatchLength: yes}
-  - ReplaceString: {Offset: 199, Find: ".KBAnnotation[writingMode=\"horizontal-tb\"] { border-bottom: 2px solid black !important; } \t", Replace: "ode=\"horizontal-tb\"]{border-bottom:2px solid black!important}.KBAnnotation[writingMode=\"ver", MustMatchLength: yes}
-  - ReplaceString: {Offset: 290, Find: ".KBAnnotation[writingMode=\"vertical-rl\"] { border-right: 2px solid black !important; } \t",      Replace: "tical-rl\"]{border-right:2px solid black!important}.KBAnnotation[writingMode=\"vertical-lr", MustMatchLength: yes}
-  - ReplaceString: {Offset: 378, Find: ".KBAnnotation[writingMode=\"vertical-lr\"] { border-left: 2px solid black !important; }",          Replace: "\"]{border-left:2px solid black!important}/******************************************/", MustMatchLength: yes}
-    #
-    # Replacement value: CSS string of exactly 44 characters in length:
-    #
-    # Example 1: Make full justification the default:
-# - FindReplaceString: {Find: "/******************************************/", Replace: "body{text-align:justify                    }", MustMatchLength: yes}
-    #
-    # Example 2: text-rendering:optimizeSpeed: Disables kerning, but fixes some
-    # rendering problems that affect fully justified text.
-# - FindReplaceString: {Find: "/******************************************/", Replace: "body{text-rendering:optimizeSpeed          }", MustMatchLength: yes}
-    #
-    # Example 3: text-rendering:optimizeLegibility: Enables ligatures, but causes
-    # some additional rendering problems for fully justified text.
-# - FindReplaceString: {Find: "/******************************************/", Replace: "body{text-rendering:optimizeLegibility     }", MustMatchLength: yes}
-    #
-    # Example 4: Make KePub font sizes match ePub font sizes: ePubs render with
-    # 1.5 times the font size of KePubs by default. With this enabled they match.
[... cut ...]
       Changes the size of indentation for each level in the TOC. This patch
       replaces the old "Increase TOC level indentation" patch from firmware
       versions before 4.24.15672 (see the note in libnickel.so.1.0.0.yaml).
-  - FindZlib: "qproperty-indentUnit"
+  - FindZlib: "qproperty-indentUnit" # qss/N3TableOfContentsWidget.qss
   - ReplaceZlibGroup:
       Replacements:
       # Change the replacement values to your desired width (I've made it half the default width as an example).
```

</details>

closed 2021-07-20 by pgaskin