pgaskin/kobopatch-patches #88: Patches for 4.28.17820

pgaskin, 2021-06-24:


pgaskin, 2021-07-18:
**Resource extraction:**

```bash
./qrc2zip --output "nickel.17820.qInitResources_resources.zip"    --recursive --verbose "nickel" 1 $((0x12205e8 - 0x0010000)) $((0x00272b8 - 0x0010000)) $((0x12200f8 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
./qrc2zip --output "nickel.17820.qInitResources_translations.zip" --recursive --verbose "nickel" 1 $((0x15d7e20 - 0x0010000)) $((0x12209d0 - 0x0010000)) $((0x15d7b98 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
./qrc2zip --output "nickel.17820.qInitResources_styles.zip"       --recursive --verbose "nickel" 1 $((0x15f6640 - 0x0010000)) $((0x15d7f38 - 0x0010000)) $((0x15f3d90 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
./qrc2zip --output "nickel.17820.qInitResources_certificates.zip" --recursive --verbose "nickel" 1 $((0x161db90 - 0x0010000)) $((0x15f71c0 - 0x0010000)) $((0x161cb40 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
```

<details><summary><b>Output</b></summary>

```
nickel@17820/qInitResources_resources
DIR     "qss" (0x12109BC + 14)
FILE    "qss/TouchMenu.qss" (0x120FDDF + 189)
DIR     "html" (0x12109A0 + 28)
FILE    "html/RGBFrontLightAboutView.html" (0x120F88C + 1144)
FILE    "html/RGBFrontLightAboutViewSnow.html" (0x120F3BC + 1228)
DIR     "keys" (0x1210984 + 28)
FILE    "keys/device.qmap" (0x121007C + 124)
FILE    "keys/avrcp.qmap" (0x120FEA0 + 472)
FILE    "omfTemplate.html" (0x1190CDF + 231)
DIR     "pika.rcc" (0x410D + 14)
DIR     "pika.rcc/images" (0x411B + 42)
DIR     "pika.rcc/images/fte" (0x4161 + 14)
FILE    "pika.rcc/images/fte/quicktour_diagram.png" (0x18 + 3841)
DIR     "pika.rcc/images/home" (0x4153 + 14)
FILE    "pika.rcc/images/home/quicktour_device.png" (0xF1D + 4337)
DIR     "pika.rcc/images/about" (0x4145 + 14)
FILE    "pika.rcc/images/about/global_certification.png" (0x2012 + 8221)
FILE    "verifier.pub" (0x77C89C + 451)
DIR     "daylight.rcc" (0xD7B74 + 14)
DIR     "daylight.rcc/images" (0xD7B82 + 336)
DIR     "daylight.rcc/images/fte" (0xD93F4 + 280)
FILE    "daylight.rcc/images/fte/missingbooks_factoryreset.png" (0x3D73F + 9065)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg4_adjust_brightness.png" (0x262A + 24664)
FILE    "daylight.rcc/images/fte/missingbooks_account.png" (0x31AA0 + 22560)
FILE    "daylight.rcc/images/fte/missingbooks_repair.png" (0x452EA + 20737)
FILE    "daylight.rcc/images/fte/quicktour_nightmode.png" => "daylight.rcc/images/fte/quicktour_nightmode[language!Japanese].png" (0x372C4 + 24646)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg2.png" => "daylight.rcc/images/fte/quicktour_reading_pg2[language!Japanese].png" (0x13ACB + 23892)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg2.png" (0x19823 + 23662)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg3.png" => "daylight.rcc/images/fte/quicktour_reading_pg3[language!Japanese].png" (0x1F495 + 23716)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg3.png" (0x2513D + 23519)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg5.png" (0x2CD85 + 19735)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg1.png" (0xE1B1 + 22806)
FILE    "daylight.rcc/images/fte/missingbooks_sync.png" (0x3FAAC + 6061)
FILE    "daylight.rcc/images/fte/fte_oob_connected.png" => "daylight.rcc/images/fte/fte_oob_connected[language!Japanese].png" (0x42177 + 9157)
FILE    "daylight.rcc/images/fte/fte_oob_plugin.svg" (0x44540 + 3494)
FILE    "daylight.rcc/images/fte/pocket_logo_large.png" (0x4125D + 3862)
FILE    "daylight.rcc/images/fte/quicktour_texticon_search.png" (0x2588 + 158)
FILE    "daylight.rcc/images/fte/quicktour_diagram.png" (0x2AD20 + 8289)
FILE    "daylight.rcc/images/fte/missingbooks_help.png" (0x8686 + 23335)
FILE    "daylight.rcc/images/fte/overflow_small.png" (0x3D30E + 1069)
FILE    "daylight.rcc/images/fte/missingbooks_intro.png" (0x4A3EF + 17630)
DIR     "daylight.rcc/images/book" (0xD9384 + 112)
FILE    "daylight.rcc/images/book/dustbin_delete_icon_inverted.png" (0x83D2B + 257)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_highlight.png" (0x83A25 + 447)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_dogear.png" (0x833D6 + 409)
FILE    "daylight.rcc/images/book/markup_icon_transparent_padding.png" (0x8363F + 994)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_note.png" (0x841A8 + 202)
FILE    "daylight.rcc/images/book/note_icon_transparent_padding.png" (0x83573 + 200)
FILE    "daylight.rcc/images/book/dustbin_delete_icon.png" (0x83BE8 + 319)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_markup.png" (0x83E30 + 884)
DIR     "daylight.rcc/images/help" (0xD933E + 70)
FILE    "daylight.rcc/images/help/help_large.png" (0xCD2F6 + 1138)
FILE    "daylight.rcc/images/help/help_icon.png" (0xD01DD + 1022)
FILE    "daylight.rcc/images/help/kobopluslogo_help_sp.png" (0xD05DF + 7449)
FILE    "daylight.rcc/images/help/kobopluslogo_eLeya.png" (0xCD76C + 10861)
FILE    "daylight.rcc/images/help/kobopluslogo_help_version.png" (0xCB8E9 + 6665)
DIR     "daylight.rcc/images/home" (0xD9234 + 266)
[... 2474 lines cut ...]
FILE    "dragon.rcc/images/widgets/settings_page_turning_option1_off.png" (0x5693B + 378)
FILE    "dragon.rcc/images/widgets/settings_page_turning_option2_off.png" (0x57B59 + 367)
FILE    "dragon.rcc/images/widgets/settings_page_turning_option3_off.png" (0x56679 + 402)
FILE    "dragon.rcc/images/widgets/global_keypad_delete_hit.png" (0x55D2B + 291)
[... cut ...]
FILE    "dragon.rcc/images/navigation/discover_dismiss_hit.png" (0x7868 + 322)
FILE    "dragon.rcc/images/navigation/browser_reload_off.png" (0x756A + 388)
FILE    "dragon.rcc/images/navigation/global_error.png" (0x7B3B + 441)
FILE    "dragon.rcc/images/navigation/discover_readit_off.png" (0x79AE + 393)
FILE    "dragon.rcc/images/navigation/browser_reload_hit.png" (0x7E4A + 332)
FILE    "dragon.rcc/images/navigation/discover_dismiss_off.png" (0x7CF8 + 334)
FILE    "kepub-book.css" (0xF1D2AE + 185)
DIR     "pixie.rcc" (0x2A20 +

pgaskin, 2021-07-18:
<details><summary><b>$ git diff --no-index 17623 17820</b></summary>

```diff
diff --git a/17623/certificates/17b51fe6.der b/17623/certificates/17b51fe6.der
deleted file mode 100644
index ce76c05..0000000
Binary files a/17623/certificates/17b51fe6.der and /dev/null differ
diff --git a/17623/certificates/21855f49.der b/17623/certificates/21855f49.der
deleted file mode 100644
index d5d03fa..0000000
Binary files a/17623/certificates/21855f49.der and /dev/null differ
diff --git a/17623/certificates/4e18c148.der b/17623/certificates/4e18c148.der
deleted file mode 100644
index d5e2a12..0000000
Binary files a/17623/certificates/4e18c148.der and /dev/null differ
diff --git a/17623/certificates/559f7c71.der b/17623/certificates/559f7c71.der
deleted file mode 100644
index fcf7bbc..0000000
Binary files a/17623/certificates/559f7c71.der and /dev/null differ
diff --git a/17820/certificates/869fbf79.der b/17820/certificates/869fbf79.der
new file mode 100644
index 0000000..1ff0c9c
Binary files /dev/null and b/17820/certificates/869fbf79.der differ
diff --git a/17820/certificates/b74d2bd5.der b/17820/certificates/b74d2bd5.der
new file mode 100644
index 0000000..235e186
Binary files /dev/null and b/17820/certificates/b74d2bd5.der differ
diff --git a/17820/certificates/bd43e1dd.der b/17820/certificates/bd43e1dd.der
new file mode 100644
index 0000000..91d8e47
Binary files /dev/null and b/17820/certificates/bd43e1dd.der differ
diff --git a/17623/certificates/d6e6eab9.der b/17623/certificates/d6e6eab9.der
deleted file mode 100644
index cd218e5..0000000
Binary files a/17623/certificates/d6e6eab9.der and /dev/null differ
diff --git a/17623/certificates/d8317ada.der b/17623/certificates/d8317ada.der
deleted file mode 100644
index c2ce7d6..0000000
Binary files a/17623/certificates/d8317ada.der and /dev/null differ
diff --git a/17820/certificates/dbff3a01.der b/17820/certificates/dbff3a01.der
new file mode 100644
index 0000000..5a3dc50
Binary files /dev/null and b/17820/certificates/dbff3a01.der differ
diff --git a/17820/certificates/f0cd152c.der b/17820/certificates/f0cd152c.der
new file mode 100644
index 0000000..1c74b84
Binary files /dev/null and b/17820/certificates/f0cd152c.der differ
diff --git a/17820/certificates/f459871d.der b/17820/certificates/f459871d.der
new file mode 100644
index 0000000..675cbeb
Binary files /dev/null and b/17820/certificates/f459871d.der differ
diff --git a/17623/daylight.rcc/images/book/markup_icon_transparent_padding.png b/17820/daylight.rcc/images/book/markup_icon_transparent_padding.png
index ca0b91f..b79ba6e 100644
Binary files a/17623/daylight.rcc/images/book/markup_icon_transparent_padding.png and b/17820/daylight.rcc/images/book/markup_icon_transparent_padding.png differ
diff --git a/17623/daylight.rcc/images/cobranding/splash_orbile.png b/17820/daylight.rcc/images/cobranding/splash_orbile.png
index 034be2c..1a3bf84 100644
Binary files a/17623/daylight.rcc/images/cobranding/splash_orbile.png and b/17820/daylight.rcc/images/cobranding/splash_orbile.png differ
diff --git a/17820/daylight.rcc/images/help/help_large.png b/17820/daylight.rcc/images/help/help_large.png
new file mode 100644
index 0000000..e367a1d
Binary files /dev/null and b/17820/daylight.rcc/images/help/help_large.png differ
diff --git a/17623/daylight.rcc/images/library/library_download_lg_hit.png b/17820/daylight.rcc/images/library/library_download_lg_hit.png
index 1968dc8..06fcda7 100644
Binary files a/17623/daylight.rcc/images/library/library_download_lg_hit.png and b/17820/daylight.rcc/images/library/library_download_lg_hit.png differ
diff --git a/17623/daylight.rcc/images/library/library_download_lg_off.png b/17820/daylight.rcc/images/library/library_download_lg_off.png
index 933196a..f4267c1 100644
Binary files a/17623/daylight.rcc/images/library/library_download_lg_off.png and b/17820/daylight.rcc/images/library/library_download_lg_off.png differ
diff --git a/17623/daylight.rcc/images/library/library_upload_lg_hit.png b/17820/daylight.rcc/images/library/library_upload_lg_hit.png
index 13dcd67..8efd78b 100644
Binary files a/17623/daylight.rcc/images/library/library_upload_lg_hit.png and b/17820/daylight.rcc/images/library/library_upload_lg_hit.png differ
[... 1396 lines cut ...]
index 0000000..d1cd1d9
Binary files /dev/null and b/17820/trilogy.rcc/images/mynotes/stylus_swipe_up.png differ
diff --git a/17820/trilogy.rcc/images/mynotes/tap_hand.png b/17820/trilogy.rcc/images/mynotes/tap_hand.png
new file mode 100644
index 0000000..7c87f39
Binary files /dev/null and b/17820/trilogy.rcc/images/mynotes/tap_hand.png differ
diff --git a/17820/trilogy.rcc/images/mynotes/underline.png b/17820/trilogy.rcc/images/mynotes/underline.png
new file mode 100644
index 0000000..ee443bc
Binary files /dev/null and b/17820/trilogy.rcc/images/mynotes/underline.png differ
diff --git a/17820/trilogy.rcc/images/mynotes/undo.png b/17820/trilogy.rcc/images/mynotes/undo.png
new file mode 100644
index 0000000..d573d23
Binary files /dev/null and b/17820/trilogy.rcc/images/mynotes/undo.png differ
diff --git a/17820/trilogy.rcc/images/widgets/add_page.png b/17820/trilogy.rcc/images/widgets/add_page.png
new file mode 100644
index 0000000..a343626
Binary files /dev/null and b/17820/trilogy.rcc/images/widgets/add_page.png differ
diff --git a/17820/trilogy.rcc/images/widgets/add_page_selected.png b/17820/trilogy.rcc/images/widgets/add_page_selected.png
new file mode 100644
index 0000000..f3a9183
Binary files /dev/null and b/17820/trilogy.rcc/images/widgets/add_page_selected.png differ
```

</details>

pgaskin, 2021-07-18:
```
Errors:
  /home/patrick/src/kobopatch-patches/src/versions/4.28.17820/libnickel.so.1.0.0.yaml/pgaskin.yaml: Customize ComfortLight settings: could not apply patch "Customize ComfortLight settings": line 566: inst 4: ReplaceInt: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.28.17820/libnickel.so.1.0.0.yaml/pgaskin.yaml: PowerSettings - UnlockEnabled: could not apply patch "PowerSettings - UnlockEnabled": line 642: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.28.17820/nickel.yaml/pgaskin.yaml: Remove forgot pin button from lock screen: could not apply patch "Remove forgot pin button from lock screen": line 20: inst 4: ReplaceZlib: not a zlib stream
  /home/patrick/src/kobopatch-patches/src/versions/4.28.17820/nickel.yaml/oren64.yaml: New home screen subtitle custom font: could not apply patch "New home screen subtitle custom font": line 31: inst 4: ReplaceZlib: not a zlib stream
  /home/patrick/src/kobopatch-patches/src/versions/4.28.17820/libnickel.so.1.0.0.yaml/geoffr.yaml: Set font scale factor: could not apply patch "Set font scale factor": line 503: inst 3: ReplaceInt: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.28.17820/libnickel.so.1.0.0.yaml/geoffr.yaml: Set KePub hyphenation: could not apply patch "Set KePub hyphenation": line 320: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.28.17820/libnickel.so.1.0.0.yaml/geoffr.yaml: ePub uniform font scale: could not apply patch "ePub uniform font scale": line 489: inst 3: FindBaseAddressHex: FindBaseAddress: could not find bytes
```

As of 17623 @ a0c0273eeaa7fe1e0880b23d601010e2f12b2419.

pgaskin, 2021-07-18:
> see if ePub uniform font scale is still necessary

So... unless it's something to do with the few books I tested, the difference between EPUB and KEPUB is now much larger... it looks like EPUB is now more than double the size. I don't know much about the original font scale patch or the intricacies of RMSDK (I usually read Kobo EPUBs), so I'm just going to remove it for now and do it from scratch later if necessary.

NiLuJe, 2021-07-18:
Doesn't ring a bell on nickel's end, no, but we do default to 96dpi in KOReader for that very reason ;).

pgaskin, 2021-07-20:
Released: [`v71`](https://github.com/pgaskin/kobopatch-patches/releases/v71), MR:thread-340739.

pgaskin, 2021-07-20:
<details><summary><b>$ git diff --no-index src/versions/{4.28.17623,4.28.17820}</b></summary>

```diff
diff --git a/src/versions/4.28.17623/libnickel.so.1.0.0.yaml/geoffr.yaml b/src/versions/4.28.17820/libnickel.so.1.0.0.yaml/geoffr.yaml
index 16365d8..b83797f 100644
--- a/src/versions/4.28.17623/libnickel.so.1.0.0.yaml/geoffr.yaml
+++ b/src/versions/4.28.17820/libnickel.so.1.0.0.yaml/geoffr.yaml
@@ -480,16 +480,6 @@ Ignore .otf fonts:
       Embedded .otf fonts should continue to work as normal.
   - FindReplaceString: {Find: "*.otf", Replace: "_.otf"}
 
-ePub uniform font scale:
-  - Enabled: no
-  - Description: |
-      Unlike the KePub reader, the ePub reader uses a ~50% larger font scale on
-      dragon (AuraHD/H2O/GloHD/H2O2/Clara/Libra) devices. This patch changes
-      changes the ePub reader to match the KePub reader in having the same font
-      scale on all devices.
-  - FindBaseAddressHex: 00 00 C0 42 00 00 10 43
-  - ReplaceBytes: {Offset: 6, FindH: 10 43, ReplaceH: C0 42}
-
 Set font scale factor:
   - Enabled: no
   - Description: |
@@ -501,5 +491,5 @@ Set font scale factor:
   # Original value X=14, so original scale factor is (font size slider) / 46.
   # Allowed replacement values for X: 0-15 (smaller X --> larger scale factor)
   # Note: vmov.f32 s1, #15.0 before getReadingFontSizeScaleFactor
-  - ReplaceInt: {Offset: 0x5E98FA, Find: 14, Replace: 4} # ePub
-  - ReplaceInt: {Offset: 0xC5E776, Find: 14, Replace: 4} # KePub
+  - ReplaceInt: {Offset: 0x5E74A6, Find: 14, Replace: 4} # ePub
+  - ReplaceInt: {Offset: 0xC5C292, Find: 14, Replace: 4} # KePub
diff --git a/src/versions/4.28.17623/libnickel.so.1.0.0.yaml/pgaskin.yaml b/src/versions/4.28.17820/libnickel.so.1.0.0.yaml/pgaskin.yaml
index 55ad431..2235655 100644
--- a/src/versions/4.28.17623/libnickel.so.1.0.0.yaml/pgaskin.yaml
+++ b/src/versions/4.28.17820/libnickel.so.1.0.0.yaml/pgaskin.yaml
@@ -544,6 +544,39 @@ Enable advanced settings for all fonts:
       FindInstBLX:    {SymPLT: "N3FontTypeUtil::isMonotypeFont(QString const&)"}
       ReplaceH:       4F F0 01 00
 
+ePub uniform font scale:
+  - Enabled: no
+  - Description: |
+      By default, the EPUB reader uses larger fonts than the KEPUB reader. This
+      patch forces the EPUB reader to always render fonts at 96dpi so the font
+      sizes match. This is a rewritten version of GeoffR's patch, which stopped
+      working on 4.28.17820 since the dpi was changed to use QScreen instead of
+      144dpi on dragon and 96dpi on other devices (GeoffR's patch forced it to
+      always be 96dpi). This patch has not been tested on the Elipsa.
+  - BaseAddress: 0x5E74CE
+  - ReplaceBytes:
+      Offset:      0
+      FindInstBLX: {SymPLT: "QGuiApplication::primaryScreen()"} # unused
+      ReplaceH:    4F F0 60 00 # mov r0, #96 (int32 r0 = 96)
+  - ReplaceBytes:
+      Offset:      4
+      FindInstBLX: {SymPLT: "QScreen::logicalDotsPerInchX() const"} # unused
+      ReplaceH:    00 EE 10 0A # vmov s0, r0 (int32 s0 = r0)
+  - ReplaceBytes:
+      Offset:      16
+      FindH:       F7 EE E7 0A # vcvt.f64.f32 d16, s15 (s15 is a float32 from getReadingFontSizeScaleFactor())
+      ReplaceH:    B8 EE C0 0A # vcvt.f32.s32 s0, s0 (float32 s0)
+  - ReplaceBytes:
+      Offset:      20
+      FindH:       20 EE 20 0B # vmul.f64 d0, d0, d16
+      ReplaceH:    20 EE 27 0A # vmul.f32 s0, s0, s15
+  - ReplaceBytes:
+      Offset:      24
+      FindH:       BD EE C0 0B # vcvt.s32.f64 s0, d0 (s0 is the int32 output)
+      ReplaceH:    BD EE C0 0A # vcvt.s32.f32 s0, s0
[... 27 lines cut ...]
@@ -27,7 +27,7 @@ New home screen subtitle custom font:
   - Description: |
       Increase subtitle font size, font color black. Works best when used
       with with patch `New home screen increasing cover size`
-  - BaseAddress: 0x15D2292 # qss/GenericHomeWidget.qss
+  - BaseAddress: 0x15E1C08 # qss/GenericHomeWidget.qss
   - ReplaceZlib: {Find: "MetaDataLabel {\n  color: #666666;\n  qproperty-indent: 0;\n}", Replace: "MetaDataLabel {\n  color: #000000;\n  qproperty-indent: 0;\n}"}
   - ReplaceZlib:
       Find:    "MetaDataLabel[qApp_deviceIsTrilogy=true] {\n  font-size: 13px;\n}"
diff --git a/src/versions/4.28.17623/nickel.yaml/pgaskin.yaml b/src/versions/4.28.17820/nickel.yaml/pgaskin.yaml
index 18ef66a..9eb546b 100644
--- a/src/versions/4.28.17623/nickel.yaml/pgaskin.yaml
+++ b/src/versions/4.28.17820/nickel.yaml/pgaskin.yaml
@@ -15,7 +15,7 @@ Remove forgot pin button from lock screen:
       Removes the Forgot PIN -> Sign Out button from the lock screen.
       If this is enabled and you forget your pin, you will need to
       hard reset your Kobo.
-  - BaseAddress: 0x15CC9E6 # qss/PinCodeInputDialog.qss
+  - BaseAddress: 0x15D1B12 # qss/PinCodeInputDialog.qss
   - ReplaceZlib:
       Find: "#lblForgotPin[qApp_deviceIsDragon=true],\n#lblSignOut[qApp_deviceIsDragon=true] {\n  font-size: 26px;\n}"
       Replace: "#lblForgotPin,#lblSignOut{qproperty-visible:false;}"
```

</details>

closed 2021-07-20 by pgaskin