pgaskin/kobopatch-patches #90: Patches for 4.28.17925

pgaskin, 2021-07-17:
TODO

NiLuJe, 2021-07-17:
On a mildly related subject: this one explained the optional -TFW suffix: it's for when an update ships a touch panel FW update.

```diff
┌─(niluje@illyria:pts/13)────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────(/var/tmp/niluje)─┐
└─(0.10:34%:15:41:%)── diff -Nuarp rcS etc/init.d/rcS                                                                                                                                                                                                                                                                                                  ──(Thu, Jul 15)─┘
--- rcS 2021-07-15 15:41:59.525885955 +0200
+++ etc/init.d/rcS      2021-07-13 17:27:25.000000000 +0200
@@ -117,6 +117,7 @@ if [ $? != 0 ] || [ $FS_CORRUPT == 1 ] |
 
        write_uboot_env $PLATFORM $UBOOT_RECOVERY
        reboot
+       exit
 fi
 
 mount -t vfat -o noatime,nodiratime,shortname=mixed,utf8 /dev/mmcblk0p3 /mnt/onboard
@@ -159,6 +160,7 @@ if [ $CAN_UPGRADE == 1 ] && [ -e /mnt/on
        write_uboot_env $PLATFORM $UBOOT_MMC
        sync
        reboot
+       exit
 fi
 
 ( 
@@ -168,22 +170,6 @@ fi
        /etc/init.d/on-animator.sh
 ) &
 
-if [ $PRODUCT == europa ] && [ -e /lib/firmware/elants_i2c.bin ]; then
-       TOUCH_FW_VERSION=`cat /sys/class/i2c-adapter/i2c-0/0-0010/elan_ktf/fw_version`
-       if [ "$TOUCH_FW_VERSION" != "6417" ]; then
-               echo > /sys/devices/platform/soc/twi0/i2c-0/0-0010/elan_ktf/update_fw
-
-               killall -9 on-animator.sh
-               sleep 1;
-
-               zcat /etc/images/$PREFIX\ghostbuster.raw.gz | /usr/local/Kobo/pickel showpic
-               zcat /etc/images/$PREFIX\my_first_time.raw.gz | /usr/local/Kobo/pickel showpic
-               sync
-               sleep 2;
-               poweroff
-       fi
-fi
-
 (
        /usr/local/Kobo/pickel disable.rtc.alarm
```

pgaskin, 2021-07-17:
Yep, that makes sense. It also explains why this update is only delivered to devices running 17826 (the last TouchFW update). If there are more like this, I'll need to update the firmware page to properly support and display update dependencies.

NiLuJe, 2021-07-17:
FWIW, I had to switch *away* from the `kobo` affiliate to get it (to `fnac`, which is my "official" affiliate) (~ the day it was released, might have changed since) (from 17826).

jackiew1, 2021-07-18:
@pgaskin You gave the impression on MR that you were waiting for me to do something with my patches ... whereas I believe I'm waiting for you. I need the resource map text output file for 4.28.17925 before I can start.

pgaskin, 2021-07-18:
> whereas I believe I'm waiting for you

I'd recommend starting with #86, since I expect the recent versions will be similar, and it's easier to keep track of it incrementally.

pgaskin, 2021-07-18:
**Resource extraction:**

```bash
./qrc2zip --output "nickel.17925.qInitResources_resources.zip"    --recursive --verbose "nickel" 1 $((0x12205d0 - 0x0010000)) $((0x00272a0 - 0x0010000)) $((0x12200e0 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
./qrc2zip --output "nickel.17925.qInitResources_translations.zip" --recursive --verbose "nickel" 1 $((0x15d89a0 - 0x0010000)) $((0x12209b8 - 0x0010000)) $((0x15d8718 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
./qrc2zip --output "nickel.17925.qInitResources_styles.zip"       --recursive --verbose "nickel" 1 $((0x15f71c0 - 0x0010000)) $((0x15d8ab8 - 0x0010000)) $((0x15f4910 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
./qrc2zip --output "nickel.17925.qInitResources_certificates.zip" --recursive --verbose "nickel" 1 $((0x161e710 - 0x0010000)) $((0x15f7d40 - 0x0010000)) $((0x161d6c0 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
```

<details><summary><b>Output</b></summary>

```
nickel@17925/qInitResources_resources
DIR     "qss" (0x12109A4 + 14)
FILE    "qss/TouchMenu.qss" (0x120F6D7 + 189)
DIR     "html" (0x1210988 + 28)
FILE    "html/RGBFrontLightAboutView.html" (0x120F798 + 1144)
FILE    "html/RGBFrontLightAboutViewSnow.html" (0x120FC14 + 1228)
DIR     "keys" (0x121096C + 28)
FILE    "keys/device.qmap" (0x120F657 + 124)
FILE    "keys/avrcp.qmap" (0x120F47B + 472)
FILE    "omfTemplate.html" (0x29837 + 231)
DIR     "pika.rcc" (0x410D + 14)
DIR     "pika.rcc/images" (0x411B + 42)
DIR     "pika.rcc/images/fte" (0x4161 + 14)
FILE    "pika.rcc/images/fte/quicktour_diagram.png" (0x312E + 3841)
DIR     "pika.rcc/images/home" (0x4153 + 14)
FILE    "pika.rcc/images/home/quicktour_device.png" (0x18 + 4337)
DIR     "pika.rcc/images/about" (0x4145 + 14)
FILE    "pika.rcc/images/about/global_certification.png" (0x110D + 8221)
FILE    "verifier.pub" (0x977DE0 + 451)
DIR     "daylight.rcc" (0xD7B74 + 14)
DIR     "daylight.rcc/images" (0xD7B82 + 336)
DIR     "daylight.rcc/images/fte" (0xD93F4 + 280)
FILE    "daylight.rcc/images/fte/missingbooks_factoryreset.png" (0x75D3A + 9065)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg4_adjust_brightness.png" (0x7D1AC + 24664)
FILE    "daylight.rcc/images/fte/missingbooks_account.png" (0xB18AE + 22560)
FILE    "daylight.rcc/images/fte/missingbooks_repair.png" (0x780A7 + 20737)
FILE    "daylight.rcc/images/fte/quicktour_nightmode.png" => "daylight.rcc/images/fte/quicktour_nightmode[language!Japanese].png" (0xB70D2 + 24646)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg2.png" => "daylight.rcc/images/fte/quicktour_reading_pg2[language!Japanese].png" (0x8FE13 + 23892)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg2.png" (0x95B6B + 23662)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg3.png" (0xA1485 + 23519)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg3.png" => "daylight.rcc/images/fte/quicktour_reading_pg3[language!Japanese].png" (0x9B7DD + 23716)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg5.png" (0xACB93 + 19735)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg1.png" (0x8A4F9 + 22806)
FILE    "daylight.rcc/images/fte/missingbooks_sync.png" (0xBD5EF + 6061)
FILE    "daylight.rcc/images/fte/fte_oob_connected.png" => "daylight.rcc/images/fte/fte_oob_connected[language!Japanese].png" (0x72A57 + 9157)
FILE    "daylight.rcc/images/fte/fte_oob_plugin.svg" (0x876EA + 3494)
FILE    "daylight.rcc/images/fte/pocket_logo_large.png" (0x74E20 + 3862)
FILE    "daylight.rcc/images/fte/quicktour_texticon_search.png" (0xBD54D + 158)
FILE    "daylight.rcc/images/fte/quicktour_diagram.png" (0x88494 + 8289)
FILE    "daylight.rcc/images/fte/missingbooks_help.png" (0xA7068 + 23335)
FILE    "daylight.rcc/images/fte/overflow_small.png" (0xBD11C + 1069)
FILE    "daylight.rcc/images/fte/missingbooks_intro.png" (0x83208 + 17630)
DIR     "daylight.rcc/images/book" (0xD9384 + 112)
FILE    "daylight.rcc/images/book/dustbin_delete_icon_inverted.png" (0x5ABA + 257)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_highlight.png" (0x5C8B + 447)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_dogear.png" (0x5F91 + 409)
FILE    "daylight.rcc/images/book/markup_icon_transparent_padding.png" (0x535C + 994)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_note.png" (0x612E + 202)
FILE    "daylight.rcc/images/book/note_icon_transparent_padding.png" (0x5BBF + 200)
FILE    "daylight.rcc/images/book/dustbin_delete_icon.png" (0x5E4E + 319)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_markup.png" (0x5742 + 884)
DIR     "daylight.rcc/images/help" (0xD933E + 70)
FILE    "daylight.rcc/images/help/help_large.png" (0x30146 + 1138)
FILE    "daylight.rcc/images/help/help_icon.png" (0x2E027 + 1022)
FILE    "daylight.rcc/images/help/kobopluslogo_help_sp.png" (0x2E429 + 7449)
FILE    "daylight.rcc/images/help/kobopluslogo_eLeya.png" (0x305BC + 10861)
FILE    "daylight.rcc/images/help/kobopluslogo_help_version.png" (0x2C61A + 6665)
DIR     "daylight.rcc/images/home" (0xD9234 + 266)
[... 2480 lines cut ...]
FILE    "dragon.rcc/images/widgets/settings_page_turning_option3_on.png" (0x4A00F + 381)
FILE    "dragon.rcc/images/widgets/add_page.png" (0x47BFA + 1625)
FILE    "dragon.rcc/images/widgets/settings_checkbox_on_disabled.png" (0x48BA5 + 173)
FILE    "dragon.rcc/images/widgets/settings_checkbox_off_disabled.png" (0x49CD9 + 98)
FILE    "dragon.rcc/images/widgets/dropbox_folder.png" (0x489FB + 250)
[... cut ...]
FILE    "dragon.rcc/images/navigation/browser_reload_hit.png" (0x47334 + 332)
FILE    "dragon.rcc/images/navigation/discover_dismiss_off.png" (0x47484 + 334)
FILE    "kepub-book.css" (0x80DA26 + 185)
DIR     "pixie.rcc" (0x2A20 + 14)
DIR     "pixie.rcc/images" (0x2A2E + 28)
DIR     "pixie.rcc/images/fte" (0x2A58 + 14)
FILE    "pixie.rcc/images/fte/quicktour_diagram.png" (0x18 + 4688)
DIR     "pixie.rcc/images/home" (0x2A4A + 14)
FILE    "pixie.rcc/images/home/quicktour_device.png" (0x126C + 5916)
FILE    "smil.xslt" (0

pgaskin, 2021-07-18:
@jackiew1, I've uploaded the resources for 17820+. Note that there are quite a few new resources in 17820 (https://github.com/pgaskin/kobopatch-patches/issues/88#issuecomment-881987259).

pgaskin, 2021-07-18:
```
Errors:
  /home/patrick/src/kobopatch-patches/src/versions/4.28.17925/nickel.yaml/pgaskin.yaml: Remove forgot pin button from lock screen: could not apply patch "Remove forgot pin button from lock screen": line 20: inst 4: ReplaceZlib: not a zlib stream
  /home/patrick/src/kobopatch-patches/src/versions/4.28.17925/libnickel.so.1.0.0.yaml/pgaskin.yaml: Customize ComfortLight settings: could not apply patch "Customize ComfortLight settings": line 566: inst 4: ReplaceInt: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.28.17925/nickel.yaml/oren64.yaml: New home screen subtitle custom font: could not apply patch "New home screen subtitle custom font": line 31: inst 4: ReplaceZlib: not a zlib stream
  /home/patrick/src/kobopatch-patches/src/versions/4.28.17925/libnickel.so.1.0.0.yaml/geoffr.yaml: Set font scale factor: could not apply patch "Set font scale factor": line 496: inst 3: ReplaceInt: could not find specified bytes at offset
```

As of 17826 @ eab3714a36d5a958683fcbeb64d2993321bf95d8.

The same as usual.

jackiew1, 2021-07-18:
As far as I'm concerned fw versions 4.28.xxxx prior to this one are beta versions and no-one should need a kobopatch pack for them. I don't intend to waste any more time on patches for them. I'm not even convinced 4.28.17925 is anything other than Kobo's most recent beta version.

Having said all that I believe all my official nickel patches will run as-is on any of them. The Elipsa has the same screen pixels as the KA1 so I assume all the current Daylight settings will apply.

The only patches I know of which did need attention were:
- `ePub uniform font scale`
- `KePub stylesheet additions`

It looks like you've already rewritten the first and removed the second.

I did investigate patch `KePub stylesheet additions` in some detail when 4.28.17623 came out. In case  you don't know this already the problem has arisen because the CSS code in libnickel which it currently uses has been split into 2 parts separated by binary zeros. It looks more like Kobo "tidying up" rather than function change.

Unfortunately, the consequence for this patch is that if we now have to restrict ourselves to patching either part1 or part2 (rather than the 2 parts together) there is no longer quite enough cosmetic whitespace which can be reclaimed to accommodate each of the 4 alternate Replaces. I think I could rewrite the patch so that any of these 3 alts would still work:
- `body{text-align:justify}`
- `body{text-rendering:optimizeSpeed}`
- `#book-inner{font-size:1.5em}`

That leaves the `body{text-rendering:optimizeLegibility}` option which is a few chars too long.

I don't know how many users use this option (enables ligatures/kerning in kepub) but as it's the only one I, personally, want I plan to use a workaround which gets rid of one of the `!important` occurrences on a style aimed at languages using vertical text. I won't be reading any Chinese/Japanese vertical text so I'm hoping I'll get away with it, but possibly not safe as a published patch. 

Thoughts?

pgaskin, 2021-07-18:
> I don't know how many users use this option

I'm not sure either.

> restrict ourselves to patching either part1 or part2 (rather than the 2 parts together) there is no longer quite enough cosmetic whitespace which can be reclaimed to accommodate each of the 4 alternate 

Note that this probably won't be an issue anymore with the next major kobopatch version (late next year at the earliest, though), since I'm planning to implement the ability to move strings to a new memory location, to allocate arbitrary amounts of space in the executable, and to support instruction replacements via trampolines.

> Replaces. I think I could rewrite the patch so that any of these 3 alts would still work:

That's probably a good idea, and it's also a better fit for the kobopatch way of doing things (multiple individual options in a PatchGroup to make it easier to use overrides and reduce the amount of copy/paste needed).

> It looks like you've already rewritten the first [ePub uniform font scale patch] and removed the second.

Yep, that's correct. It would be helpful if you could do a quick test on on some of your devices to determine the optimal multiplier, but it's not an issue if you can't since the current value shouldn't be any worse than it was before.

> As far as I'm concerned fw versions 4.28.xxxx prior to this one are beta versions and no-one should need a kobopatch pack for them

That's why I'm not testing on them directly. I decided to update the old ones so I could look at the changes incrementally (for both the firmware and the patches), and for completeness. Notice how the last 3 versions took around 7 minutes to update for.

If you just want to update the latest one, I'll backport any changes as necessary.

> The Elipsa has the same screen pixels as the KA1 so I assume all the current Daylight settings will apply.

That was my initial guess, but I when I looked into it, I found that dragon is the primary codename. I think it has more to do with the ppi than the resolution. @NiLuJe might have some relevant insight.

> I plan to use a workaround which gets rid of one of the !important occurrences on a style aimed at languages using vertical text. I won't be reading any Chinese/Japanese vertical text so I'm hoping I'll get away with it, but possibly not safe as a published patch

If you don't think it should be included in the releases, feel free to put it as a comment here so it doesn't get lost and so it can be referenced if someone wants it specifically.

NiLuJe, 2021-07-18:
> That was my initial guess, but I when I looked into it, I found that dragon is the primary codename. I think it has more to do with the ppi than the resolution. @NiLuJe might have some relevant [...]

No idea, really: I was already puzzled by how the Forma was handled on that front ;).

jackiew1, 2021-07-18:
> Yep, that's correct. It would be helpful if you could do a quick test on on some of your devices to determine the optimal multiplier, but it's not an issue if you can't since the current value shouldn't be any worse than it was before.

I've just realised again how long "a quick test" takes with these patches 😃 

I think the multiplier for the old `KePub stylesheet additions` option 4 should now be about 2.75 instead of 1.5, at least on the old AuraHD I used as the guinea-pig. 

I haven't got the patience to test on a different device tonight. I'll check out what happens on a KA1 (i.e. non-Dragon device) tomorrow. The KA1 didn't used to have a epub/kepub font-size mismatch.

As far as I can tell new `ePub uniform font scale` plus old `Set font scale factor` works the same as before. Given that these 2 patches exist is there any real need to keep a `KePub stylesheet additions` font-size scaling option? If there's only one way to do font-size matching we wouldn't need to answer questions about the pros and cons of using one way rather than the other.

pgaskin, 2021-07-18:
> I've just realised again how long "a quick test" takes with these patches

Hopefully that should be reduced whenever I get around to finishing the Kobo emulator...

> I think the multiplier for the old `KePub stylesheet additions` option 4 should now be about 2.75 instead of 1.5, at least on the old AuraHD I used as the guinea-pig.

I was actually referring to the new `ePub uniform font scale` patch, but I suppose the final optional replacement in it I was referring to ends up having the same effect as `Set font scale factor`. The optional replacement I added to `ePub uniform font scale` is a bit easier to tune, though.

> I haven't got the patience to test on a different device tonight. I'll check out what happens on a KA1 (i.e. non-Dragon device) tomorrow. The KA1 didn't used to have a epub/kepub font-size mismatch.

:+1: 

> As far as I can tell new `ePub uniform font scale` plus old `Set font scale factor` works the same as before.

:+1: 

> Given that these 2 patches exist is there any real need to keep a `KePub stylesheet additions` font-size scaling option? If there's only one way to do font-size matching we wouldn't need to answer questions about the pros and cons of using one way rather than the other.

Personally, I never understood the point of the font scaling option in that patch, but I think it's there since that patch predates the font scale ones by quite a bit. I'd vote for removing it.

Are you planning to add patches for the other options to `jackie_w.yaml`, or are we leaving those out for now?

---

I'm going to do the final testing for the libnickel patches later tonight, then I'll make the release once you're ready.

NiLuJe, 2021-07-18:
Yeah, +1 on removing the duplicate historical font scaling thingy ;).

jackiew1, 2021-07-19:
That's 3-nil for getting rid of it, then. In that case the old `KePub stylesheet additions` will be split into 2 new patches:
- `KePub stylesheet additions - text justify`
- `KePub stylesheet additions - optimizeSpeed`

which share the same PatchGroup and which don't need to be customised. I'll put them in `jackie.yaml` shortly.

For the missing `optimizeLegibility` option (which no longer fits in the available space) I'm inclined to just leave it out for now to get a better idea of how many (_if any_) will miss it. In my brief testing, the not-ideal workaround I mentioned earlier seems to work OK for me (_as an English-only horizontal-left-to-right top-to-bottom reader_).

pgaskin, 2021-07-19:
I'm ready to release the patches. @jackiew1?

jackiew1, 2021-07-19:
I've now tested this fw on a KA1 and can confirm that when unpatched it has the same huge font-size mismatch between epub and kepub as seen on the AuraHD Dragon device. Applying patch `ePub uniform font scale` with or without patch `Set font scale factor` removes the problem.

When I first bought the KA1 (on launch day) this font-size mismatch didn't exist. When things changed I don't know because I've been using `ePub uniform font scale` since it was created. I used to think it was "unnecessary but harmless" on my KA1/Forma but I'd say it's now become "essential" for me.

I'm not prepared to load this mark8 4.28.17925 on my beloved Forma but I might have a look later to see whether the official mark7 4.26.16704 also has the font-size mismatch when `ePub uniform font scale` is disabled.

Release the patches if you really want to. Personally I'd wait until Kobo show some sign that they have a stable mark8 firmware. If they keep releasing new versions every few days I'm not going to be happy re-testing every time. I'm sure we've both got better things to do. As neither of us has an Elipsa (_and no prospect of that changing, in my case_) troubleshooting may be problematic.

NiLuJe, 2021-07-19:
I've got access to one (for now at least), if you need a guinea pig ;). (I won't be available next week, though).

jackiew1, 2021-07-19:
> I've got access to one (for now at least), if you need a guinea pig ;). (I won't be available next week, though).

Let's hope you won't regret valunteering 😄

jackiew1, 2021-07-19:
> I'm not prepared to load this mark8 4.28.17925 on my beloved Forma but I might have a look later to see whether the official mark7 4.26.16704 also has the font-size mismatch when `ePub uniform font scale` is disabled.
> 
Well, "later" turned out to be "quite soon". 

In fw 4.26.16704 the Forma has no font-size mismatch between epub/kepub when both `ePub uniform font scale` and `Set font scale factor` are disabled. That looks like a brand new negative enhancement feature awaiting non-patchers using non-Dragon devices at the next official fw update.

pgaskin, 2021-07-19:
> That looks like a brand new negative enhancement feature awaiting non-patchers using non-Dragon devices at the next official fw update.

It's due to the new method of calculating the dpi multiplier which gets the DPI from the QScreen (which might be incorrect; @NiLuJe?).

pgaskin, 2021-07-20:
Released: [`v71`](https://github.com/pgaskin/kobopatch-patches/releases/v71), MR:thread-340741.

pgaskin, 2021-07-20:
<details><summary><b>$ git diff --no-index src/versions/{4.28.17826,4.28.17925}</b></summary>

```diff
diff --git a/src/versions/4.28.17826/libnickel.so.1.0.0.yaml/geoffr.yaml b/src/versions/4.28.17925/libnickel.so.1.0.0.yaml/geoffr.yaml
index 0d9e843..ec96194 100644
--- a/src/versions/4.28.17826/libnickel.so.1.0.0.yaml/geoffr.yaml
+++ b/src/versions/4.28.17925/libnickel.so.1.0.0.yaml/geoffr.yaml
@@ -491,5 +491,5 @@ Set font scale factor:
   # Original value X=14, so original scale factor is (font size slider) / 46.
   # Allowed replacement values for X: 0-15 (smaller X --> larger scale factor)
   # Note: vmov.f32 s1, #15.0 before getReadingFontSizeScaleFactor
-  - ReplaceInt: {Offset: 0x5E74A6, Find: 14, Replace: 4} # ePub
-  - ReplaceInt: {Offset: 0xC5C29A, Find: 14, Replace: 4} # KePub
+  - ReplaceInt: {Offset: 0x5E73FE, Find: 14, Replace: 4} # ePub
+  - ReplaceInt: {Offset: 0xC5CB82, Find: 14, Replace: 4} # KePub
diff --git a/src/versions/4.28.17826/libnickel.so.1.0.0.yaml/pgaskin.yaml b/src/versions/4.28.17925/libnickel.so.1.0.0.yaml/pgaskin.yaml
index d70cfd1..e7de9a6 100644
--- a/src/versions/4.28.17826/libnickel.so.1.0.0.yaml/pgaskin.yaml
+++ b/src/versions/4.28.17925/libnickel.so.1.0.0.yaml/pgaskin.yaml
@@ -553,7 +553,7 @@ ePub uniform font scale:
       working on 4.28.17820 since the dpi was changed to use QScreen instead of
       144dpi on dragon and 96dpi on other devices (GeoffR's patch forced it to
       always be 96dpi). This patch has not been tested on the Elipsa.
-  - BaseAddress: 0x5E74CE
+  - BaseAddress: 0x5E7426
   - ReplaceBytes:
       Offset:      0
       FindInstBLX: {SymPLT: "QGuiApplication::primaryScreen()"} # unused
@@ -592,7 +592,7 @@ Customize ComfortLight settings:
   # In an unnamed subroutine two layers into FrontLightPopupController::loadView
   # (find it by going back from QTime::addSecs), the times for the dropdown are
   # generated into a QVector<QPair<QString, QTime>> with a simple loop.
-  - BaseAddress: 0xBAE960 # find the base of the unnamed subroutine with the x-ref to _ZN5QTimeC1Eiiii
+  - BaseAddress: 0xBAF248 # find the base of the unnamed subroutine with the x-ref to _ZN5QTimeC1Eiiii
   #
   # Change the initial hour / first bedtime dropdown item passed to the QTime
   # constructor (mov r1, #21):
diff --git a/src/versions/4.28.17826/nickel.yaml/oren64.yaml b/src/versions/4.28.17925/nickel.yaml/oren64.yaml
index 16e57fe..5184928 100644
--- a/src/versions/4.28.17826/nickel.yaml/oren64.yaml
+++ b/src/versions/4.28.17925/nickel.yaml/oren64.yaml
@@ -27,7 +27,7 @@ New home screen subtitle custom font:
   - Description: |
       Increase subtitle font size, font color black. Works best when used
       with with patch `New home screen increasing cover size`
-  - BaseAddress: 0x15E1E38 # qss/GenericHomeWidget.qss
+  - BaseAddress: 0x15DED55 # qss/GenericHomeWidget.qss
   - ReplaceZlib: {Find: "MetaDataLabel {\n  color: #666666;\n  qproperty-indent: 0;\n}", Replace: "MetaDataLabel {\n  color: #000000;\n  qproperty-indent: 0;\n}"}
   - ReplaceZlib:
       Find:    "MetaDataLabel[qApp_deviceIsTrilogy=true] {\n  font-size: 13px;\n}"
diff --git a/src/versions/4.28.17826/nickel.yaml/pgaskin.yaml b/src/versions/4.28.17925/nickel.yaml/pgaskin.yaml
index 1bcc091..9f05039 100644
--- a/src/versions/4.28.17826/nickel.yaml/pgaskin.yaml
+++ b/src/versions/4.28.17925/nickel.yaml/pgaskin.yaml
@@ -15,7 +15,7 @@ Remove forgot pin button from lock screen:
       Removes the Forgot PIN -> Sign Out button from the lock screen.
       If this is enabled and you forget your pin, you will need to
       hard reset your Kobo.
-  - BaseAddress: 0x15D1D42 # qss/PinCodeInputDialog.qss
+  - BaseAddress: 0x15D8400 # qss/PinCodeInputDialog.qss
   - ReplaceZlib:
       Find: "#lblForgotPin[qApp_deviceIsDragon=true],\n#lblSignOut[qApp_deviceIsDragon=true] {\n  font-size: 26px;\n}"
       Replace: "#lblForgotPin,#lblSignOut{qproperty-visible:false;}"
```

</details>

closed 2021-07-20 by pgaskin

NiLuJe, 2021-07-21:
Lab rat reporting in ;p.

I applied my usual set on the Elipsa a few hours ago, nothing horribly blew up ;).