pgaskin/kobopatch-patches #92: Patches for 4.28.18220

pgaskin, 2021-08-10:
TODO

pgaskin, 2021-08-10:
So first, a few interesting things I noticed about this version:
- The Freescale and Sunxi boards have their own rootfs.
  - Luckily, libnickel and the boot scripts appear to be identical.
  - kobo8 doesn't include:
    - Elipsa DBus services used by nickel:
      - bluealsa (the audio thingy)
      - foxit (for PDF rendering instead of RMSDK)
      - iink (for markup)
    - GStreamer
    - pickel-aw (pickel for Sunxi boards)
    - non-Elipsa kmods, bootloader files, and boot images
  - kobo7- doesn't include:
    - pickel-freescale (pickel for Sunxi boards)
    - Elipsa kmods, bootloader files, and boot images

Basically, the only thing which really matters for us is that we need to delete `the following Elipsa-specific files` if we installed kobo8 firmware on non-kobo8 devices (or nickel may behave differently) (there's also more files than these, but they should not make any difference).

pgaskin, 2021-08-10:
```
Errors:
  /home/patrick/src/kobopatch-patches/src/versions/4.28.18220/nickel.yaml/pgaskin.yaml: Remove forgot pin button from lock screen: could not apply patch "Remove forgot pin button from lock screen": line 20: inst 4: ReplaceZlib: not a zlib stream
  /home/patrick/src/kobopatch-patches/src/versions/4.28.18220/libnickel.so.1.0.0.yaml/pgaskin.yaml: Customize ComfortLight settings: could not apply patch "Customize ComfortLight settings": line 599: inst 4: ReplaceInt: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.28.18220/libnickel.so.1.0.0.yaml/pgaskin.yaml: ePub uniform font scale: could not apply patch "ePub uniform font scale": line 558: inst 4: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.28.18220/libnickel.so.1.0.0.yaml/geoffr.yaml: Set font scale factor: could not apply patch "Set font scale factor": line 494: inst 3: ReplaceInt: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.28.18220/libnickel.so.1.0.0.yaml/geoffr.yaml: Custom font sizes: could not apply patch "Custom font sizes": line 193: inst 4: ReplaceInt: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.28.18220/nickel.yaml/oren64.yaml: New home screen subtitle custom font: could not apply patch "New home screen subtitle custom font": line 31: inst 4: ReplaceZlib: not a zlib stream
```

In addition to the usual:
- ePub uniform font scale
- Custom font sizes
- New home screen subtitle custom font

Maybe they fixed the epub font size issue? I'll need to do some additional testing.

pgaskin, 2021-08-10:
**Resource extraction:**

```bash
./qrc2zip --output "nickel.18220.qInitResources_resources.zip"    --recursive --verbose "nickel" 1 $((0x12205d0 - 0x0010000)) $((0x00272a0 - 0x0010000)) $((0x12200e0 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
./qrc2zip --output "nickel.18220.qInitResources_translations.zip" --recursive --verbose "nickel" 1 $((0x15d89a0 - 0x0010000)) $((0x12209b8 - 0x0010000)) $((0x15d8718 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
./qrc2zip --output "nickel.18220.qInitResources_styles.zip"       --recursive --verbose "nickel" 1 $((0x15f71f0 - 0x0010000)) $((0x15d8ab8 - 0x0010000)) $((0x15f4940 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
./qrc2zip --output "nickel.18220.qInitResources_certificates.zip" --recursive --verbose "nickel" 1 $((0x161e740 - 0x0010000)) $((0x15f7d70 - 0x0010000)) $((0x161d6f0 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
```

<details><summary><b>Output</b></summary>

```
nickel@18220/qInitResources_resources
DIR     "qss" (0x12109A4 + 14)
FILE    "qss/TouchMenu.qss" (0x120F6D7 + 189)
DIR     "html" (0x1210988 + 28)
FILE    "html/RGBFrontLightAboutView.html" (0x120F798 + 1144)
FILE    "html/RGBFrontLightAboutViewSnow.html" (0x120FC14 + 1228)
DIR     "keys" (0x121096C + 28)
FILE    "keys/device.qmap" (0x120F657 + 124)
FILE    "keys/avrcp.qmap" (0x120F47B + 472)
FILE    "omfTemplate.html" (0x29837 + 231)
DIR     "pika.rcc" (0x410D + 14)
DIR     "pika.rcc/images" (0x411B + 42)
DIR     "pika.rcc/images/fte" (0x4161 + 14)
FILE    "pika.rcc/images/fte/quicktour_diagram.png" (0x312E + 3841)
DIR     "pika.rcc/images/home" (0x4153 + 14)
FILE    "pika.rcc/images/home/quicktour_device.png" (0x18 + 4337)
DIR     "pika.rcc/images/about" (0x4145 + 14)
FILE    "pika.rcc/images/about/global_certification.png" (0x110D + 8221)
FILE    "verifier.pub" (0x977DE0 + 451)
DIR     "daylight.rcc" (0xD7B74 + 14)
DIR     "daylight.rcc/images" (0xD7B82 + 336)
DIR     "daylight.rcc/images/fte" (0xD93F4 + 280)
FILE    "daylight.rcc/images/fte/missingbooks_factoryreset.png" (0x75D3A + 9065)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg4_adjust_brightness.png" (0x7D1AC + 24664)
FILE    "daylight.rcc/images/fte/missingbooks_account.png" (0xB18AE + 22560)
FILE    "daylight.rcc/images/fte/missingbooks_repair.png" (0x780A7 + 20737)
FILE    "daylight.rcc/images/fte/quicktour_nightmode.png" => "daylight.rcc/images/fte/quicktour_nightmode[language!Japanese].png" (0xB70D2 + 24646)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg2.png" => "daylight.rcc/images/fte/quicktour_reading_pg2[language!Japanese].png" (0x8FE13 + 23892)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg2.png" (0x95B6B + 23662)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg3.png" (0xA1485 + 23519)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg3.png" => "daylight.rcc/images/fte/quicktour_reading_pg3[language!Japanese].png" (0x9B7DD + 23716)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg5.png" (0xACB93 + 19735)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg1.png" (0x8A4F9 + 22806)
FILE    "daylight.rcc/images/fte/missingbooks_sync.png" (0xBD5EF + 6061)
FILE    "daylight.rcc/images/fte/fte_oob_connected.png" => "daylight.rcc/images/fte/fte_oob_connected[language!Japanese].png" (0x72A57 + 9157)
FILE    "daylight.rcc/images/fte/fte_oob_plugin.svg" (0x876EA + 3494)
FILE    "daylight.rcc/images/fte/pocket_logo_large.png" (0x74E20 + 3862)
FILE    "daylight.rcc/images/fte/quicktour_texticon_search.png" (0xBD54D + 158)
FILE    "daylight.rcc/images/fte/quicktour_diagram.png" (0x88494 + 8289)
FILE    "daylight.rcc/images/fte/missingbooks_help.png" (0xA7068 + 23335)
FILE    "daylight.rcc/images/fte/overflow_small.png" (0xBD11C + 1069)
FILE    "daylight.rcc/images/fte/missingbooks_intro.png" (0x83208 + 17630)
DIR     "daylight.rcc/images/book" (0xD9384 + 112)
FILE    "daylight.rcc/images/book/dustbin_delete_icon_inverted.png" (0x5ABA + 257)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_highlight.png" (0x5C8B + 447)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_dogear.png" (0x5F91 + 409)
FILE    "daylight.rcc/images/book/markup_icon_transparent_padding.png" (0x535C + 994)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_note.png" (0x612E + 202)
FILE    "daylight.rcc/images/book/note_icon_transparent_padding.png" (0x5BBF + 200)
FILE    "daylight.rcc/images/book/dustbin_delete_icon.png" (0x5E4E + 319)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_markup.png" (0x5742 + 884)
DIR     "daylight.rcc/images/help" (0xD933E + 70)
FILE    "daylight.rcc/images/help/help_large.png" (0x30146 + 1138)
FILE    "daylight.rcc/images/help/help_icon.png" (0x2E027 + 1022)
FILE    "daylight.rcc/images/help/kobopluslogo_help_sp.png" (0x2E429 + 7449)
FILE    "daylight.rcc/images/help/kobopluslogo_eLeya.png" (0x305BC + 10861)
FILE    "daylight.rcc/images/help/kobopluslogo_help_version.png" (0x2C61A + 6665)
DIR     "daylight.rcc/images/home" (0xD9234 + 266)
[... 2480 lines cut ...]
FILE    "dragon.rcc/images/widgets/settings_page_turning_option3_on.png" (0x4A00F + 381)
FILE    "dragon.rcc/images/widgets/add_page.png" (0x47BFA + 1625)
FILE    "dragon.rcc/images/widgets/settings_checkbox_on_disabled.png" (0x48BA5 + 173)
FILE    "dragon.rcc/images/widgets/settings_checkbox_off_disabled.png" (0x49CD9 + 98)
FILE    "dragon.rcc/images/widgets/dropbox_folder.png" (0x489FB + 250)
[... cut ...]
FILE    "dragon.rcc/images/navigation/browser_reload_hit.png" (0x47334 + 332)
FILE    "dragon.rcc/images/navigation/discover_dismiss_off.png" (0x47484 + 334)
FILE    "kepub-book.css" (0x80DA26 + 185)
DIR     "pixie.rcc" (0x2A20 + 14)
DIR     "pixie.rcc/images" (0x2A2E + 28)
DIR     "pixie.rcc/images/fte" (0x2A58 + 14)
FILE    "pixie.rcc/images/fte/quicktour_diagram.png" (0x18 + 4688)
DIR     "pixie.rcc/images/home" (0x2A4A + 14)
FILE    "pixie.rcc/images/home/quicktour_device.png" (0x126C + 5916)
FILE    "smil.xslt" (0

pgaskin, 2021-08-10:
Doesn't appear to be any nickel changes, but does that diff mean what I think it means about Dropbox on other devices?

pgaskin, 2021-08-10:
So, I had a look at the font stuff, and it seems that they reworked the font size/scale calculations. This will need more testing.

pgaskin, 2021-08-10:
@jackiew1, can you confirm that the `ePub uniform font scale` and `Set font scale factor` patches aren't necessary anymore on your devices for consistent epub/kepub font sizes, and that `Custom font sizes` isn't required to have smaller font sizes than on older versions?

pgaskin, 2021-08-10:
I have tested the patches, and everything seems to work fine now. The font scale patches and the `Custom font sizes` patch doesn't seem to be necessary anymore.

@jackiew1, I'm ready to release this once you've tested the nickel ones, and if you have time, confirmed the font scale/size changes.

jackiew1, 2021-08-10:
@pgaskin Seems I'm a bit late to the party, but here I am now.

Based on what you said at the top of this thread, my test Kobos (KA1, AuraHD) need to get rid of some Elipsa-only files. How do I do that?

pgaskin, 2021-08-10:
> Based on what you said at the top of this thread, my test Kobos (KA1, AuraHD) need to get rid of some Elipsa-only files. How do I do that?

That shouldn't make a difference for the patches, and it probably won't make a significant difference with anything else (the only possibility is foxit being used instead of RMSDK for PDFs, but I haven't tested that yet). If you still want to remove them, you can do it with NickelMenu: `menu_item:main:Remove files:cmd_output:500:rm -fv /usr/share/dbus-1/services/com.kobo.{foxit,myscript}.*`.

jackiew1, 2021-08-10:
@pgaskin 
Thanks for the NM tip. I think I'll let the dust settle before I do any remedial work on my KA1 & AuraHD just in case you make any new discoveries in the next few days.

I'll do my testing on my H2O which has only ever had official mark5 fw installed.

pgaskin, 2021-08-10:
> I think I'll let the dust settle before I do any remedial work on my KA1 & AuraHD just in case you make any new discoveries in the next few days.

I think I've discovered everything that will make a difference. On MR:post-4145522, someone reported that the font sizes are acting oddly on the KA1. On the contrary, based on @shermp and my testing, the font size/scale issues have been solved in this firmware version. Can you at least test the font sizes/scaling on the stock FW on the KA1 to see if there is actually a problem on the KA1?

jackiew1, 2021-08-10:
I just installed 4.28.18220 on the H2O.  At first glance all my nickel patches appear to be working. I disabled
- `Custom font-sizes`
- `ePub uniform font scale`
- `Set font scale factor`

and cleared the database's content_settings table.

There is still a big font-size mismatch between epub/kepub. Are you saying this is not what you're seeing?

jackiew1, 2021-08-10:
Got to stop for about 30 mins, will be back later.

shermp, 2021-08-10:
After doing a bit more testing, can confirm that at higher font sizes, there's definitely a difference between epub/kepub

pgaskin, 2021-08-10:
> There is still a big font-size mismatch between epub/kepub. Are you saying this is not what you're seeing?

I was seeing around the same as when I had the patches applied (almost no difference at smaller font sizes, some difference at larger ones). The thing I'm not seeing anymore is the ridiculously large base EPUB font size from the regression in 17820.

The main thing I'm looking to confirm is whether the firmware by default now is as good or better than the previous version with my patches (which is much better than the previous version without them).

IIRC, the patches never completely fixed the font size mismatch at larger sizes, but correct me if I'm wrong.

jackiew1, 2021-08-11:
I don't know how you define "larger sizes". Based on screenshots I see from other people I'd describe my preferred font-size as "medium".

On the H2O my "right font-size" for an epub is 26. For the kepub (same book, same font-family, zero L/R margins on both) I needed to increase it to 44 for it to look "very similar".

Personally speaking the fw is worse without any of these 3 patches, but I'll get by one way or another even if I have to go back to force-injecting a kepub font-size scale-up during send-to-device with the calibre driver options.

pgaskin, 2021-08-11:
I can put back the two font scale patches without too much effort, but they will need to be tweaked again, and I don't have enough variety of devices to figure out the settings on. I'd rather not redo the custom font sizes patch unless absolutely necessary since that one will be a PITA to rewrite and test.

jackiew1, 2021-08-11:
I suppose absence of `Custom font-sizes` might be less of a problem if the other 2 worked. I was using it to extend the range of font-sizes in the increment-by-1 zone by reducing the increment-by-2 and increment-by-4 zones to hardly any. Even my Forma was able to always use a value in the increment-by-1 zone. If kepubs can be read at the same lower size as epub then the increment-by-4 zone may be avoided.

I've no reason to presume my usage is typical.

pgaskin, 2021-08-11:
So, a few more people are reporting font size issues on other devices. I guess I'm going to update the font scale patches again.

jackiew1, 2021-08-11:
> So, a few more people are reporting font size issues on other devices. I guess I'm going to update the font scale patches again.

I think people are mainly reporting that their perfect font-size they had set in part-read books  pre-update is no longer perfect post-update. We know the font-sizes have been re-scaled (or something) so this isn't really surprising. I'm not really sure why it's a big deal as you just have to set your "new perfect" and carry on. Eventually all new books will be back as you want them.

The bigger problem (_actually annoyance, not problem_) remains the epub/kepub mismatch for those who read both. Dragon device users will be familiar with this issue but for non-Dragon device users (KA1, Forma, Glo etc) it will be a new thing to get upset about.

I wonder what %age of Kobo users are epub-only, kepub-only versus mixture of both?

NiLuJe, 2021-08-11:
> 
> I wonder what %age of Kobo users are epub-only, kepub-only versus mixture of both?

We might have a stronger bias towards ePub on MR than the average Joe, but it's probably still a minority, although a possibly rather sizable one ;p.

jackiew1, 2021-08-14:
@NiLuJe 
If you've still got your Elipsa please can you tell me what the min/max values are on the [Aa] menu font-size slider bar. I'm going to guess that they're the same as for KA1/Forma, i.e. min=14, max=194, but confirmation would be good.

NiLuJe, 2021-08-14:
> @NiLuJe
> If you've still got your Elipsa please can you tell me what the min/max values are on the [Aa] menu font-size slider bar. I'm going to guess that they're the same as for KA1/Forma, i.e. min=14, max=194, but confirmation would be good.

Are those affected by the font size patches? (Because I'm currently using 'em).

jackiew1, 2021-08-14:
> Are those affected by the font size patches? (Because I'm currently using 'em).

It was possible to change them via the `Custom font-sizes` patch but I don't think either `ePub uniform font scale` or `Set font scale factor` affect them. None of these 3 patches work in this firmware, do they? Or have I missed something?

The reason I asked the question is because I've been doing some of my own patch hacking on `Custom font sizes`. As I don't have any de-compiling skills I've had to rely on guesswork using eyeball comparisons of hex-strings, but I thought there may be a possibility that some Elipsa-specific code had been added which wouldn't be apparent when comparing fw 4.28 and 4.26.

pgaskin, 2021-08-14:
> Custom font sizes

This code involved in this one has changed significantly, and will need some work (and a lot of testing) to update.

> ePub uniform font scale, Set font scale factor

I could update these again, and probably will do at least one of them (I won't have any preset values though, since those seem to need to be different on each device right now).

I might have time to look at the patches again within the next few days.

NiLuJe, 2021-08-14:
> It was possible to change them via the `Custom font-sizes` patch but I don't think either `ePub uniform font scale` or `Set font scale factor` affect them. None of these 3 patches work in this [...]

But I'm running the *previous* 4.28 build on the Elipsa, where they were ;).

NiLuJe, 2021-08-14:
> @NiLuJe
> If you've still got your Elipsa please can you tell me what the min/max values are on the [Aa] menu font-size slider bar. I'm going to guess that they're the same as for KA1/Forma, i.e. min=14, max=194, but confirmation would be good.

Err, stupid question is stupid: how do I even get to these numerical values?

jackiew1, 2021-08-14:
> Err, stupid question is stupid: how do I even get to these numerical values?

- Open any epub/kepub. Open [Aa] menu and slide font-size slider to minimum. (_Trying to read a kepub at this size may damage your eyesight_)
- Connect Elipsa to PC and note value of `readingFontSize` in `Kobo eReader.conf` [Reading] section.
- Repeat the above 2 steps after sliding the slider to maximum. (_Scratch head wondering who in the world would ever want to use this font-size_)

NiLuJe, 2021-08-14:
Right, that's what I thought, but I was mildly confused at not seeing the key at all in my current config ^^.

(Possibly because I haven't *actually* read anything in Nickel on that device? :D).

NiLuJe, 2021-08-14:
And apparently... `12` - `108`

```
┌─(ROOT@europa:pts/0)──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────(~)─┐
└─(2.53:21%:23:54:97%:#)── ag readingFontSize /mnt/onboard/.kobo/Kobo/Kobo\ eReader.conf                                                                                                                                                                                                                                                               ──(Sat, Aug 14)─┘
readingFontSize=12
┌─(ROOT@europa:pts/0)──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────(~)─┐
└─(2.46:22%:23:54:97%:#)── ag readingFontSize /mnt/onboard/.kobo/Kobo/Kobo\ eReader.conf                                                                                                                                                                                                                                                               ──(Sat, Aug 14)─┘
readingFontSize=108
```

jackiew1, 2021-08-14:
Hmm, well I wasn't expecting those values. I wonder why the maximum size on an Elipsa would be less than that for every other Kobo except Touch/Mini.

jackiew1, 2021-08-18:
@pgaskin
We've had an [MR request](https://www.mobileread.com/forums/showthread.php?t=341269) about the removed `optimizeLegibility` option for the old `KePub stylesheet additions` patch. I need to post my beta somewhere because it doesn't belong in the standard patch pack. Would you prefer:
- the linked MR thread
- this GitHub issue
- a separate issue in `kobopatch-patches` - which I can open then immediately close.

pgaskin, 2021-08-19:
A separate issue would probably be best.

Also, do you think I should just release the patches as-is and mark the font size/scale ones as missing for a future version? It wouldn't be the first time we've done something like that. I don't think I'll have enough time to look at them this week, and the patch stuff is starting to clutter the main firmware thread.

jackiew1, 2021-08-19:
> Also, do you think I should just release the patches as-is and mark the font size/scale ones as missing for a future version? It wouldn't be the first time we've done something like that. I don't [...]

Probably a good idea to release what we have so far if you won't have time to look at the missing patches for a while.

Re: `Custom font sizes`
I saw your reply to someone on MR this week where you said you thought it may no longer be needed. That may well be true because I'm not sure how many people ever took the time to understand what it actually did. However, for anyone (e.g. me) who ever used it to greatly extend the range of font-sizes where you could inc/dec by 1, rather than always being stuck in the inc/dec by 2 (for epub) or 4 (for kepub) ranges, then that "need" won't have changed. I'm not saying this to try and force the issue because the number of users who would miss it may be tiny.

The revised alpha patch I cobbled together seems to work OK for my needs for 4.28.18220 but I wouldn't want to release it to others given you say the code has changed significantly. FWIW my alpha doesn't look hugely different from the original but I've made no attempt to cater for the Elipsa.

- Initial `BaseAddress: "N3FontTypeUtil::fontSizes()` still works as before.
- Tweaked all offsets - which had not changed by many bytes.
- Tweaked integer constants representing model's minimum font-size. I don't think there's much to be gained from patching these, so I commented these changes out for myself.

I've posted my alpha patch below in case you decide to look into it further.
Taking the ClaraHD as an example, these are the default vs patched font-sizes I would expect to see on the [Aa] slider bar:
Unpatched:
- 10-22 in steps of 1 (13 sizes)
- 24-50 in steps of 2 (14 sizes)
- 54-150 in steps of 4 (25 sizes)

Alpha-patched:
- 10-60 in steps of 1 (51 sizes)
- 62-72 in steps of 2 (6 sizes)
- 76-80 in steps of 4 (2 sizes)

```
Custom font sizes - jackie_w:
  - Enabled: no
  - Description: |
      With this patch enabled you will not be able to select the very large font
      sizes, but will be able to make finer adjustment to the smaller sizes.
      Font sizes depend on the device's screen density. Unpatched, the sizes
      increase in steps of 1 from the smallest size up to size 22, then in steps
      of 2 up to size 50, then in steps of 4 up to the largest size:
  #
  #  AuraOne/Forma:       14px - 194px (59 sizes) new
  #  GloHD/ClaraHD/Libra: 10px - 150px (52 sizes) new (only tested ClaraHD)
  #  AuraHD/H2O:           8px - 150px (54 sizes) new (only tested H2O)
  #  Glo/Aura:             8px - 122px (47 sizes) new (only tested Glo)
  #  Touch/Mini:           8px - 90px  (39 sizes) ???
  #  Elipsa               12px - 108px (39 sizes) ??? based on NiLuJe's min/max observations
  - BaseAddress: "N3FontTypeUtil::fontSizes()"
  # Initial font size:
  #- ReplaceInt: {Offset: 390, Find: 14, Replace: 18} # AuraOne/Forma
  #- ReplaceInt: {Offset:  36, Find: 10, Replace: 18} # ClaraHD
  #- ReplaceInt: {Offset: 362, Find: 8, Replace: 18}  # H2O/Glo
  # Increment:
  - ReplaceInt: {Offset:  78, Find: 21, Replace: 59} # Add font sizes in increments of 1 until this size exceeded}
  - ReplaceInt: {Offset:  88, Find: 22, Replace: 60} # Continue from this font size}
  - ReplaceInt: {Offset: 218, Find: 49, Replace: 71} # Add font sizes in increments of 2 until this size exceeded}
  - ReplaceInt: {Offset: 224, Find: 50, Replace: 72} # Continue from this font size}
  # Now increment by +4 until final font size:
  - ReplaceInt: {Offset: 386, Find: 195, Replace: 80} # AuraOne/Forma
  - ReplaceInt: {Offset:  50, Find: 150, Replace: 80} # H2O/ClaraHD
  - ReplaceInt: {Offset: 412, Find: 122, Replace: 80} # Glo
  - ReplaceInt: {Offset: 408, Find: 90,  Replace: 80} # Touch/Mini (not tested)

```

NiLuJe, 2021-08-19:
If I actually still read in Nickel, I'd add another vote for keeping that one, because the default granularity in the "useful" range is nigh useless ;).

Dkoeddnsnp, 2021-09-14:
For me the most important patches were the uniform scale for epub and kepub, custom font sizes and ligatures for kepub. Unfortunately with all those patches missing on newer patchset releases I have to block firmware updates.

pgaskin, 2021-10-22:
<details><summary><b>$ git diff --no-index src/versions/{4.28.17925,4.28.18220}</b></summary>

```diff
diff --git a/src/versions/4.28.17925/libnickel.so.1.0.0.yaml/geoffr.yaml b/src/versions/4.28.18220/libnickel.so.1.0.0.yaml/geoffr.yaml
index ec96194..f06e363 100644
--- a/src/versions/4.28.17925/libnickel.so.1.0.0.yaml/geoffr.yaml
+++ b/src/versions/4.28.18220/libnickel.so.1.0.0.yaml/geoffr.yaml
@@ -164,45 +164,50 @@ Custom font sizes:
       the small sizes and fewer of the large sizes.
       With this patch enabled you will not be able to select the very large font
       sizes, but will be able to make finer adjustment to the smaller sizes.
-    # Font sizes depend on the device's screen density. Unpatched, the sizes
-    # increase in steps of 1 from the smallest size up to size 22, then in steps
-    # of 2 up to size 50, then in steps of 4 up to the largest size:
-    #
-    #  AuraOne/Forma:       18px - 194px (55 sizes)
-    #  GloHD/ClaraHD/Libra: 14px - 150px (48 sizes)
-    #  AuraHD/H2O:          12px - 150px (50 sizes)
-    #  Glo/Aura:            12px - 122px (43 sizes)
-    #  Touch/Mini:          12px - 90px  (35 sizes)
-    #
-    # The example replacement values in this patch result in the following ranges,
-    # with increases in steps of 1 from the smallest size up to size 44, then
-    # steps of 2 up to size 68, then steps of 4 up to the largest size:
-    #
-    #  AuraOne/Forma:       18px - 132px (55 sizes)
-    #  GloHD/ClaraHD/Libra: 14px - 108px (53 sizes)
-    #  AuraHD/H2O:          12px - 108px (55 sizes)
-    #  Glo/Aura:            12px - 88px (50 sizes)
-    #  Touch/Mini:          12px - 80px (48 sizes)
+  # Font sizes depend on the device's screen density. Unpatched, the sizes
+  # increase in steps of 1 from the smallest size up to size 44, then in steps
+  # of 2 up to size 68, then in steps of 4 up to the largest size:
+  #
+  #                     Touch/Mini:    8px -  90px  (39 steps)
+  #          AuraHD/AuraH2O/Libra2:    8px - 150px  (54 steps)
+  #                  GloHD/ClaraHD:   10px - 150px  (52 steps)
+  #                       LibraH2O:   11px - 150px  (51 steps)
+  #      AuraOne/Forma/Elipsa/Sage:   14px - 195px  (59 steps)
+  #                   Aura/Glo/Nia:    8px - 122px  (47 steps)
+  #
+  # The example replacement values in this patch result in the following ranges,
+  # with increases in steps of 1 from the smallest size up to size 44, then
+  # steps of 2 up to size 68, then steps of 4 up to the largest size:
+  #
+  #                     Touch/Mini:    8px -  80px  (52 steps)
+  #          AuraHD/AuraH2O/Libra2:    8px - 108px  (59 steps)
+  #                  GloHD/ClaraHD:   10px - 108px  (57 steps)
+  #                       LibraH2O:   11px - 108px  (56 steps)
+  #      AuraOne/Forma/Elipsa/Sage:   14px - 132px  (59 steps)
+  #                   Aura/Glo/Nia:    8px -  88px  (54 steps)
   #
   # Note (pgaskin): The device values can be determined with info from
   #    https://gist.github.com/pgaskin/613b34c23f026f7c39c50ee32f5e167e and
   #    the Device::is* calls.
   #
+  # This patch was rewritten for 18220 by @shermp.
+  #
   - BaseAddress: "N3FontTypeUtil::fontSizes()"
   # Initial font size:
-  - ReplaceInt: {Offset: 400, Find: 18, Replace: 18} # AuraOne/Forma
-  - ReplaceInt: {Offset:  36, Find: 14, Replace: 14} # GloHD/ClaraHD/Libra
-  - ReplaceInt: {Offset: 376, Find: 12, Replace: 12} # Others
+  - ReplaceInt: {Offset: 362, Find:   8, Replace:   8} # Other devices
+  - ReplaceInt: {Offset: 358, Find:  11, Replace:  11} # LibraH2O
+  - ReplaceInt: {Offset:  36, Find:  10, Replace:  10} # ClaraHD/GloHD
+  - ReplaceInt: {Offset: 390, Find:  14, Replace:  14} # Sage/Forma/AuraOne/Elipsa
   # Increment:
-  - ReplaceInt: {Offset:  76, Find: 21, Replace: 43} # Add font sizes in increments of 1 until this size exceeded}
-  - ReplaceInt: {Offset:  84, Find: 22, Replace: 44} # Continue from this font size}
[... 303 lines cut ...]
@@ -27,7 +27,7 @@ New home screen subtitle custom font:
   - Description: |
       Increase subtitle font size, font color black. Works best when used
       with with patch `New home screen increasing cover size`
-  - BaseAddress: 0x15DED55 # qss/GenericHomeWidget.qss
+  - BaseAddress: 0x15C93CF # qss/GenericHomeWidget.qss
   - ReplaceZlib: {Find: "MetaDataLabel {\n  color: #666666;\n  qproperty-indent: 0;\n}", Replace: "MetaDataLabel {\n  color: #000000;\n  qproperty-indent: 0;\n}"}
   - ReplaceZlib:
       Find:    "MetaDataLabel[qApp_deviceIsTrilogy=true] {\n  font-size: 13px;\n}"
diff --git a/src/versions/4.28.17925/nickel.yaml/pgaskin.yaml b/src/versions/4.28.18220/nickel.yaml/pgaskin.yaml
index 9f05039..d405133 100644
--- a/src/versions/4.28.17925/nickel.yaml/pgaskin.yaml
+++ b/src/versions/4.28.18220/nickel.yaml/pgaskin.yaml
@@ -15,7 +15,7 @@ Remove forgot pin button from lock screen:
       Removes the Forgot PIN -> Sign Out button from the lock screen.
       If this is enabled and you forget your pin, you will need to
       hard reset your Kobo.
-  - BaseAddress: 0x15D8400 # qss/PinCodeInputDialog.qss
+  - BaseAddress: 0x15D5D6B # qss/PinCodeInputDialog.qss
   - ReplaceZlib:
       Find: "#lblForgotPin[qApp_deviceIsDragon=true],\n#lblSignOut[qApp_deviceIsDragon=true] {\n  font-size: 26px;\n}"
       Replace: "#lblForgotPin,#lblSignOut{qproperty-visible:false;}"
```

</details>