pgaskin/kobopatch-patches #97: Patches for 4.30.18838

pgaskin, 2021-11-01:
- [ ] Upload testdata (@pgaskin)
- [ ] Extract resources (@pgaskin)
- [ ] Find libnickel offsets where necessary (@pgaskin)
- [ ] Update most patches (@pgaskin)
- [ ] Test patches (@pgaskin)
- [ ] Make release (@pgaskin)
- [ ] MR post (@pgaskin)

pgaskin, 2021-11-01:
```
Errors:
  /home/patrick/src/kobopatch-patches/src/versions/4.30.18838/libadobe.so.yaml/pgaskin.yaml: Remove PDF map widget shown during panning: could not apply patch "Remove PDF map widget shown during panning": line 9: inst 4: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.30.18838/nickel.yaml/pgaskin.yaml: Increase size of kepub chapter progress chart: could not apply patch "Increase size of kepub chapter progress chart": line 28: inst 4: ReplaceZlib: not a zlib stream
  /home/patrick/src/kobopatch-patches/src/versions/4.30.18838/nickel.yaml/pgaskin.yaml: Remove forgot pin button from lock screen: could not apply patch "Remove forgot pin button from lock screen": line 20: inst 4: ReplaceZlib: not a zlib stream
  /home/patrick/src/kobopatch-patches/src/versions/4.30.18838/libnickel.so.1.0.0.yaml/pgaskin.yaml: Customize ComfortLight settings: could not apply patch "Customize ComfortLight settings": line 551: inst 4: ReplaceInt: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.30.18838/nickel.yaml/oren64.yaml: New home screen subtitle custom font: could not apply patch "New home screen subtitle custom font": line 31: inst 4: ReplaceZlib: not a zlib stream
```

The usual version-specific patches, nothing special.

pgaskin, 2021-11-01:
**Resource extraction:**

```bash
./qrc2zip --output "nickel.18838.qInitResources_resources.zip"    --recursive --verbose "nickel" 1 $((0x12e4540 - 0x0010000)) $((0x00272b8 - 0x0010000)) $((0x12e4028 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
./qrc2zip --output "nickel.18838.qInitResources_translations.zip" --recursive --verbose "nickel" 1 $((0x1679a48 - 0x0010000)) $((0x12e4940 - 0x0010000)) $((0x16797c0 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
./qrc2zip --output "nickel.18838.qInitResources_styles.zip"       --recursive --verbose "nickel" 1 $((0x1699498 - 0x0010000)) $((0x1679b60 - 0x0010000)) $((0x16969e8 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
./qrc2zip --output "nickel.18838.qInitResources_certificates.zip" --recursive --verbose "nickel" 1 $((0x16c0a68 - 0x0010000)) $((0x169a098 - 0x0010000)) $((0x16bfa18 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
```

<details><summary><b>Output</b></summary>

```
nickel@18838/qInitResources_resources
DIR     "qss" (0x12D4930 + 14)
FILE    "qss/TouchMenu.qss" (0x12D3F67 + 189)
DIR     "html" (0x12D4914 + 28)
FILE    "html/RGBFrontLightAboutView.html" (0x12D361B + 1144)
FILE    "html/RGBFrontLightAboutViewSnow.html" (0x12D3A97 + 1228)
DIR     "keys" (0x12D48F8 + 28)
FILE    "keys/device.qmap" (0x12D34C4 + 124)
FILE    "keys/avrcp.qmap" (0x12D32E8 + 472)
FILE    "omfTemplate.html" (0xA4EC23 + 231)
DIR     "pika.rcc" (0x410D + 14)
DIR     "pika.rcc/images" (0x411B + 42)
DIR     "pika.rcc/images/fte" (0x4161 + 14)
FILE    "pika.rcc/images/fte/quicktour_diagram.png" (0x18 + 3841)
DIR     "pika.rcc/images/home" (0x4153 + 14)
FILE    "pika.rcc/images/home/quicktour_device.png" (0xF1D + 4337)
DIR     "pika.rcc/images/about" (0x4145 + 14)
FILE    "pika.rcc/images/about/global_certification.png" (0x2012 + 8221)
FILE    "verifier.pub" (0x1E487E + 451)
DIR     "daylight.rcc" (0xD5E09 + 14)
DIR     "daylight.rcc/images" (0xD5E17 + 350)
DIR     "daylight.rcc/images/fte" (0xD7715 + 280)
FILE    "daylight.rcc/images/fte/missingbooks_factoryreset.png" (0x4E089 + 9065)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg4_adjust_brightness.png" (0x5D7AA + 24664)
FILE    "daylight.rcc/images/fte/missingbooks_account.png" (0x5245B + 22560)
FILE    "daylight.rcc/images/fte/missingbooks_repair.png" (0x84C88 + 20737)
FILE    "daylight.rcc/images/fte/quicktour_nightmode.png" => "daylight.rcc/images/fte/quicktour_nightmode[language!Japanese].png" (0x4803F + 24646)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg2.png" => "daylight.rcc/images/fte/quicktour_reading_pg2[language!Japanese].png" (0x6D602 + 23892)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg2.png" (0x7335A + 23662)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg3.png" (0x7EC74 + 23519)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg3.png" => "daylight.rcc/images/fte/quicktour_reading_pg3[language!Japanese].png" (0x78FCC + 23716)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg5.png" (0x89E2F + 19735)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg1.png" (0x67CE8 + 22806)
FILE    "daylight.rcc/images/fte/missingbooks_sync.png" (0x8EB4A + 6061)
FILE    "daylight.rcc/images/fte/fte_oob_connected.png" => "daylight.rcc/images/fte/fte_oob_connected[language!Japanese].png" (0x902FB + 9157)
FILE    "daylight.rcc/images/fte/fte_oob_plugin.svg" (0x47295 + 3494)
FILE    "daylight.rcc/images/fte/pocket_logo_large.png" (0x4637B + 3862)
FILE    "daylight.rcc/images/fte/quicktour_texticon_search.png" (0x89D8D + 158)
FILE    "daylight.rcc/images/fte/quicktour_diagram.png" (0x503F6 + 8289)
FILE    "daylight.rcc/images/fte/missingbooks_help.png" (0x57C7F + 23335)
FILE    "daylight.rcc/images/fte/overflow_small.png" (0x84857 + 1069)
FILE    "daylight.rcc/images/fte/missingbooks_intro.png" (0x63806 + 17630)
DIR     "daylight.rcc/images/book" (0xD76A5 + 112)
FILE    "daylight.rcc/images/book/dustbin_delete_icon_inverted.png" (0x44B49 + 257)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_highlight.png" (0x44542 + 447)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_dogear.png" (0x43FBF + 409)
FILE    "daylight.rcc/images/book/markup_icon_transparent_padding.png" (0x4415C + 994)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_note.png" (0x43DAE + 202)
FILE    "daylight.rcc/images/book/note_icon_transparent_padding.png" (0x44705 + 200)
FILE    "daylight.rcc/images/book/dustbin_delete_icon.png" (0x43E7C + 319)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_markup.png" (0x447D1 + 884)
DIR     "daylight.rcc/images/help" (0xD767B + 42)
FILE    "daylight.rcc/images/help/help_large.png" (0xC4EB8 + 1138)
FILE    "daylight.rcc/images/help/help_icon.png" (0xC4AB6 + 1022)
FILE    "daylight.rcc/images/help/help_redeem_audiobook_credit.png" (0xC3699 + 5145)
DIR     "daylight.rcc/images/home" (0xD7571 + 266)
FILE    "daylight.rcc/images/home/main_nav_books.png" (0x3DAB5 + 637)
FILE    "daylight.rcc/images/home/my_notebooks_book.png" (0x3D96B + 201)
[... 2526 lines cut ...]
FILE    "dragon.rcc/images/statusbar/global_menu.png" (0xF574 + 146)
FILE    "dragon.rcc/images/statusbar/battery_01.png" (0xCB95 + 212)
FILE    "dragon.rcc/images/statusbar/battery_06.png" (0xB7EA + 211)
FILE    "dragon.rcc/images/statusbar/battery_08.png" (0x12934 + 211)
FILE    "dragon.rcc/images/statusbar/battery_07.png" (0x14B9A + 212)
FILE    "dragon.rcc/images/statusbar/battery_09.png" (0xF8F1 + 211)
[... cut ...]
FILE    "dragon.rcc/images/statusbar/wifi_disabled.png" (0x10FE1 + 528)
FILE    "dragon.rcc/images/statusbar/notification_icon.png" (0x11D5E + 743)
FILE    "dragon.rcc/images/statusbar/search.png" (0xF9C8 + 650)
FILE    "dragon.rcc/images/statusbar/rotate_auto.png" (0x115CC + 864)
FILE    "dragon.rcc/images/statusbar/sync_failed.png" (0x16688 + 648)
FILE    "dragon.rcc/images/statusbar/markup_icon.png" (0xF381 + 495)
FILE    "dragon.rcc/images/statusbar/wifi04.png" (0xCC6D + 708)
FILE    "dragon.rcc/images/statusbar/wifi02.p

shermp, 2021-11-01:
Do the custom font sizes patch and (unreleased) font scale patch require any updates for this version?

rtiangha, 2021-11-01:
> 
> 
> Do the custom font sizes patch and (unreleased) font scale patch require any updates for this version?

No to  custom font size, yes to the unreleased unify size patch; I just tried them myself.

shermp, 2021-11-01:
> No to custom font size

Yay, that's the PITA to update

> yes to the unreleased unify size patch

Boo hiss. Probably need to better handle the offset stuff in that patch.

pgaskin, 2021-11-01:
Later tonight, I'm going to do some testing, commit some changes from Semwize, look through the MR threads for anything else I need to fix, then make the release. I'll also look at #96 if I have time.

shermp, 2021-11-01:
> yes to the unreleased unify size patch

Was that a "yes it works"? Or a "yes, it needs updating"? Because it looks like the offsets have not changed.

rtiangha, 2021-11-01:
> 
> 
> > yes to the unreleased unify size patch
> 
> Was that a "yes it works"? Or a "yes, it needs updating"? Because it looks like the offsets have not changed.

To patching in. I haven't installed it yet since I'm waiting for my GloHD to finish charging first. The unify patch was the one that threw errors.

```
  APPLY `Unify font sizes`
    Error: could not apply patch "Unify font sizes": line 17: inst 5: ReplaceBytes: could not find specified bytes
Error: could not apply patches: error applying patch file 'src/libnickel.so.1.0.0_custom.yaml': could not apply patch "Unify font sizes": line 17: inst 5: ReplaceBytes: could not find specified bytes
```

shermp, 2021-11-01:
Yeah, found the problem. The `blx` instruction to `QScreen::logicalDotsPerInchX()` is a bit different.

@pgaskin what's the proper way to find calls to external functions? The following gives me:
```yaml
- ReplaceBytes:
      Offset: 0x16C2
      FindInstBLX: {Sym: "_ZNK7QScreen19logicalDotsPerInchXEv"} # QScreen::logicalDotsPerInchX()
      ReplaceH: B1 EE 0A 0B # vmov.f64 d0,#6.5
```
With an error of:
```
/home/sherman/dev/kobopatch-patches/src/versions/4.30.18838/libnickel.so.1.0.0.yaml/shermp.yaml: Unify font sizes: could not apply patch "Unify font sizes": line 17: inst 5: ReplaceBytes: could not find specified bytes
```
Note this is different from the `no such symbol "QScreen::logicalDotsPerInchX(void)"` I was getting when I tried the demangled version.

pgaskin, 2021-11-01:
@shermp, I think you forgot to use SymPLT instead of Sym.

shermp, 2021-11-01:
> @shermp, I think you forgot to use SymPLT instead of Sym.

Ah, that works, thanks. I'm not really sure what the difference is between the different `Sym`'s are to be honest.

I'll be force updating my branch.

pgaskin, 2021-11-02:
Released: [`v73`](https://github.com/pgaskin/kobopatch-patches/releases/v73), MR:thread-342717.

pgaskin, 2021-11-02:
<details><summary><b>$ git diff --no-index src/versions/{4.29.18730,4.30.18838}</b></summary>

```diff
diff --git a/src/versions/4.29.18730/libadobe.so.yaml/pgaskin.yaml b/src/versions/4.30.18838/libadobe.so.yaml/pgaskin.yaml
index 93df323..78af60a 100644
--- a/src/versions/4.29.18730/libadobe.so.yaml/pgaskin.yaml
+++ b/src/versions/4.30.18838/libadobe.so.yaml/pgaskin.yaml
@@ -6,5 +6,5 @@ Remove PDF map widget shown during panning:
   - BaseAddress:  {Sym: "N3AdobeReader::showMapWidget()"}
   # tail: N3AdobeReader::updatePanningMap() -> N3AdobeReader::hideMapWidget()
   # TODO: figure out what broke the plt parsing in kobopatch for libadobe in 18730+
-  - ReplaceBytes: {Offset:  80, FindInstBW: 0x1BF94, ReplaceInstBW: 0x1B2F8}
-  - ReplaceBytes: {Offset: 192, FindInstBW: 0x1BF94, ReplaceInstBW: 0x1B2F8}
+  - ReplaceBytes: {Offset:  80, FindInstBW: 0x1BD1C, ReplaceInstBW: 0x1B08C}
+  - ReplaceBytes: {Offset: 192, FindInstBW: 0x1BD1C, ReplaceInstBW: 0x1B08C}
diff --git a/src/versions/4.29.18730/libnickel.so.1.0.0.yaml/pgaskin.yaml b/src/versions/4.30.18838/libnickel.so.1.0.0.yaml/pgaskin.yaml
index 28902f1..37e0f58 100644
--- a/src/versions/4.29.18730/libnickel.so.1.0.0.yaml/pgaskin.yaml
+++ b/src/versions/4.30.18838/libnickel.so.1.0.0.yaml/pgaskin.yaml
@@ -546,7 +546,7 @@ Customize ComfortLight settings:
   # In an unnamed subroutine two layers into FrontLightPopupController::loadView
   # (find it by going back from QTime::addSecs), the times for the dropdown are
   # generated into a QVector<QPair<QString, QTime>> with a simple loop.
-  - BaseAddress: 0xBFFC68 # find the base of the unnamed subroutine with the x-ref to _ZN5QTimeC1Eiiii
+  - BaseAddress: 0xC104A0 # find the base of the unnamed subroutine with the x-ref to _ZN5QTimeC1Eiiii
   #
   # Change the initial hour / first bedtime dropdown item passed to the QTime
   # constructor (mov r1, #21):
diff --git a/src/versions/4.29.18730/nickel.yaml/oren64.yaml b/src/versions/4.30.18838/nickel.yaml/oren64.yaml
index 628dca6..baef22f 100644
--- a/src/versions/4.29.18730/nickel.yaml/oren64.yaml
+++ b/src/versions/4.30.18838/nickel.yaml/oren64.yaml
@@ -27,7 +27,7 @@ New home screen subtitle custom font:
   - Description: |
       Increase subtitle font size, font color black. Works best when used
       with with patch `New home screen increasing cover size`
-  - BaseAddress: 0x16923FE # qss/GenericHomeWidget.qss
+  - BaseAddress: 0x168120B # qss/GenericHomeWidget.qss
   - ReplaceZlib: {Find: "MetaDataLabel {\n  color: #666666;\n  qproperty-indent: 0;\n}", Replace: "MetaDataLabel {\n  color: #000000;\n  qproperty-indent: 0;\n}"}
   - ReplaceZlib:
       Find:    "MetaDataLabel[qApp_deviceIsTrilogy=true] {\n  font-size: 13px;\n}"
diff --git a/src/versions/4.29.18730/nickel.yaml/pgaskin.yaml b/src/versions/4.30.18838/nickel.yaml/pgaskin.yaml
index e787033..d24eaec 100644
--- a/src/versions/4.29.18730/nickel.yaml/pgaskin.yaml
+++ b/src/versions/4.30.18838/nickel.yaml/pgaskin.yaml
@@ -15,7 +15,7 @@ Remove forgot pin button from lock screen:
       Removes the Forgot PIN -> Sign Out button from the lock screen.
       If this is enabled and you forget your pin, you will need to
       hard reset your Kobo.
-  - BaseAddress: 0x16A214E # qss/PinCodeInputDialog.qss
+  - BaseAddress: 0x1683BDA # qss/PinCodeInputDialog.qss
   - ReplaceZlib:
       Find: "#lblForgotPin[qApp_deviceIsDragon=true],\n#lblSignOut[qApp_deviceIsDragon=true] {\n  font-size: 26px;\n}"
       Replace: "#lblForgotPin,#lblSignOut{qproperty-visible:false;}"
@@ -23,7 +23,7 @@ Remove forgot pin button from lock screen:
 Increase size of kepub chapter progress chart:
   - Enabled: no
   - Description: Originally by oren64, rewritten for 4.16.13337 by pgaskin (geek1011).
-  - BaseAddress: 0x16A520E # qss/ReadingMenuStatsView.qss
+  - BaseAddress: 0x167D310 # qss/ReadingMenuStatsView.qss
   - ReplaceZlibGroup:
       Replacements:
       # Top padding is already set to 15px, 25px, 33px, vertical aligned to middle.
```

</details>

closed 2021-11-02 by pgaskin