pgaskin/kobopatch-patches #100: nickel.yaml patches and Libra2/Sage/Elipsa

jackiew1, 2021-11-06:
@pgaskin 
I think some of the comments in nickel.yaml are probably wrong for fw 4.30.18838. See [MR post 26](https://www.mobileread.com/forums/showpost.php?p=4169272&postcount=26) and my reply in post 31.

I can't test but I suspect the Libra2 belongs in the `[qApp_deviceIsStorm=true]` sections, if/when they exist, rather than the "Dragon" sections.

Similarly, I'd expect the Elipsa to use the values in the "Daylight" sections. I see you've already put the Sage in the "Daylight" sections.

How would you like to proceed?

pgaskin, 2021-11-06:
I might have made some mistakes when rushing to get those updated. I am planning on allowing these to be automatically generated in the future, but I don't know when I'll get around to that.

Regarding the Libra2, @shermp was originally looking into the codenames. His work seemed correct, but I'll check if there are edge cases as soon as I have time.

If you have any corrections you are certain about, feel free to make them, and I'll backport them as necessary.

jackiew1, 2021-11-06:
> If you have any corrections you are certain about, feel free to make them, and I'll backport them as necessary.

I can't be certain about them because I don't have Libra2/Sage/Elipsa. I may get a Libra2, but not just yet. I think you said the original LibraH2O is a "Dragon" device, but it's a "Storm" for nickel patch purposes. The Elipsa has the same screen pixels as a KA1 which we know is "Daylight" for nickel CSS patches.

For nickel CSS patches I think screen resolution is probably the main determining factor for which "group" it's in rather than the technicalities of chips and kernel (and other things I don't really understand 😄).

Patches for libnickel may be a different story.

shermp, 2021-11-07:
Had another look at `libnickel`. It seems that between `4.28.18220` and `4.30.18838`, `Device::isStorm()` sprouted a new  `bool` parameter.

So yeah, for CSS purposes, `storm` may be the correct class to use for Libra 2. Thanks for investigating this @jackiew1 

I'm still rather new to this stuff.

jackiew1, 2021-11-27:
There is now official confirmation on [MR](https://www.mobileread.com/forums/showpost.php?p=4175919&postcount=76) that, for at least one nickel patch, the Libra2 uses Storm values not Dragon. That's good enough for me to say that all nickel.yaml comments should point:
- Libra2 to Storm if/when it exists, otherwise Dragon.
- Sage to Daylight

I'll make the necessary changes to my own file for the latest firmware, i.e.
`kobopatch-patches/src/versions/4.30.18838/nickel.yaml/jackie_w.yaml`

jackiew1, 2021-11-28:
@pgaskin I have some time today, would you like me to correct the GeoffR and oren64 nickel files too?

pgaskin, 2021-11-28:
Yes, that would be greatly appreciated, since I'm extremely short on time right now.

jackiew1, 2021-11-28:
> @pgaskin I have some time today, would you like me to correct the GeoffR and oren64 nickel files too?

Done. 

Your nickel.yaml file doesn't need correcting because you were smart enough not to include comments containing device names 😁

pgaskin, 2021-11-29:
Thanks! I'll backport your changes to 4.28.18220+ before the next release.

closed 2022-04-16 by pgaskin