pgaskin/kobopatch-patches #101: Patch request: overwrite notes/annotations API endpoint

a1ex4, 2021-11-10:
Over at https://github.com/janeczku/calibre-web/issues/2155 we are trying to figure out how to redirect the syncing of notes and annotations to a different URL, to replace the remote sync server.

As you can see on the issue this particular endpoint cannot be configured so I though that maybe the only solution would be to use a new patch for it ? Are you by any chance familiar with this built-in feature ? Do you know if this would be a viable option ?

Apologize if this is not the right place, I saw on MR that we could also open requests here.

Thanks in advance for any help

pgaskin, 2021-11-11:
I've already been playing around with the annotations API (back in August), but I don't remember the details about the endpoints. I'll take another look when I have time.

Chickensoupwithrice, 2023-04-16:
Really interested in this and want to help make this happen, but will need a holding hand. I'm not familiar with reverse-engineering but am familiar with the rest of the development process (For Linux, not Kobo). 

I suppose I need to find the function, and specific location in that function that defines the URL Endpoint (https://readingservices.kobo.com/, you mentioned it in your [notes](https://github.com/janeczku/calibre-web/issues/2155#issuecomment-966546094), and then I can define a patch in the src file that overwrites or replaces this data. 

I don't know what to do in order to find this location though.

geneccx, 2024-10-03:
> Really interested in this and want to help make this happen, but will need a holding hand. I'm not familiar with reverse-engineering but am familiar with the rest of the development process (For [...]

I did some reverse engineering on this:

The `reading_services_host` is sent through the regular store API as part of the [initialization command](https://github.com/janeczku/calibre-web/blob/94df767c28858376164cfaf380ef5ee39a476c92/cps/kobo.py#L1051). Maybe it's empty for you because you're syncing through calibre-web? The good news is that this means it doesn't require any firmware patches to swap the endpoint.